b-rad-NDi/Ubuntu-media-tree-kernel-builder #3: CentOS 7 support

jheimbouch, 2017-06-26:
Any chance CentOS 7 support can be added?

b-rad-NDi, 2017-06-26:
Yes, multiple operating system support is on my list. However, unless Hauppauge received multiple support requests for this, or the effort was sponsored externally, it's not very high priority for me to work on.

Just looked at CentOS 7...kernel 3.10. Ouch :)

jheimbouch, 2017-06-26:
I did speak with Hauppauge support today, so there should be at least 1 request in the queue! Thanks for humoring me.

jheimbouch, 2017-06-26:
@b-rad-NDi , what is the level of effort you would expect in CentOS 7 support? 20+ 50+ 100+ hours? I haven't the slightest clue on how long an effort like this would be.

b-rad-NDi, 2017-06-26:
I'd estimate 20-40hr to get the possibility of multiple arbitrary OS support implemented, including the time to figure out the RH build and packaging system parts I'd need to adopt to enable the full stack deployment.

To support kernel 3.10 might take a much larger amount of hours depending on factors such as: how much RH has tinkered (hint: a lot) with their 3.10, how many mainline drivers require extra back ports to compile+work in 3.10, and then how many fixes are required to glue it all together happily. This could be a rabbit hole, although 3.10 is not ancient, it is old enough (4yr!) to be a potential head ache.

b-rad-NDi, 2017-07-19:
@jheimbouch : I pushed up a new branch which is a start to generic os support. CentOS7 will still take a generous sponsor, but I'm going to try and get another distro supported sometime soon.

wcn00, 2018-09-29:
"Me Too" on this.  I must run centos because the machine uses many other rpm type packages.  I want to use mythtv to store pre-recorded shows for eventual distribution via Plex.  
I'd pitch in to do this work (if public contributions are allowed) but I just don't have the time :(