b-rad-NDi/Ubuntu-media-tree-kernel-builder #4: WinTV-dualHD query

tphoney, 2017-07-07:
Thanks for the work you have put into this, i appreciate it a lot. 

I have been running the WinTV-dualHD on different versions of the linux kernerl 4.10, 4.11 ... and i only get a single tuner. When I use your kernel patching i get 2 tuners, wonderful.
Question 1: Will your patches / fixes be merged into the mainline kernel ? probably too late for 4.12
Question 2: When scanning for HD channels in the UK they is not picked up by w_scan or by plex, however if i use dvbv5-scan and a valid channel scanning template eg uk-Divis I find the HD channels. Are you aware of this issue ? Have you got HD channels (DVB-T2) working with the device.

apologies for having so many questions.

tphoney, 2017-07-07:
The issue i have of not finding DVB-T2 channels is echoed in this excellent posting http://blog.palosaari.fi/2014/09/linux-dvb-t2-tuning-problems.html

From what i understand, when scanning the application should specify DVB-T or DVB-T2, most scanners only scan for DVB-T assuming that it will autoswitch to DVB-T2.

muzena, 2017-07-08:
@tphoney 
I had same problem with scanning DVB-T2 channels on WinTV-HVR-2205 dual tuner.
Only software that is capable scanning DVB-T2 channels that i found is TVHEADNED.
https://github.com/tvheadend/tvheadend

b-rad-NDi, 2017-07-13:
@tphoney : 
Q1) The intention of this work is to gather, polish, and then upstream all of our hardware support which is lacking mainline. I have quite a lot of other responsibilities as well however, so finding the time a slow process.
Q2) I do not have T2 signal to test, I'll ask one of our Euro testers.

tphoney, 2017-07-13:
Thanks for the update, i understand the time constraints.

Have a good one :)

b-rad-NDi, 2017-07-19:
@tphoney @muzena : would you guys mind trying out a si2168.ko module with a T/T2 autodetect patch applied? That is if your device in question is using that driver :) I'll put it on dropbox or something to test.

b-rad-NDi, 2017-07-19:
Actually, it is attached here. Just (maybe) save the old module, then replace it with this one.

[si2168-T2-autodetect.zip](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1160370/si2168-T2-autodetect.zip)

muzena, 2017-07-21:
@b-rad-NDi  @tphoney 
When I bought it in 2015 year  WinTV-HVR-2205 dual tuner card it was only partly work (DVB-T)
I use it for DVB-T2 and DVB-C.
Then i contact and help Linixtv developer Olli Salonen  (trsqr)

tphoney, 2017-07-21:
I will hopefully get to apply this, over the weekend. I'll keep you updated. 

Also THANKS !!! 👍

muzena, 2017-07-21:
@b-rad-NDi  @tphoney 
When I bought it in 2015 year  WinTV-HVR-2205 dual tuner card it was only partly work (DVB-T)
I use it for DVB-T2 and DVB-C.
Then i contact and help Linixtv developer [Olli Salonen  (trsqr)](https://github.com/OpenELEC/dvb-firmware/commit/fb88eed23c12745bc264f5169f68a7f282b0ee25) in testing WinTV-HVR-2205 and after half of year he managed to patch DVB-T2 DVB-C to work, only analog part of card need to be patched.
We scan DVB-T2 channels with w-scan and dvbv5-scan without sucess.
The problem was in w-scan and dvbv5-scan utilities.
DVB-T2 scanning only worked in [Tvheadend](https://tvheadend.org/)
So @tphoney , i think you should first try to scaning DVB-T2 channels with tvheadend to see if it works.

b-rad-NDi, 2017-07-21:
@muzena : Using tvheadend is not a proper solution. The driver was set to scan T only by default. We have now set the driver to T/T2 auto detect. That should work in *any* scanner, if it works.

b-rad-NDi, 2017-07-21:
@muzena @tphoney : if either of you are building yourself, the DVB-T/T2 auto detect patch is in the DVB-T-T2-autodetect branch

3785d9a91131bc9da8d14a9b68ed2d4a95f14c5d

Will not be adding this patch to the main builds until it is confirmed working and not adversely affecting DVB-T.

tphoney, 2017-07-22:
@b-rad-NDi Apologies, how do i apply the patch. I have cloned the repository, i am running the 4.12 kernel

b-rad-NDi, 2017-07-22:
After cloning:
<pre>
# patch infrastructure
./mediatree -p
# apply extra patches
./mediatree -x
# clean
./mediatree -c
# build
./mediatree -b</pre>

That will produce a set of four .deb's for you.

There's loads of packages you'll have to install to be able to compile all the packages. This will only build the default kernel for your flavour of Ubuntu. So xenial 4.4, yakkety 4.8, and zesty 4.10.

tphoney, 2017-07-23:
HI @b-rad-NDi been trying to get this working took a while to figure out i had to manually set UBUNTU_VERSION=zesty.
Running the script with -i seems to run ok

`➜  Ubuntu-media-tree-kernel-builder git:(3785d9a) ./mediatree-kbuild.sh -i
Initializing/Updating clean patch gen repo
Cloning into '.media-tree-clean-patch-repo'...
remote: Counting objects: 5342142, done.
remote: Compressing objects: 100% (784698/784698), done.
remote: Total 5342142 (delta 4517397), reused 5340929 (delta 4516242)
Receiving objects: 100% (5342142/5342142), 1.22 GiB | 9.15 MiB/s, done.
Resolving deltas: 100% (4517397/4517397), done.
Checking connectivity... done.
Checking out files: 100% (58593/58593), done.
Already up-to-date.

Initializing Ubuntu work repo
Cloning into 'ubuntu-zesty'...
remote: Counting objects: 5342142, done.
remote: Compressing objects: 100% (784698/784698), done.
remote: Total 5342142 (delta 4517397), reused 5340929 (delta 4516242)
Receiving objects: 100% (5342142/5342142), 1.22 GiB | 9.15 MiB/s, done.
Resolving deltas: 100% (4517397/4517397), done.
Checking connectivity... done.
Checking out files: 100% (58593/58593), done.
Already up-to-date.

Initializing/Updating media_build system
Cloning into 'media_build'...
remote: Counting objects: 3766, done.
remote: Compressing objects: 100% (1035/1035), done.
remote: Total 3766 (delta 2693), reused 3756 (delta 2688)
Receiving objects: 100% (3766/3766), 667.91 KiB | 0 bytes/s, done.
Resolving deltas: 100% (2693/2693), done.
Checking connectivity... done.`

Running the script with -p fails.
`#############################################################
Applying: Mainline Ubuntu build fixes
Checking patch drivers/dma-buf/sync_debug.c...
Checking patch drivers/dma-buf/sync_file.c...
Checking patch drivers/hid/hid-picolcd_cir.c...
Checking patch drivers/net/ethernet/via/via-velocity.c...
Applied patch drivers/dma-buf/sync_debug.c cleanly.
Applied patch drivers/dma-buf/sync_file.c cleanly.
Applied patch drivers/hid/hid-picolcd_cir.c cleanly.
Applied patch drivers/net/ethernet/via/via-velocity.c cleanly.
#############################################################
Applying: Packaging updates
Checking patch debian.master/control.d/flavour-control.stub...
Checking patch debian.master/control.stub.in...
Checking patch debian.master/copyright...
Checking patch debian.master/rules.d/amd64.mk...
Checking patch debian.master/rules.d/arm64.mk...
Checking patch debian.master/rules.d/armhf.mk...
Checking patch debian.master/rules.d/i386.mk...
Checking patch debian.master/rules.d/ppc64el.mk...
Checking patch debian.master/rules.d/s390x.mk...
Checking patch debian/rules...
Checking patch debian/rules.d/0-common-vars.mk...
Checking patch debian/rules.d/2-binary-arch.mk...
Applied patch debian.master/control.d/flavour-control.stub cleanly.
Applied patch debian.master/control.stub.in cleanly.
Applied patch debian.master/copyright cleanly.
Applied patch debian.master/rules.d/amd64.mk cleanly.
Applied patch debian.master/rules.d/arm64.mk cleanly.
Applied patch debian.master/rules.d/armhf.mk cleanly.
Applied patch debian.master/rules.d/i386.mk cleanly.
Applied patch debian.master/rules.d/ppc64el.mk cleanly.
Applied patch debian.master/rules.d/s390x.mk cleanly.
Applied patch debian/rules cleanly.
Applied patch debian/rules.d/0-common-vars.mk cleanly.
Applied patch debian/rules.d/2-binary-arch.mk cleanly.
#############################################################
Applying: Changelog
Checking patch debian.master/changelog...
error: while searching for:
linux (4.10.0-26.30) zesty; urgency=low

  * linux: 4.10.0-26.30 -proposed tracker (LP: #1700528)

error: patch failed: debian.master/changelog:1
Applying patch debian.master/changelog with 1 reject...
Rejected hunk #1.
Patch failed at 0001 Changelog
The copy of the patch that failed is found in: .git/rebase-apply/patch
When you have resolved this problem, run "git am --continue".
If you prefer to skip this patch, run "git am --skip" instead.
To restore the original branch and stop patching, run "git am --abort".
You must manually fix a conflict, and then mark complete with:
    git am --continue
patch failure, exiting 2
➜  Ubuntu-media-tree-kernel-builder git:(3785d9a) uname -a
Linux nasty 4.10.0-27-generic #30~16.04.2-Ubuntu SMP Thu Jun 29 16:07:46 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`

b-rad-NDi, 2017-07-23:
Looks like new 4.10 kernel version, so the changelog must be regenerated.
<pre>
cd ubuntu-zesty
git am --abort
cd ../
./mediatree -r    # enter YES
UPDATE_MT_KBUILD_VER=YES ./mediatree -p
</pre>

b-rad-NDi, 2017-07-24:
Builds are located here in case there has been further issues:
https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0
You will need to download and install the 4 debs for the kernel you are testing.

tphoney, 2017-07-24:
Many thanks for the upload. I grabbed the 4 dev packages for 4.10 from dropbox. and installed them. Rebooted to find
`[    5.898051] em28xx 3-2:1.0: New device HCW dualHD @ 480 Mbps (2040:0265, interface 0, class 0)
[    5.898056] em28xx 3-2:1.0: DVB interface 0 found: isoc
[    5.898109] em28xx 3-2:1.0: chip ID is em28174
[    5.906051] usbcore: registered new interface driver btusb
[    5.909264] usb-storage 3-9.4.1:1.0: USB Mass Storage device detected
[    5.909345] scsi host6: usb-storage 3-9.4.1:1.0
[    5.909462] usbcore: registered new interface driver usb-storage
[    5.911694] usbcore: registered new interface driver uas
[    6.017989] Bluetooth: hci0: BCM: chip id 63
[    6.033979] Bluetooth: hci0: nasty
[    6.034985] Bluetooth: hci0: BCM20702A1 (001.002.014) build 0000
[    6.037570] bluetooth hci0: Direct firmware load for brcm/BCM20702A1-0b05-17cb.hcd failed with error -2
[    6.037572] Bluetooth: hci0: BCM: Patch brcm/BCM20702A1-0b05-17cb.hcd not found
[    6.832210] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    6.832212] Bluetooth: BNEP filters: protocol multicast
[    6.832215] Bluetooth: BNEP socket layer initialized
[    6.921636] scsi 6:0:0:0: Direct-Access     Generic- SD/MMC/MS/MSPRO  1.00 PQ: 0 ANSI: 4
[    6.924349] sd 6:0:0:0: Attached scsi generic sg5 type 0
[    6.929731] sd 6:0:0:0: [sdf] Attached SCSI removable disk
[    7.089723] e1000e: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[    7.089758] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[    7.113971] em28xx 3-2:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0xb2dcc1c8
[    7.113972] em28xx 3-2:1.0: EEPROM info:
[    7.113973] em28xx 3-2:1.0: 	microcode start address = 0x0004, boot configuration = 0x01
[    7.120211] em28xx 3-2:1.0: 	AC97 audio (5 sample rates)
[    7.120212] em28xx 3-2:1.0: 	500mA max power
[    7.120214] em28xx 3-2:1.0: 	Table at offset 0x27, strings=0x0e6a, 0x1888, 0x087e
[    7.120328] em28xx 3-2:1.0: Identified as Hauppauge WinTV-dualHD DVB (card=99)
[    7.124289] tveeprom: Hauppauge model 204109, rev B3I6, serial# 13909791
[    7.124291] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    7.124292] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[    7.124292] tveeprom: audio processor is None (idx 0)
[    7.124293] tveeprom: has no radio, has IR receiver, has no IR transmitter
[    7.124297] em28xx 3-2:1.0: dvb set to isoc mode.
[    7.124356] em28xx 3-2:1.0: chip ID is em28174
[    8.329829] em28xx 3-2:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0xb2dcc1c8
[    8.329831] em28xx 3-2:1.0: EEPROM info:
[    8.329832] em28xx 3-2:1.0: 	microcode start address = 0x0004, boot configuration = 0x01
[    8.335925] em28xx 3-2:1.0: 	AC97 audio (5 sample rates)
[    8.335927] em28xx 3-2:1.0: 	500mA max power
[    8.335928] em28xx 3-2:1.0: 	Table at offset 0x27, strings=0x0e6a, 0x1888, 0x087e
[    8.336021] em28xx 3-2:1.0: Identified as Hauppauge WinTV-dualHD DVB (card=99)
[    8.336984] tveeprom: Hauppauge model 204109, rev B3I6, serial# 13909791
[    8.336985] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    8.336986] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[    8.336987] tveeprom: audio processor is None (idx 0)
[    8.336987] tveeprom: has no radio, has IR receiver, has no IR transmitter
[    8.336990] em28xx 3-2:1.0: dvb ts2 set to isoc mode.
[    8.536758] usbcore: registered new interface driver em28xx
[    8.545372] em28xx 3-2:1.0: Binding DVB extension
[    8.552509] i2c i2c-7: Added multiplexed i2c bus 10
[    8.552512] si2168 7-0064: Silicon Labs Si2168-B40 successfully identified
[    8.552513] si2168 7-0064: firmware version: B 4.0.2
[    8.555881] si2157: probe of 10-0060 failed with error -11
[    8.556727] em28xx 3-2:1.0: Binding DVB extension
[    8.560450] i2c i2c-9: Added multiplexed i2c bus 10
[    8.560453] si2168 9-0067: Silicon Labs Si2168-B40 successfully identified
[    8.560454] si2168 9-0067: firmware version: B 4.0.2
[    8.561877] si2157: probe of 10-0063 failed with error -11
[    8.561977] em28xx: Registered (Em28xx dvb Extension) extension
[    8.567572] em28xx 3-2:1.0: Registering input extension
[    8.595772] Registered IR keymap rc-hauppauge
[    8.595977] rc rc0: 3-2:1.0 IR as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/rc/rc0
[    8.596027] input: 3-2:1.0 IR as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/rc/rc0/input15
[    8.596101] em28xx 3-2:1.0: Input extension successfully initalized
[    8.596102] em28xx 3-2:1.0: Registering input extension
[    8.596204] Registered IR keymap rc-hauppauge
[    8.596348] rc rc1: 3-2:1.0 IR as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/rc/rc1
[    8.596403] input: 3-2:1.0 IR as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/rc/rc1/input16
[    8.596490] em28xx 3-2:1.0: Input extension successfully initalized
[    8.596491] em28xx: Registered (Em28xx Input Extension) extension
[   23.055296] ata2.00: configured for UDMA/133
[   23.055299] ata2: EH complete
[   23.056327] ata3.00: configured for UDMA/133
[   23.056331] ata3: EH complete
[   23.126058] Bluetooth: RFCOMM TTY layer initialized
[   23.126065] Bluetooth: RFCOMM socket layer initialized
[   23.126071] Bluetooth: RFCOMM ver 1.11
[   23.456485] ata5.00: configured for UDMA/133
[   23.456490] ata5: EH complete
[   23.461787] ata4.00: configured for UDMA/133
[   23.461791] ata4: EH complete`

The tuner is not being picked up any software. above is the output from dmesg.

b-rad-NDi, 2017-07-24:
Everything appears ok, except for this:
<pre> [ 8.555881] si2157: probe of 10-0060 failed with error -11</pre>
I will investigate, there shouldn't be any failures.

tphoney, 2017-07-24:
that is one of the chips on the stick, unless i am wrong.

b-rad-NDi, 2017-07-24:
You are definitely correct, hence why that failure shouldn't be there. I don't think I have one of these, but digging through my stuff. I've passed this on to our testers so they can try and verify that there is indeed some issue.

tphoney, 2017-07-25:
many thanks for looking at this, and apologies for causing hassle

b-rad-NDi, 2017-07-25:
Thanks for testing and helping out. You're causing a welcome hassle by finding oopses and issues before I push releases to the ppa and get yelled at by hundreds :)

b-rad-NDi, 2017-07-25:
We think we figured this out. I will have new builds up sometime today.

tphoney, 2017-07-25:
Sweet. Out of curiousity what was the issue ?

b-rad-NDi, 2017-07-25:
We added error bit check to cmd_execute, but that bit can be set if nothing has been done to the tuner yet. So that must be checked.

b-rad-NDi, 2017-07-25:
New, hopefully fixed, packages are available here for 4.4 and 4.10:
 https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0

Please comment on your findings.

tphoney, 2017-07-25:
Unfortunately i have the same error message. (kernel version, and showing packages are installed. are shown below the dmesg output)

[  210.565826] usb 3-2: new high-speed USB device number 23 using xhci_hcd
[  210.706504] usb 3-2: New USB device found, idVendor=2040, idProduct=0265
[  210.706509] usb 3-2: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[  210.706511] usb 3-2: Product: dualHD
[  210.706514] usb 3-2: Manufacturer: HCW
[  210.706516] usb 3-2: SerialNumber: 0013909791
[  210.707357] em28xx 3-2:1.0: New device HCW dualHD @ 480 Mbps (2040:0265, interface 0, class 0)
[  210.707367] em28xx 3-2:1.0: DVB interface 0 found: isoc
[  210.707429] em28xx 3-2:1.0: chip ID is em28174
[  211.904235] em28xx 3-2:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0xb2dcc1c8
[  211.904238] em28xx 3-2:1.0: EEPROM info:
[  211.904241] em28xx 3-2:1.0: 	microcode start address = 0x0004, boot configuration = 0x01
[  211.910705] em28xx 3-2:1.0: 	AC97 audio (5 sample rates)
[  211.910708] em28xx 3-2:1.0: 	500mA max power
[  211.910712] em28xx 3-2:1.0: 	Table at offset 0x27, strings=0x0e6a, 0x1888, 0x087e
[  211.911047] em28xx 3-2:1.0: Identified as Hauppauge WinTV-dualHD DVB (card=99)
[  211.913395] tveeprom: Hauppauge model 204109, rev B3I6, serial# 13909791
[  211.913398] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[  211.913401] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[  211.913402] tveeprom: audio processor is None (idx 0)
[  211.913404] tveeprom: has no radio, has IR receiver, has no IR transmitter
[  211.913408] em28xx 3-2:1.0: dvb set to isoc mode.
[  211.913506] em28xx 3-2:1.0: chip ID is em28174
[  213.120096] em28xx 3-2:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0xb2dcc1c8
[  213.120099] em28xx 3-2:1.0: EEPROM info:
[  213.120103] em28xx 3-2:1.0: 	microcode start address = 0x0004, boot configuration = 0x01
[  213.126571] em28xx 3-2:1.0: 	AC97 audio (5 sample rates)
[  213.126574] em28xx 3-2:1.0: 	500mA max power
[  213.126578] em28xx 3-2:1.0: 	Table at offset 0x27, strings=0x0e6a, 0x1888, 0x087e
[  213.126913] em28xx 3-2:1.0: Identified as Hauppauge WinTV-dualHD DVB (card=99)
[  213.129246] tveeprom: Hauppauge model 204109, rev B3I6, serial# 13909791
[  213.129249] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[  213.129252] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[  213.129253] tveeprom: audio processor is None (idx 0)
[  213.129255] tveeprom: has no radio, has IR receiver, has no IR transmitter
[  213.129259] em28xx 3-2:1.0: dvb ts2 set to isoc mode.
[  213.327827] em28xx 3-2:1.0: Binding DVB extension
[  213.330889] i2c i2c-7: Added multiplexed i2c bus 10
[  213.330891] si2168 7-0064: Silicon Labs Si2168-B40 successfully identified
[  213.330893] si2168 7-0064: firmware version: B 4.0.2
[  213.332474] si2157: probe of 10-0060 failed with error -11
[  213.332593] em28xx 3-2:1.0: Binding DVB extension
[  213.335906] i2c i2c-9: Added multiplexed i2c bus 10
[  213.335909] si2168 9-0067: Silicon Labs Si2168-B40 successfully identified
[  213.335910] si2168 9-0067: firmware version: B 4.0.2
[  213.337511] si2157: probe of 10-0063 failed with error -11
[  213.337625] em28xx 3-2:1.0: Registering input extension
[  213.337798] Registered IR keymap rc-hauppauge
[  213.337954] rc rc0: 3-2:1.0 IR as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/rc/rc0
[  213.338008] input: 3-2:1.0 IR as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/rc/rc0/input24
[  213.338152] em28xx 3-2:1.0: Input extension successfully initalized
[  213.338154] em28xx 3-2:1.0: Registering input extension
[  213.338295] Registered IR keymap rc-hauppauge
[  213.338442] rc rc1: 3-2:1.0 IR as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/rc/rc1
[  213.338495] input: 3-2:1.0 IR as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/rc/rc1/input25
[  213.338608] em28xx 3-2:1.0: Input extension successfully initalized

➜  ~ uname -a 
Linux nasty 4.10.0-26201707251251-generic #0+mediatree+hauppauge SMP Tue Jul 25 12:52:31 CDT 2017 x86_64 x86_64 x86_64 GNU/Linux

➜  ~ dpkg -l | grep 26201707251251
ii  linux-headers-4.10.0-26201707251251             4.10.0-26201707251251.0+mediatree+hauppauge all          Header files related to Linux kernel version 4.10.0
ii  linux-headers-4.10.0-26201707251251-generic     4.10.0-26201707251251.0+mediatree+hauppauge amd64        Linux kernel headers with slipstreamed mediatree drivers for version 4.10.0 on 64 bit x86 SMP
ii  linux-image-4.10.0-26201707251251-generic       4.10.0-26201707251251.0+mediatree+hauppauge amd64        Linux kernel image for version 4.10.0 on 64 bit x86 SMP
ii  linux-image-extra-4.10.0-26201707251251-generic 4.10.0-26201707251251.0+mediatree+hauppauge amd64        Linux kernel extra modules with slipstreamed mediatree drivers for version 4.10.0 on 64 bit x86 SMP

b-rad-NDi, 2017-07-25:
Ugh. Of course I went and rebuilt 4.10 *without* applying the new patch... /headsmack

Apologies

b-rad-NDi, 2017-07-25:
New 4.10 packages *with patch applied* is in the dropbox.

tphoney, 2017-07-25:
sorry, there was no link :)

b-rad-NDi, 2017-07-25:
Same link as before, it's my staging area:

 https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0

tphoney, 2017-07-25:
So it is being picked up ok now, by software. First scan did not get and DVBT-2  channels then i things stopped scanning, about to reboot, but got this message. 
[  918.880811] CPU: 2 PID: 4135 Comm: kworker/2:3 Not tainted 4.10.0-26201707251504-generic #0+mediatree+hauppauge
[  918.880836] Hardware name: ASUS All Series/H87I-PLUS, BIOS 2003 11/05/2014
[  918.880858] Workqueue: events request_module_async [em28xx]
[  918.880873] task: ffff8a200e20ad00 task.stack: ffffb52e42774000
[  918.880890] RIP: 0010:mutex_lock+0x1d/0x40
[  918.880902] RSP: 0018:ffffb52e42777920 EFLAGS: 00010246
[  918.880916] RAX: 0000000000000000 RBX: 0000000000000030 RCX: 0000000000000006
[  918.880935] RDX: ffff8a200e20ad00 RSI: ffffb52e42777990 RDI: 0000000000000030
[  918.880953] RBP: ffffb52e42777928 R08: 0000000000000001 R09: 0000000000000486
[  918.880972] R10: 0000000000000060 R11: 0000000000000000 R12: ffff8a1fb433b000
[  918.880990] R13: ffff8a1fb433b000 R14: 0000000000000030 R15: ffff8a201196e800
[  918.881009] FS:  0000000000000000(0000) GS:ffff8a201fb00000(0000) knlGS:0000000000000000
[  918.881030] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  918.881045] CR2: 0000000000000030 CR3: 0000000212a52000 CR4: 00000000001406e0
[  918.881063] Call Trace:
[  918.881074]  si2157_cmd_execute+0x2f/0x190 [si2157]
[  918.881089]  si2157_attach+0x205/0x497 [si2157]
[  918.881102]  si2157_probe+0x77/0x140 [si2157]
[  918.881116]  ? si2157_setup_configuration.isra.7+0xc0/0xc0 [si2157]
[  918.881134]  i2c_device_probe+0x1c2/0x220
[  918.881148]  driver_probe_device+0x2bb/0x460
[  918.881161]  __device_attach_driver+0x8c/0x100
[  918.881174]  ? __driver_attach+0xf0/0xf0
[  918.881187]  bus_for_each_drv+0x67/0xb0
[  918.881198]  __device_attach+0xdd/0x160
[  918.881210]  device_initial_probe+0x13/0x20
[  918.881223]  bus_probe_device+0x92/0xa0
[  918.881235]  device_add+0x373/0x630
[  918.881246]  device_register+0x1a/0x20
[  918.881257]  i2c_new_device+0x196/0x240
[  918.881270]  em28xx_dvb_init.part.5+0x316/0x3940 [em28xx_dvb]
[  918.881287]  ? load_balance+0x180/0xa00
[  918.881299]  ? sched_clock+0x9/0x10
[  918.881311]  ? sched_clock_cpu+0x99/0xb0
[  918.881323]  ? sched_clock+0x9/0x10
[  918.881333]  ? sched_clock_cpu+0x99/0xb0
[  918.881345]  ? pick_next_task_fair+0x2ef/0x4b0
[  918.881358]  ? __switch_to+0x23c/0x520
[  918.881370]  em28xx_dvb_init+0x24/0x30 [em28xx_dvb]
[  918.881385]  em28xx_init_extension+0x66/0xa0 [em28xx]
[  918.881400]  request_module_async+0x22/0x100 [em28xx]
[  918.881415]  process_one_work+0x1fc/0x4b0
[  918.881427]  worker_thread+0x4b/0x500
[  918.881438]  kthread+0x109/0x140
[  918.881448]  ? process_one_work+0x4b0/0x4b0
[  918.881461]  ? kthread_create_on_node+0x60/0x60
[  918.881474]  ret_from_fork+0x2c/0x40
[  918.881485] Code: e9 74 fe ff ff e8 f4 17 7b ff 0f 1f 40 00 0f 1f 44 00 00 55 48 89 e5 53 48 89 fb e8 7e e7 ff ff 65 48 8b 14 25 00 d3 00 00 31 c0 <f0> 48 0f b1 13 48 85 c0 74 08 48 89 df e8 31 fd ff ff 5b 5d c3 
[  918.881547] RIP: mutex_lock+0x1d/0x40 RSP: ffffb52e42777920
[  918.881561] CR2: 0000000000000030
[  918.907559] ---[ end trace 57a416c83251e4fa ]---
[  952.007992] usb 3-2: USB disconnect, device number 23
[  952.008098] em28xx 3-2:1.0: Disconnecting em28xx #1
[  952.008103] em28xx 3-2:1.0: Disconnecting em28xx

b-rad-NDi, 2017-07-25:
one step closer...Interesting back trace, it would seem to have happened in the probe. I will investigate.

tphoney, 2017-07-25:
[    5.848424] em28xx 3-2:1.0: New device HCW dualHD @ 480 Mbps (2040:0265, interface 0, class 0)
[    5.848427] em28xx 3-2:1.0: DVB interface 0 found: isoc
[    5.848481] em28xx 3-2:1.0: chip ID is em28174
[    5.894524] Bluetooth: hci0: BCM: chip id 63
[    5.910526] Bluetooth: hci0: BCM20702A
[    5.911524] Bluetooth: hci0: BCM20702A1 (001.002.014) build 0000
[    5.913025] bluetooth hci0: Direct firmware load for brcm/BCM20702A1-0b05-17cb.hcd failed with error -2
[    5.913026] Bluetooth: hci0: BCM: Patch brcm/BCM20702A1-0b05-17cb.hcd not found
[    6.778151] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    6.778152] Bluetooth: BNEP filters: protocol multicast
[    6.778157] Bluetooth: BNEP socket layer initialized
[    6.788836] scsi 6:0:0:0: Direct-Access     Generic- SD/MMC/MS/MSPRO  1.00 PQ: 0 ANSI: 4
[    6.789254] sd 6:0:0:0: Attached scsi generic sg5 type 0
[    6.795767] sd 6:0:0:0: [sdf] Attached SCSI removable disk
[    6.866306] e1000e: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[    6.866337] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[    7.050466] em28xx 3-2:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0xb2dcc1c8
[    7.050467] em28xx 3-2:1.0: EEPROM info:
[    7.050468] em28xx 3-2:1.0: 	microcode start address = 0x0004, boot configuration = 0x01
[    7.056835] em28xx 3-2:1.0: 	AC97 audio (5 sample rates)
[    7.056837] em28xx 3-2:1.0: 	500mA max power
[    7.056839] em28xx 3-2:1.0: 	Table at offset 0x27, strings=0x0e6a, 0x1888, 0x087e
[    7.056954] em28xx 3-2:1.0: Identified as Hauppauge WinTV-dualHD DVB (card=99)
[    7.062192] tveeprom: Hauppauge model 204109, rev B3I6, serial# 13909791
[    7.062193] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    7.062195] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[    7.062196] tveeprom: audio processor is None (idx 0)
[    7.062196] tveeprom: has no radio, has IR receiver, has no IR transmitter
[    7.062201] em28xx 3-2:1.0: dvb set to isoc mode.
[    7.062271] em28xx 3-2:1.0: chip ID is em28174
[    8.262826] em28xx 3-2:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0xb2dcc1c8
[    8.262827] em28xx 3-2:1.0: EEPROM info:
[    8.262828] em28xx 3-2:1.0: 	microcode start address = 0x0004, boot configuration = 0x01
[    8.269208] em28xx 3-2:1.0: 	AC97 audio (5 sample rates)
[    8.269210] em28xx 3-2:1.0: 	500mA max power
[    8.269211] em28xx 3-2:1.0: 	Table at offset 0x27, strings=0x0e6a, 0x1888, 0x087e
[    8.269332] em28xx 3-2:1.0: Identified as Hauppauge WinTV-dualHD DVB (card=99)
[    8.270163] tveeprom: Hauppauge model 204109, rev B3I6, serial# 13909791
[    8.270164] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    8.270165] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[    8.270166] tveeprom: audio processor is None (idx 0)
[    8.270166] tveeprom: has no radio, has IR receiver, has no IR transmitter
[    8.270168] em28xx 3-2:1.0: dvb ts2 set to isoc mode.
[    8.468714] usbcore: registered new interface driver em28xx
[    8.475230] em28xx 3-2:1.0: Binding DVB extension
[    8.480622] i2c i2c-7: Added multiplexed i2c bus 10
[    8.480625] si2168 7-0064: Silicon Labs Si2168-B40 successfully identified
[    8.480627] si2168 7-0064: firmware version: B 4.0.2
[    8.482944] si2157_probe: probing si2157 tuner fe=ffff96c8ce89b828 cfg=ffffbf53c21d7ae8 addr=0X60
[    8.482945] si2157_setup_configuration: client=ffff96c8ca4f0800
[    8.482946] si2157_setup_configuration: dev=          (null)
[    8.482950] si2157_attach (10-0060)
[    8.482951] si2157 10-0060: creating new instance
[    8.482951] si2157_attach: instance=1
[    8.482952] si2157_attach(): new instance for tuner @0x60
[    8.482953] si2157_setup_configuration: client=ffff96c8ca4f0800
[    8.482953] si2157_setup_configuration: dev=ffff96c8ca4f2c00
[    8.482954] si2157_setup_configuration(0x60): dvb driver submitted configuration; port=1 invert=0
[    8.483383] si2157 10-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    8.483393] dvbdev: DVB: registering new adapter (3-2:1.0)
[    8.483397] em28xx 3-2:1.0: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[    8.483399] dvbdev: dvb_create_media_entity: media entity 'Silicon Labs Si2168' registered.
[    8.483747] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[    8.485454] em28xx 3-2:1.0: DVB extension successfully initialized
[    8.485457] em28xx 3-2:1.0: Binding DVB extension
[    8.488349] i2c i2c-9: Added multiplexed i2c bus 11
[    8.488352] si2168 9-0067: Silicon Labs Si2168-B40 successfully identified
[    8.488352] si2168 9-0067: firmware version: B 4.0.2
[    8.489410] si2157_probe: probing si2157 tuner fe=ffff96c8cca10828 cfg=ffffbf53c21d7ae8 addr=0X63
[    8.489412] si2157_setup_configuration: client=ffff96c8cf862400
[... 18 lines cut ...]
[    8.524652] input: 3-2:1.0 IR as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/rc/rc0/input15
[    8.524704] em28xx 3-2:1.0: Input extension successfully initalized
[    8.524706] em28xx 3-2:1.0: Registering input extension
[    8.524833] Registered IR keymap rc-hauppauge
[    8.524972] rc rc1: 3-2:1.0 IR as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/rc/rc1
[    8.525017] input: 3-2:1.0 IR as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/rc/rc1/input16
[    8.525086] em28xx 3-2:1.0: Input extension successfully initalized
[... cut ...]
[   34.018507] ata4.00: configured for UDMA/133
[   34.018509] ata4: EH complete
[   34.450603] ata5.00: configured for UDMA/133
[   34.450607] ata5: EH complete
[   34.454638] ata3.00: configured for UDMA/133
[   34.454642] ata3: EH complete
[   68.062376] si2168 7-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   68.260100] si2168 7-0064: firmware version: B 4.0.11
[   68.263503] si2157 10-0060: found a 'Silicon Labs Si2157-A30'
[   68.327424] si2157 10-0060: firmware version: 3.0.5

DMESG from a clean boot.

tphoney, 2017-07-25:
out of interest what CLI tool would you recommend for running scans ? i am using 
dvbv5-scan uk-Divis
it accepts an input file, it specifies whether it scans using DVB-T or DVB-T2. So i get everything. 
I also use plex, but it will only return DVB-T channels.

b-rad-NDi, 2017-07-25:
Thanks for the dmesg. It appears to be good on clean boot, up to the point your log stops. I would have expected an oops already if there was a problem in the probe. Please post if you encounter another oops, including dmesg messages that happen around the same time.

For scanning our testers typically use w_scan on the cli. I will inquire if our euro testers are using something else.

skykinetic, 2017-07-27:
Thanks a lot for the patch, it works very well for me, however, I do notice that the patch not only enables 2 tuners, it also duplicates the IR device, resulting in 2 input devices being registered by the kernel (all the dmesg reflects this so far so I won't post another). Will there be any fix for this to prevent repeated inputs when using the IR remote (which is also working very great without the patch)?

b-rad-NDi, 2017-07-27:
@skykinetic : please post file a separate issue with this ir problem, including snippet of related dmesg and I'll sort it out.

b-rad-NDi, 2017-07-29:
Latest packages are here:
https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0

This contains the T/T2 scanning fix, as well as everything else in master. @tphoney : if you could confirm T and T2 both work for you, then you may close this issue. Thanks.

tphoney, 2017-07-30:
Unfortunately no dice, i still cannot get any DVB-T2 channels. 
With plex i get only DVB-T channels.
With w_scan i dont get anything. 

One thing is 
[   87.669106] si2168 7-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   87.866569] si2168 7-0064: firmware version: B 4.0.11

Is there any particular version of this firmware i should be using, this looks peculiar ?

tphoney, 2017-07-30:
sudo w_scan -X -c GB
[sudo] password for tp:
w_scan -X -c GB
w_scan version 20161022 (compiled for DVB API 5.10)
using settings for UNITED KINGDOM
DVB aerial
DVB-T GB
scan type TERRESTRIAL, channellist 6
output format czap/tzap/szap/xine
WARNING: could not guess your codepage. Falling back to 'UTF-8'
output charset 'UTF-8', use -C <charset> to override
Info: using DVB adapter auto detection.
        /dev/dvb/adapter0/frontend0 -> TERRESTRIAL "Silicon Labs Si2168": very good :-))

Using TERRESTRIAL frontend (adapter /dev/dvb/adapter0/frontend0)
-_-_-_-_ Getting frontend capabilities-_-_-_-_
Using DVB API 5.10
frontend 'Silicon Labs Si2168' supports
DVB-T2
INVERSION_AUTO
QAM_AUTO
TRANSMISSION_MODE_AUTO
GUARD_INTERVAL_AUTO
HIERARCHY_AUTO
FEC_AUTO
BANDWIDTH_AUTO not supported, trying 6/7/8 MHz.
FREQ (42.00MHz ... 870.00MHz)
-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_
Scanning DVB-T...
Scanning 8MHz frequencies...
474000: (time: 00:00.080)
474167: (time: 00:02.084)
473833: (time: 00:04.088)
482000: (time: 00:06.092)
482167: (time: 00:08.096)
481833: (time: 00:10.100)
490000: (time: 00:12.104)
490167: (time: 00:14.104)
489833: (time: 00:16.108)
498000: (time: 00:18.116)
498167: (time: 00:20.116)
497833: (time: 00:22.124)
506000: (time: 00:24.124)
506167: (time: 00:26.128)
505833: (time: 00:28.132)
514000: (time: 00:30.140)
514167: (time: 00:32.144)
513833: (time: 00:34.148)
522000: (time: 00:36.152) ^T
522167: (time: 00:38.152)
521833: (time: 00:40.156)
530000: (time: 00:42.164)
530167: (time: 00:44.168)
529833: (time: 00:46.172)
538000: (time: 00:48.176)
538167: (time: 00:50.176)
537833: (time: 00:52.180)
546000: (time: 00:54.184)
546167: (time: 00:56.188)
545833: (time: 00:58.192)
554000: (time: 01:00.196)
554167: (time: 01:02.200)
553833: (time: 01:04.204)
562000: (time: 01:06.208)
562167: (time: 01:08.212)
561833: (time: 01:10.216)
570000: (time: 01:12.220)
570167: (time: 01:14.224)
569833: (time: 01:16.228)
578000: (time: 01:18.232)
[... 233 lines cut ...]
801833: (time: 09:05.176)
810000: (time: 09:07.180)
810167: (time: 09:09.184)
809833: (time: 09:11.192)
818000: (time: 09:13.196)
818167: (time: 09:15.196)
817833: (time: 09:17.200)
826000: (time: 09:19.208)
826167: (time: 09:21.212)
825833: (time: 09:23.212)
834000: (time: 09:25.220)
834167: (time: 09:27.220)
833833: (time: 09:29.228)
842000: (time: 09:31.232)
842167: (time: 09:33.236)
841833: (time: 09:35.240)
850000: (time: 09:37.244)
850167: (time: 09:39.248)
849833: (time: 09:41.256)
858000: (time: 09:43.260)
858167: (time: 09:45.264)
857833: (time: 09:47.268)

ERROR: Sorry - i couldn't get any working frequency/transponder
 Nothing to scan!!

b-rad-NDi, 2017-07-31:
If you could try some scanning software other than tvheadend and report back with command lines and output I'd appreciate it @skykinetic

tphoney, 2017-07-31:
Thanks for looking into this @skykinetic

tphoney, 2017-07-31:
@b-rad-NDi  Connecting and reattaching the usb tuner causes issues:

[88230.409030] tveeprom: Hauppauge model 204109, rev B3I6, serial# 13909791
[88230.409031] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[88230.409032] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[88230.409032] tveeprom: audio processor is None (idx 0)
[88230.409032] tveeprom: has no radio, has IR receiver, has no IR transmitter
[88230.409034] em28xx 3-2:1.0: dvb ts2 set to isoc mode.
[88230.607558] em28xx 3-2:1.0: Binding DVB extension
[88230.614165] i2c i2c-7: Added multiplexed i2c bus 10
[88230.614168] si2168 7-0064: Silicon Labs Si2168-B40 successfully identified
[88230.614169] si2168 7-0064: firmware version: B 4.0.2
[88230.614986] si2157_probe: probing si2157 tuner fe=ffff895b8e40d028 cfg=ffffb8fe081b7c88 addr=0X60
[88230.614987] si2157_setup_configuration: client=ffff895b8fc7dc00
[88230.614988] si2157_setup_configuration: dev=          (null)
[88230.614995] si2157_attach (10-0060)
[88230.615010] BUG: unable to handle kernel NULL pointer dereference at 0000000000000018
[88230.615047] IP: si2157_attach+0xc9/0x497 [si2157]
[88230.615061] PGD 0

[88230.615074] Oops: 0000 [#1] SMP
[88230.615084] Modules linked in: bluetooth rc_hauppauge em28xx_rc rc_core si2157 si2168 i2c_mux em28xx_dvb dvb_core em28xx tveeprom v4l2_common videodev media msr binfmt_misc intel_rapl x86_pkg_temp_thermal snd_hda_codec_hdmi snd_hda_codec_realtek intel_powerclamp snd_hda_codec_generic eeepc_wmi snd_hda_intel coretemp asus_wmi sparse_keymap snd_hda_codec kvm snd_hda_core irqbypass snd_hwdep snd_pcm crct10dif_pclmul crc32_pclmul ghash_clmulni_intel snd_seq_midi snd_seq_midi_event pcbc aesni_intel snd_rawmidi snd_seq aes_x86_64 snd_seq_device crypto_simd snd_timer glue_helper cryptd intel_cstate intel_rapl_perf snd soundcore mei_me shpchp mei lpc_ich mac_hid parport_pc ppdev lp parport autofs4 dm_mirror dm_region_hash dm_log i915 i2c_algo_bit e1000e ahci drm_kms_helper libahci syscopyarea sysfillrect
[88230.615275]  sysimgblt fb_sys_fops drm ptp pps_core wmi fjes video
[88230.615296] CPU: 2 PID: 4140 Comm: kworker/2:0 Not tainted 4.10.0-28201707281149-generic #0+mediatree+hauppauge
[88230.615321] Hardware name: ASUS All Series/H87I-PLUS, BIOS 2003 11/05/2014
[88230.615342] Workqueue: events request_module_async [em28xx]
[88230.615357] task: ffff895998b4c380 task.stack: ffffb8fe081b4000
[88230.615374] RIP: 0010:si2157_attach+0xc9/0x497 [si2157]
[88230.615389] RSP: 0018:ffffb8fe081b7978 EFLAGS: 00010286
[88230.615403] RAX: ffff895b95ea0938 RBX: ffff895b8e40d028 RCX: 0000000000000006
[88230.615421] RDX: 0000000000000000 RSI: 0000000000000247 RDI: ffffffffc08ae100
[88230.615440] RBP: ffffb8fe081b79e8 R08: ffff895b94520800 R09: 00000000000003b4
[88230.615459] R10: 0000000000000060 R11: 0000000000000000 R12: 0000000000000060
[88230.615477] R13: ffff895b8fc7dc00 R14: 0000000000000000 R15: ffff895b8e40c000
[88230.615496] FS:  0000000000000000(0000) GS:ffff895b9fb00000(0000) knlGS:0000000000000000
[88230.615517] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[88230.615532] CR2: 0000000000000018 CR3: 00000000c3c09000 CR4: 00000000001406e0
[88230.615551] Call Trace:
[88230.615562]  si2157_probe+0x77/0x140 [si2157]
[88230.615575]  ? si2157_setup_configuration.isra.7+0xc0/0xc0 [si2157]
[88230.615594]  i2c_device_probe+0x1c2/0x220
[88230.615607]  driver_probe_device+0x2bb/0x460
[88230.615620]  __device_attach_driver+0x8c/0x100
[88230.615633]  ? __driver_attach+0xf0/0xf0
[88230.615645]  bus_for_each_drv+0x67/0xb0
[88230.615657]  __device_attach+0xdd/0x160
[88230.615669]  device_initial_probe+0x13/0x20
[88230.615682]  bus_probe_device+0x92/0xa0
[88230.615693]  device_add+0x373/0x630
[88230.615704]  device_register+0x1a/0x20
[88230.615715]  i2c_new_device+0x196/0x240
[88230.615728]  em28xx_dvb_init.part.5+0x316/0x3940 [em28xx_dvb]
[88230.615744]  ? load_balance+0x180/0xa00
[88230.615757]  ? update_load_avg+0x6b/0x510
[88230.615770]  ? update_curr+0xf3/0x170
[88230.615781]  ? dequeue_task_fair+0x5ab/0xaa0
[88230.615794]  ? pick_next_task_fair+0x10d/0x4b0
[88230.615807]  ? __switch_to+0x23c/0x520
[88230.615819]  em28xx_dvb_init+0x24/0x30 [em28xx_dvb]
[88230.615834]  em28xx_init_extension+0x66/0xa0 [em28xx]
[88230.615849]  request_module_async+0x22/0x100 [em28xx]
[88230.615864]  process_one_work+0x1fc/0x4b0
[88230.615876]  worker_thread+0x4b/0x500
[88230.615887]  kthread+0x109/0x140
[88230.615897]  ? process_one_work+0x4b0/0x4b0
[88230.615910]  ? kthread_create_on_node+0x60/0x60
[88230.615924]  ret_from_fork+0x2c/0x40
[88230.615935] Code: 8e 17 00 00 49 81 f8 e0 e0 8a c0 0f 84 42 02 00 00 4d 89 c6 eb 10 4d 8b 36 49 81 fe e0 e0 8a c0 0f 84 2d 02 00 00 4d 85 ff 74 eb <49> 8b 46 18 48 85 c0 74 e2 8b 90 68 03 00 00 41 3b 97 68 03 00
[88230.615996] RIP: si2157_attach+0xc9/0x497 [si2157] RSP: ffffb8fe081b7978
[88230.616014] CR2: 0000000000000018
[88230.640714] ---[ end trace 2e6500b52c710cdd ]---

b-rad-NDi, 2017-07-31:
@tphoney : Please make a new issue with this reconnect issue. I will look into it asap tomorrow.

b-rad-NDi, 2017-08-02:
@tphoney : I've put new packages in:
https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0
With a modified T2 auto scan patch that does indeed appear to work now, using standard w_scan.

We are addressing the reconnection oops now.

skykinetic, 2017-08-03:
My apologies for the delay in testing, wasn't around to conduct the test. It seems like to me that T2 channels now can be detected even in T mode, is that the intended behaviour? (my test setup is located somewhere that the signals are weak, but still enough to detect some channels)
```
uname -a
Linux ubuntu-tuner 4.10.0-28201708021632-generic #0+mediatree+hauppauge SMP Wed Aug 2 16:39:16 CDT 2017 x86_64 x86_64 x86_64 GNU/Linux
```
```
sudo w_scan -X -c SG
w_scan -X -c SG
w_scan version 20161022 (compiled for DVB API 5.10)
using settings for SINGAPORE
Country identifier SG not defined. Using defaults.
scan type TERRESTRIAL, channellist 4
output format czap/tzap/szap/xine
WARNING: could not guess your codepage. Falling back to 'UTF-8'
output charset 'UTF-8', use -C <charset> to override
Info: using DVB adapter auto detection.
        /dev/dvb/adapter0/frontend0 -> TERRESTRIAL "Silicon Labs Si2168": very good :-))

Using TERRESTRIAL frontend (adapter /dev/dvb/adapter0/frontend0)
-_-_-_-_ Getting frontend capabilities-_-_-_-_
Using DVB API 5.10
frontend 'Silicon Labs Si2168' supports
DVB-T2
INVERSION_AUTO
QAM_AUTO
TRANSMISSION_MODE_AUTO
GUARD_INTERVAL_AUTO
HIERARCHY_AUTO
FEC_AUTO
BANDWIDTH_AUTO not supported, trying 6/7/8 MHz.
FREQ (42.00MHz ... 870.00MHz)
-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_
Scanning DVB-T...
Scanning 8MHz frequencies...
474000: (time: 00:00.487)
482000: (time: 00:02.507)
490000: (time: 00:04.535)
498000: (time: 00:06.555)
506000: (time: 00:08.619)
514000: (time: 00:10.659)
522000: (time: 00:12.683)
530000: (time: 00:14.731)
538000: (time: 00:16.743)
546000: (time: 00:21.123)
554000: (time: 00:23.175)         signal ok:    QAM_AUTO f = 554000 kHz I999B8C999D999T999G999Y999 (0:0:0)
        QAM_AUTO f = 554000 kHz I999B8C999D999T999G999Y999 (0:0:0) : updating transport_stream_id: -> (0:0:3)
WARNING: received garbage data: crc = 0x57eb8d63; expected crc = 0x65204654
increasing filter timeout to 42 secs (pid:16 table_id:64 table_id_ext:-1).
        QAM_AUTO f = 554000 kHz I999B8C999D999T999G999Y999 (0:0:3) : updating network_id -> (0:12817:3)
        new transponder: (QAM_AUTO f =      0 kHz I999B999C999D0T999G999Y0P0 (8894:12817:1)) 0x4004
        new transponder: (QAM_AUTO f =      0 kHz I999B999C999D0T999G999Y0P0 (8894:12817:2)) 0x4004
        new transponder: (QAM_AUTO f =      0 kHz I999B999C999D0T999G999Y0P0 (8894:12817:3)) 0x4004
        new transponder: (QAM_AUTO f =      0 kHz I999B999C999D0T999G999Y0P0 (8894:12817:4)) 0x4004
562000: (time: 00:32.759)
570000: (time: 00:34.763)         signal ok:    QAM_AUTO f = 570000 kHz I999B8C999D999T999G999Y999 (0:0:0)
        QAM_AUTO f = 570000 kHz I999B8C999D999T999G999Y999 (0:0:0) : updating transport_stream_id: -> (0:0:4)
        QAM_AUTO f = 570000 kHz I999B8C999D999T999G999Y999 (0:0:4) : updating network_id -> (0:12817:4)
578000: (time: 00:35.787)
586000: (time: 00:37.827)         signal ok:    QAM_AUTO f = 586000 kHz I999B8C999D999T999G999Y999 (0:0:0)
        QAM_AUTO f = 586000 kHz I999B8C999D999T999G999Y999 (0:0:0) : updating transport_stream_id: -> (0:0:3)
        QAM_AUTO f = 586000 kHz I999B8C999D999T999G999Y999 (0:0:3) : updating network_id -> (0:12817:3)
594000: (time: 00:38.767)
602000: (time: 00:40.775)
610000: (time: 00:42.790)
618000: (time: 00:44.822)
626000: (time: 00:46.830)
634000: (time: 00:48.858)
642000: (time: 00:50.914)
650000: (time: 00:52.954)
658000: (time: 00:54.974)
[... 64 lines cut ...]
762000: (time: 02:58.717)
770000: (time: 03:00.733)
778000: (time: 03:02.757)
786000: (time: 03:04.817)
794000: (time: 03:06.821)
802000: (time: 03:08.921)
810000: (time: 03:10.953)
818000: (time: 03:12.961)
826000: (time: 03:15.013)
834000: (time: 03:17.053)
842000: (time: 03:19.093)
850000: (time: 03:21.105)
858000: (time: 03:23.173)
tune to: QAM_AUTO f = 554000 kHz I999B999C999D0T999G999Y0P0 (8894:12817:3) (time: 03:25.193)
----------no signal----------
tune to: QAM_AUTO f = 554000 kHz I999B999C999D0T999G999Y0P0 (8894:12817:3) (time: 03:31.245)  (no signal)
----------no signal----------
tune to: QAM_AUTO f = 570000 kHz I999B999C999D0T999G999Y0P0 (8894:12817:4) (time: 03:37.281)
----------no signal----------
tune to: QAM_AUTO f = 570000 kHz I999B999C999D0T999G999Y0P0 (8894:12817:4) (time: 03:43.308)  (no signal)
----------no signal----------

ERROR: Sorry - i couldn't get any working frequency/transponder
 Nothing to scan!!
```

b-rad-NDi, 2017-08-03:
Thank you @skykinetic, the intention is as you describe. To do a T scan/tune, but pick up if a T2 channel if one is available.

tphoney, 2017-08-03:
[image]
WINNER !! the latest files worked perfectly. Apologies for taking time to get back to you

b-rad-NDi, 2017-08-03:
Awesome, closing this issue. I have opened another ticket for the reconnection oops #9

closed 2017-08-03 by b-rad-NDi