b-rad-NDi/Ubuntu-media-tree-kernel-builder #5: WinTV-HVR-1265 Does not work

apgoetz, 2017-07-13:
I stumbled across this project because it is listed on the Hauppage website as supporting the WinTV-HVR-1265 tuner card. However, when I installed the custom kernel in my PC, my tuner card is not recognized. What is necessary to use this card?

`uname -a` output:

    Linux ubuntu-vm 4.10.0-26201707011420-generic #0+mediatree+hauppauge-Ubuntu SMP Sun Jul 2 20:44:17 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

output of dmesg: 

```
[    2.593133] cx23885: cx23885 driver version 0.0.4 loaded
[    3.047657] cx23885: cx23885[0]: Your board isn't known (yet) to the driver.
               cx23885[0]: Try to pick one of the existing card configs via
               cx23885[0]: card=<n> insmod option.  Updating to the latest
               cx23885[0]: version might help as well.
[    3.047658] cx23885: cx23885[0]: Here is a list of valid choices for the card=<n> insmod option:
[    3.047658] cx23885: cx23885[0]:    card=0 -> UNKNOWN/GENERIC
[    3.047658] cx23885: cx23885[0]:    card=1 -> Hauppauge WinTV-HVR1800lp
[    3.047659] cx23885: cx23885[0]:    card=2 -> Hauppauge WinTV-HVR1800
[    3.047659] cx23885: cx23885[0]:    card=3 -> Hauppauge WinTV-HVR1250
[    3.047659] cx23885: cx23885[0]:    card=4 -> DViCO FusionHDTV5 Express
[    3.047659] cx23885: cx23885[0]:    card=5 -> Hauppauge WinTV-HVR1500Q
[    3.047660] cx23885: cx23885[0]:    card=6 -> Hauppauge WinTV-HVR1500
[    3.047660] cx23885: cx23885[0]:    card=7 -> Hauppauge WinTV-HVR1200
[    3.047661] cx23885: cx23885[0]:    card=8 -> Hauppauge WinTV-HVR1700
[    3.047661] cx23885: cx23885[0]:    card=9 -> Hauppauge WinTV-HVR1400
[    3.047661] cx23885: cx23885[0]:    card=10 -> DViCO FusionHDTV7 Dual Express
[    3.047662] cx23885: cx23885[0]:    card=11 -> DViCO FusionHDTV DVB-T Dual Express
[    3.047662] cx23885: cx23885[0]:    card=12 -> Leadtek Winfast PxDVR3200 H
[    3.047662] cx23885: cx23885[0]:    card=13 -> Compro VideoMate E650F
[    3.047663] cx23885: cx23885[0]:    card=14 -> TurboSight TBS 6920
[    3.047663] cx23885: cx23885[0]:    card=15 -> TeVii S470
[    3.047663] cx23885: cx23885[0]:    card=16 -> DVBWorld DVB-S2 2005
[    3.047663] cx23885: cx23885[0]:    card=17 -> NetUP Dual DVB-S2 CI
[    3.047664] cx23885: cx23885[0]:    card=18 -> Hauppauge WinTV-HVR1270
[    3.047664] cx23885: cx23885[0]:    card=19 -> Hauppauge WinTV-HVR1275
[    3.047664] cx23885: cx23885[0]:    card=20 -> Hauppauge WinTV-HVR1255
[    3.047664] cx23885: cx23885[0]:    card=21 -> Hauppauge WinTV-HVR1210
[    3.047665] cx23885: cx23885[0]:    card=22 -> Mygica X8506 DMB-TH
[    3.047665] cx23885: cx23885[0]:    card=23 -> Magic-Pro ProHDTV Extreme 2
[    3.047665] cx23885: cx23885[0]:    card=24 -> Hauppauge WinTV-HVR1850
[    3.047665] cx23885: cx23885[0]:    card=25 -> Compro VideoMate E800
[    3.047666] cx23885: cx23885[0]:    card=26 -> Hauppauge WinTV-HVR1290
[    3.047666] cx23885: cx23885[0]:    card=27 -> Mygica X8558 PRO DMB-TH
[    3.047666] cx23885: cx23885[0]:    card=28 -> LEADTEK WinFast PxTV1200
[    3.047667] cx23885: cx23885[0]:    card=29 -> GoTView X5 3D Hybrid
[    3.047667] cx23885: cx23885[0]:    card=30 -> NetUP Dual DVB-T/C-CI RF
[    3.047667] cx23885: cx23885[0]:    card=31 -> Leadtek Winfast PxDVR3200 H XC4000
[    3.047667] cx23885: cx23885[0]:    card=32 -> MPX-885
[    3.047668] cx23885: cx23885[0]:    card=33 -> Mygica X8502/X8507 ISDB-T
[    3.047668] cx23885: cx23885[0]:    card=34 -> TerraTec Cinergy T PCIe Dual
[    3.047668] cx23885: cx23885[0]:    card=35 -> TeVii S471
[    3.047669] cx23885: cx23885[0]:    card=36 -> Hauppauge WinTV-HVR1255
[    3.047669] cx23885: cx23885[0]:    card=37 -> Prof Revolution DVB-S2 8000
[    3.047669] cx23885: cx23885[0]:    card=38 -> Hauppauge WinTV-HVR4400/HVR5500
[    3.047669] cx23885: cx23885[0]:    card=39 -> AVerTV Hybrid Express Slim HC81R
[    3.047670] cx23885: cx23885[0]:    card=40 -> TurboSight TBS 6981
[    3.047670] cx23885: cx23885[0]:    card=41 -> TurboSight TBS 6980
[    3.047670] cx23885: cx23885[0]:    card=42 -> Leadtek Winfast PxPVR2200
[    3.047671] cx23885: cx23885[0]:    card=43 -> Hauppauge ImpactVCB-e
[    3.047671] cx23885: cx23885[0]:    card=44 -> DViCO FusionHDTV DVB-T Dual Express2
[    3.047671] cx23885: cx23885[0]:    card=45 -> DVBSky T9580
[    3.047672] cx23885: cx23885[0]:    card=46 -> DVBSky T980C
[    3.047672] cx23885: cx23885[0]:    card=47 -> DVBSky S950C
[    3.047672] cx23885: cx23885[0]:    card=48 -> Technotrend TT-budget CT2-4500 CI
[    3.047672] cx23885: cx23885[0]:    card=49 -> DVBSky S950
[    3.047673] cx23885: cx23885[0]:    card=50 -> DVBSky S952
[    3.047673] cx23885: cx23885[0]:    card=51 -> DVBSky T982
[    3.047673] cx23885: cx23885[0]:    card=52 -> Hauppauge WinTV-HVR5525
[    3.047673] cx23885: cx23885[0]:    card=53 -> Hauppauge WinTV Starburst
[    3.047674] cx23885: cx23885[0]:    card=54 -> ViewCast 260e
[    3.047674] cx23885: cx23885[0]:    card=55 -> ViewCast 460e
[    3.047674] cx23885: cx23885[0]:    card=56 -> Hauppauge WinTV-QuadHD-DVB
[    3.047674] cx23885: cx23885[0]:    card=57 -> Hauppauge WinTV-QuadHD-ATSC
[    3.047683] cx23885: CORE cx23885[0]: subsystem: 0070:2a18, board: UNKNOWN/GENERIC [card=0,autodetected]
[    3.225292] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xd0
[    3.225310] cx23885: cx23885[0]/0: found at 0000:00:09.0, rev: 4, irq: 11, latency: 0, mmio: 0x800000000
```

Output of lspci

```
00:09.0 Multimedia video controller [0400]: Conexant Systems, Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb [14f1:8880] (rev 04)
	Subsystem: Hauppauge computer works Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb [0070:2a18]
	Physical Slot: 9
	Flags: bus master, fast devsel, latency 0, IRQ 11
	Memory at 800000000 (64-bit, non-prefetchable) [size=2M]
	Capabilities: [40] Express Endpoint, MSI 00
	Capabilities: [80] Power Management version 3
	Capabilities: [90] Vital Product Data
	Capabilities: [a0] MSI: Enable- Count=1/1 Maskable- 64bit+
	Kernel driver in use: cx23885
	Kernel modules: cx23885
```

b-rad-NDi, 2017-07-13:
1265 support is actively being worked on right now. It is listed prematurely on our website, but the patches are still being reviewed and tested. With any luck I'll push that support up by the weekend.

b-rad-NDi, 2017-07-13:
@apgoetz : 1265 support appears to be working, I hope to have the patches committed tomorrow.

apgoetz, 2017-07-13:
Awesome! Thanks for all your work supporting these cards on linux. It's a lot of work for a small amount of customers, but it is appreciated. 

Let me know when the patches land and I will test them on my machine.

b-rad-NDi, 2017-07-21:
@apgoetz : 1265k4 support has been added to the repo for kernel 4.4. I will be adding all the new patches to 4.8 and 4.10 today.

apgoetz, 2017-07-21:
Awesome! I will give it a try when I get home tonight.

b-rad-NDi, 2017-07-21:
Already for you to compile and test out. I will probably not push a build to the PPA until I hear back from you and our testers.

b-rad-NDi, 2017-07-24:
Builds are located here if you would like to test and report back success:
https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0
You will need to download and install the 4 debs for the kernel you are testing.

apgoetz, 2017-07-25:
Thanks for packaging this, saves a step figuring out the build system. (By the way, what are the steps to build the debian packages from the repo? I looked at the help for the build script but didn't know what order to use the flags). 

Anyways, I installed the packages in a VM  with PCI-E passthrough to my HVR-1265 card, with ubuntu 17.04 and kernel 4.10. Upon boot, I got the following OOPS in the cx23885 driver:

```
[    2.768175] cx23885: CORE cx23885[0]: subsystem: 0070:2a18, board: Hauppauge WinTV-HVR-1265(161111) [card=58,autodetected]
[    2.881979] hid-generic 0003:0627:0001.0001: input,hidraw0: USB HID v0.01 Pointer [QEMU QEMU USB Tablet] on usb-0000:00:05.7-1/input0
[    3.108830] tveeprom: Hauppauge model 161111, rev A1I6, serial# 4035217864
[    3.108832] tveeprom: MAC address is 00:0d:fe:84:89:c8
[    3.108834] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    3.108836] tveeprom: TV standards NTSC(M) ATSC/DVB Digital (eeprom 0x88)
[    3.108837] tveeprom: audio processor is CX23888 (idx 40)
[    3.108838] tveeprom: decoder processor is CX23888 (idx 34)
[    3.108839] tveeprom: has no radio, has IR receiver, has no IR transmitter
[    3.108841] cx23885: cx23885[0]: hauppauge eeprom: model=161111
[    3.118920] cx25840 3-0044: cx23888 A/V decoder found @ 0x88 (cx23885[0])
[    3.734522] cx25840 3-0044: loaded v4l-cx23885-avcore-01.fw firmware (16382 bytes)
[    3.815283] cx23885: cx23885[0]: registered device video0 [v4l2]
[    3.815328] cx23885: cx23885[0]: registered device vbi0
[    3.815442] cx23885: cx23885[0]: alsa: registered ALSA audio device
[    3.815443] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    3.815444] cx23885: cx23885[0]: cx23885 based dvb card
[    3.815444] dvb_register(): port=2
[    3.825349] si2157: probe of 2-0060 failed with error -11
[    3.825354] BUG: unable to handle kernel NULL pointer dereference at 00000000000000a8
[    3.825547] IP: dvb_register+0x2d51/0x4060 [cx23885]
[    3.825694] PGD 0 

[    3.825893] Oops: 0000 [#1] SMP
[    3.825985] Modules linked in: si2157 lgdt3306a tuner cx25840 joydev ppdev hid_generic snd_hda_codec_generic nls_iso8859_1 cx23885(+) altera_ci tda18271 altera_stapl m88ds3103 tveeprom cx2341x videobuf2_dvb dvb_core rc_core v4l2_common i2c_mux videobuf2_dma_sg videobuf2_memops snd_hda_intel videobuf2_v4l2 input_leds serio_raw snd_hda_codec videobuf2_core snd_hda_core snd_hwdep videodev media snd_pcm snd_timer snd soundcore parport_pc i2c_piix4 parport qemu_fw_cfg mac_hid usbhid hid ib_iser rdma_cm iw_cm ib_cm ib_core configfs iscsi_tcp libiscsi_tcp libiscsi scsi_transport_iscsi ip_tables x_tables autofs4 btrfs raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear crct10dif_pclmul crc32_pclmul ghash_clmulni_intel pcbc qxl aesni_intel
[    3.826853]  aes_x86_64 crypto_simd glue_helper cryptd psmouse virtio_blk virtio_net ttm drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm floppy pata_acpi
[    3.827084] CPU: 0 PID: 432 Comm: systemd-udevd Not tainted 4.10.0-26201707211336-generic #0+mediatree+hauppauge
[    3.827206] Hardware name: QEMU Standard PC (i440FX + PIIX, 1996), BIOS 0.0.0 02/06/2015
[    3.827325] task: ffff8ed4fbe82f00 task.stack: ffff9f3940598000
[    3.827447] RIP: 0010:dvb_register+0x2d51/0x4060 [cx23885]
[    3.827562] RSP: 0018:ffff9f394059b840 EFLAGS: 00010246
[    3.827678] RAX: 0000000000000000 RBX: ffff8ed4fb1e97ac RCX: 0000000000000000
[    3.827798] RDX: ffff8ed4ff835400 RSI: ffff8ed4f5c121a8 RDI: ffff8ed4ff835468
[    3.827919] RBP: ffff9f394059b9f8 R08: 0000000000000000 R09: 0000000000000269
[    3.828040] R10: 000000000000a14b R11: 0000000000000000 R12: ffff8ed4fb1e9e90
[    3.828162] R13: ffff8ed4fb1e8000 R14: ffff8ed4ff365000 R15: ffff8ed4fb1e88a0
[    3.828286] FS:  00007f0bbca348c0(0000) GS:ffff8ed4fea00000(0000) knlGS:0000000000000000
[    3.828411] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    3.828533] CR2: 00000000000000a8 CR3: 000000003c2ec000 CR4: 00000000003406f0
[    3.828662] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[    3.828789] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[    3.828915] Call Trace:
[    3.829037]  ? console_unlock+0x2b1/0x4b0
[    3.829162]  ? kvm_sched_clock_read+0x1e/0x30
[    3.829286]  ? sched_clock+0x9/0x10
[    3.829407]  ? sched_clock_cpu+0x99/0xb0
[    3.829529]  ? console_unlock+0x2b6/0x4b0
[    3.829650]  ? vprintk_emit+0x312/0x4a0
[    3.829773]  cx23885_dvb_register+0x145/0x190 [cx23885]
[    3.829898]  cx23885_dev_setup+0xb69/0xfa0 [cx23885]
[    3.830021]  cx23885_initdev+0x158/0x310 [cx23885]
[    3.830144]  local_pci_probe+0x45/0xa0
[    3.830265]  pci_device_probe+0x103/0x150
[    3.830387]  driver_probe_device+0x2bb/0x460
[    3.830508]  __driver_attach+0xdf/0xf0
[    3.830675]  ? driver_probe_device+0x460/0x460
[    3.830796]  bus_for_each_dev+0x6c/0xc0
[    3.830917]  driver_attach+0x1e/0x20
[    3.831080]  bus_add_driver+0x170/0x270
[    3.831200]  driver_register+0x60/0xe0
[    3.831318]  ? 0xffffffffc077c000
[    3.831435]  __pci_register_driver+0x4c/0x50
[    3.831555]  cx23885_init+0x31/0x1000 [cx23885]
[    3.831674]  do_one_initcall+0x52/0x1b0
[    3.831792]  ? __vunmap+0x81/0xd0
[    3.831906]  ? kmem_cache_alloc_trace+0xd7/0x190
[    3.832020]  do_init_module+0x5f/0x200
[    3.832131]  load_module+0x190b/0x1c70
[    3.832238]  ? __symbol_put+0x60/0x60
[... cut ...]
[    3.833480] R13: 0000563e5677ab80 R14: 0000000000020000 R15: 0000563e55c67e30
[    3.833575] Code: 85 c0 74 27 48 8b 78 10 48 89 95 58 fe ff ff e8 46 9e fb d8 84 c0 48 8b 95 58 fe ff ff 0f 85 1e 0f 00 00 48 89 d7 e8 bf 4a 5b d9 <48> 8b 04 25 a8 00 00 00 48 8b 78 10 e8 de 97 fb d8 31 ff e8 a7 
[    3.833796] RIP: dvb_register+0x2d51/0x4060 [cx23885] RSP: ffff9f394059b840
[    3.833897] CR2: 00000000000000a8
[    3.833994] ---[ end trace ffae5677a65482d7 ]---
[   76.234565] random: crng init done
```

b-rad-NDi, 2017-07-25:
I seem to have an issue in the si2157 driver. There's a couple affected devices, I hope to track it down asap. I do have a 1265k4 and another hybrid device now, so I shouldn't have much issue.

I will fix the readme with detailed building steps too, essentially it's this:
<pre># initialize
./mediatree.sh -i
# patch infrastructure and generate new changelog in case of new kernel rev
UPDATE_MT_KBUILD_VER=YES ./mediatree.sh -p
# apply extra patches
./mediatree.sh -x
# clean
./mediatree.sh -c
# build
./mediatree.sh -b
</pre>

b-rad-NDi, 2017-07-25:
I accidentally left out a core chunk of the si2157 analog patch. That is fixed and I am rebuilding 4.4 and 4.10 now, I will post again when they are in the dropbox.

b-rad-NDi, 2017-07-25:
New, hopefully fixed, packages are available here for 4.4 and 4.10:
 https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0

Please comment on your findings.

apgoetz, 2017-07-26:
I was able to install the new kernel, and at a cursory glance, it seems to be working: I see /deb/dvb populated, and no more errors in the kernel log. I will report back once I have had a chance to verify that the card actually works.

b-rad-NDi, 2017-07-26:
Thanks for the update, lemme know if it completely works when you have a chance.

apgoetz, 2017-07-27:
I was able to confirm that the ATSC tuner works! Can scan and stream my local over-the-air channels without any problems. Thanks for all your work to support this hardware. It is nice to finally be able to use this card under Ubuntu!

I think this bug can be marked as closed (at least, it is definitely closed for my use case). I do wonder though, what are the long term plans for this patchset? Are you planning on trying to merge 1265 support into the mediatree project? Currently, I am running this custom Hauppage kernel in an ubuntu vm under a debian hypervisor, but if the HVR-1265 support was moved upstream I would be able to use the default kernel in my distribution.

b-rad-NDi, 2017-07-27:
It is the goal of this project to gather all our hardware support into one spot, and eventually upstream it. I have other products to manage as well, and we lack a dedicated upstreamer, so the process is a little slower than it could be. Rest assured, upstreaming everything in on my list :)

Closing this issue, if you find anything wrong please file another issue.

closed 2017-07-27 by b-rad-NDi