b-rad-NDi/Ubuntu-media-tree-kernel-builder #6: Ooops with Linuxtv.org 2017-07-01-04-20 and new analog patch

b-rad-NDi, 2017-07-21:
`[    0.000000] Linux version 4.4.0-83201706301514-generic (root@mkultra) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4) ) #0+mediatree+hauppauge SMP Fri Jun 30 15:15:47 CDT 2017 (Ubuntu 4.4.0-83201706301514.0+mediatree+hauppauge-generic 4.4.70)

[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.4.0-83201706301514-generic root=UUID=6faa5d35-041a-49f4-8bb3-28ff6bf9f2da ro quiet splash vt.handoff=7

[    1.952747] Linux video capture interface: v2.00
[    1.952754] WARNING: You are using an experimental version of the media stack.
                As the driver is backported to an older kernel, it doesn't offer
                enough quality for its usage in production.
                Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
                2748e76ddb2967c4030171342ebdd3faa6a5e8e8 media: staging: cxd2099: Activate cxd2099 buffer mode
                3e8d8a085ff454527a422aafe9b7dcfa95c09cc5 media: staging: cxd2099: Removed printing in write_block
                6475483734d86f88848fcd61db3d4c35123abcba media: staging: cxd2099: Removed useless printing in cxd2099 driver

[    1.981818] v4l2_common: unknown parameter 'debug' ignored
[    1.987570] WARNING: You are using an experimental version of the media stack.
                As the driver is backported to an older kernel, it doesn't offer
                enough quality for its usage in production.
                Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
                2748e76ddb2967c4030171342ebdd3faa6a5e8e8 media: staging: cxd2099: Activate cxd2099 buffer mode
                3e8d8a085ff454527a422aafe9b7dcfa95c09cc5 media: staging: cxd2099: Removed printing in write_block
                6475483734d86f88848fcd61db3d4c35123abcba media: staging: cxd2099: Removed useless printing in cxd2099 driver

[    2.027605] WARNING: You are using an experimental version of the media stack.
                As the driver is backported to an older kernel, it doesn't offer
                enough quality for its usage in production.
                Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
                2748e76ddb2967c4030171342ebdd3faa6a5e8e8 media: staging: cxd2099: Activate cxd2099 buffer mode
                3e8d8a085ff454527a422aafe9b7dcfa95c09cc5 media: staging: cxd2099: Removed printing in write_block
                6475483734d86f88848fcd61db3d4c35123abcba media: staging: cxd2099: Removed useless printing in cxd2099 driver

[    2.080397] cx23885: cx23885 driver version 0.0.4 loaded
[    2.080492] cx23885: cx23885_dev_setup: cx23885_dev_setup() Memory configured for PCIe bridge type 887
[    2.080493] cx23885: cx23885_init_tsport: cx23885_init_tsport(portno=2)
[    2.080501] cx23885: CORE cx23885[0]: subsystem: 0070:2211, board: Hauppauge WinTV-HVR1270 [card=18,autodetected]
[    2.080502] cx23885: cx23885_pci_quirks: cx23885_pci_quirks()
[    2.080502] cx23885: cx23885_dev_setup: cx23885_dev_setup() tuner_type = 0x0 tuner_addr = 0x0 tuner_bus = 0
[    2.080503] cx23885: cx23885_dev_setup: cx23885_dev_setup() radio_type = 0x0 radio_addr = 0x0
[    2.080504] cx23885: cx23885_reset: cx23885_reset()

[    2.180182] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() Configuring channel [VID A]
[    2.180187] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() Configuring channel [VID A (VBI)]
[    2.180210] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() Configuring channel [TS1 B]
[    2.180213] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() Erasing channel [ch4]
[    2.180215] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() Erasing channel [ch5]
[    2.180216] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() Configuring channel [TS2 C]
[    2.180231] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() Configuring channel [TV Audio]
[    2.180249] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() Erasing channel [ch8]
[    2.180250] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() Erasing channel [ch9]

[    2.229684] tveeprom: Hauppauge model 22651, rev C1F5, serial# 4031911606
[    2.229685] tveeprom: MAC address is 00:0d:fe:52:16:b6
[    2.229686] tveeprom: tuner model is NXP 18271C2 (idx 155, type 54)
[    2.229686] tveeprom: TV standards NTSC(M) ATSC/DVB Digital (eeprom 0x88)
[    2.229687] tveeprom: audio processor is CX23888 (idx 40)
[    2.229687] tveeprom: decoder processor is CX23888 (idx 34)
[    2.229688] tveeprom: has no radio, has no IR receiver, has no IR transmitter
[    2.229689] cx23885: cx23885[0]: warning: unknown hauppauge model #22651
[    2.229689] cx23885: cx23885[0]: hauppauge eeprom: model=22651
[    2.245511] cx25840 2-0044: cx23888 A/V decoder found @ 0x88 (cx23885[0])

[    2.245522] general protection fault: 0000 [#1] SMP
[    2.245524] Modules linked in: cx25840 btusb btrtl nvidia_uvm(POE) snd_hda_codec_hdmi(+) cx23885(+) snd_hda_codec_realtek(+) altera_ci snd_hda_codec_generic tda18271 altera_stapl arc4 m88ds3103 intel_rapl tveeprom i915_bpo cx2341x videobuf2_dvb x86_pkg_temp_thermal intel_powerclamp dvb_core rtl8723be coretemp rc_core btcoexist v4l2_common rtl8723_common i2c_mux rtl_pci intel_ips videobuf2_dma_sg i2c_algo_bit rtlwifi kvm videobuf2_memops videobuf2_v4l2 mac80211 videobuf2_core snd_hda_intel snd_hda_codec videodev snd_hda_core irqbypass media hp_wmi snd_hwdep sparse_keymap cfg80211 crct10dif_pclmul snd_pcm crc32_pclmul snd_seq_midi ghash_clmulni_intel snd_seq_midi_event snd_rawmidi aesni_intel snd_seq aes_x86_64 lrw snd_seq_device gf128mul snd_timer glue_helper ablk_helper snd cryptd hci_uart input_leds
[    2.245541]  soundcore btbcm shpchp mei_me serio_raw btqca mei btintel wmi bluetooth acpi_als kfifo_buf industrialio intel_lpss_acpi mac_hid tpm_crb intel_lpss acpi_pad parport_pc ppdev lp parport autofs4 hid_generic usbhid nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) drm_kms_helper syscopyarea sysfillrect psmouse sysimgblt fb_sys_fops r8169 ahci drm mii libahci video i2c_hid pinctrl_sunrisepoint pinctrl_intel hid fjes
[    2.245552] CPU: 6 PID: 289 Comm: systemd-udevd Tainted: P           OE   4.4.0-83201706301514-generic #0+mediatree+hauppauge
[    2.245553] Hardware name: HP 750-175se/2B4B, BIOS A0.05 09/08/2015
[    2.245554] task: ffff8800350e1e00 ti: ffff88035ee2c000 task.ti: ffff88035ee2c000
[    2.245555] RIP: 0010:[<ffffffffc113c085>]  [<ffffffffc113c085>] find_ref+0x55/0xc0 [videodev]
[... 46 lines cut ...]
[    2.245710]  [<ffffffffc1327000>] ? 0xffffffffc1327000
[    2.245712]  [<ffffffff81563da0>] driver_register+0x60/0xe0
[    2.245714]  [<ffffffff8144b17c>] __pci_register_driver+0x4c/0x50
[    2.245719]  [<ffffffffc1327031>] cx23885_init+0x31/0x1000 [cx23885]
[    2.245722]  [<ffffffff81002123>] do_one_initcall+0xb3/0x200
[    2.245739]  [<ffffffff811d1541>] ? __vunmap+0x91/0xe0
[    2.245742]  [<ffffffff811ed753>] ? kmem_cache_alloc_trace+0x183/0x1f0
[    2.245744]  [<ffffffff811ee54a>] ? kfree+0x13a/0x150
[    2.245746]  [<ffffffff8118e013>] do_init_module+0x5f/0x1cf
[    2.245749]  [<ffffffff8110acff>] load_module+0x166f/0x1c10
[    2.245750]  [<ffffffff811072a0>] ? __symbol_put+0x60/0x60
[    2.245753]  [<ffffffff812153d0>] ? kernel_read+0x50/0x80
[    2.245755]  [<ffffffff8110b4e4>] SYSC_finit_module+0xb4/0xe0
[    2.245757]  [<ffffffff8110b52e>] SyS_finit_module+0xe/0x10
[    2.245760]  [<ffffffff81841e72>] entry_SYSCALL_64_fastpath+0x16/0x71
[    2.245761] Code: 8b 4f 58 48 85 c9 74 45 44 0f b7 47 70 31 d2 89 f0 41 f7 f0 48 63 d2 48 8b 04 d1 48 85 c0 75 0b eb 2b 48 8b 40 10 48 85 c0 74 22 <48> 8b 50 18 3b 72 48 75 ee 48 89 47 50 5d c3 48 8b 47 40 48 83
[    2.245782] RIP  [<ffffffffc113c085>] find_ref+0x55/0xc0 [videodev]
[    2.245788]  RSP <ffff88035ee2f598>
[    2.245789] ---[ end trace 63698be803353a1b ]---
[    2.286718] atkbd serio0: Unknown key released (translated set 2, code 0x7c on isa0060/serio0).
[    2.286721] atkbd serio0: Use 'setkeycodes 7c <keycode>' to make it known.

[   67.907725] do_trap: 36 callbacks suppressed
[   67.907735] traps: pool[2012] trap int3 ip:7f5d35603a6b sp:7f5d12c276f0 error:0
`

b-rad-NDi, 2017-07-21:
Something happened between 06/20 and 07/01, awaiting hardware to debug and get back in sync with Linuxtv.org media tree.

mspieth, 2017-09-23:
I have some patches for this (using all 22). How would you like them?
They arent perfect for the quadhd 885 (166200/1) 2157A30 2168B40 but they don't crash anymore.
1. rmmod cx23885 used to fail with an OOPS.
this is due to a huge bug in cx23885-dvb.c and i2c platform data. this data is stored in the i2c_new_device but is allocated on the stack for register. I suspect this is prevalent throughout the whole set of drivers.
2. read_status failed due to auto dvbt/t2. this is due to a change in status between 8701 and 5001 thus making the actual fetch fail.
3. also missing is legacy read_signal strength et al. not done yet.
4. also noted that release on 2157 is being called twice which also caused problems.
Note I'm running mythtv with 4.12.6 (debian testing)
regards
Mark

b-rad-NDi, 2017-09-25:
Thanks Mark. I have been extremely busy all summer and hadn't gotten a chance to dedicate time to this. I will review your patch set.

b-rad-NDi, 2017-09-26:
Hi Mark. The i2c lifetime bug is fairly savage. Since it is a driver wide issue I'm going to apply it before I make my changes, then adapt my changes to fit the new allocator.

For the T2 auto fix, i'm going to work that into my T2 auto patch and get our euro testers to give it a shot.

I'll be updating the linux-oops branch with the latest work as I'm testing this out. Thanks again, fixing this had just made its way to the front of my queue. If you want to make separate issues for the other pieces you say are missing or broken still I can look into whatever you're not currently sorting.

b-rad-NDi, 2017-10-05:
@mspieth : I have applied your cx23885 i2c alloc patch. I added some missing kfrees. It is 0012-cx23885-i2c-platform-data-cannot-be-stack-allocated.patch in zesty extra.

It is in the linuxtv-oops-with-extra-applied branch.

mspieth, 2017-10-06:
You missed a couple. 

- & when not required/should not be there
- extra spaces
- double ;;
- null check on si2157 remove

Patch attached.

[cx23885_i2c_lifetime.patch.zip](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1361240/cx23885_i2c_lifetime.patch.zip)

b-rad-NDi, 2017-10-06:
The si2157 bit is part of the corresponding patch, as a sanity check. I will fix the other three parts tomorrow.

b-rad-NDi, 2017-10-06:
I fixed those 3 issues. I also added the same i2c platform data fix to cx231xx driver, as it has the same problem. 

Unfortunately the i2c lifetime issue is not the sole source of the oops in this issue. I am continuing to sort out why I get a very similar oops with one of our usb devices.

b-rad-NDi, 2017-10-06:
I reversed this patch:
https://patchwork.linuxtv.org/patch/41960
and now all the devices which were previously oopsing are perfectly fine.
Once I noticed this was affecting a lot of devices I had, each using independent unrelated drivers, I identified that it had to be a core issue. This patch was coincidentally applied briefly after the last working linuxtv patch I made, so it was reviewed quickly and immediately made me suspicious.

b-rad-NDi, 2017-10-07:
I am closing this issue. The reverse of the kvmalloc patch has fixed this particular oops. Everything appears all good with the various devices tested. I will continue to investigate your other issues.

All changes are applied and a fresh linuxtv.org patch was generated in the linuxtv-oops-with-extra-applied directory.

closed 2017-10-07 by b-rad-NDi