b-rad-NDi/Ubuntu-media-tree-kernel-builder #7: Duplicate input device for WinTV-dualHD

skykinetic, 2017-07-27:
I have the DVB version of the device, dvb-t2 is working correctly with both tuners active. Without the patch only one tuner works, along with one IR input device. With the patch, two tuners work but there is an additional IR input device, resulting in duplicate input events when the IR remote is used. IR remote works well without additional configuration when patch is not applied. 
```
[    3.980948] usb 1-1.4: New USB device found, idVendor=2040, idProduct=0265
[    3.996450] usb 1-1.4: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[    4.012267] usb 1-1.4: Product: dualHD                                                                       
[    4.024591] usb 1-1.4: Manufacturer: HCW                                                                     
[    4.036599] usb 1-1.4: SerialNumber: 0011513864                                                              
[    4.481791] systemd[1]: System time before build time, advancing clock.                                      
[    4.579405] NET: Registered protocol family 10                                                               
[    4.605017] ip_tables: (C) 2000-2006 Netfilter Core Team                                                     
[    4.661160] systemd[1]: systemd 233 running in system mode. (+PAM -AUDIT -SELINUX -IMA -APPARMOR +SMACK -SYSV
INIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 -SECCOMP +BLKID +ELFUTILS +KMOD +IDN default-hierarchy=h
ybrid)                                                                                                          
[    4.698583] systemd[1]: Detected architecture arm.                                                           
[    4.742739] systemd[1]: Set hostname to <tvheadend2>.                                                          
[    4.841565] uart-pl011 3f201000.serial: no DMA platform data                                                 
[    5.226826] systemd[1]: /usr/lib/systemd/system/tvheadend.service:6: Unknown lvalue 'OnBootSec' in section 'S
ervice'                                                                                                         
[    5.308406] systemd[1]: Set up automount Arbitrary Executable File Formats File System Automount Point.      
[    5.337200] systemd[1]: Listening on /dev/initctl Compatibility Named Pipe.                                  
[    5.363008] systemd[1]: Listening on Journal Socket (/dev/log).                                              
[    5.390859] systemd[1]: Created slice User and Session Slice.                                                
[    5.415691] systemd[1]: Started Dispatch Password Requests to Console Directory Watch.                       
[    5.442771] systemd[1]: Listening on udev Control Socket.                                                                                            
[    6.154109] systemd-journald[168]: Received request to flush runtime journal from PID 1                      
[    6.224719] systemd-journald[168]: File /var/log/journal/27e15a9e19f94de287e28f78982584a9/system.journal corr
upted or uncleanly shut down, renaming and replacing.                                                           
[    6.992419] gpiomem-bcm2835 3f200000.gpiomem: Initialised: Registers at 0x3f200000                           
[    7.310251] media: Linux media interface: v0.10                                                              
[    7.384338] Linux video capture interface: v2.00                                                             
[    7.422841] em28xx: New device HCW dualHD @ 480 Mbps (2040:0265, interface 0, class 0)                       
[    7.438136] em28xx: DVB interface 0 found: isoc                                                              
[    7.450159] em28xx: chip ID is em28174                                                                                                
[    8.791710] em28174 #0: EEPROM ID = 26 00 01 00, EEPROM hash = 0x68e3c8c8                                    
[    8.805354] em28174 #0: EEPROM info:                                                                         
[    8.815532] em28174 #0:      microcode start address = 0x0004, boot configuration = 0x01                     
[    8.837585] em28174 #0:      AC97 audio (5 sample rates)                                                     
[    8.849367] em28174 #0:      500mA max power                                                                 
[    8.859699] em28174 #0:      Table at offset 0x27, strings=0x0e6a, 0x1888, 0x087e                            
[    8.877345] em28174 #0: Identified as Hauppauge WinTV-dualHD DVB (card=99)                                   
[    8.900017] tveeprom 4-0050: Hauppauge model 204109, rev B2I6, serial# 11513864                              
[    8.913899] tveeprom 4-0050: tuner model is SiLabs Si2157 (idx 186, type 4)                                  
[    8.927272] tveeprom 4-0050: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)                                                                                                     
[    8.944897] tveeprom 4-0050: audio processor is None (idx 0)                                                 
[    8.957090] tveeprom 4-0050: has no radio, has IR receiver, has no IR transmitter                            
[    8.971380] em28174 #0: dvb set to isoc mode.                                                                
[    8.983163] em28xx: chip ID is em28174                                                                                      
[   10.307837] em28174 #1: EEPROM ID = 26 00 01 00, EEPROM hash = 0x68e3c8c8                                    
[   10.314966] em28174 #1: EEPROM info:                                                                         
[   10.318680] em28174 #1:      microcode start address = 0x0004, boot configuration = 0x01                     
[   10.337810] em28174 #1:      AC97 audio (5 sample rates)                                                     
[   10.343029] em28174 #1:      500mA max power                                                                 
[   10.347083] em28174 #1:      Table at offset 0x27, strings=0x0e6a, 0x1888, 0x087e                            
[   10.358075] em28174 #1: Identified as Hauppauge WinTV-dualHD DVB (card=99)                                   
[   10.375399] tveeprom 6-0050: Hauppauge model 204109, rev B2I6, serial# 11513864                              
[   10.378946] random: crng init done                                                                           
[   10.386614] tveeprom 6-0050: tuner model is SiLabs Si2157 (idx 186, type 4)                                  
[   10.393909] tveeprom 6-0050: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)                                                                                                     
[   10.405214] tveeprom 6-0050: audio processor is None (idx 0)                                                 
[   10.411141] tveeprom 6-0050: has no radio, has IR receiver, has no IR transmitter                            
[   10.418871] em28174 #0: dvb ts2 set to isoc mode.                                                            
[   10.625515] usbcore: registered new interface driver em28xx                                                  
[   10.673576] em28174 #0: Binding DVB extension                                                                
[   10.705272] i2c i2c-4: Added multiplexed i2c bus 7                                                           
[   10.710497] si2168 4-0064: Silicon Labs Si2168-B40 successfully identified                                   
[   10.717636] si2168 4-0064: firmware version: B 4.0.2                                                         
[   10.745709] si2157 7-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached                          
[   10.753948] DVB: registering new adapter (em28174 #0)                                                        
[   10.759254] usb 1-1.4: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...                        
[   10.774747] em28174 #0: DVB extension successfully initialized                                               
[... 7 lines cut ...]
[   10.881516] em28174 #1: DVB extension successfully initialized                                               
[... cut ...]
[   14.218483] si2168 4-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'                       
[   14.600342] si2168 4-0064: firmware version: B 4.0.11                                                        
[   14.613436] si2157 7-0060: found a 'Silicon Labs Si2157-A30'                                                 
[   14.669099] si2157 7-0060: firmware version: 3.0.5                                                           
[   14.674263] usb 1-1.4: DVB: adapter 0 frontend 0 frequency 0 out of range (42000000..870000000)
```
Thanks a lot for working on the Linux drivers.

b-rad-NDi, 2017-07-27:
Thanks for T2 confirmation. I will sort out the duplicate IR device asap.

b-rad-NDi, 2017-07-27:
There are packages you can test here, duplicated IR should be fixed:
https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0

For Xenial grab the 4.4.0-87 packages.

For Zesty grab the 4.10.0-28 packages.

Please let me know this takes care of you.

skykinetic, 2017-07-28:
I'm running a fresh install of Zesty server edition for testing, installed lirc to test the IR using the `irw` command, disabled the lircd daemon to prevent any other source of inputs. 

# Stock kernel
### dmesg
```
[    5.139095] usb 1-1: New USB device found, idVendor=2040, idProduct=0265
[    5.139098] usb 1-1: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[    5.139100] usb 1-1: Product: dualHD
[    5.139102] usb 1-1: Manufacturer: HCW
[    5.139103] usb 1-1: SerialNumber: 0011513864
[    5.192574] usb 2-1: not running at top speed; connect to a high speed hub
[   14.224875] piix4_smbus 0000:00:01.3: SMBus Host Controller at 0x700, revision 0
[   14.510728] shpchp: Standard Hot Plug PCI Controller Driver version: 0.4
[   14.517682] media: Linux media interface: v0.10
[   14.611009] Linux video capture interface: v2.00
[   15.029349] em28xx 1-1:1.0: New device HCW dualHD @ 480 Mbps (2040:0265, interface 0, class 0)
[   15.033131] em28xx 1-1:1.0: DVB interface 0 found: isoc
[   15.036877] em28xx 1-1:1.0: chip ID is em28174
[   16.311467] em28xx 1-1:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0x68e3c8c8
[   16.311469] em28xx 1-1:1.0: EEPROM info:
[   16.311472] em28xx 1-1:1.0:  microcode start address = 0x0004, boot configuration = 0x01
[   16.433378] em28xx 1-1:1.0:  AC97 audio (5 sample rates)
[   16.433382] em28xx 1-1:1.0:  500mA max power
[   16.433387] em28xx 1-1:1.0:  Table at offset 0x27, strings=0x0e6a, 0x1888, 0x087e
[   16.433674] em28xx 1-1:1.0: Identified as Hauppauge WinTV-dualHD DVB (card=99)
[   16.439828] tveeprom: Hauppauge model 204109, rev B2I6, serial# 11513864
[   16.439833] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[   16.439835] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[   16.439837] tveeprom: audio processor is None (idx 0)
[   16.439838] tveeprom: has no radio, has IR receiver, has no IR transmitter
[   16.439847] em28xx 1-1:1.0: dvb set to isoc mode.
[   16.474885] usbcore: registered new interface driver em28xx
[   16.564931] em28xx 1-1:1.0: Binding DVB extension
[   16.774383] i2c i2c-2: Added multiplexed i2c bus 3
[   16.774389] si2168 2-0064: Silicon Labs Si2168-B40 successfully identified
[   16.774391] si2168 2-0064: firmware version: B 4.0.2
[   16.914539] si2157 3-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[   16.914564] dvbdev: DVB: registering new adapter (1-1:1.0)
[   16.914572] em28xx 1-1:1.0: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[   16.915574] em28xx 1-1:1.0: DVB extension successfully initialized
[   16.915579] em28xx: Registered (Em28xx dvb Extension) extension
[   16.997605] em28xx 1-1:1.0: Registering input extension
[   17.071815] Registered IR keymap rc-hauppauge
[   17.072099] input: 1-1:1.0 IR as /devices/pci0000:00/0000:00:1d.7/usb1/1-1/1-1:1.0/rc/rc0/input6
[   17.072405] rc rc0: 1-1:1.0 IR as /devices/pci0000:00/0000:00:1d.7/usb1/1-1/1-1:1.0/rc/rc0
[   17.081055] em28xx 1-1:1.0: Input extension successfully initalized
[   17.081060] em28xx: Registered (Em28xx Input Extension) extension
[   17.109328] cgroup: new mount options do not match the existing superblock, will be ignored
```
`cat /proc/bus/input/devices`
```
I: Bus=0003 Vendor=2040 Product=0265 Version=0001
N: Name="1-1:1.0 IR"
P: Phys=usb-0000:00:1d.7-1/input0
S: Sysfs=/devices/pci0000:00/0000:00:1d.7/usb1/1-1/1-1:1.0/rc/rc0/input6
U: Uniq=
H: Handlers=kbd event5
B: PROP=0
B: EV=100013
B: KEY=10afc312 214201700000000 0 118000 41a800004801 9e16c000000000 10000ffc
B: MSC=10
```

Running `irw` showed that the IR remote works and are responsive, keys will not have any duplicates when short pressing, keys repeat correctly when held down. The button repeat behaviour is such that long presses without letting go repeats keypresses at constant intervals, upon releasing the button will have an addtional repeat registered. None of the keys become a keyboard input at any point in time. I will take this behavious as the reference one when comparing. 

# kernel version 4.10.0-26 from ppa
### dmesg
```
[    5.121498] usb 1-1: Product: dualHD
[    5.121500] usb 1-1: Manufacturer: HCW
[... 198 lines cut ...]
N: Name="1-1:1.0 IR"
P: Phys=usb-0000:00:1d.7-1/input0
S: Sysfs=/devices/pci0000:00/0000:00:1d.7/usb1/1-1/1-1:1.0/rc/rc0/input6
U: Uniq=
H: Handlers=kbd event5
B: PROP=0
B: EV=100013
B: KEY=10afc312 214201700000000 0 118000 41a800004801 9e16c000000000 10000ffc
B: MSC=10

I: Bus=0003 Vendor=2040 Product=0265 Version=0001
N: Name="1-1:1.0 IR"
P: Phys=usb-0000:00:1d.7-1/input0
S: Sysfs=/devices/pci0000:00/0000:00:1d.7/usb1/1-1/1-1:1.0/rc/rc1/input7
U: Uniq=
H: Handlers=kbd event6
B: PROP=0
B: EV=100013
B: KEY=10afc312 214201700000000 0 118000 41a800004801 9e16c000000000 10000ffc
B: MSC=10
```

Running `irw` shows that the IR remote works, however some button presses are not responsive somehow, as if it is not registered. Long presses work correctly with repeats at somewhat constant intervals like stock kernel, just that sometimes the first key is missed. The missing keys because keyboard inputs into the terminal most of the time, or sometimes become registered by both irw and in the terminal. 

My observation is that the IR device is still being duplicated. Is it possible to prevent the duplication from happening at all?

b-rad-NDi, 2017-07-28:
Interesting it is still being doubled. After em28xx_duplicate_dev I am setting ir_codes = 0 and has_ir_i2c = 0, and that seems like it should have disabled the second input extension. I will look further to see why that's not working.

The dropbox is a staging area, so it's usually best to only try out the version with the newest time stamp.

b-rad-NDi, 2017-07-28:
Ok, I see what's wrong. The values are reset a little after I unset them.

b-rad-NDi, 2017-07-28:
New packages are in dropbox:
https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0

You can inspect the current 4.10 patch here:
566d8f2c4c43851fd07854cb3caa1da06894a668

its after em28xx_init_dev() now, so the values shouldn't get overwritten and the second IR input should not be created.

skykinetic, 2017-07-29:
Yup, this did the trick, thanks a lot.

closed 2017-07-29 by skykinetic