b-rad-NDi/Ubuntu-media-tree-kernel-builder #9: SoloHD/DualHD oops on reconnect

b-rad-NDi, 2017-08-03:
[  110.650858] Oops: 0002 [#1] SMP
[  110.650860] Modules linked in: nls_utf8 udf rc_hauppauge em28xx_rc rc_core si2157 lgdt3306a i2c_mux em28xx_dvb dvb_core snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep snd_pcm edac_mce_amd edac_core snd_seq_midi snd_seq_midi_event snd_rawmidi kvm irqbypass em28xx tveeprom input_leds v4l2_common videodev media serio_raw k10temp snd_seq nvidia_uvm(POE) snd_seq_device snd_timer snd wmi soundcore shpchp i2c_piix4 mac_hid parport_pc ppdev lp parport ip_tables x_tables autofs4 hid_generic usbhid hid nvidia_drm(POE) nvidia_modeset(POE) pata_acpi nvidia(POE) drm_kms_helper syscopyarea r8169 mii pata_jmicron firewire_ohci sysfillrect sysimgblt firewire_core fb_sys_fops crc_itu_t pata_atiixp drm ahci libahci fjes

[  110.650903] CPU: 2 PID: 44 Comm: kworker/2:1 Tainted: P           OE   4.10.0-26201707251504-generic #0+mediatree+hauppauge
[  110.650905] Hardware name: Gigabyte Technology Co., Ltd. GA-870A-UD3/GA-870A-UD3, BIOS F4 11/10/2010
[  110.650914] Workqueue: events request_module_async [em28xx]
[  110.650916] task: ffff9271aa67bfc0 task.stack: ffffb4d7c07e8000
[  110.650919] RIP: 0010:mutex_lock+0x1d/0x40
[  110.650921] RSP: 0018:ffffb4d7c07eb920 EFLAGS: 00010246
[  110.650923] RAX: 0000000000000000 RBX: 0000000000000030 RCX: 0000000000000006
[  110.650925] RDX: ffff9271aa67bfc0 RSI: ffffb4d7c07eb990 RDI: 0000000000000030
[  110.650927] RBP: ffffb4d7c07eb928 R08: 0000000000000001 R09: 000000000000042e
[  110.650929] R10: 0000000000000060 R11: 0000000000000000 R12: ffff92713e7d3400
[  110.650931] R13: ffff92713e7d3400 R14: 0000000000000030 R15: ffff9271a475a000
[  110.650934] FS:  0000000000000000(0000) GS:ffff9271afc80000(0000) knlGS:0000000000000000
[  110.650936] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  110.650938] CR2: 0000000000000030 CR3: 0000000122001000 CR4: 00000000000006e0
[  110.650940] Call Trace:
[  110.650946]  si2157_cmd_execute+0x2f/0x190 [si2157]
[  110.650950]  si2157_attach+0x205/0x497 [si2157]
[  110.650953]  si2157_probe+0x77/0x140 [si2157]
[  110.650956]  ? si2157_setup_configuration.isra.7+0xc0/0xc0 [si2157]
[  110.650961]  i2c_device_probe+0x1c2/0x220
[  110.650965]  driver_probe_device+0x2bb/0x460
[  110.650968]  __device_attach_driver+0x8c/0x100
[  110.650971]  ? __driver_attach+0xf0/0xf0
[  110.650974]  bus_for_each_drv+0x67/0xb0
[  110.650977]  __device_attach+0xdd/0x160
[  110.650980]  device_initial_probe+0x13/0x20
[  110.650983]  bus_probe_device+0x92/0xa0
[  110.650986]  device_add+0x373/0x630
[  110.650989]  device_register+0x1a/0x20
[  110.650991]  i2c_new_device+0x196/0x240
[  110.650995]  em28xx_dvb_init.part.5+0x316/0x3940 [em28xx_dvb]
[  110.650999]  ? __srcu_notifier_call_chain+0x5b/0x70
[  110.651003]  ? __wake_up+0x44/0x50
[  110.651006]  em28xx_dvb_init+0x24/0x30 [em28xx_dvb]
[  110.651011]  em28xx_init_extension+0x66/0xa0 [em28xx]
[  110.651015]  request_module_async+0x22/0x100 [em28xx]
[  110.651018]  process_one_work+0x1fc/0x4b0
[  110.651021]  worker_thread+0x4b/0x500
[  110.651024]  kthread+0x109/0x140
[  110.651027]  ? process_one_work+0x4b0/0x4b0
[  110.651030]  ? kthread_create_on_node+0x60/0x60
[  110.651032]  ret_from_fork+0x2c/0x40
[  110.651034] Code: e9 74 fe ff ff e8 f4 17 7b ff 0f 1f 40 00 0f 1f 44 00 00 55 48 89 e5 53 48 89 fb e8 7e e7 ff ff 65 48 8b 14 25 00 d3 00 00 31 c0 <f0> 48 0f b1 13 48 85 c0 74 08 48 89 df e8 31 fd ff ff 5b 5d c3

[  110.651064] RIP: mutex_lock+0x1d/0x40 RSP: ffffb4d7c07eb920
[  110.651066] CR2: 0000000000000030
[  110.651068] ---[ end trace b2da95bd29d44b98 ]---

b-rad-NDi, 2017-08-03:
<pre>[88230.409030] tveeprom: Hauppauge model 204109, rev B3I6, serial# 13909791
[88230.409031] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[88230.409032] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[88230.409032] tveeprom: audio processor is None (idx 0)
[88230.409032] tveeprom: has no radio, has IR receiver, has no IR transmitter
[88230.409034] em28xx 3-2:1.0: dvb ts2 set to isoc mode.
[88230.607558] em28xx 3-2:1.0: Binding DVB extension
[88230.614165] i2c i2c-7: Added multiplexed i2c bus 10
[88230.614168] si2168 7-0064: Silicon Labs Si2168-B40 successfully identified
[88230.614169] si2168 7-0064: firmware version: B 4.0.2
[88230.614986] si2157_probe: probing si2157 tuner fe=ffff895b8e40d028 cfg=ffffb8fe081b7c88 addr=0X60
[88230.614987] si2157_setup_configuration: client=ffff895b8fc7dc00
[88230.614988] si2157_setup_configuration: dev= (null)
[88230.614995] si2157_attach (10-0060)
[88230.615010] BUG: unable to handle kernel NULL pointer dereference at 0000000000000018
[88230.615047] IP: si2157_attach+0xc9/0x497 [si2157]
[88230.615061] PGD 0

[88230.615074] Oops: 0000 [#1] SMP
[88230.615084] Modules linked in: bluetooth rc_hauppauge em28xx_rc rc_core si2157 si2168 i2c_mux em28xx_dvb dvb_core em28xx tveeprom v4l2_common videodev media msr binfmt_misc intel_rapl x86_pkg_temp_thermal snd_hda_codec_hdmi snd_hda_codec_realtek intel_powerclamp snd_hda_codec_generic eeepc_wmi snd_hda_intel coretemp asus_wmi sparse_keymap snd_hda_codec kvm snd_hda_core irqbypass snd_hwdep snd_pcm crct10dif_pclmul crc32_pclmul ghash_clmulni_intel snd_seq_midi snd_seq_midi_event pcbc aesni_intel snd_rawmidi snd_seq aes_x86_64 snd_seq_device crypto_simd snd_timer glue_helper cryptd intel_cstate intel_rapl_perf snd soundcore mei_me shpchp mei lpc_ich mac_hid parport_pc ppdev lp parport autofs4 dm_mirror dm_region_hash dm_log i915 i2c_algo_bit e1000e ahci drm_kms_helper libahci syscopyarea sysfillrect
[88230.615275] sysimgblt fb_sys_fops drm ptp pps_core wmi fjes video
[88230.615296] CPU: 2 PID: 4140 Comm: kworker/2:0 Not tainted 4.10.0-28201707281149-generic #0+mediatree+hauppauge
[88230.615321] Hardware name: ASUS All Series/H87I-PLUS, BIOS 2003 11/05/2014
[88230.615342] Workqueue: events request_module_async [em28xx]
[88230.615357] task: ffff895998b4c380 task.stack: ffffb8fe081b4000
[88230.615374] RIP: 0010:si2157_attach+0xc9/0x497 [si2157]
[88230.615389] RSP: 0018:ffffb8fe081b7978 EFLAGS: 00010286
[88230.615403] RAX: ffff895b95ea0938 RBX: ffff895b8e40d028 RCX: 0000000000000006
[88230.615421] RDX: 0000000000000000 RSI: 0000000000000247 RDI: ffffffffc08ae100
[88230.615440] RBP: ffffb8fe081b79e8 R08: ffff895b94520800 R09: 00000000000003b4
[88230.615459] R10: 0000000000000060 R11: 0000000000000000 R12: 0000000000000060
[88230.615477] R13: ffff895b8fc7dc00 R14: 0000000000000000 R15: ffff895b8e40c000
[88230.615496] FS: 0000000000000000(0000) GS:ffff895b9fb00000(0000) knlGS:0000000000000000
[88230.615517] CS: 0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[88230.615532] CR2: 0000000000000018 CR3: 00000000c3c09000 CR4: 00000000001406e0
[88230.615551] Call Trace:
[88230.615562] si2157_probe+0x77/0x140 [si2157]
[88230.615575] ? si2157_setup_configuration.isra.7+0xc0/0xc0 [si2157]
[88230.615594] i2c_device_probe+0x1c2/0x220
[88230.615607] driver_probe_device+0x2bb/0x460
[88230.615620] __device_attach_driver+0x8c/0x100
[88230.615633] ? __driver_attach+0xf0/0xf0
[88230.615645] bus_for_each_drv+0x67/0xb0
[88230.615657] __device_attach+0xdd/0x160
[88230.615669] device_initial_probe+0x13/0x20
[88230.615682] bus_probe_device+0x92/0xa0
[88230.615693] device_add+0x373/0x630
[88230.615704] device_register+0x1a/0x20
[88230.615715] i2c_new_device+0x196/0x240
[88230.615728] em28xx_dvb_init.part.5+0x316/0x3940 [em28xx_dvb]
[88230.615744] ? load_balance+0x180/0xa00
[88230.615757] ? update_load_avg+0x6b/0x510
[88230.615770] ? update_curr+0xf3/0x170
[88230.615781] ? dequeue_task_fair+0x5ab/0xaa0
[88230.615794] ? pick_next_task_fair+0x10d/0x4b0
[88230.615807] ? __switch_to+0x23c/0x520
[88230.615819] em28xx_dvb_init+0x24/0x30 [em28xx_dvb]
[88230.615834] em28xx_init_extension+0x66/0xa0 [em28xx]
[88230.615849] request_module_async+0x22/0x100 [em28xx]
[88230.615864] process_one_work+0x1fc/0x4b0
[88230.615876] worker_thread+0x4b/0x500
[88230.615887] kthread+0x109/0x140
[88230.615897] ? process_one_work+0x4b0/0x4b0
[88230.615910] ? kthread_create_on_node+0x60/0x60
[88230.615924] ret_from_fork+0x2c/0x40
[88230.615935] Code: 8e 17 00 00 49 81 f8 e0 e0 8a c0 0f 84 42 02 00 00 4d 89 c6 eb 10 4d 8b 36 49 81 fe e0 e0 8a c0 0f 84 2d 02 00 00 4d 85 ff 74 eb <49> 8b 46 18 48 85 c0 74 e2 8b 90 68 03 00 00 41 3b 97 68 03 00
[88230.615996] RIP: si2157_attach+0xc9/0x497 [si2157] RSP: ffffb8fe081b7978
[88230.616014] CR2: 0000000000000018
[88230.640714] ---[ end trace 2e6500b52c710cdd ]---</pre>

b-rad-NDi, 2017-08-03:
@tphoney : 
We believe we've sorted the reconnect oops, if you'd like to test out the packages are here:
https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0

b-rad-NDi, 2017-08-08:
Confirmed (truly) sorted now, new builds will be in the ppa later today.

closed 2017-08-08 by b-rad-NDi

tphoney, 2017-08-12:
Sorry about the delay. Everything is working great many thanks for your help

tphoney, 2017-08-12:
I used the PPA to be clear