b-rad-NDi/Ubuntu-media-tree-kernel-builder #11: Kernel 4.12

macsrok, 2017-08-09:
Thanks for your work on this! I've been using a WinTV-dualHD in my ubuntu based plex media server for two months now. I'm upgrading my server to Ryzen. Ryzen requires kernel 4.11 or greater. Could you provide instructions on how to build a  4.12 kernel with needed drivers the WinTV-dualHD.

Thanks!

b-rad-NDi, 2017-08-09:
I'll spin up an Artful container soon and start tracking it. Looks like it is on 4.11 now and is aiming for 4.13.

All the patches are in patches/{distro}-{codename}/extra, and should apply against mainline cleanly. If you're looking to compile 4.12 yourself, just apply all the patches from 4.10 before you compile. It is possible that will work.

b-rad-NDi, 2017-08-17:
I've added patches to support Ubuntu 17.10. If you are running artful and would like to try them out I can put the debs somewhere for you. I probably won't add it to the ppa until Beta 1 is released.

macsrok, 2017-08-18:
Thanks so much! Unfortunately I wasn't able to get Ubuntu, Ryzen, and my media server software to work together and switched to Fedora for the time being. When Artful is more mature I'll try again.

b-rad-NDi, 2017-08-18:
No worries. Fedora is on the list to possibly support in the future.

closed 2017-08-18 by b-rad-NDi