b-rad-NDi/Ubuntu-media-tree-kernel-builder #12: Not able to attach to tuner

Hengjie, 2017-08-11:
It started a few days ago, I ran apt update, and upgrade everything and it started failing.

dmesg
```
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.0-89201708091011-generic (buildd@lgw01-03) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4) ) #0+mediatree+hauppauge-Ubuntu SMP Wed Aug 9 17:23:11 UTC 2017 (Ubuntu 4.4.0-89201708091011.0+mediatree+hauppauge-generic 4.4.76)
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-4.4.0-89201708091011-generic root=/dev/mapper/plex--vg-root ro
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] Disabled fast string operations
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x04: 'AVX registers'
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] x86/fpu: Using 'eager' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009f3ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f400-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000dc000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000bfeeffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bfef0000-0x00000000bfefefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000bfeff000-0x00000000bfefffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bff00000-0x00000000bfffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec0ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffe0000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000013fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 04/05/2016
[    0.000000] Hypervisor detected: VMware
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x140000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-CBFFF write-protect
[    0.000000]   CC000-EFFFF uncachable
[    0.000000]   F0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000C0000000 mask 7FFC0000000 uncachable
[    0.000000]   1 base 00000000000 mask 7FE00000000 write-back
[    0.000000]   2 disabled
[    0.000000]   3 disabled
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT
[    0.000000] original variable MTRRs
[    0.000000] reg 0, base: 3GB, range: 1GB, type UC
[    0.000000] reg 1, base: 0GB, range: 8GB, type WB
[    0.000000] total RAM covered: 7168M
[    0.000000] Found optimal setting for mtrr clean up
[    0.000000]  gran_size: 64K 	chunk_size: 64K 	num_reg: 3  	lose cover RAM: 0G
[    0.000000] New variable MTRRs
[    0.000000] reg 0, base: 0GB, range: 2GB, type WB
[    0.000000] reg 1, base: 2GB, range: 1GB, type WB
[    0.000000] reg 2, base: 4GB, range: 4GB, type WB
[    0.000000] e820: update [mem 0xc0000000-0xffffffff] usable ==> reserved
[    0.000000] e820: last_pfn = 0xc0000 max_arch_pfn = 0x400000000
[    0.000000] found SMP MP-table at [mem 0x000f6a80-0x000f6a8f] mapped at [ffff8800000f6a80]
[... 1614 lines cut ...]
[   15.245163] si2157_attach: instance=1
[   15.245164] si2157_attach(): new instance for tuner @0x62
[   15.245165] si2157_setup_configuration: client=ffff8800ba7f2c00
[   15.245166] si2157_setup_configuration: dev=ffff8800b6bf2000
[   15.245168] si2157_setup_configuration(0x62): dvb driver submitted configuration; port=1 invert=0
[   15.245419] si2157 5-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[   15.245425] dvbdev: DVB: registering new adapter (cx23885[1])
[   15.245427] cx23885 0000:1b:00.0: DVB: registering adapter 1 frontend 0 (Silicon Labs Si2168)...
[   15.245586] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xa5
[   15.245604] cx23885: cx23885[1]/0: found at 0000:1b:00.0, rev: 4, irq: 17, latency: 64, mmio: 0xfd000000
[   15.687987] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[   15.688407] NFSD: starting 90-second grace period (net ffffffff81ef5f00)
[   15.840431] FS-Cache: Loaded
[   15.850870] FS-Cache: Netfs 'nfs' registered for caching
[   15.898539] NFS: Registering the id_resolver key type
[   15.898545] Key type id_resolver registered
[   15.898546] Key type id_legacy registered
```

```
uname -a
Linux plex 4.4.0-89201708091011-generic #0+mediatree+hauppauge-Ubuntu SMP Wed Aug 9 17:23:11 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```

I'm not able to use Plex to attach to it anymore. Dmesg says `set_type: attaching si2157 tuner failed`. It didn't use to say that before when I first setup everything.

b-rad-NDi, 2017-08-11:
Investigating.

b-rad-NDi, 2017-08-11:
Reproduced by testers. Working on it.

b-rad-NDi, 2017-08-11:
Only the DVB QuadHD models are affected. We are on the case.

b-rad-NDi, 2017-08-12:
@Hengjie : you may try these packages, which contain a possible fix:
https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0

Hengjie, 2017-08-13:
It doesn't work, dmesg does not seem to recognize QuadHD anymore.

```
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.0-89201708111931-generic (root@4d00612cbce2) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4) ) #0+mediatree+hauppauge SMP Fri Aug 11 19:33:51 CDT 2017 (Ubuntu 4.4.0-89201708111931.0+mediatree+hauppauge-generic 4.4.76)
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-4.4.0-89201708111931-generic root=/dev/mapper/plex--vg-root ro
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] Disabled fast string operations
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x04: 'AVX registers'
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] x86/fpu: Using 'eager' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009f3ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f400-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000dc000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000bfeeffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bfef0000-0x00000000bfefefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000bfeff000-0x00000000bfefffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bff00000-0x00000000bfffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec0ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffe0000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000013fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 04/05/2016
[    0.000000] Hypervisor detected: VMware
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x140000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-CBFFF write-protect
[    0.000000]   CC000-EFFFF uncachable
[    0.000000]   F0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000C0000000 mask 7FFC0000000 uncachable
[    0.000000]   1 base 00000000000 mask 7FE00000000 write-back
[    0.000000]   2 disabled
[    0.000000]   3 disabled
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT
[    0.000000] original variable MTRRs
[    0.000000] reg 0, base: 3GB, range: 1GB, type UC
[    0.000000] reg 1, base: 0GB, range: 8GB, type WB
[    0.000000] total RAM covered: 7168M
[    0.000000] Found optimal setting for mtrr clean up
[    0.000000]  gran_size: 64K 	chunk_size: 64K 	num_reg: 3  	lose cover RAM: 0G
[    0.000000] New variable MTRRs
[    0.000000] reg 0, base: 0GB, range: 2GB, type WB
[    0.000000] reg 1, base: 2GB, range: 1GB, type WB
[    0.000000] reg 2, base: 4GB, range: 4GB, type WB
[    0.000000] e820: update [mem 0xc0000000-0xffffffff] usable ==> reserved
[    0.000000] e820: last_pfn = 0xc0000 max_arch_pfn = 0x400000000
[    0.000000] found SMP MP-table at [mem 0x000f6a80-0x000f6a8f] mapped at [ffff8800000f6a80]
[    0.000000] Scanning 1 areas for low memory corruption
[... 1464 lines cut ...]
[   12.849991] audit: type=1400 audit(1502616276.679:2): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/bin/lxc-start" pid=1041 comm="apparmor_parser"
[   12.852964] audit: type=1400 audit(1502616276.683:3): apparmor="STATUS" operation="profile_load" profile="unconfined" name="lxc-container-default" pid=1039 comm="apparmor_parser"
[   12.852969] audit: type=1400 audit(1502616276.683:4): apparmor="STATUS" operation="profile_load" profile="unconfined" name="lxc-container-default-cgns" pid=1039 comm="apparmor_parser"
[   12.852972] audit: type=1400 audit(1502616276.683:5): apparmor="STATUS" operation="profile_load" profile="unconfined" name="lxc-container-default-with-mounting" pid=1039 comm="apparmor_parser"
[   12.852975] audit: type=1400 audit(1502616276.683:6): apparmor="STATUS" operation="profile_load" profile="unconfined" name="lxc-container-default-with-nesting" pid=1039 comm="apparmor_parser"
[   12.853105] audit: type=1400 audit(1502616276.683:7): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/sbin/dhclient" pid=1040 comm="apparmor_parser"
[... cut ...]
[   13.307480] NET: Registered protocol family 40
[   15.050723] floppy0: no floppy controllers found
[   15.706298] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[   15.706731] NFSD: starting 90-second grace period (net ffffffff81ef5f00)
[   15.855124] FS-Cache: Loaded
[   15.942634] FS-Cache: Netfs 'nfs' registered for caching
[   15.957748] NFS: Registering the id_resolver key type
[   15.957755] Key type id_resolver registered
[   15.957757] Key type id_legacy registered
```

b-rad-NDi, 2017-08-14:
As you found out, all four packages are mandatory. Good to hear it works.

closed 2017-08-14 by b-rad-NDi

Hengjie, 2017-08-15:
When will they be rolled out to your apt-get repo?

b-rad-NDi, 2017-08-16:
Pushing it out today.

Hengjie, 2017-08-21:
I'm still getting errors such as this:

```
[320890.464315] si2168 5-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[320891.953613] si2168 5-0066: firmware version: B 4.0.25
[320891.971371] si2157 6-0062: found a 'Silicon Labs Si2157-A30'
[320892.031065] si2157 6-0062: firmware version: 3.0.5
[320932.858268] cx23885: cx23885[1]: mpeg risc op code error
[320932.858293] cx23885: cx23885[1]: TS2 C - dma channel status dump
[320932.858298] cx23885: cx23885[1]:   cmds: init risc lo   : 0xb9a98000
[320932.858301] cx23885: cx23885[1]:   cmds: init risc hi   : 0x00000000
[320932.858305] cx23885: cx23885[1]:   cmds: cdt base       : 0x000105e0
[320932.858309] cx23885: cx23885[1]:   cmds: cdt size       : 0x0000000a
[320932.858312] cx23885: cx23885[1]:   cmds: iq base        : 0x00010440
[320932.858315] cx23885: cx23885[1]:   cmds: iq size        : 0x00000010
[320932.858319] cx23885: cx23885[1]:   cmds: risc pc lo     : 0xb9a980f0
[320932.858323] cx23885: cx23885[1]:   cmds: risc pc hi     : 0x00000000
[320932.858326] cx23885: cx23885[1]:   cmds: iq wr ptr      : 0x0000411a
[320932.858329] cx23885: cx23885[1]:   cmds: iq rd ptr      : 0x0000411e
[320932.858333] cx23885: cx23885[1]:   cmds: cdt current    : 0x000105e8
[320932.858336] cx23885: cx23885[1]:   cmds: pci target lo  : 0x371c0eb0
[320932.858340] cx23885: cx23885[1]:   cmds: pci target hi  : 0x00000000
[320932.858343] cx23885: cx23885[1]:   cmds: line / byte    : 0x000e0000
[320932.858347] cx23885: cx23885[1]:   risc0:
[320932.858348] 0x1c0002f0 [ write
[320932.858348]  sol eol count=752 ]
[320932.858353] cx23885: cx23885[1]:   risc1:
[320932.858353] 0x371c0eb0 [ INVALID
[320932.858354]  eol irq2 irq1 20 19 18 count=3760 ]
[320932.858358] cx23885: cx23885[1]:   risc2:
[320932.858359] 0x00000000 [ INVALID
[320932.858360]  count=0 ]
[320932.858363] cx23885: cx23885[1]:   risc3:
[320932.858363] 0x1c0002f0 [ write
[320932.858364]  sol eol count=752 ]
[320932.858368] cx23885: cx23885[1]:   (0x00010440) iq 0:
[320932.858368] 0x00000000 [ INVALID
[320932.858369]  count=0 ]
[320932.858372] cx23885: cx23885[1]:   (0x00010444) iq 1:
[320932.858372] 0x1c0002f0 [ write
[320932.858373]  sol eol count=752 ]
[320932.858376] cx23885: cx23885[1]:   iq 2: 0x371c1780 [ arg #1 ]
[320932.858378] cx23885: cx23885[1]:   iq 3: 0x00000000 [ arg #2 ]
[320932.858381] cx23885: cx23885[1]:   (0x00010450) iq 4:
[320932.858382] 0x1c0002f0 [ write
[320932.858382]  sol eol count=752 ]
[320932.858386] cx23885: cx23885[1]:   iq 5: 0x371c31f0 [ arg #1 ]
[320932.858390] cx23885: cx23885[1]:   iq 6: 0x00000000 [ arg #2 ]
[320932.858393] cx23885: cx23885[1]:   (0x0001045c) iq 7:
[320932.858393] 0x1c0002f0 [ write
[320932.858394]  sol eol count=752 ]
[320932.858397] cx23885: cx23885[1]:   iq 8: 0x371c34e0 [ arg #1 ]
[320932.858401] cx23885: cx23885[1]:   iq 9: 0x00000000 [ arg #2 ]
[320932.858404] cx23885: cx23885[1]:   (0x00010468) iq a:
[320932.858405] 0x00000000 [ INVALID
[320932.858405]  count=0 ]
[320932.858409] cx23885: cx23885[1]:   (0x0001046c) iq b:
[320932.858409] 0x1c0002f0 [ write
[320932.858409]  sol eol count=752 ]
[320932.858413] cx23885: cx23885[1]:   iq c: 0x371c0eb0 [ arg #1 ]
[320932.858416] cx23885: cx23885[1]:   iq d: 0x00000000 [ arg #2 ]
[320932.858420] cx23885: cx23885[1]:   (0x00010478) iq e:
[320932.858420] 0x1c0002f0 [ write
[320932.858421]  sol eol count=752 ]
[320932.858424] cx23885: cx23885[1]:   iq f: 0x371c1490 [ arg #1 ]
[320932.858428] cx23885: cx23885[1]:   iq 10: 0xf73abfde [ arg #2 ]
[320932.858428] cx23885: cx23885[1]: fifo: 0x00006000 -> 0x7000
[320932.858429] cx23885: cx23885[1]: ctrl: 0x00010440 -> 0x104a0
[320932.858432] cx23885: cx23885[1]:   ptr1_reg: 0x000060d0
[320932.858435] cx23885: cx23885[1]:   ptr2_reg: 0x000105e8
[... 46 lines cut ...]
[320980.471854]  sol irq1 23 18 resync 14 13 count=832 ]
[320980.471859] cx23885: cx23885[1]:   iq 7: 0x00000000 [ arg #1 ]
[320980.471863] cx23885: cx23885[1]:   iq 8: 0x1c0002f0 [ arg #2 ]
[320980.471866] cx23885: cx23885[1]:   (0x00010424) iq 9:
[320980.471866] 0xb984e630 [ writerm
[320980.471866]  sol irq1 23 18 resync 14 13 count=1584 ]
[320980.471871] cx23885: cx23885[1]:   iq a: 0x00000000 [ arg #1 ]
[320980.471875] cx23885: cx23885[1]:   iq b: 0x00000000 [ arg #2 ]
[320980.471878] cx23885: cx23885[1]:   (0x00010430) iq c:
[320980.471878] 0x1c0002f0 [ write
[320980.471879]  sol eol count=752 ]
[320980.471883] cx23885: cx23885[1]:   iq d: 0xb984da70 [ arg #1 ]
[320980.471886] cx23885: cx23885[1]:   iq e: 0x00000000 [ arg #2 ]
[320980.471889] cx23885: cx23885[1]:   (0x0001043c) iq f:
[320980.471890] 0x1c0002f0 [ write
[320980.471890]  sol eol count=752 ]
[320980.471894] cx23885: cx23885[1]:   iq 10: 0x00000000 [ arg #1 ]
[320980.471897] cx23885: cx23885[1]:   iq 11: 0x1c0002f0 [ arg #2 ]
[320980.471898] cx23885: cx23885[1]: fifo: 0x00005000 -> 0x6000
[320980.471898] cx23885: cx23885[1]: ctrl: 0x00010400 -> 0x10460
[320980.471902] cx23885: cx23885[1]:   ptr1_reg: 0x00005d10
[320980.471906] cx23885: cx23885[1]:   ptr2_reg: 0x000105c8
[320980.471909] cx23885: cx23885[1]:   cnt1_reg: 0x00000015
[320980.471912] cx23885: cx23885[1]:   cnt2_reg: 0x00000001
```

b-rad-NDi, 2017-08-22:
@Hengjie: Did you sort this out? I cannot see this comment on the issue.
Are you by chance running the tuner in a virtual machine?

Hengjie, 2017-08-22:
@b-rad-NDi I did have issues with it, but I suspect it's because of a weak signal since I switched to TV ears antennas recently. I'm using it with Vmware EXSI and direct PCIe passthrough.

Hengjie, 2017-08-22:
I am noticing that my tuner server seems to crash more since moving to the Dropbox kernels. Crash as in completely non-responsive even though the virtual machine is powered on. It's unresponsive even if I plug a 'VMWare virtual display and keyboard',

Hengjie, 2017-08-25:
Any update to my question? Thanks

b-rad-NDi, 2017-08-25:
Using the 4.12 kernel is experimental as Ubuntu Artful doesn't even have a beta released yet. I cannot vouch for the stability of it. I will not be pushing an Artful kernel to the launchpad until Beta 1 is released.

As for unresponsive crashing, you would have to recover a syslog from before the lockup for me to even theorize. I do know that I have had multiple reports of PCIe passthru working, but have also had a person or two say they've had an issue. Whether the issue is with ESXI, the chipset of the motherboard, or what, is unknown, but the cards work fine and do not display any similar issues when not-passed through.