b-rad-NDi/Ubuntu-media-tree-kernel-builder #13: lag/jitter esxi 6, ubuntu 16.04.1 plex haupauge 955

vandammages, 2017-09-06:
Hi, using the PPA install, under my environment this really lags, jitters .

plex  and ubuntu see the device, able to scan.. but not able to watch to much jitter.

same setup but using esxi 6, windows 8 vm and plexs  is ok..

b-rad-NDi, 2017-09-08:
This has been reported to us and we can reproduce it. Unfortunately it doesn't seem to be a problem in the DVB driver, but in the USB passthru system that VMware has implemented. You wouldn't think in a so call 'passthru' implementation that there would be such issues, but there is a problem somewhere in the lower level internals.
On some systems if you set the VM USB to be USB3 then it works better, but if you start adding load to the host then it starts behaving similar. I will try out KVM/qemu, which is what I use, when I have a chance and see if I get similar behaviour.

vandammages, 2017-09-08:
Hey thanks for he update.
let me know. much appreciated

b-rad-NDi, 2017-10-27:
Please file a support ticket with vmware. I spoke with them this week and they suggested that we suggest any end users to file tickets with vmware. I will file one as well and offer hardware if anyone there happens to be interested in the issue, but they need support tickets for the record.

b-rad-NDi, 2017-12-15:
@vandammages : I finally got around to testing this with kvm/qemu. My 955Q has zero issues scanning and viewing in 16.04 kvm/qemu VM. Scan works with atsc and qam via w_scan, scanning and viewing also worked via kaffeine.

This indeed appears to be specific to vmware's Linux USB stack. Please file a ticket with vmware.

b-rad-NDi, 2017-12-17:
@vandammages : Reports indicate that an optimization of bulk transfer size completely fixed em28xx based tuners in vmware. I am going to see if cx231xx can be similarly optimized.

b-rad-NDi, 2017-12-19:
So, 955Q is an ISOC device. Therefore it is at the full mercy of vmware's sub-optimal usb passthru system. Vmware would be the only one able to solve this problem. Closing the issue.

closed 2017-12-19 by b-rad-NDi