b-rad-NDi/Ubuntu-media-tree-kernel-builder #14: Hauppauge QuadTV can't scan channels

leigh-richardson, 2017-09-22:
On 2017-09-21 03:51, Leigh Richardson wrote:
> Hi Brad.
>
> I'm running a fresh install of Ubuntu 17.04. The only applications 
> I've installed since are Chrome, Teamview, and Plex Media Server.
>
> I've installed your PPA and run all updates.
>
> here is the output of uname -a
>
> Linux Leigh-HTPC 4.10.0-32201708161632-generic #0+mediatree+hauppauge- 
> Ubuntu SMP Thu Aug 17 00:27:10 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
>
> Whe I open plex, the quadtv appears, but scanning for channels does 
> returns nothing, from an antenna that I regularly use for digital TV.
>
> Can you suggest some troubleshooting steps I can take, I'm tearing my 
> hair out!

On Friday, 22 September 2017 1:12 AM, Brad wrote:
Hi Leigh,

I need more information than uname. Please follow the link to my github and file an issue. Include the full output from dmesg as part of the issue.

Cheers,

Brad
_________________________________________________________________________________

Hi Brad,

I have attached the full output of dmesg to this thread. Thanks again for your assistance.

[    0.000000] Linux version 4.10.0-32201708161632-generic (buildd@lgw01-45) (gcc version 6.3.0 20170406 (Ubuntu 6.3.0-12ubuntu2) ) #0+mediatree+hauppauge-Ubuntu SMP Thu Aug 17 00:27:10 UTC 2017 (Ubuntu 4.10.0-32201708161632.0+mediatree+hauppauge-generic 4.10.17)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.10.0-32201708161632-generic root=UUID=27e064d5-449d-48d2-876f-d77c885e1c07 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'compacted' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009cfffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009d00000-0x0000000009ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000a000000-0x00000000dadeafff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dadeb000-0x00000000dc354fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dc355000-0x00000000dc447fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dc448000-0x00000000dc819fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000dc81a000-0x00000000dd1e7fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dd1e8000-0x00000000dd23ffff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000dd240000-0x00000000deffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000df000000-0x00000000dfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fdf00000-0x00000000fdffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fea00000-0x00000000fea0ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feb80000-0x00000000fec01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec30000-0x00000000fec30fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedc2000-0x00000000fedcffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedd4000-0x00000000fedd5fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000feefffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[... 1541 lines cut ...]
[34358.689727] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[34673.740732] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[34988.795471] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[35303.856486] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[35618.915080] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[35933.969280] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[36249.023569] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[36564.075410] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[36879.125685] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[37194.166769] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[37509.210352] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[37824.256746] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[38139.305650] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[38454.354616] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[38769.403023] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[39084.450156] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[39399.493044] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[39714.538816] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[40029.585544] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[40344.635034] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[40659.684191] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[40974.732886] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready
[41289.784290] IPv6: ADDRCONF(NETDEV_UP): wlp9s0: link is not ready

Leigh

b-rad-NDi, 2017-09-22:
Hi,

Could you paste the dmesg log that are generated when you try and run a Plex scan?
This will show you your current time spot:
<pre>dmesg | tail</pre>
I would like everything that shows up after that.

vandammages, 2017-09-22:
i seems my issue is with the way VMware passes the USB controller to ubuntu
..work fine in windows
If I add a hardware USB controller card. and pass it trough to ubuntu VM
 all works fine.

so issue is VMware and USB passthrough on Ubuntu. with default onboard
controllers.

Thanks

leigh-richardson, 2017-09-23:
Hi Brad, I'm so sorry to waste your time, I've just realised the stupidest thing. I've been sold the wrong format tuner card for Australia. The shop stocks both and they sent me the wrong one.

Thanks for your prompt follow up.

Regards,

Leigh Richardson
National Sales Manager

[http://files.ordermatepos.com/email-signature/ordermate-logo.gif]<http://www.ordermate.com.au/>

OrderMate POS
Level 6 / 69 Reservoir St, Surry Hills, NSW 2010
Tel: 1300 667 067<tel:1300%20667%20067> | Fax: 1300 667 101<tel:1300%20667%20101> | Mob: +61 447 554 446<tel:+61%20447%20554%20446> | Web: www.ordermate.com.au<http://www.ordermate.com.au/>

This email has been scanned by LANserve Email Defence.
For more information please visit www.emergingit.com.au<http://www.emergingit.com.au/>

Report this message as spam<https://console.mailguard.com.au/ras/1RJaXJeXC3/6LzvkZELSmkMc4fU31oEPn/3.3>

This email has been scanned by LANserve Email Defence.
For more information please visit www.emergingit.com.au

b-rad-NDi, 2017-09-25:
oops :)

closed 2017-09-25 by b-rad-NDi

Saentist, 2018-02-05:
I have same problem with correct device. I hope
cx23885: CORE cx23885[0]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB [card=56,autodetected]
(56 | Hauppauge WinTV-QuadHD-DVB | 0070:6a28, 0070:6b28)

`dmesg |grep si2
[   15.423656] si2168 5-0064: Silicon Labs Si2168-B40 successfully identified
[   15.423657] si2168 5-0064: firmware version: B 4.0.2
[   15.550144] si2157 6-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[   15.556503] si2168 5-0066: Silicon Labs Si2168-B40 successfully identified
[   15.556504] si2168 5-0066: firmware version: B 4.0.2
[   15.557836] si2157 6-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[   15.906135] si2168 10-0064: Silicon Labs Si2168-B40 successfully identified
[   15.906136] si2168 10-0064: firmware version: B 4.0.2
[   15.907470] si2157 11-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[   15.914310] si2168 10-0066: Silicon Labs Si2168-B40 successfully identified
[   15.914311] si2168 10-0066: firmware version: B 4.0.2
[   15.917898] si2157 11-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[   22.979202] si2168 10-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   23.590260] si2168 10-0066: firmware version: B 4.0.11
[   23.596547] si2157 11-0062: found a 'Silicon Labs Si2157-A30'
[   23.645326] si2157 11-0062: firmware version: 3.0.5
[   23.652146] si2168 10-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   24.257463] si2168 10-0064: firmware version: B 4.0.11
[   24.263755] si2157 11-0060: found a 'Silicon Labs Si2157-A30'
[   24.312752] si2157 11-0060: firmware version: 3.0.5
[   24.317277] si2168 5-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   24.922612] si2168 5-0066: firmware version: B 4.0.11
[   24.928898] si2157 6-0062: found a 'Silicon Labs Si2157-A30'
[   24.977426] si2157 6-0062: firmware version: 3.0.5
[   24.984257] si2168 5-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   25.590301] si2168 5-0064: firmware version: B 4.0.11
[   25.596598] si2157 6-0060: found a 'Silicon Labs Si2157-A30'
[   25.645388] si2157 6-0060: firmware version: 3.0.5
`
**lspci**
`03:00.0 Multimedia video controller: Conexant Systems, Inc. CX23885 PCI Video and Audio Decoder (rev 04)
        Subsystem: Hauppauge computer works Inc. CX23885 PCI Video and Audio Decoder
        Flags: bus master, fast devsel, latency 0, IRQ 19
        Memory at ff400000 (64-bit, non-prefetchable) [size=2M]
        Capabilities: [40] Express Endpoint, MSI 00
        Capabilities: [80] Power Management version 2
        Capabilities: [90] Vital Product Data
        Capabilities: [a0] MSI: Enable- Count=1/1 Maskable- 64bit+
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [200] Virtual Channel
        Kernel driver in use: cx23885
        Kernel modules: cx23885

04:00.0 Multimedia video controller: Conexant Systems, Inc. CX23885 PCI Video and Audio Decoder (rev 04)
        Subsystem: Hauppauge computer works Inc. CX23885 PCI Video and Audio Decoder
        Flags: bus master, fast devsel, latency 0, IRQ 16
        Memory at ff200000 (64-bit, non-prefetchable) [size=2M]
        Capabilities: [40] Express Endpoint, MSI 00
        Capabilities: [80] Power Management version 2
        Capabilities: [90] Vital Product Data
        Capabilities: [a0] MSI: Enable- Count=1/1 Maskable- 64bit+
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [200] Virtual Channel
        Kernel driver in use: cx23885
        Kernel modules: cx23885
`
tuner reported as Silicon Labs Si2168
I scan DVB-C

`w_scan -f c -F -t 3 -Q 2 -e 1 -v
w_scan version 20141122 (compiled for DVB API 5.10)
WARNING: could not guess your country. Falling back to 'DE'
guessing country 'DE', use -c <country> to override
using settings for GERMANY
DVB cable
[... 173 lines cut ...]
Problem retrieving frontend information: Operation not supported
status       | signal  40% | snr  45% | ber 1869833334 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal  40% | snr  45% | ber 1869833334 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal  40% | snr  45% | ber 1869833334 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal  40% | snr  45% | ber 1869833334 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal  40% | snr  45% | ber 1869833334 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal  40% | snr  45% | ber 1869833334 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal  40% | snr  45% | ber 1869833334 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal  40% | snr  45% | ber 1869833334 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal  40% | snr  45% | ber 1869833334 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal  40% | snr  45% | ber 1869833334 | unc 1 |
`

no data from PAT or poll TIMEOUT 
SNR 38dB Signal -41dBm
but no reception

b-rad-NDi, 2018-02-05:
Please include dmesg.

If you have a DVB model, did you install the required firmware so the demod works?

If you are missing firmware, and using my ppa then:
``sudo apt-get install linux-firmware-hauppauge``
*should* fix you up.

Saentist, 2018-02-05:
I use Ubuntu 16.04 server x64
standard kernel 4.4 no card recognized
install PPA path kernel 4.4-somelongnumberstring
card recognized no channels
currently
```
uname -a
Linux tvserv2 4.10.0-42-generic #46~16.04.1-Ubuntu SMP Mon Dec 4 15:57:59 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```
card recognied no channels 
but some strange errors appear in dmesg somtimes from cx23885

 
`[    0.000000] Linux version 4.10.0-42-generic (buildd@lgw01-amd64-007) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.5) ) #46~16.04.1-Ubuntu SMP Mon Dec 4 15:57:59 UTC 2017 (Ubuntu 4.10.0-42.46~16.04.1-generic 4.10.17)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.10.0-42-generic root=UUID=fe0308a0-b6a1-4825-817c-24a08016516e ro
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009f7ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f800-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000009d76bfff] usable
[    0.000000] BIOS-e820: [mem 0x000000009d76c000-0x000000009d7effff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000009d7f0000-0x000000009d7fdfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000009d7fe000-0x000000009d831fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000009d832000-0x00000000adb15fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000adb16000-0x00000000ae4ebfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ae4ec000-0x00000000ae4ecfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ae4ed000-0x00000000ae6f2fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000ae6f3000-0x00000000aeb31fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000aeb32000-0x00000000aeff2fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000aeff3000-0x00000000aeffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff800000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100001000-0x000000011effffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: Hewlett-Packard HP Compaq Pro 6305 SFF/1850, BIOS K06 v02.74 10/24/2016
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x11f000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF write-through
[    0.000000]   C0000-D3FFF write-protect
[    0.000000]   D4000-FFFFF uncachable
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000000000000 mask FFFF80000000 write-back
[    0.000000]   1 base 000080000000 mask FFFFC0000000 write-back
[    0.000000]   2 base 0000AF000000 mask FFFFFF000000 uncachable
[    0.000000]   3 base 0000B0000000 mask FFFFF0000000 uncachable
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] TOM2: 000000011f000000 aka 4592M
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT
[    0.000000] total RAM covered: 2800M
[... 856 lines cut ...]
[   15.034839] dvbdev: DVB: registering new adapter (cx23885[1])
[   15.034842] cx23885 0000:04:00.0: DVB: registering adapter 3 frontend 0 (Silicon Labs Si2168)...
[   15.035076] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xa5
[   15.035080] cx23885: cx23885[1]/0: found at 0000:04:00.0, rev: 4, irq: 16, latency: 0, mmio: 0xff200000
[   15.360995] ppdev: user-space parallel port driver
[   15.367275] IPv6: ADDRCONF(NETDEV_UP): enp7s0: link is not ready
[   15.461042] input: HP WMI hotkeys as /devices/virtual/input/input13
[   16.146335] Adding 3355644k swap on /dev/sda5.  Priority:-1 extents:1 across:3355644k FS
[   17.383706] cgroup: new mount options do not match the existing superblock, will be ignored
[   18.133994] tg3 0000:07:00.0 enp7s0: Link is up at 1000 Mbps, full duplex
[   18.133996] tg3 0000:07:00.0 enp7s0: Flow control is off for TX and off for RX
[   18.134013] IPv6: ADDRCONF(NETDEV_CHANGE): enp7s0: link becomes ready
[   24.366465] si2168 10-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   24.979264] si2168 10-0066: firmware version: B 4.0.11
[   24.981683] cx23885 0000:04:00.0: DVB: adapter 3 frontend 0 frequency 0 out of range (42000000..870000000)
[... cut ...]
[   25.726156] si2168 5-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   26.333195] si2168 5-0066: firmware version: B 4.0.11
[   26.335624] cx23885 0000:03:00.0: DVB: adapter 1 frontend 0 frequency 0 out of range (42000000..870000000)
[   26.379278] si2168 5-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   26.985455] si2168 5-0064: firmware version: B 4.0.11
[   26.987882] cx23885 0000:03:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (42000000..870000000)
`

b-rad-NDi, 2018-02-05:
You're using w_scan correct? Or is this Plex? The tail end of your logs would seem to imply something is trying, and failing, tuning to 0MHz.

I'll ask support if they've recall seeing something like this.

Saentist, 2018-02-05:
I'm not using Plex, I have TVHeadend ;)
This is how TVHedend detect frontend capability 
(this can be ignored it appear on each service restart)

estimadarocha, 2018-02-06:
@b-rad-NDi i am probably with problems also!

i am also reporting on issue #17 using same cards but on the amd platform!

so i get back cards to z270 asus board with i7-7700k and brand new installation SO.

uname -a
Linux srvtv 4.14.16-041416-generic #201801310931 SMP Wed Jan 31 14:33:23 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

lspci
00:00.0 Host bridge: Intel Corporation Device 591f (rev 05)
00:01.0 PCI bridge: Intel Corporation Sky Lake PCIe Controller (x16) (rev 05)
00:01.1 PCI bridge: Intel Corporation Sky Lake PCIe Controller (x8) (rev 05)
00:14.0 USB controller: Intel Corporation Device a2af
00:16.0 Communication controller: Intel Corporation Device a2ba
00:17.0 SATA controller: Intel Corporation Device a282
00:1b.0 PCI bridge: Intel Corporation Device a2e7 (rev f0)
00:1c.0 PCI bridge: Intel Corporation Device a290 (rev f0)
00:1c.1 PCI bridge: Intel Corporation Device a291 (rev f0)
00:1c.2 PCI bridge: Intel Corporation Device a292 (rev f0)
00:1c.4 PCI bridge: Intel Corporation Device a294 (rev f0)
00:1c.6 PCI bridge: Intel Corporation Device a296 (rev f0)
00:1d.0 PCI bridge: Intel Corporation Device a298 (rev f0)
00:1f.0 ISA bridge: Intel Corporation Device a2c5
00:1f.2 Memory controller: Intel Corporation Device a2a1
00:1f.3 Audio device: Intel Corporation Device a2f0
00:1f.4 SMBus: Intel Corporation Device a2a3
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (2) I219-V
01:00.0 VGA compatible controller: NVIDIA Corporation Device 1c03 (rev a1)
01:00.1 Audio device: NVIDIA Corporation Device 10f1 (rev a1)
02:00.0 PCI bridge: Pericom Semiconductor Device 2304 (rev 05)
03:01.0 PCI bridge: Pericom Semiconductor Device 2304 (rev 05)
03:02.0 PCI bridge: Pericom Semiconductor Device 2304 (rev 05)
04:00.0 Multimedia video controller: Conexant Systems, Inc. CX23885 PCI Video and Audio Decoder (rev 04)
05:00.0 Multimedia video controller: Conexant Systems, Inc. CX23885 PCI Video and Audio Decoder (rev 04)
08:00.0 PCI bridge: Pericom Semiconductor Device 2304 (rev 05)
09:01.0 PCI bridge: Pericom Semiconductor Device 2304 (rev 05)
09:02.0 PCI bridge: Pericom Semiconductor Device 2304 (rev 05)
0a:00.0 Multimedia video controller: Conexant Systems, Inc. CX23885 PCI Video and Audio Decoder (rev 04)
0b:00.0 Multimedia video controller: Conexant Systems, Inc. CX23885 PCI Video and Audio Decoder (rev 04)
0c:00.0 Network controller: Qualcomm Atheros QCA6174 802.11ac Wireless Network Adapter (rev 32)
0d:00.0 USB controller: ASMedia Technology Inc. Device 2142
0e:00.0 USB controller: ASMedia Technology Inc. Device 2142
0f:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd Device a804

On previous SO reinstall i just have to update kernel to a recent one and put on /lib/firmware this https://github.com/OpenELEC/dvb-firmware/blob/master/firmware/dvb-demod-si2168-b40-01.fw and everything it's fine.

dvb-fe-tool
Device Silicon Labs Si2168 (/dev/dvb/adapter0/frontend0) capabilities:
     CAN_2G_MODULATION
     CAN_FEC_1_2
     CAN_FEC_2_3
     CAN_FEC_3_4
     CAN_FEC_5_6
     CAN_FEC_7_8
     CAN_FEC_AUTO
     CAN_GUARD_INTERVAL_AUTO
     CAN_HIERARCHY_AUTO
     CAN_INVERSION_AUTO
     CAN_MULTISTREAM
     CAN_MUTE_TS
     CAN_QAM_16
     CAN_QAM_32
     CAN_QAM_64
     CAN_QAM_128
     CAN_QAM_256
     CAN_QAM_AUTO
     CAN_QPSK
     CAN_TRANSMISSION_MODE_AUTO
DVB API Version 5.10, Current v5 delivery system: DVBT
Supported delivery systems:
    [DVBT]
     DVBT2
     DVBC/ANNEX_A

 femon -H -a 1
FE: Silicon Labs Si2168 (DVBT)
Problem retrieving frontend information: Operation not supported
status       | signal  38% | snr  45% | ber 1819042149 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal  38% | snr  45% | ber 1819042149 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal  38% | snr  45% | ber 1819042149 | unc 1 |

but can't use it on tvheadend also!

estimadarocha, 2018-02-06:
how is this possible?

root@srvtv:/# femon -H -a 0 -c 2
FE: Silicon Labs Si2168 (DVBT)
Problem retrieving frontend information: Operation not supported
status       | signal   0% | snr   0% | ber 0 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal   0% | snr   0% | ber 0 | unc 1 |
root@srvtv:/# femon -H -a 1 -c 2
FE: Silicon Labs Si2168 (DVBT)
Problem retrieving frontend information: Operation not supported
status       | signal   0% | snr   0% | ber 0 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal   0% | snr   0% | ber 0 | unc 1 |
root@srvtv:/# femon -H -a 2 -c 2
FE: Silicon Labs Si2168 (DVBT)
Problem retrieving frontend information: Operation not supported
status       | signal  38% | snr  45% | ber 1819042149 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal  38% | snr  45% | ber 1819042149 | unc 1 |
root@srvtv:/# femon -H -a 3 -c 2
FE: Silicon Labs Si2168 (DVBT)
Problem retrieving frontend information: Operation not supported
status       | signal  38% | snr  45% | ber 1819042149 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal  38% | snr  45% | ber 1819042149 | unc 1 |
root@srvtv:/# femon -H -a 4 -c 2
FE: Silicon Labs Si2168 (DVBT)
Problem retrieving frontend information: Operation not supported
status       | signal   0% | snr   0% | ber 0 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal   0% | snr   0% | ber 0 | unc 1 |
root@srvtv:/# femon -H -a 5 -c 2
FE: Silicon Labs Si2168 (DVBT)
Problem retrieving frontend information: Operation not supported
status       | signal   0% | snr   0% | ber 0 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal   0% | snr   0% | ber 0 | unc 1 |
root@srvtv:/# femon -H -a 6 -c 2
FE: Silicon Labs Si2168 (DVBT)
Problem retrieving frontend information: Operation not supported
status       | signal  38% | snr  45% | ber 1819042149 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal  38% | snr  45% | ber 1819042149 | unc 1 |
root@srvtv:/# femon -H -a 7 -c 2
FE: Silicon Labs Si2168 (DVBT)
Problem retrieving frontend information: Operation not supported
status       | signal  38% | snr  45% | ber 1819042149 | unc 1 |
Problem retrieving frontend information: Operation not supported
status       | signal  38% | snr  45% | ber 1819042149 | unc 1 |