b-rad-NDi/Ubuntu-media-tree-kernel-builder #15: Read Status fails on autodetect switchover

mspieth, 2017-09-26:
Currently with extras patches, read status reads 8701 and then uses this to read 5001 or a001 depending on the result of the previous.
If the si2168 FE is in switchover transition, this causes the a001 to fail.
This causes readstatus to fail the ioctl and thus stops many progs from trying again.

mspieth, 2017-09-26:
Failure logs pre patch

Sep 22 10:23:35 moonbeam kernel: [81493.380988] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1)
Sep 22 10:23:35 moonbeam kernel: [81493.380990] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1) addr = 0x64  len = 0x2
Sep 22 10:23:35 moonbeam kernel: [81493.380990] cx23885: i2c_sendbytes: i2c:i2c_sendbytes(msg->len=2)
Sep 22 10:23:35 moonbeam kernel: [81493.381202]  <W c8 87 01 >
Sep 22 10:23:35 moonbeam kernel: [81493.381347] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1)
Sep 22 10:23:35 moonbeam kernel: [81493.381348] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1) addr = 0x64  len = 0x8
Sep 22 10:23:35 moonbeam kernel: [81493.381348] cx23885: i2c_readbytes: i2c:i2c_readbytes(msg->len=8)
Sep 22 10:23:35 moonbeam kernel: [81493.381349] cx23885: i2c_readbytes: i2c: <R c9
Sep 22 10:23:35 moonbeam kernel: [81493.381563] cx23885: i2c_readbytes: i2c: 80
Sep 22 10:23:35 moonbeam kernel: [81493.381673] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.381783] cx23885: i2c_readbytes: i2c: 08
Sep 22 10:23:35 moonbeam kernel: [81493.381893] cx23885: i2c_readbytes: i2c: 07
Sep 22 10:23:35 moonbeam kernel: [81493.382003] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.382113] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.382223] cx23885: i2c_readbytes: i2c: 20
Sep 22 10:23:35 moonbeam kernel: [81493.382367] cx23885: i2c_readbytes: i2c: 03
Sep 22 10:23:35 moonbeam kernel: [81493.382367] cx23885: i2c_readbytes: i2c: >
Sep 22 10:23:35 moonbeam kernel: [81493.382367] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1)
Sep 22 10:23:35 moonbeam kernel: [81493.382368] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1) addr = 0x64  len = 0x2
Sep 22 10:23:35 moonbeam kernel: [81493.382368] cx23885: i2c_sendbytes: i2c:i2c_sendbytes(msg->len=2)
Sep 22 10:23:35 moonbeam kernel: [81493.382580]  <W c8 50 01 >
Sep 22 10:23:35 moonbeam kernel: [81493.382723] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1)
Sep 22 10:23:35 moonbeam kernel: [81493.382723] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1) addr = 0x64  len = 0xe
Sep 22 10:23:35 moonbeam kernel: [81493.382724] cx23885: i2c_readbytes: i2c:i2c_readbytes(msg->len=14)
Sep 22 10:23:35 moonbeam kernel: [81493.382724] cx23885: i2c_readbytes: i2c: <R c9
Sep 22 10:23:35 moonbeam kernel: [81493.382938] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.383048] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.383158] cx23885: i2c_readbytes: i2c: 08
Sep 22 10:23:35 moonbeam kernel: [81493.383268] cx23885: i2c_readbytes: i2c: dc
Sep 22 10:23:35 moonbeam kernel: [81493.383377] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.383487] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.383596] cx23885: i2c_readbytes: i2c: 42
Sep 22 10:23:35 moonbeam kernel: [81493.383706] cx23885: i2c_readbytes: i2c: ff
Sep 22 10:23:35 moonbeam kernel: [81493.383815] cx23885: i2c_readbytes: i2c: 03
Sep 22 10:23:35 moonbeam kernel: [81493.383925] cx23885: i2c_readbytes: i2c: 1b
Sep 22 10:23:35 moonbeam kernel: [81493.384035] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.384144] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.384254] cx23885: i2c_readbytes: i2c: 01
Sep 22 10:23:35 moonbeam kernel: [81493.384397] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.384398] cx23885: i2c_readbytes: i2c: >
Sep 22 10:23:35 moonbeam kernel: [81493.384398] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1)
Sep 22 10:23:35 moonbeam kernel: [81493.384399] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1) addr = 0x64  len = 0xe
Sep 22 10:23:35 moonbeam kernel: [81493.384399] cx23885: i2c_readbytes: i2c:i2c_readbytes(msg->len=14)
Sep 22 10:23:35 moonbeam kernel: [81493.384399] cx23885: i2c_readbytes: i2c: <R c9
Sep 22 10:23:35 moonbeam kernel: [81493.384613] cx23885: i2c_readbytes: i2c: 80
Sep 22 10:23:35 moonbeam kernel: [81493.384722] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.384832] cx23885: i2c_readbytes: i2c: 08
Sep 22 10:23:35 moonbeam kernel: [81493.384942] cx23885: i2c_readbytes: i2c: dc
Sep 22 10:23:35 moonbeam kernel: [81493.385052] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.385161] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.385277] cx23885: i2c_readbytes: i2c: 42
Sep 22 10:23:35 moonbeam kernel: [81493.385386] cx23885: i2c_readbytes: i2c: ff
Sep 22 10:23:35 moonbeam kernel: [81493.385496] cx23885: i2c_readbytes: i2c: 03
Sep 22 10:23:35 moonbeam kernel: [81493.385606] cx23885: i2c_readbytes: i2c: 1b
Sep 22 10:23:35 moonbeam kernel: [81493.385716] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.385826] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.385936] cx23885: i2c_readbytes: i2c: 01
Sep 22 10:23:35 moonbeam kernel: [81493.386080] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.386081] cx23885: i2c_readbytes: i2c: >
Sep 22 10:23:35 moonbeam kernel: [81493.437146] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1)
Sep 22 10:23:35 moonbeam kernel: [81493.437147] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1) addr = 0x64  len = 0x2
Sep 22 10:23:35 moonbeam kernel: [81493.437148] cx23885: i2c_sendbytes: i2c:i2c_sendbytes(msg->len=2)
Sep 22 10:23:35 moonbeam kernel: [81493.437373]  <W c8 87 01 >
Sep 22 10:23:35 moonbeam kernel: [81493.437517] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1)
[... cut ...]
Sep 22 10:23:35 moonbeam kernel: [81493.439874] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.439983] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.440093] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.440203] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.440312] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.440456] cx23885: i2c_readbytes: i2c: 00
Sep 22 10:23:35 moonbeam kernel: [81493.440456] cx23885: i2c_readbytes: i2c: >

mspieth, 2017-09-26:
logs after patch

Sep 23 16:27:41 moonbeam kernel: [ 2532.783329]  <W c8 87 01 >
Sep 23 16:27:41 moonbeam kernel: [ 2532.783472] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1)
Sep 23 16:27:41 moonbeam kernel: [ 2532.783473] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1) addr = 0x64  len = 0x8
Sep 23 16:27:41 moonbeam kernel: [ 2532.783473] cx23885: i2c_readbytes: i2c:i2c_readbytes(msg->len=8)
Sep 23 16:27:41 moonbeam kernel: [ 2532.783473] cx23885: i2c_readbytes: i2c: <R c9
Sep 23 16:27:41 moonbeam kernel: [ 2532.783686] cx23885: i2c_readbytes: i2c: 80
Sep 23 16:27:41 moonbeam kernel: [ 2532.783795] cx23885: i2c_readbytes: i2c: 00
Sep 23 16:27:41 moonbeam kernel: [ 2532.783905] cx23885: i2c_readbytes: i2c: 06
Sep 23 16:27:41 moonbeam kernel: [ 2532.784014] cx23885: i2c_readbytes: i2c: 02
Sep 23 16:27:41 moonbeam kernel: [ 2532.784123] cx23885: i2c_readbytes: i2c: 01
Sep 23 16:27:41 moonbeam kernel: [ 2532.784232] cx23885: i2c_readbytes: i2c: 09
Sep 23 16:27:41 moonbeam kernel: [ 2532.784341] cx23885: i2c_readbytes: i2c: ec
Sep 23 16:27:41 moonbeam kernel: [ 2532.784485] cx23885: i2c_readbytes: i2c: 09
Sep 23 16:27:41 moonbeam kernel: [ 2532.784485] cx23885: i2c_readbytes: i2c: >
Sep 23 16:27:41 moonbeam kernel: [ 2532.784486] si2168_cmd_execute: cmd execution took -2 ms
Sep 23 16:27:41 moonbeam kernel: [ 2532.784511] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1)
Sep 23 16:27:41 moonbeam kernel: [ 2532.784512] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1) addr = 0x64  len = 0x2
Sep 23 16:27:41 moonbeam kernel: [ 2532.784512] cx23885: i2c_sendbytes: i2c:i2c_sendbytes(msg->len=2)
Sep 23 16:27:41 moonbeam kernel: [ 2532.784722]  <W c8 50 01 >
Sep 23 16:27:41 moonbeam kernel: [ 2532.784864] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1)
Sep 23 16:27:41 moonbeam kernel: [ 2532.784865] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1) addr = 0x64  len = 0xe
Sep 23 16:27:41 moonbeam kernel: [ 2532.784865] cx23885: i2c_readbytes: i2c:i2c_readbytes(msg->len=14)
Sep 23 16:27:41 moonbeam kernel: [ 2532.784865] cx23885: i2c_readbytes: i2c: <R c9
Sep 23 16:27:41 moonbeam kernel: [ 2532.785078] cx23885: i2c_readbytes: i2c: c0
Sep 23 16:27:41 moonbeam kernel: [ 2532.785187] cx23885: i2c_readbytes: i2c: 10
Sep 23 16:27:41 moonbeam kernel: [ 2532.785296] cx23885: i2c_readbytes: i2c: 00
Sep 23 16:27:41 moonbeam kernel: [ 2532.785406] cx23885: i2c_readbytes: i2c: 00
Sep 23 16:27:41 moonbeam kernel: [ 2532.785515] cx23885: i2c_readbytes: i2c: 00
Sep 23 16:27:41 moonbeam kernel: [ 2532.785623] cx23885: i2c_readbytes: i2c: 00
Sep 23 16:27:41 moonbeam kernel: [ 2532.785732] cx23885: i2c_readbytes: i2c: 00
Sep 23 16:27:41 moonbeam kernel: [ 2532.785841] cx23885: i2c_readbytes: i2c: 00
Sep 23 16:27:41 moonbeam kernel: [ 2532.785941] cx23885: cx23885_wakeup: [ffff8cfc3776e400/24] wakeup reg=249 buf=249
Sep 23 16:27:41 moonbeam kernel: [ 2532.785950] cx23885: i2c_readbytes: i2c: 00
Sep 23 16:27:41 moonbeam kernel: [ 2532.786059] cx23885: i2c_readbytes: i2c: 00
Sep 23 16:27:41 moonbeam kernel: [ 2532.786096] cx23885: cx23885_buf_prepare: cx23885_buf_prepare: ffff8cfc3776e400
Sep 23 16:27:41 moonbeam kernel: [ 2532.786101] cx23885: cx23885_buf_queue: [ffff8cfc3776e400/24] cx23885_buf_queue - append to active
Sep 23 16:27:41 moonbeam kernel: [ 2532.786168] cx23885: i2c_readbytes: i2c: 00
Sep 23 16:27:41 moonbeam kernel: [ 2532.786277] cx23885: i2c_readbytes: i2c: 00
Sep 23 16:27:41 moonbeam kernel: [ 2532.786386] cx23885: i2c_readbytes: i2c: 00
Sep 23 16:27:41 moonbeam kernel: [ 2532.786530] cx23885: i2c_readbytes: i2c: 00
Sep 23 16:27:41 moonbeam kernel: [ 2532.786530] cx23885: i2c_readbytes: i2c: >
Sep 23 16:27:41 moonbeam kernel: [ 2532.786531] si2168_cmd_execute: cmd execution took -2 ms
Sep 23 16:27:41 moonbeam kernel: [ 2532.786554] si2168_cmd_execute: failed=-121
Sep 23 16:27:41 moonbeam kernel: [ 2532.786575] si2168_read_status: last arg=0xc0 -121
Sep 23 16:27:41 moonbeam kernel: [ 2532.786597] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1)
Sep 23 16:27:41 moonbeam kernel: [ 2532.786597] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1) addr = 0x64  len = 0x2
Sep 23 16:27:41 moonbeam kernel: [ 2532.786598] cx23885: i2c_sendbytes: i2c:i2c_sendbytes(msg->len=2)
Sep 23 16:27:41 moonbeam kernel: [ 2532.786808]  <W c8 a0 01 >
Sep 23 16:27:41 moonbeam kernel: [ 2532.786953] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1)
Sep 23 16:27:41 moonbeam kernel: [ 2532.786954] cx23885: i2c_xfer: i2c:i2c_xfer(num = 1) addr = 0x64  len = 0xd
Sep 23 16:27:41 moonbeam kernel: [ 2532.786954] cx23885: i2c_readbytes: i2c:i2c_readbytes(msg->len=13)
Sep 23 16:27:41 moonbeam kernel: [ 2532.786955] cx23885: i2c_readbytes: i2c: <R c9
Sep 23 16:27:41 moonbeam kernel: [ 2532.787169] cx23885: i2c_readbytes: i2c: 40
Sep 23 16:27:41 moonbeam kernel: [ 2532.787279] cx23885: i2c_readbytes: i2c: 00
Sep 23 16:27:41 moonbeam kernel: [ 2532.787388] cx23885: i2c_readbytes: i2c: 06
Sep 23 16:27:41 moonbeam kernel: [ 2532.787497] cx23885: i2c_readbytes: i2c: 80
Sep 23 16:27:41 moonbeam kernel: [ 2532.787606] cx23885: i2c_readbytes: i2c: 03
Sep 23 16:27:41 moonbeam kernel: [ 2532.787714] cx23885: i2c_readbytes: i2c: 00
Sep 23 16:27:41 moonbeam kernel: [ 2532.787823] cx23885: i2c_readbytes: i2c: f6
Sep 23 16:27:41 moonbeam kernel: [ 2532.787932] cx23885: i2c_readbytes: i2c: ff
Sep 23 16:27:41 moonbeam kernel: [ 2532.788041] cx23885: i2c_readbytes: i2c: 09
Sep 23 16:27:41 moonbeam kernel: [ 2532.788150] cx23885: i2c_readbytes: i2c: 13
Sep 23 16:27:41 moonbeam kernel: [ 2532.788260] cx23885: i2c_readbytes: i2c: 2d
[... cut ...]
Sep 23 16:27:41 moonbeam kernel: [ 2532.789817] cx23885: i2c_readbytes: i2c: 2d
Sep 23 16:27:41 moonbeam kernel: [ 2532.789926] cx23885: i2c_readbytes: i2c: 01
Sep 23 16:27:41 moonbeam kernel: [ 2532.790069] cx23885: i2c_readbytes: i2c: 17
Sep 23 16:27:41 moonbeam kernel: [ 2532.790069] cx23885: i2c_readbytes: i2c: >
Sep 23 16:27:41 moonbeam kernel: [ 2532.790070] si2168_cmd_execute: cmd execution took -2 ms
Sep 23 16:27:41 moonbeam kernel: [ 2532.790093] si2168_read_status: status=1f args=80 00 06 80 03 00 f6 ff 09 13 2d 01 17

mspieth, 2017-09-26:
patch that works (with 4.0.11 and 4.0.25 fw)

[20170927.force-status-autodetect.patch.zip](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1335336/20170927.force-status-autodetect.patch.zip)

b-rad-NDi, 2017-09-29:
Thank you for the detailed logs, we are looking into this. I will point out to our testers that you are experiencing this with a dvb quadHD.

b-rad-NDi, 2018-02-15:
@mspieth: This should be handled now by: 35de9b590b78b708481af7af5fc913bf519b53f4

closed 2018-02-15 by b-rad-NDi