b-rad-NDi/Ubuntu-media-tree-kernel-builder #16: No support in cx23885 for legacy v3 status ioctls

mspieth, 2017-09-27:
There is no support for tuner FE_READ_SIGNAL_STRENGTH and other FE ones like FE_READ_SNR etc.
This is specifically with si2168 FE and si2157 tuner

The v5 interface using FE_GET_PROPERTY will only work 

1. FE_READ_STATUS is called first
2. for DTV_STATUS_SIGNAL_STRENGTH only if the initial tune that kicked of stat_work. Most will poll this only during tuning and more often than every 2 seconds. propose start poll immediately and more often or make it demand driven. no point accessing stale status or discarding expensive I2C accesses.
3. for POST_ERROR_COUNT and TOTAL, these are not available until signal is present. propose they be set to available but 0 irrespective of valid mode unless this is against the philosophy of v5 status.

b-rad-NDi, 2017-09-29:
Issues noted, I will look at implementing anything missing.

b-rad-NDi, 2018-02-14:
@mspieth : Mark, is your si2157-faster-signal-strength.patch your solution to handling your 1&2 points? The current 1s delay to start stat_work, then only polling it every 2s is rather absurd. I will integrate this.

mspieth, 2018-02-14:
#2 definitely. the poll immediately and more often (0.5 sec) was implemented.
#1 cant remember it was so long ago. probably not by that patch but may now implicitly occur.

mspieth, 2018-02-14:
Note that the original problem in this issue was no v3 support for get sig strength and read snr.

b-rad-NDi, 2018-02-14:
Yes, I revisited this issue because I was fixing cx23885 analog where tuner is TUNER_ABSENT. That added various ioctls, so I came here to check if that took care of this. Your v3 issue is still not handled.

b-rad-NDi, 2018-02-15:
Unfortunately this is guarded by NDA, I would have to reverse engineer the snr & ber registers. Leaving open, but lacking necessary information to implement.

b-rad-NDi, 2018-02-16:
Actually...what I need is in read_status already, so I'll do this tomorrow.

b-rad-NDi, 2018-02-17:
@mspieth : Here is a preliminary v3 ioctl patch. Thankfully everything required was included in the open source driver already. I included on demand rf signal check in 2157 and then various 2168 stat functions. Lemme know if it works suitably for you, or if there is still something missing.

[0001-si21xx-Add-some-various-v3-ioctls.zip](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1733309/0001-si21xx-Add-some-various-v3-ioctls.zip)

mspieth, 2018-02-18:
It looks good. Still have to test. 
I assume there are not locking issues when 2 threads access any one on different devices simultaneously. Thats the only thing I thought of.

b-rad-NDi, 2018-02-20:
I'm not sure how the dvb subsytsem handles locking. I was going to do some looking around and testing before I committed this.

MikeB2013, 2018-02-20:
@b-rad-NDi 

If it helps, I can test this using mythtv (which currently complains about not being able to get signal strength, bit error etc.) if you can add the patches to your 4.15 build which I am currently running on my Ryzen system and put it in your dropbox

Mike

b-rad-NDi, 2018-02-20:
@MikeB2013 : They're in the dropbox ready for testing. Cheers.

b-rad-NDi, 2018-02-20:
Let me know if any of the units don't make sense. I converted the dvbv5 representation to v3, where the translation was obvious. For snr though I left it in dvbv5 format instead of reversing the transformation they applied. So, if snr is astronomical it probably needs be divided by 250.

MikeB2013, 2018-02-21:
Installed new build 4.15.0-rc6+ #6 SMP Tue Feb 20 13:33:47 CST 2018 x86_64 x86_64 x86_64 GNU/Linux

I ran femon as follows on all 4 tuners. However the femon with -H (Human output) shows 0% for signal, and I am not convinced by the high snr % (it could correct but I don't know of any way to confirm/deny). I suspect it is the usual problem of interpretation of the raw values. Maybe  @mspieth has some views on this.

Mythtv no longer complains about I/O errors as it now gets something!

> femon -a0 -c 5
FE: Silicon Labs Si2168 (DVBT)
status SCVYL | signal 004e | snr 733c | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 004e | snr 7436 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 004e | snr 7436 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 004e | snr 733c | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 004f | snr 733c | ber 00000000 | unc 00000000 | FE_HAS_LOCK
femon -a0 -c 5 -H
FE: Silicon Labs Si2168 (DVBT)
status SCVYL | signal   0% | snr  45% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal   0% | snr  45% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal   0% | snr  45% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal   0% | snr  45% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal   0% | snr  45% | ber 0 | unc 0 | FE_HAS_LOCK
femon -a1 -c 5
FE: Silicon Labs Si2168 (DVBT)
status SCVYL | signal 0055 | snr 781e | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 0055 | snr 762a | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 0055 | snr 762a | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 0054 | snr 781e | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 0054 | snr 762a | ber 00000000 | unc 00000000 | FE_HAS_LOCK
femon -a1 -c 5 -H
FE: Silicon Labs Si2168 (DVBT)
status SCVYL | signal   0% | snr  46% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal   0% | snr  46% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal   0% | snr  46% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal   0% | snr  46% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal   0% | snr  46% | ber 0 | unc 0 | FE_HAS_LOCK
femon -a2 -c 5
FE: Silicon Labs Si2168 (DVBT)
status SCVYL | signal 0055 | snr 7530 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 0055 | snr 762a | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 0055 | snr 781e | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 0055 | snr 7530 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 0056 | snr 781e | ber 00000000 | unc 00000000 | FE_HAS_LOCK
femon -a2 -c 5 -H
FE: Silicon Labs Si2168 (DVBT)
status SCVYL | signal   0% | snr  48% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal   0% | snr  48% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal   0% | snr  46% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal   0% | snr  46% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal   0% | snr  46% | ber 0 | unc 0 | FE_HAS_LOCK
femon -a3 -c 5
FE: Silicon Labs Si2168 (DVBT)
status SCVYL | signal 0056 | snr 762a | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 0056 | snr 762a | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 0056 | snr 762a | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 0056 | snr 762a | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 0056 | snr 781e | ber 00000000 | unc 00000000 | FE_HAS_LOCK
femon -a3 -c 5 -H
FE: Silicon Labs Si2168 (DVBT)
status SCVYL | signal   0% | snr  46% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal   0% | snr  47% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal   0% | snr  48% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal   0% | snr  47% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal   0% | snr  46% | ber 0 | unc 0 | FE_HAS_LOCK

Previously (i.e. no patches) the femon output was

> mike@ssd-AB350-Gaming-3:~/haupp-ioctl$ femon
FE: Silicon Labs Si2168 (DVBT)
Problem retrieving frontend information: Unknown error 524
status SCVYL | signal 0000 | snr 0000 | ber 00000000 | unc 00000001 | FE_HAS_LOCK
Problem retrieving frontend information: Unknown error 524
status SCVYL | signal 0000 | snr 0000 | ber 00000000 | unc 00000001 | FE_HAS_LOCK
Problem retrieving frontend information: Unknown error 524
status SCVYL | signal 0000 | snr 0000 | ber 00000000 | unc 00000001 | FE_HAS_LOCK
Problem retrieving frontend information: Unknown error 524
status SCVYL | signal 0000 | snr 0000 | ber 00000000 | unc 00000001 | FE_HAS_LOCK
Problem retrieving frontend information: Unknown error 524
status SCVYL | signal 0000 | snr 0000 | ber 00000000 | unc 00000001 | FE_HAS_LOCK

Mike

b-rad-NDi, 2018-02-21:
I'll check out how femon is processing the raw data. The signal strength *should* have been interpreting 0x55 as 85%. Strange it shows snr in % instead of db too. I'll see what I get from an atsc driver w/stats and see if I can massage the raw values.

b-rad-NDi, 2018-02-21:
Ahhh...gotta love it when femon and dvb-fe-tool produce completely different output from the same data...

MikeB2013, 2018-02-21:
Here is output for dvb-fe-tool on my DVB 885

>mike@ssd-AB350-Gaming-3:~$ dvb-fe-tool -m -a0
Lock   (0x1f) Signal= -39.00dBm C/N= 31.25dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -39.00dBm C/N= 31.25dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -39.00dBm C/N= 31.50dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -39.00dBm C/N= 31.50dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -39.00dBm C/N= 31.25dB UCB= 0 postBER= 0
mike@ssd-AB350-Gaming-3:~$ dvb-fe-tool -m -a1
Lock   (0x1f) Signal= -42.00dBm C/N= 29.75dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -43.00dBm C/N= 29.50dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -42.00dBm C/N= 29.50dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -43.00dBm C/N= 29.75dB UCB= 0 postBER= 0
mike@ssd-AB350-Gaming-3:~$ dvb-fe-tool -m -a2
Lock   (0x1f) Signal= -42.00dBm C/N= 29.50dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -42.00dBm C/N= 29.75dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -42.00dBm C/N= 29.00dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -42.00dBm C/N= 29.75dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -42.00dBm C/N= 29.75dB UCB= 0 postBER= 0
mike@ssd-AB350-Gaming-3:~$ dvb-fe-tool -m -a3
Lock   (0x1f) Signal= -33.00dBm C/N= 32.75dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -34.00dBm C/N= 32.75dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -34.00dBm C/N= 33.00dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -33.00dBm C/N= 33.00dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -33.00dBm C/N= 32.75dB UCB= 0 postBER= 0

Mike

b-rad-NDi, 2018-02-21:
Thanks. Perusing source code for proper formats. Working with an atsc device, but if I can get suitable output from it, from both tools I'll know how to fix up the dvb frontend.

MikeB2013, 2018-02-22:
@b-rad-NDi 

Don't know if this helps or hinders, TBS Open source drivers have implemented stats for si2168, see https://github.com/tbsdtv/linux_media/blob/latest/drivers/media/dvb-frontends/si2168.c

Mike

b-rad-NDi, 2018-02-22:
That looks pretty much like the mainline file. The difficulty here is the v3 ioctls produce wishy washy 1-100% values and v5 produces actual numbers. I've reviewed an assortment of other drivers and will be doing this:
- signal strength : 0 is at -80db, anything above 0db is 100%
- snr/cnr : will report in x10 db, so femon will show 0% in human format, but in hex will show the value.

Trying to get the lgdt3306a to have matching v3 and v5, then I'll fix up 2168 for another attempt.

b-rad-NDi, 2018-02-23:
@MikeB2013 : there will be new build in my dropbox whenever it finishes building. I had to add a v5 stat to lgdt3306a, but now I see the same from it via v3 as I do v5. Signal strength will still be a bit subjective, I did it as described above, but v3 snr should be v5 (cnr / 100). I think si2168 should return values for v3 and v5 that correlate fairly well for you now.

b-rad-NDi, 2018-02-23:
I chose -80db since that is below I can sync on anything out here in the middle of nowhere. I can get lock at -75db with atsc. I'll have to test qam tomorrow to see how it compares.

MikeB2013, 2018-02-23:
Here is output from build 4.15.0-rc6+ 7 SMP Thu Feb 22 18:13:27 CST 2018 x86_64 x86_64 x86_64 GNU/Linux

The only thing that sticks out to me is "snr   0%" in femon -H output.

> femon -a0 -c 5
FE: Silicon Labs Si2168 (DVBT)
status SCVYL | signal a147 | snr 0140 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal a147 | snr 0140 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal a147 | snr 0140 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal a147 | snr 0140 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal a147 | snr 0140 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
femon -a0 -c 5 -H
FE: Silicon Labs Si2168 (DVBT)
status SCVYL | signal  61% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  61% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  62% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  61% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  61% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
femon -a1 -c 5
FE: Silicon Labs Si2168 (DVBT)
status SCVYL | signal 9eb7 | snr 0140 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 9eb7 | snr 0140 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 9eb7 | snr 013d | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 9eb7 | snr 0142 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 9eb7 | snr 0140 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
femon -a1 -c 5 -H
FE: Silicon Labs Si2168 (DVBT)
status SCVYL | signal  61% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  61% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  61% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  61% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  61% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
femon -a2 -c 5
FE: Silicon Labs Si2168 (DVBT)
status SCVYL | signal 9999 | snr 0129 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 9999 | snr 0129 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 9999 | snr 0129 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 9709 | snr 0129 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 9999 | snr 0129 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
femon -a2 -c 5 -H
FE: Silicon Labs Si2168 (DVBT)
status SCVYL | signal  60% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  60% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  60% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  60% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  60% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
femon -a3 -c 5
FE: Silicon Labs Si2168 (DVBT)
status SCVYL | signal a3d6 | snr 013d | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal a3d6 | snr 013d | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal a665 | snr 013d | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal a3d6 | snr 013d | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal a3d6 | snr 0140 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
femon -a3 -c 5 -H
FE: Silicon Labs Si2168 (DVBT)
status SCVYL | signal  63% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  63% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  63% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  63% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  63% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
mike@ssd-AB350-Gaming-3:~/haupp-ioctl$ dvb-fe-tool -m -a0
Lock   (0x1f) Signal= -38.00dBm C/N= 32.00dB UCB= 63 postBER= 0
Lock   (0x1f) Signal= -37.00dBm C/N= 32.00dB UCB= 63 postBER= 0
Lock   (0x1f) Signal= -37.00dBm C/N= 32.00dB UCB= 63 postBER= 0
Lock   (0x1f) Signal= -37.00dBm C/N= 32.00dB UCB= 63 postBER= 0
Lock   (0x1f) Signal= -37.00dBm C/N= 32.00dB UCB= 63 postBER= 0
Lock   (0x1f) Signal= -38.00dBm C/N= 32.00dB UCB= 63 postBER= 0
^Cmike@ssd-AB350-Gaming-3:~/haupp-ioctl$ dvb-fe-tool -m -a1
Lock   (0x1f) Signal= -37.00dBm C/N= 33.00dB UCB= 54 postBER= 0
Lock   (0x1f) Signal= -37.00dBm C/N= 33.00dB UCB= 54 postBER= 0
Lock   (0x1f) Signal= -37.00dBm C/N= 33.25dB UCB= 54 postBER= 0
Lock   (0x1f) Signal= -37.00dBm C/N= 33.25dB UCB= 54 postBER= 0
Lock   (0x1f) Signal= -37.00dBm C/N= 33.25dB UCB= 54 postBER= 0
^Cmike@ssd-AB350-Gaming-3:~/haupp-ioctl$ dvb-fe-tool -m -a2
Lock   (0x1f) Signal= -36.00dBm C/N= 31.75dB UCB= 96 postBER= 0
Lock   (0x1f) Signal= -36.00dBm C/N= 31.75dB UCB= 96 postBER= 0
Lock   (0x1f) Signal= -36.00dBm C/N= 31.75dB UCB= 96 postBER= 0
Lock   (0x1f) Signal= -36.00dBm C/N= 31.75dB UCB= 96 postBER= 0
Lock   (0x1f) Signal= -36.00dBm C/N= 31.75dB UCB= 96 postBER= 0
^Cmike@ssd-AB350-Gaming-3:~/haupp-ioctl$ dvb-fe-tool -m -a3
Lock   (0x1f) Signal= -35.00dBm C/N= 31.75dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -36.00dBm C/N= 31.75dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -35.00dBm C/N= 32.00dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -36.00dBm C/N= 31.75dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -35.00dBm C/N= 31.75dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -36.00dBm C/N= 32.00dB UCB= 0 postBER= 0

Mike

b-rad-NDi, 2018-02-23:
Everything looks good in v3 and v5. If you look at the source code of femon you'll see that in human format they convert the value to a scale 0-100 for 0x00-0xffff. If you convert the non-human 0x140 to decimal you'll get 320, divide by 10 and you get 32.0dB. The formatting femon uses there is just fubar.

MikeB2013, 2018-02-23:
Yes, femon is strange and pretty much obsoleted (deprecated even) by dvb-fe-tool. Suggest we go with what you have now.

Mike

b-rad-NDi, 2018-02-23:
Will get this added to main builds today.

b-rad-NDi, 2018-03-02:
@MikeB2013 : Unsure if you use the ppa, but the si2168 v3 stat patch has been updated. I had to change it to compile in 32bit environments. The math should still work out in 64bit, but I do not have a dvb generator to verify.

MikeB2013, 2018-03-02:
@b-rad-NDi 
No problems with the revised stat patch. 

I ran with 4.4 from your ppa on Xubuntu 16.04.

I also applied the complete patch set from tip to current v4l  on stock Ubuntu 4.13.0-36 kernel, and mainline kernel 4.15.7, both worked with femon and dvb-fe-tool giving expected results.

Mike

b-rad-NDi, 2018-03-02:
Excellent, closing this ticket finally then :)

closed 2018-03-02 by b-rad-NDi