b-rad-NDi/Ubuntu-media-tree-kernel-builder #17: quadHD 885 mpeg risc op code error

mspieth, 2017-09-27:
Quite often there is an mpeg risc opcode error in my quadHD 885 any channel. 
example log of one.

Sep 27 17:34:56 moonbeam kernel: [ 1463.264161] cx23885: cx23885_start_dma: cx23885_start_dma() w: 0, h: 0, f: 0
Sep 27 17:34:56 moonbeam kernel: [ 1463.264165] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() Configuring channel [TS1 B]
Sep 27 17:34:56 moonbeam kernel: [ 1463.264370] cx23885: cx23885_start_dma: cx23885_start_dma() enabling TS int's and DMA
Sep 27 17:34:56 moonbeam kernel: [ 1463.264498] dmxdev: dvb_dvr_poll: dvb_dvr_poll
Sep 27 17:34:56 moonbeam kernel: [ 1463.314937] dmxdev: dvb_dvr_poll: dvb_dvr_poll
Sep 27 17:34:56 moonbeam kernel: [ 1463.459029] si2157 6-0060: si2157_stat_work: 
Sep 27 17:34:56 moonbeam kernel: [ 1463.460900] si2157 6-0060: si2157_cmd_execute: cmd execution took 0 ms, status=85
Sep 27 17:34:56 moonbeam kernel: [ 1463.460987] si2157 6-0060: si2157_stat_work: n=1 strength=-51000dB args=85 01 01 cd 28 f7 6b 0b 00 c8 e4 0c
Sep 27 17:34:56 moonbeam kernel: [ 1463.523046] cx23885: cx23885[1]: mpeg risc op code error
Sep 27 17:34:56 moonbeam kernel: [ 1463.523147] cx23885: cx23885[1]: TS1 B - dma channel status dump
Sep 27 17:34:56 moonbeam kernel: [ 1463.523244] cx23885: cx23885[1]:   cmds: init risc lo   : 0xfaf0a000
Sep 27 17:34:56 moonbeam kernel: [ 1463.523342] cx23885: cx23885[1]:   cmds: init risc hi   : 0x00000000
Sep 27 17:34:56 moonbeam kernel: [ 1463.523442] cx23885: cx23885[1]:   cmds: cdt base       : 0x00010580
Sep 27 17:34:56 moonbeam kernel: [ 1463.523541] cx23885: cx23885[1]:   cmds: cdt size       : 0x0000000a
Sep 27 17:34:56 moonbeam kernel: [ 1463.523642] cx23885: cx23885[1]:   cmds: iq base        : 0x00010400
Sep 27 17:34:56 moonbeam kernel: [ 1463.523723] cx23885: cx23885[1]:   cmds: iq size        : 0x00000010
Sep 27 17:34:56 moonbeam kernel: [ 1463.523802] cx23885: cx23885[1]:   cmds: risc pc lo     : 0xfaf0a018
Sep 27 17:34:56 moonbeam kernel: [ 1463.523880] cx23885: cx23885[1]:   cmds: risc pc hi     : 0x00000000
Sep 27 17:34:56 moonbeam kernel: [ 1463.523959] cx23885: cx23885[1]:   cmds: iq wr ptr      : 0x00004106
Sep 27 17:34:56 moonbeam kernel: [ 1463.524040] cx23885: cx23885[1]:   cmds: iq rd ptr      : 0x00004100
Sep 27 17:34:56 moonbeam kernel: [ 1463.524119] cx23885: cx23885[1]:   cmds: cdt current    : 0x000105c8
Sep 27 17:34:56 moonbeam kernel: [ 1463.524198] cx23885: cx23885[1]:   cmds: pci target lo  : 0xd8219780
Sep 27 17:34:56 moonbeam kernel: [ 1463.524277] cx23885: cx23885[1]:   cmds: pci target hi  : 0x00000000
Sep 27 17:34:56 moonbeam kernel: [ 1463.524353] cx23885: cx23885[1]:   cmds: line / byte    : 0x00080000
Sep 27 17:34:56 moonbeam kernel: [ 1463.524406] si2168 10-0064: si2168_cmd_execute: cmd execution took -2 ms
Sep 27 17:34:56 moonbeam kernel: [ 1463.524506] cx23885: cx23885[1]:   risc0: 
Sep 27 17:34:56 moonbeam kernel: [ 1463.524507] 0x1c0002f0 [ write sol eol count=752 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.524580] cx23885: cx23885[1]:   risc1: 
Sep 27 17:34:56 moonbeam kernel: [ 1463.524580] 0xd8219490 [ writecr sol 21 cnt0 resync 12 count=1168 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.524655] cx23885: cx23885[1]:   risc2: 
Sep 27 17:34:56 moonbeam kernel: [ 1463.524655] 0x00000000 [ INVALID count=0 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.524729] cx23885: cx23885[1]:   risc3: 
Sep 27 17:34:56 moonbeam kernel: [ 1463.524729] 0x1c0002f0 [ write sol eol count=752 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.524803] cx23885: cx23885[1]:   (0x00010400) iq 0: 
Sep 27 17:34:56 moonbeam kernel: [ 1463.524804] 0x1c0002f0 [ write sol eol count=752 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.524880] cx23885: cx23885[1]:   iq 1: 0xd8219a70 [ arg #1 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.524957] cx23885: cx23885[1]:   iq 2: 0x00000000 [ arg #2 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.525032] cx23885: cx23885[1]:   (0x0001040c) iq 3: 
Sep 27 17:34:56 moonbeam kernel: [ 1463.525032] 0x1c0002f0 [ write sol eol count=752 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.525109] cx23885: cx23885[1]:   iq 4: 0xd8219d60 [ arg #1 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.525184] cx23885: cx23885[1]:   iq 5: 0x00000000 [ arg #2 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.525259] cx23885: cx23885[1]:   (0x00010418) iq 6: 
Sep 27 17:34:56 moonbeam kernel: [ 1463.525260] 0xd8218eb0 [ writecr sol 21 cnt0 resync count=3760 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.525336] cx23885: cx23885[1]:   iq 7: 0x00000000 [ arg #1 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.525411] cx23885: cx23885[1]:   iq 8: 0x1c0002f0 [ arg #2 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.525487] cx23885: cx23885[1]:   iq 9: 0xd82191a0 [ arg #3 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.525565] cx23885: cx23885[1]:   (0x00010428) iq a: 
Sep 27 17:34:56 moonbeam kernel: [ 1463.525565] 0x00000000 [ INVALID count=0 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.525641] cx23885: cx23885[1]:   (0x0001042c) iq b: 
Sep 27 17:34:56 moonbeam kernel: [ 1463.525641] 0x1c0002f0 [ write sol eol count=752 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.525719] cx23885: cx23885[1]:   iq c: 0xd8219490 [ arg #1 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.525797] cx23885: cx23885[1]:   iq d: 0x00000000 [ arg #2 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.525873] cx23885: cx23885[1]:   (0x00010438) iq e: 
Sep 27 17:34:56 moonbeam kernel: [ 1463.525873] 0x1c0002f0 [ write sol eol count=752 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.525951] cx23885: cx23885[1]:   iq f: 0xd8219780 [ arg #1 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.526028] cx23885: cx23885[1]:   iq 10: 0xdfefb7d0 [ arg #2 ]
Sep 27 17:34:56 moonbeam kernel: [ 1463.526101] cx23885: cx23885[1]: fifo: 0x00005000 -> 0x6000
Sep 27 17:34:56 moonbeam kernel: [ 1463.526175] cx23885: cx23885[1]: ctrl: 0x00010400 -> 0x10460
Sep 27 17:34:56 moonbeam kernel: [ 1463.526252] cx23885: cx23885[1]:   ptr1_reg: 0x00005050
Sep 27 17:34:56 moonbeam kernel: [ 1463.526326] cx23885: cx23885[1]:   ptr2_reg: 0x00010588
Sep 27 17:34:56 moonbeam kernel: [ 1463.526403] cx23885: cx23885[1]:   cnt1_reg: 0x00000005
Sep 27 17:34:56 moonbeam kernel: [ 1463.526477] cx23885: cx23885[1]:   cnt2_reg: 0x00000009

mspieth, 2017-09-27:
When its good this happens

Sep 27 17:35:06 moonbeam kernel: [ 1473.052595] cx23885: cx23885_start_dma: cx23885_start_dma() w: 0, h: 0, f: 0
Sep 27 17:35:06 moonbeam kernel: [ 1473.052598] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() Configuring channel [TS1 B]
Sep 27 17:35:06 moonbeam kernel: [ 1473.052803] cx23885: cx23885_start_dma: cx23885_start_dma() enabling TS int's and DMA
Sep 27 17:35:06 moonbeam kernel: [ 1473.052931] dmxdev: dvb_dvr_poll: dvb_dvr_poll
Sep 27 17:35:06 moonbeam kernel: [ 1473.053344] si2157 6-0060: si2157_cmd_execute: cmd execution took 0 ms, status=85
Sep 27 17:35:06 moonbeam kernel: [ 1473.053430] si2157 6-0060: si2157_stat_work: n=1 strength=-49000dB args=85 01 01 cf 60 6f 94 0a 00 c6 d2 f6
Sep 27 17:35:06 moonbeam kernel: [ 1473.061728] cx23885: cx23885_wakeup: [ffff9e9f450cb000/0] wakeup reg=1 buf=1
Sep 27 17:35:06 moonbeam kernel: [ 1473.061792] cx23885: cx23885_buf_prepare: cx23885_buf_prepare: ffff9e9f450cb000
Sep 27 17:35:06 moonbeam kernel: [ 1473.061798] cx23885: cx23885_buf_queue: [ffff9e9f450cb000/0] cx23885_buf_queue - append to active
Sep 27 17:35:06 moonbeam kernel: [ 1473.070077] cx23885: cx23885_wakeup: [ffff9e9ee931e800/1] wakeup reg=2 buf=2
Sep 27 17:35:06 moonbeam kernel: [ 1473.070131] cx23885: cx23885_buf_prepare: cx23885_buf_prepare: ffff9e9ee931e800
Sep 27 17:35:06 moonbeam kernel: [ 1473.070136] cx23885: cx23885_buf_queue: [ffff9e9ee931e800/1] cx23885_buf_queue - append to active

mspieth, 2017-09-27:
Another behavior is with no error but may have impacted from a previous error and not cleared down.
Case with no subsequent wakeups.

Sep 27 17:44:22 moonbeam kernel: [ 2029.063511] cx23885: cx23885_start_dma: cx23885_start_dma() w: 0, h: 0, f: 0
Sep 27 17:44:22 moonbeam kernel: [ 2029.063514] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() Configuring channel [TS1 B]
Sep 27 17:44:22 moonbeam kernel: [ 2029.063721] cx23885: cx23885_start_dma: cx23885_start_dma() enabling TS int's and DMA
Sep 27 17:44:22 moonbeam kernel: [ 2029.063855] dmxdev: dvb_dvr_poll: dvb_dvr_poll
Sep 27 17:44:22 moonbeam kernel: [ 2029.113932] dmxdev: dvb_dvr_poll: dvb_dvr_poll

mspieth, 2017-09-27:
Interesting the when I enabled level 7 debug on cx23885 driver, all mpeg risc no opcode errors disappeared. lots of debug prints in syslog.
turned it back to 0 and thet came back. The test was for over 1h.
This means its either a race or a cache issue. 
My bet is on cache not flushed to ram before DMA starts.
Happens on a Ryzen 1700 with MSI PC MATE B350 M/B, RAM @2666, no overclocking for anything.
Will chase this if no one has any ideas on this. 
Suspect PCI (sram) memory is not non-cachable.

b-rad-NDi, 2017-09-29:
We've seen this specific issue and it appears to be amd specific. I don't think we've received reports of this using intel chipsets. I will see about setting up a system I can test pcie on next week.

mspieth, 2017-09-30:
I have a workaround which is to perform all cx_reads that are part of cx23885_tsport_reg_dump at the same point with no printks.
This completely removes the mpeg risc error. 
Not sure why though as wmb doesnt work.

ManiacTwister, 2017-10-20:
I have the same problem, the system is an "hp proliant dl380 g6" - So obviously this isn't an AMD only problem.

b-rad-NDi, 2018-01-24:
I just put in a request we get a Ryzen system. This problem has only been reported with Ryzen (or newer) and Xeon systems.

MikeB2013, 2018-01-24:
Brad,
If it helps, I can run any tests you would like done on my system - it is a test system so anything goes.

Mike

b-rad-NDi, 2018-01-24:
I'll review the patches Mark produced so far and see if I can work on this sans equipment. I have a dropbox for distributing testing packages. Will let you know if I have some for you.

b-rad-NDi, 2018-01-24:
@MikeB2013 : Want to swap for an 888 version of your QuadHD? The difference is 888 has additional analog capability. We produced 885 models for a while, due to chip shortages, but are back to 888 now.

We'd like to see if this is an 885 specific issue, since the 885 is an older chip than the 888.

mspieth, 2018-01-25:
Ill upload my current patchset which I updated a couple of days ago for 
kernel 4.14.13 (generic debian) and current media tarball. This has all 
my hacks for the delays required to be robust on my ryzen msi pcmate MB. 
It doest emit that much to the logs so that should not be a problem.

More on ryzen with this kernel and disabling CState and CoollnQuieet 
modes in the bios is seems reasonably robust and doesnt reboot itself 
any more (I hope) so YMMV.

Pre driver updating, the driver works but it can cause a pair to not 
pass data any more though it tunes ok. This can be exacerbated  by using 
continual EIT updates which are short and frequent. This is what mythtv 
does. Once a pair gets in this mode you have to reboot. There does not 
seem to be any way to reset the data path. I suspect it may be tied to 
the PCI channel but this is only a guess.

Mark

MikeB2013, 2018-01-25:
Just some more info, a mythtv user in USA is having similar (but maybe not the same) issue with ATSC version of QuadHD, again this is on Ryzen.

See https://forum.mythtv.org/viewtopic.php?f=2&t=2462#p11964

Mike

MikeB2013, 2018-01-25:
@b-rad-NDi 

Yes, I can test a model 888 DVB

mspieth, 2018-01-25:
Latest reasonably stable patchset including brads set for current media tarball and kernel 4.14.13.
lots of debug on open and close seems to help with ryzen. (dup from #6 )
[latest_patches_20180126.zip](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1664113/latest_patches_20180126.zip)

b-rad-NDi, 2018-01-26:
Help is on the way, I have a Ryzen3 system coming :)

b-rad-NDi, 2018-01-31:
Hello everyone. I received a report today that a user who was experiencing this issue in Linux as well as Windows did some tests. They found the QuadHD did not work for them in any of their 1x slots, but worked fine in a 16x slot. They reached out to their motherboard manufacturer, ASRock, and received a bios update. This bios update completely fixed them up.
The changelog blurb was along the lines of:
`update PCIE routing to improve PCIE3 compatibility`

I suggest reaching out to motherboard manufacturer for bios updates while we investigate.

estimadarocha, 2018-02-02:
@b-rad-NDi another one with problems.... i was running a pair of this quad cards on asus motherboard with i7-7700k and everything was fine. Today i upgraded to asus x370 prime pro and ryzen 7 1800x and i start getting these errors.

I can't do anything with cards!!!

Any hints?

b-rad-NDi, 2018-02-02:
1) Please reach out to Asus and complain. This is a potential BIOS issue.
2) We are getting hardware to work on this.

The only solution found so far is putting this:
``options cx23885 debug=7``
in /etc/modprobe.d/cx23885.conf
BUT, this also spams your dmesg thereafter.

estimadarocha, 2018-02-02:
@b-rad-NDi thanks for reply, indeed could you please give me some points that help make my email to them so i can explain correctly whats going on?

I understand that after enabling debug the card works but can the machine work with that enabled?

by the way since i have 2 cards i did make the tests of work with each one at a time and also no luck... and i changed the card from 1x to 16x and at least on my board no luck... so i really don't know if the asus will have something to do!

One thing it's clear from what i read the % of cases with this problems is almost 95% on amd platform!

Thanks

b-rad-NDi, 2018-02-02:
The machine will work, but you will have immensely growing logs that would have to be trimmed if you were using the devices constantly. It is not at all an optimal solution.

Can anyone reporting on this thread please include your motherboard/cpu as well as model of QuadHD as can be found in dmesg:
``[    5.397605] cx23885: CORE cx23885[0]: subsystem: 0070:6a18, board: Hauppauge WinTV-QuadHD-ATSC [card=57,autodetected]``

Card will either be 56, 57, 60, or 61.

estimadarocha, 2018-02-02:
my board like i said before it's asus prime x370 pro and amd ryzen 1800X... will get the card info in one hour

b-rad-NDi, 2018-02-02:
Yup, got your mobo/cpu down. Just trying to get this info from anyone affected so we can repro.

mspieth, 2018-02-02:
cx23885: CORE cx23885[0]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB(885) [card=60,autodetected]
M/B is a MSI PCMATE B350 chipset with ryzen 7 1700 no overclock except to run ram at rated speed. v7 bios. (latest stable/beta)

estimadarocha, 2018-02-05:
@b-rad-NDi does the ryzen system already there? can you reproduce the problem? any news?

b-rad-NDi, 2018-02-05:
I have not received word that the main office got a system yet. I will ping them and ask.

estimadarocha, 2018-02-06:
my both cards:

cx23885: CORE cx23885[3]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB [card=56,autodetected]

b-rad-NDi, 2018-02-06:
Ok, one thing appears to be settled after receiving a handful of reports. Both 885 and 888 versions of the board[s] are affected.

Saentist, 2018-02-07:
+1 with possible AMD problem
HP Compaq Pro 6305 Small Form Factor PC with AMD A6-5400B
enabled 
```
options cx23885 debug=7
in /etc/modprobe.d/cx23885.conf 
```
and start to work
but cpu eaten by log
22.3 % | /lib/systemd/systemd-journald
11.2 % | /usr/sbin/rsyslogd -n

chanell serving eat
1.8 % | /usr/bin/tvheadend -f -p /var/run/tvheadend.pid -u hts -g video

estimadarocha, 2018-02-08:
@b-rad-NDi any news ?

b-rad-NDi, 2018-02-09:
My boss had not still not ordered a system as of last night :/ But I gave him an exact mobo/cpu combo to pick up in case he was hesitating because of that.

When my two launchpad test-builds complete I am going to rebuild 4.4, 4.13, and 4.15 with Marks current patch applied. I will put all three package sets in a dropbox for you to test. This should at least get you working until I can actually work on this.

b-rad-NDi, 2018-02-09:
The packages will be in here as they are built:
https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0

b-rad-NDi, 2018-02-10:
All packages are up for testing. The only patch from Marks set that is applied is the cx23885-robustness one.

MikeB2013, 2018-02-10:
Hi Brad,
Tested the 4.13.0-32201802091806-generic build from your dropbox. No channels are found by w_scan, not really surprising, as the only change was cx23885-robustness.

Mike

b-rad-NDi, 2018-02-10:
Yesterdays builds have been replaced, si2168 ts bus is no longer tri stated when going to sleep. Potential fix to i2c working, but not streaming.

Saentist, 2018-02-12:
`options cx23885 debug=6`
Also work
this not eat logging space as debug=7
now ~100Mb hour
with debug=7 ~6Gb

Witch OS Kernel work without this debug mode

b-rad-NDi, 2018-02-12:
Did you try the packages that are in the dropbox a few comments up?

b-rad-NDi, 2018-02-12:
The packages in dropbox are built after applying Marks preliminary fixes, that he says work for him. You all are the one with the platform. You tell me if the issue is fixed in these builds.

mspieth, 2018-02-12:
My workaround only spams during open and close of the channel. no per hour unless you open and close all the time like when you continually read EIT.

Saentist, 2018-02-13:
there is EPG monitoring and few client watching 
`/var/log# ll
total 3728944
drwxrwxr-x  4 root   syslog       4096 Feb 13 06:25 ./
drwxr-xr-x 15 root   root         4096 Feb  4 23:21 ../
drwxr-xr-x  2 root   root         4096 Feb 12 13:19 apt/
-rw-r-----  1 syslog adm          1494 Feb 13 09:00 auth.log
-rw-r-----  1 syslog adm        223909 Feb 13 06:25 auth.log.1
-rw-rw----  1 root   utmp            0 Feb 11 19:40 btmp
-rw-r--r--  1 root   root         5518 Feb 12 13:19 dpkg.log
-rw-r-----  1 syslog adm             0 Feb 13 06:25 kern.log
-rw-r-----  1 syslog adm    1908449488 Feb 13 02:44 kern.log.1
-rw-rw-r--  1 root   utmp       292292 Feb 13 08:59 lastlog
-rw-r--r--  1 root   root         7297 Feb 13 02:09 oscam.log
-rw-r--r--  1 root   root        10378 Feb 13 00:02 oscam.log-prev
-rw-r-----  1 syslog adm         67996 Feb 13 09:00 syslog
-rw-r-----  1 syslog adm    1909548717 Feb 13 06:25 syslog.1
drwxr-xr-x  2 root   root         4096 Feb 11 19:38 unattended-upgrades/
-rw-rw-r--  1 root   utmp        62592 Feb 13 08:59 wtmp
`
this is for 12h (previous was deleted)

MikeB2013, 2018-02-13:
@b-rad-NDi 

I have been successfully running test build from your dropbox 
`linux-headers-4.13.0-32201802101823_4.13.0-32201802101823.0+mediatree+hauppauge_all.deb
linux-headers-4.13.0-32201802101823-generic_4.13.0-32201802101823.0+mediatree+hauppauge_amd64.deb
linux-image-4.13.0-32201802101823-generic_4.13.0-32201802101823.0+mediatree+hauppauge_amd64.deb
linux-image-extra-4.13.0-32201802101823-generic_4.13.0-32201802101823.0+mediatree+hauppauge_amd64.deb
`

on my Gigabyte AB350 Gaming 3 motherboard with AMD Ryzen 3 1200 Quad-Core Processor, base o/s is Xubuntu 16.04 (HWE) with mythtv 30pre

All 4 QuadHD tuners are in use, with EIT (so there are many cx23885 prints in logs). I also have a TBS 6981 (Dual DVB-S/S2) in the configuration (again EIT is running) as this also uses cx23885 driver. Note there are no options set in /etc/modprobe.d/ for cx23885 driver

Mike

b-rad-NDi, 2018-02-13:
@Saentist : This is not a kernel specifically for a single device. This is the entire mainline media tree backported.

@MikeB2013 : Thanks for testing, the debug spew will disappear after I pull out and clean up the vital parts from Marks patch.

b-rad-NDi, 2018-02-13:
As for 'lets other brand tuners not work' wtf does that mean? This is _**everything**_ in the media tree enabled, and provided from, _**every**_ manufacturer. Any other manufacturers devices will work with these kernels if their source code is upstream.

b-rad-NDi, 2018-02-15:
I put up a new 4.15 build for testing. This should have the main sources of debug spam dialed back. I'm waiting for one last piece to come in so I can assemble my Ryzen system and experience this issue myself.

MikeB2013, 2018-02-15:
@b-rad-NDi 
New 4.15 build running ok on my Ryzen configuration. Attached is dmesg output for info, spam virtually gone.
[dmesg-4.15.0-rc6+.txt](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1728350/dmesg-4.15.0-rc6.txt)

Mike

b-rad-NDi, 2018-02-15:
Excellent :)

Saentist, 2018-02-15:
>b-rad-NDi 
>This is not a kernel specifically for a single device. This is the entire mainline media tree backported.
Then how to use iQSV ?
one kernel from intel one from you
how to run two kernels simultaneously ?

Media tree include hundred of modules, dvb tv tuners use few.
try to make install ubuntu server on 8gb drive

b-rad-NDi, 2018-02-15:
If you only want the dvb drivers, then download the media build system from linuxtv.org, take whatever patches of mine you want, apply them to the media build system and build whatever drivers you want yourself. No one is forcing you to use these packages.

Saentist, 2018-02-15:
Ok, but this is not embedded system 
any new path = new rom.
Or when you get new TV to replace old, you build new house for it?

estimadarocha, 2018-02-15:
@b-rad-NDi you probably doing your best... and i am sure of that... and i believe @Saentist is not trying devalue your work!

I also uso custom kernel's for somethings so i understand the goal of @Saentist... it's hard to work with things like they are... don't want to compare but it's almost as TBS (with their driver's problem also). I also think to develop something as Digital Devices will be better for all.

This kind of arguing it's useless!!!

Saying this... will give a try on ryzen probably this night... thanks for the efforts!

b-rad-NDi, 2018-02-15:
This kind of arguing is pointless, because I will not be devoting endless personal time for every individuals personal desire.

b-rad-NDi, 2018-02-15:
I mean seriously, what kind of complaints will I get when all this work is finally accepted upstream and there are no patches in my repo?

estimadarocha, 2018-02-15:
my last tought.... because like i say i have to appreciate your efforts but also for this and for everything else i think easier is better....

imagine i have a kernel 4.10 user X have kernel Y and user W as a Z one. what's easier to do? force (you are not forcing no one of course) all to use a specific kernel or give all them the oportunity to mantain their ones and install something that allow's them to work with hauppauge cards?

Thanks, and once more no one (especialy me) want you to devote your personal time just for me!

b-rad-NDi, 2018-02-15:
This is exactly why this is a build system. All patches are provided and mechanisms in place to apply them against any kernel with a git repo. Anyone can build their own kernel using this system.

The problem here is manpower. You are at the personal github page of an employee of Hauppauge. I have many other responsibilities.

estimadarocha, 2018-02-15:
I believe so... and i also believe my last post will be the last about this... but since you gave a cordial and sincere answer i think i could ask one more thing? whats easier for you? provide entire kernel builds or make samething like the dkms of nVidia?

thanks

b-rad-NDi, 2018-02-15:
Considering I'd have to learn dkms and then maintain completely separate code bases than upstream submissions to handle any kernel compiled. This is easier. By far.

b-rad-NDi, 2018-02-15:
Anyone with the knowledge to compile drivers themselves should just be using the media build system with my patches on top. This is for not-those-people.

b-rad-NDi, 2018-02-15:
Who knows, maybe some day I will look into a minimal DKMS option. It will happen on personal time however, and that is a finite resource.

estimadarocha, 2018-02-15:
that's something nice to hear... we apreciate the WILLpower

thanks

b-rad-NDi, 2018-03-07:
So I've been trying to repro this to trim down the patch to the bare required, BUT I cannot encounter any issues at all.

I have an MSI b350m gaming pro + R3 1300X. I leave all four tuners on and streaming, but encounter no issues on even long running tests. Currently running 4.16, without the interrupt and dma patches applied.

Any hints on how to repro? My impression was this happened fairly quickly.

b-rad-NDi, 2018-03-07:
AHA! After not seeing it whatsoever for days, I just got the op code error while running w_scan on a single tuner. I was starting to feel like I picked one of the few Ryzen boards without an issue.

[27 more replies]

closed 2018-05-07 by b-rad-NDi