b-rad-NDi/Ubuntu-media-tree-kernel-builder #18: Error recognizing si2157 with linux-4.12.12 + 0008-si2157-check-error-status-bit....patch

usergeorge, 2017-09-27:
Hi,

recently I bought a "Hauppauge WinTV dualHD Stick", USB ID 2040:0265.
I tried to patch my linux kernel tree with all patches from the path Ubuntu-media-tree-kernel-builder/patches/ubuntu-artful-4.12.0/extra/
I applied every patch in numerical order, i.e. 0002-em28xx-bulk-transfer-implemen... after  	0001-em28xx-Hauppauge-DualHD-DVB... and so on. All patches apply clean.

But after apply 0008-si2157-check-error-status-bit...patch the chip si2157 is not recognized anymore.
Log from dmesg:
```
[ 7512.576647] em28xx 1-13:1.0: dvb ts2 set to isoc mode.
[ 7512.776950] usbcore: registered new interface driver em28xx
[ 7512.777921] em28xx 1-13:1.0: Binding DVB extension
[ 7512.780901] i2c i2c-12: Added multiplexed i2c bus 15
[ 7512.780903] si2168 12-0064: Silicon Labs Si2168-B40 successfully identified
[ 7512.780903] si2168 12-0064: firmware version: B 4.0.2
[ 7512.781909] si2157: probe of 15-0060 failed with error -11
[ 7512.781938] em28xx 1-13:1.0: Binding DVB extension
[ 7512.784487] i2c i2c-14: Added multiplexed i2c bus 15
[ 7512.784488] si2168 14-0067: Silicon Labs Si2168-B40 successfully identified
[ 7512.784488] si2168 14-0067: firmware version: B 4.0.2
[ 7512.785360] si2157: probe of 15-0063 failed with error -11
[ 7512.785401] em28xx: Registered (Em28xx dvb Extension) extension
```

With patches 0001-* up to 0007-* the message reads
```
[ 7621.755054] usbcore: registered new interface driver em28xx
[ 7621.756053] em28xx 1-13:1.0: Binding DVB extension
[ 7621.758920] i2c i2c-12: Added multiplexed i2c bus 15
[ 7621.758921] si2168 12-0064: Silicon Labs Si2168-B40 successfully identified
[ 7621.758922] si2168 12-0064: firmware version: B 4.0.2
[ 7621.760163] si2157 15-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
...
[ 7621.760885] em28xx 1-13:1.0: Binding DVB extension
[ 7621.763852] i2c i2c-14: Added multiplexed i2c bus 16
[ 7621.763854] si2168 14-0067: Silicon Labs Si2168-B40 successfully identified
[ 7621.763854] si2168 14-0067: firmware version: B 4.0.2
[ 7621.764732] si2157 16-0063: Silicon Labs Si2147/2148/2157/2158 successfully attached"
```

I can provide more tests on request

best regards
Jörg

b-rad-NDi, 2017-09-29:
I will set up an artful vm and see if I can repro this. Might not happen until Monday.

b-rad-NDi, 2017-10-05:
btw, did you apply 0009-si2157-briefly-wait-for-tuning-operation-to-complete.patch ? Patches 8 & 9 are directly related.

b-rad-NDi, 2017-12-14:
Sorry for the long delay here, Autumn was very busy. I have acquired a DualHD DVB in my hands, it is a bulk transfer model (2040:8265), but the internals are identical. I do not see the failure you encounter on Gentoo on Ubuntu 17.10 running 4.13.

Can you test this again if possible and report if you still encounter any issue?

b-rad-NDi, 2017-12-19:
Bisecting all the patches trying to find the cause of a GPF. I do see now that the patch is question is indeed missing some special sauce. I'm sorting out that sauce now.

b-rad-NDi, 2017-12-20:
Fixed by f4b348876
Thank you for reporting. Please re-open if still having issue.

closed 2017-12-20 by b-rad-NDi