b-rad-NDi/Ubuntu-media-tree-kernel-builder #19: QuadHD missed interrupts

mspieth, 2017-09-30:
Im seeing 

Sep 29 17:15:01 moonbeam kernel: [  527.738941] cx23885: cx23885_wakeup: [ffff99b384b83c00/28] wakeup reg=5406 buf=5405
Sep 29 17:15:01 moonbeam kernel: [  527.739013] cx23885: cx23885_wakeup: [ffff99b40bf79400/31] wakeup reg=9537 buf=9536

where it looks like an interrupt is missed. patch attached.

[wakeup_gpcnt.patch.zip](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1346002/wakeup_gpcnt.patch.zip)

b-rad-NDi, 2017-10-02:
I'm setting up a system today to see if I can repro these issues and test your various patches.

b-rad-NDi, 2018-02-15:
This is finally integrated: ff92a3604efa538d7721ae240862eabc99bc2e1a

If you would like signed-off-by, reported-by, or any other tags added to patches you've contributed before I submit them upstream let me know.

closed 2018-02-15 by b-rad-NDi