b-rad-NDi/Ubuntu-media-tree-kernel-builder #20: quadhd 885 lockup

mspieth, 2017-09-30:
After a while of intermittant use (EIT crawling) on 1 channel (TS1), the quadhd locks up and is no longer responsive. This affects the paired channel TS2 too. A reboot is required to clear the fault (currently).
logs from cx23885 debug = 7

notice the gpcnt register is not reset.

Sep 30 20:47:22 moonbeam kernel: [29928.818722]  ^O^_D: cx23885: cx23885_irq: pci_status: 0x000f8000  pci_mask: 0x00001f02
Sep 30 20:47:22 moonbeam kernel: [29928.818724]  ^O^_D: cx23885: cx23885_irq: vida_status: 0x00000000 vida_mask: 0x00000000 count: 0x0
Sep 30 20:47:22 moonbeam kernel: [29928.818724]  ^O^_D: cx23885: cx23885_irq: audint_status: 0x00000000 audint_mask: 0x00000000 count: 0x0
Sep 30 20:47:22 moonbeam kernel: [29928.818725]  ^O^_D: cx23885: cx23885_irq: ts1_status: 0x00000000  ts1_mask: 0x00000000 count: 0x12e
Sep 30 20:47:22 moonbeam kernel: [29928.818726]  ^O^_D: cx23885: cx23885_irq: ts2_status: 0x00000000  ts2_mask: 0x00000000 count: 0x0
Sep 30 20:47:22 moonbeam kernel: [29928.826371] cx23885: cx23885_buf_prepare: cx23885_buf_prepare: ffff8e3316014c00
Sep 30 20:47:22 moonbeam kernel: [29928.826373] cx23885: cx23885_buf_prepare: cx23885_buf_prepare: ffff8e33ba7a2000
Sep 30 20:47:22 moonbeam kernel: [29928.826374] cx23885: cx23885_buf_prepare: cx23885_buf_prepare: ffff8e33c7ed0800
......
Sep 30 20:47:22 moonbeam kernel: [29928.826414] cx23885: cx23885_buf_queue: [ffff8e3317c39c00/30] cx23885_buf_queue - append to active
Sep 30 20:47:22 moonbeam kernel: [29928.826415] cx23885: cx23885_buf_queue: [ffff8e3317c39800/31] cx23885_buf_queue - append to active
Sep 30 20:47:22 moonbeam kernel: [29928.826416] cx23885: cx23885_start_dma: cx23885_start_dma() w: 0, h: 0, f: 0
Sep 30 20:47:22 moonbeam kernel: [29928.826419] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() Configuring channel [TS1 B]
Sep 30 20:47:22 moonbeam kernel: [29928.826420] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() 0x00010580 <- 0x00005000
Sep 30 20:47:22 moonbeam kernel: [29928.826420] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() 0x00010590 <- 0x000052f0
Sep 30 20:47:22 moonbeam kernel: [29928.826421] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() 0x000105a0 <- 0x000055e0
Sep 30 20:47:22 moonbeam kernel: [29928.826422] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() 0x000105b0 <- 0x000058d0
Sep 30 20:47:22 moonbeam kernel: [29928.826423] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() 0x000105c0 <- 0x00005bc0
Sep 30 20:47:22 moonbeam kernel: [29928.826424] cx23885: cx23885_sram_channel_setup: [bridge 885] sram setup TS1 B: bpl=752 lines=5
Sep 30 20:47:22 moonbeam kernel: [29928.826424] cx23885: cx23885[1]: TS1 B - dma channel status dump
Sep 30 20:47:22 moonbeam kernel: [29928.826453] cx23885: cx23885[1]:   cmds: init risc lo   : 0x9bf6c000
Sep 30 20:47:22 moonbeam kernel: [29928.826482] cx23885: cx23885[1]:   cmds: init risc hi   : 0x00000000
Sep 30 20:47:22 moonbeam kernel: [29928.826510] cx23885: cx23885[1]:   cmds: cdt base       : 0x00010580
Sep 30 20:47:22 moonbeam kernel: [29928.826538] cx23885: cx23885[1]:   cmds: cdt size       : 0x0000000a
Sep 30 20:47:22 moonbeam kernel: [29928.826566] cx23885: cx23885[1]:   cmds: iq base        : 0x00010400
Sep 30 20:47:22 moonbeam kernel: [29928.826595] cx23885: cx23885[1]:   cmds: iq size        : 0x00000010
Sep 30 20:47:22 moonbeam kernel: [29928.826623] cx23885: cx23885[1]:   cmds: risc pc lo     : 0x00000000
Sep 30 20:47:22 moonbeam kernel: [29928.826651] cx23885: cx23885[1]:   cmds: risc pc hi     : 0x00000000
Sep 30 20:47:22 moonbeam kernel: [29928.826680] cx23885: cx23885[1]:   cmds: iq wr ptr      : 0x00000000
Sep 30 20:47:22 moonbeam kernel: [29928.826708] cx23885: cx23885[1]:   cmds: iq rd ptr      : 0x00000000
Sep 30 20:47:22 moonbeam kernel: [29928.826736] cx23885: cx23885[1]:   cmds: cdt current    : 0x00000000
Sep 30 20:47:22 moonbeam kernel: [29928.826765] cx23885: cx23885[1]:   cmds: pci target lo  : 0x00000000
Sep 30 20:47:22 moonbeam kernel: [29928.826793] cx23885: cx23885[1]:   cmds: pci target hi  : 0x00000000
Sep 30 20:47:22 moonbeam kernel: [29928.826821] cx23885: cx23885[1]:   cmds: line / byte    : 0x00000000
Sep 30 20:47:22 moonbeam kernel: [29928.826849] cx23885: cx23885[1]:   risc0: 
Sep 30 20:47:22 moonbeam kernel: [29928.826849] 0x00000000 [ INVALID count=0 ]
Sep 30 20:47:22 moonbeam kernel: [29928.826873] cx23885: cx23885[1]:   risc1: 
Sep 30 20:47:22 moonbeam kernel: [29928.826873] 0x00000000 [ INVALID count=0 ]
Sep 30 20:47:22 moonbeam kernel: [29928.826896] cx23885: cx23885[1]:   risc2: 
Sep 30 20:47:22 moonbeam kernel: [29928.826896] 0x00000000 [ INVALID count=0 ]
Sep 30 20:47:22 moonbeam kernel: [29928.826920] cx23885: cx23885[1]:   risc3: 
Sep 30 20:47:22 moonbeam kernel: [29928.826920] 0x00000000 [ INVALID count=0 ]
Sep 30 20:47:22 moonbeam kernel: [29928.826943] cx23885: cx23885[1]:   (0x00010400) iq 0: 
Sep 30 20:47:22 moonbeam kernel: [29928.826943] 0xdc6dcbc0 [ writecr sol eol 22 21 19 18 cnt0 resync 14 count=3008 ]
Sep 30 20:47:22 moonbeam kernel: [29928.826971] cx23885: cx23885[1]:   iq 1: 0x00000000 [ arg #1 ]
Sep 30 20:47:22 moonbeam kernel: [29928.826998] cx23885: cx23885[1]:   iq 2: 0x1c0002f0 [ arg #2 ]
Sep 30 20:47:22 moonbeam kernel: [29928.827025] cx23885: cx23885[1]:   iq 3: 0xdc6dceb0 [ arg #3 ]
Sep 30 20:47:22 moonbeam kernel: [29928.827051] cx23885: cx23885[1]:   (0x00010410) iq 4: 
Sep 30 20:47:22 moonbeam kernel: [29928.827052] 0x00000000 [ INVALID count=0 ]
Sep 30 20:47:22 moonbeam kernel: [29928.827077] cx23885: cx23885[1]:   (0x00010414) iq 5: 
Sep 30 20:47:22 moonbeam kernel: [29928.827077] 0x1c0002f0 [ write sol eol count=752 ]
Sep 30 20:47:22 moonbeam kernel: [29928.827103] cx23885: cx23885[1]:   iq 6: 0xdc6dd1a0 [ arg #1 ]
Sep 30 20:47:22 moonbeam kernel: [29928.827130] cx23885: cx23885[1]:   iq 7: 0x00000000 [ arg #2 ]
Sep 30 20:47:22 moonbeam kernel: [29928.827157] cx23885: cx23885[1]:   (0x00010420) iq 8: 
Sep 30 20:47:22 moonbeam kernel: [29928.827157] 0x1c0002f0 [ write sol eol count=752 ]
Sep 30 20:47:22 moonbeam kernel: [29928.827183] cx23885: cx23885[1]:   iq 9: 0xdc6dd490 [ arg #1 ]
Sep 30 20:47:22 moonbeam kernel: [29928.827210] cx23885: cx23885[1]:   iq a: 0x00000000 [ arg #2 ]
Sep 30 20:47:22 moonbeam kernel: [29928.827236] cx23885: cx23885[1]:   (0x0001042c) iq b: 
Sep 30 20:47:22 moonbeam kernel: [29928.827237] 0x00000000 [ INVALID count=0 ]
Sep 30 20:47:22 moonbeam kernel: [29928.827262] cx23885: cx23885[1]:   (0x00010430) iq c: 
Sep 30 20:47:22 moonbeam kernel: [29928.827262] 0x1c0002f0 [ write sol eol count=752 ]
Sep 30 20:47:22 moonbeam kernel: [29928.827288] cx23885: cx23885[1]:   iq d: 0xdc6dc8d0 [ arg #1 ]
Sep 30 20:47:22 moonbeam kernel: [29928.827315] cx23885: cx23885[1]:   iq e: 0x00000000 [ arg #2 ]
Sep 30 20:47:22 moonbeam kernel: [29928.827342] cx23885: cx23885[1]:   (0x0001043c) iq f: 
Sep 30 20:47:22 moonbeam kernel: [29928.827342] 0x1c0002f0 [ write sol eol count=752 ]
[... 57 lines cut ...]

Sep 30 20:47:32 moonbeam kernel: [29938.885477]  ^O^_D: cx23885: cx23885_irq: pci_status: 0x000f8000  pci_mask: 0x00001f02
Sep 30 20:47:32 moonbeam kernel: [29938.885480]  ^O^_D: cx23885: cx23885_irq: vida_status: 0x00000000 vida_mask: 0x00000000 count: 0x0
Sep 30 20:47:32 moonbeam kernel: [29938.885481]  ^O^_D: cx23885: cx23885_irq: audint_status: 0x00000000 audint_mask: 0x00000000 count: 0x0
Sep 30 20:47:32 moonbeam kernel: [29938.885481]  ^O^_D: cx23885: cx23885_irq: ts1_status: 0x00000000  ts1_mask: 0x00000000 count: 0x12e
Sep 30 20:47:32 moonbeam kernel: [29938.885482]  ^O^_D: cx23885: cx23885_irq: ts2_status: 0x00000000  ts2_mask: 0x00000000 count: 0x0
Sep 30 20:47:32 moonbeam kernel: [29938.901371] cx23885: cx23885_cancel_buffers: cx23885_cancel_buffers()
Sep 30 20:47:32 moonbeam kernel: [29938.901372] cx23885: cx23885_stop_dma: cx23885_stop_dma()
Sep 30 20:47:32 moonbeam kernel: [29938.901382] cx23885: do_cancel_buffers: [ffff8e3316014c00/0] cancel - dma=0x9bf6c000
Sep 30 20:47:32 moonbeam kernel: [29938.901383] cx23885: do_cancel_buffers: [ffff8e33ba7a2000/1] cancel - dma=0x9bf6d000
..................
Sep 30 20:47:32 moonbeam kernel: [29938.901407] cx23885: do_cancel_buffers: [ffff8e3353f11c00/29] cancel - dma=0xdcaa9000
Sep 30 20:47:32 moonbeam kernel: [29938.901408] cx23885: do_cancel_buffers: [ffff8e3317c39c00/30] cancel - dma=0xdcece000
[... cut ...]
Sep 30 20:47:32 moonbeam kernel: [29938.929544]  ^O^_D: cx23885: cx23885_irq: audint_status: 0x00000000 audint_mask: 0x00000000 count: 0x0
Sep 30 20:47:32 moonbeam kernel: [29938.929545]  ^O^_D: cx23885: cx23885_irq: ts1_status: 0x00000000  ts1_mask: 0x00000000 count: 0x12e
Sep 30 20:47:32 moonbeam kernel: [29938.929546]  ^O^_D: cx23885: cx23885_irq: ts2_status: 0x00000000  ts2_mask: 0x00000000 count: 0x0
Sep 30 20:47:32 moonbeam kernel: [29938.952280]  ^O^_D: cx23885: cx23885_irq: pci_status: 0x000f8000  pci_mask: 0x00001f02

mspieth, 2017-09-30:
Eventually, valid data stops. Log of setup is the same as before.

Sep 30 20:03:04 moonbeam kernel: [27270.139325] cx23885: cx23885_start_dma: cx23885_start_dma() w: 0, h: 0, f: 0
Sep 30 20:03:04 moonbeam kernel: [27270.139328] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() Configuring channel [TS1 B]
Sep 30 20:03:04 moonbeam kernel: [27270.139329] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() 0x00010580 <- 0x00005000
Sep 30 20:03:04 moonbeam kernel: [27270.139330] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() 0x00010590 <- 0x000052f0
Sep 30 20:03:04 moonbeam kernel: [27270.139331] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() 0x000105a0 <- 0x000055e0
Sep 30 20:03:04 moonbeam kernel: [27270.139331] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() 0x000105b0 <- 0x000058d0
Sep 30 20:03:04 moonbeam kernel: [27270.139332] cx23885: cx23885_sram_channel_setup: cx23885_sram_channel_setup() 0x000105c0 <- 0x00005bc0
Sep 30 20:03:04 moonbeam kernel: [27270.139333] cx23885: cx23885_sram_channel_setup: [bridge 885] sram setup TS1 B: bpl=752 lines=5
Sep 30 20:03:04 moonbeam kernel: [27270.139333] cx23885: cx23885[1]: TS1 B - dma channel status dump
Sep 30 20:03:04 moonbeam kernel: [27270.139363] cx23885: cx23885[1]:   cmds: init risc lo   : 0x316f7000
Sep 30 20:03:04 moonbeam kernel: [27270.139390] cx23885: cx23885[1]:   cmds: init risc hi   : 0x00000000
Sep 30 20:03:04 moonbeam kernel: [27270.139418] cx23885: cx23885[1]:   cmds: cdt base       : 0x00010580
Sep 30 20:03:04 moonbeam kernel: [27270.139446] cx23885: cx23885[1]:   cmds: cdt size       : 0x0000000a
Sep 30 20:03:04 moonbeam kernel: [27270.139473] cx23885: cx23885[1]:   cmds: iq base        : 0x00010400
Sep 30 20:03:04 moonbeam kernel: [27270.139501] cx23885: cx23885[1]:   cmds: iq size        : 0x00000010
Sep 30 20:03:04 moonbeam kernel: [27270.139529] cx23885: cx23885[1]:   cmds: risc pc lo     : 0x00000000
Sep 30 20:03:04 moonbeam kernel: [27270.139556] cx23885: cx23885[1]:   cmds: risc pc hi     : 0x00000000
Sep 30 20:03:04 moonbeam kernel: [27270.139584] cx23885: cx23885[1]:   cmds: iq wr ptr      : 0x00000000
Sep 30 20:03:04 moonbeam kernel: [27270.139612] cx23885: cx23885[1]:   cmds: iq rd ptr      : 0x00000000
Sep 30 20:03:04 moonbeam kernel: [27270.139640] cx23885: cx23885[1]:   cmds: cdt current    : 0x00000000
Sep 30 20:03:04 moonbeam kernel: [27270.139668] cx23885: cx23885[1]:   cmds: pci target lo  : 0x00000000
Sep 30 20:03:04 moonbeam kernel: [27270.139696] cx23885: cx23885[1]:   cmds: pci target hi  : 0x00000000
Sep 30 20:03:04 moonbeam kernel: [27270.139724] cx23885: cx23885[1]:   cmds: line / byte    : 0x00000000
Sep 30 20:03:04 moonbeam kernel: [27270.139752] cx23885: cx23885[1]:   risc0: 
Sep 30 20:03:04 moonbeam kernel: [27270.139753] 0x00000000 [ INVALID count=0 ]
Sep 30 20:03:04 moonbeam kernel: [27270.139776] cx23885: cx23885[1]:   risc1: 
Sep 30 20:03:04 moonbeam kernel: [27270.139776] 0x00000000 [ INVALID count=0 ]
Sep 30 20:03:04 moonbeam kernel: [27270.139799] cx23885: cx23885[1]:   risc2: 
Sep 30 20:03:04 moonbeam kernel: [27270.139799] 0x00000000 [ INVALID count=0 ]
Sep 30 20:03:04 moonbeam kernel: [27270.139822] cx23885: cx23885[1]:   risc3: 
Sep 30 20:03:04 moonbeam kernel: [27270.139823] 0x00000000 [ INVALID count=0 ]
Sep 30 20:03:04 moonbeam kernel: [27270.139846] cx23885: cx23885[1]:   (0x00010400) iq 0: 
Sep 30 20:03:04 moonbeam kernel: [27270.139846] 0xdc6dcbc0 [ writecr sol eol 22 21 19 18 cnt0 resync 14 count=3008 ]
Sep 30 20:03:04 moonbeam kernel: [27270.139873] cx23885: cx23885[1]:   iq 1: 0x00000000 [ arg #1 ]
Sep 30 20:03:04 moonbeam kernel: [27270.139900] cx23885: cx23885[1]:   iq 2: 0x1c0002f0 [ arg #2 ]
Sep 30 20:03:04 moonbeam kernel: [27270.139926] cx23885: cx23885[1]:   iq 3: 0xdc6dceb0 [ arg #3 ]
Sep 30 20:03:04 moonbeam kernel: [27270.139953] cx23885: cx23885[1]:   (0x00010410) iq 4: 
Sep 30 20:03:04 moonbeam kernel: [27270.139953] 0x00000000 [ INVALID count=0 ]
Sep 30 20:03:04 moonbeam kernel: [27270.139978] cx23885: cx23885[1]:   (0x00010414) iq 5: 
Sep 30 20:03:04 moonbeam kernel: [27270.139979] 0x1c0002f0 [ write sol eol count=752 ]
Sep 30 20:03:04 moonbeam kernel: [27270.140004] cx23885: cx23885[1]:   iq 6: 0xdc6dd1a0 [ arg #1 ]
Sep 30 20:03:04 moonbeam kernel: [27270.140031] cx23885: cx23885[1]:   iq 7: 0x00000000 [ arg #2 ]
Sep 30 20:03:04 moonbeam kernel: [27270.140057] cx23885: cx23885[1]:   (0x00010420) iq 8: 
Sep 30 20:03:04 moonbeam kernel: [27270.140057] 0x1c0002f0 [ write sol eol count=752 ]
Sep 30 20:03:04 moonbeam kernel: [27270.140083] cx23885: cx23885[1]:   iq 9: 0xdc6dd490 [ arg #1 ]
Sep 30 20:03:04 moonbeam kernel: [27270.140109] cx23885: cx23885[1]:   iq a: 0x00000000 [ arg #2 ]
Sep 30 20:03:04 moonbeam kernel: [27270.140136] cx23885: cx23885[1]:   (0x0001042c) iq b: 
Sep 30 20:03:04 moonbeam kernel: [27270.140136] 0x00000000 [ INVALID count=0 ]
Sep 30 20:03:04 moonbeam kernel: [27270.140161] cx23885: cx23885[1]:   (0x00010430) iq c: 
Sep 30 20:03:04 moonbeam kernel: [27270.140162] 0x1c0002f0 [ write sol eol count=752 ]
Sep 30 20:03:04 moonbeam kernel: [27270.140187] cx23885: cx23885[1]:   iq d: 0xdc6dc8d0 [ arg #1 ]
Sep 30 20:03:04 moonbeam kernel: [27270.140214] cx23885: cx23885[1]:   iq e: 0x00000000 [ arg #2 ]
Sep 30 20:03:04 moonbeam kernel: [27270.140240] cx23885: cx23885[1]:   (0x0001043c) iq f: 
[... cut ...]
Sep 30 20:03:04 moonbeam kernel: [27270.140885] cx23885: cx23885_tsport_reg_dump: cx23885_tsport_reg_dump() VID_C_INT_SSTAT        0x00000000

except pci_status starts at 0x080F8000

no interrupts occur ever again.

after this even after module unload and reload, the chip is dead until a reboot/reset. This does not need to be a power cycle. It does not affect the other chip on the board.
pci reset (after unload) does not work. Controlled pci power cycle functionality seems disabled and was not tried.

Disabling the iommu in the bios did not help.

mspieth, 2017-10-01:
I have noticed on failure, 0x40090 and 0x40094 are set to 1 where they were previously 0

Oct  2 09:26:11 moonbeam kernel: [13170.877623] cx23885 0000:27:00.0: cx23885_reg_dump: 00040090: 00000001 00000001 00000000 00000000
Oct  2 09:26:11 moonbeam kernel: [13170.877633] cx23885 0000:27:00.0: cx23885_reg_dump: 000400a0: ffffffff 00000000 00000000 00000000

This is very deterministic. These registers are undocumented in the code as to their purpose.

mspieth, 2017-10-02:
mega patch that fixes various things

[cx23885-robustness.patch.zip](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1347664/cx23885-robustness.patch.zip)

mspieth, 2017-10-02:
now seeing

Oct  3 01:21:48 moonbeam kernel: [70508.943410] cx23885 0000:27:00.0: cx23885_reg_dump: 00040090: 00000001 00000001 00000000 00000000
Oct  3 01:21:48 moonbeam kernel: [70508.943420] cx23885 0000:27:00.0: cx23885_reg_dump: 000400a0: ffffffff 00010000 00000000 00000000

which teh previous method doesnt clear. notice 0x400a4

mspieth, 2017-10-04:
A new solution to this problem is to delay stop_dma by 20ms + wait for 0x40090/4 not both non zero before completing stop_dma. This has been working for 12h with EIT open/grab/close so this may be a suitable solution. Patch will follow when I am happy. Not there is virtually no delay between one close and the next open (10ms in logs).
Also previous solution to mpeg risc error is not a complete fix. Adding printks does fix it but is not a desirable solution.

mspieth, 2018-01-25:
Latest reasonably stable patchset including brads set for current media tarball and kernel 4.14.13.
lots of debug on open and close seems to help with ryzen.

[latest_patches_20180126.zip](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1664098/latest_patches_20180126.zip)

b-rad-NDi, 2018-02-15:
@mspieth : you have an 885 version of this board, correct? Oddly 0x40090 and 0x40094 are only listed in the cx23888 datasheet. Unfortunately not much documentation past their names: TC_REQ and RC_REQ_SET

You did interpret these correctly though, TC_REQ says it interrupts the RISC engine when set.

mspieth, 2018-02-18:
@b-rad-NDi : 885 I believe so. see other issue where I posted when you asked for the device.

If you have a datasheet and an NDA you could send an info request to connexant/howmever for clarification on TC_REQ and RC_REQ_SET.

The other choice is to get the source for the windows driver (which you probably have access to as part of hauppauge) and see how things are done there. If not there is always disassembling the windows driver to gain secrets.
If you need analysis help on any of this, let me know but I suspect not.

closed 2018-05-07 by b-rad-NDi