b-rad-NDi/Ubuntu-media-tree-kernel-builder #25: Hauppauge WinTV HVR 950Q Ubuntu Not Working (weird IR error in dmesg)

niftylettuce, 2017-10-17:
As a side project experiment I wanted to try this out so I got a 950Q off eBay for cheap.

I've attempted to build from git source per instructions here https://www.linuxtv.org/wiki/index.php/Hauppauge_WinTV-HVR-950Q, but that didn't work so I tried this PPA which seems to have the same firmware, but that didn't seem to work either.

Not sure whether PPA or Git source is more maintained?

**The error I receive is that the channels scan once, but then further actions upon the device fail. 
 The only way to access device again is to disconnect and reconnect it.**

Upon a quick Google search (e.g "w_scan output empty 950q", "950q not working after scan"), you can see this yields dozens of results in forums of people with similar issues with the 950Q model.

I'm not certain what the issue is, or how I can help to debug, but I've provided some output from `dmesg` seemingly related to infrared controller.  Is that causing a conflict somehow?

[image]

In order to get this to work to begin with, I had to write a `udev` rule for the device - otherwise my user could not read device info over `lsusb -v` for instance.  I think this should be added to the instructions of this repo and also the LinuxTV link above?

I've installed `w_scan` and was able to scan channels to a `channels.conf` file via:

```sh
w_scan -fa > channels.conf
```

However upon attempting to open `me-tv` or `kaffeine`, it seems the device has stopped working.

The only way to get the device to scan or seemingly connect again is by unplugging it.

I also seem to get `segfaults` as well:

[image]

b-rad-NDi, 2017-10-17:
Hi. My PPA is built directly off this git source, so at least this project and the PPA are directly related. The PPA is often times behind the git though, as we test all the additional patches.

As for your 950Q issues, I cannot say I have encountered this myself. I'll ask support if they know about this issue. I've never had to add udev rules or jump through any hoops personally. What type of signal are you tuning?

Based off your diagnosis there must be something wrong. Could you enable debug for the affected drivers and do a plug/unplug/plug for me? Also I'd like log of the scanning process as well as what is produced when you try to use the tuner during one of these 'dead' states.

I'll update this issue when I hear anything.

niftylettuce, 2017-10-17:
I'm scanning ATSC in US and what command can I run to get you output you
need?

b-rad-NDi, 2017-10-17:
This for example enables debug for si2157 and lgdt3306:
`$ cat si2157.conf
options si2157 dyndbg==pfl
$ cat lgdt3306a.conf
options lgdt3306a dyndbg==pfl
options lgdt3306a debug=1`
You'll want to add dyndb and debug lines to the proper driver conf file, then after a reboot you'll see everything I need in dmesg.

niftylettuce, 2017-10-17:
Unfortunately I decided I'm going to return this device.  Thanks for your
work in the open source community on these things.

b-rad-NDi, 2017-10-17:
I know you've returned this already, but support and our other engineers say this is indicative of a failed dongle. I use a 950Q with ATSC on occasion and I've never had to jump through the hoops you describe. I wish I had one that operated like yours though, so I can robustify (??) the driver however.

niftylettuce, 2017-10-17:
@b-rad-NDi if you email me your address I'll mail you it for testing.  niftylettuce@gmail.com

b-rad-NDi, 2017-10-17:
Will do.

closed 2017-10-31 by b-rad-NDi