b-rad-NDi/Ubuntu-media-tree-kernel-builder #26: Kernel Oops

mrevjd, 2017-10-21:
Hi

After a recent update I started receiving the below error:

[  152.356805] BUG: unable to handle kernel paging request at ffffb1bf420cfad3
[  152.356879] IP: si2168_set_frontend+0x35d/0x5f0 [si2168]
[  152.356911] PGD 237099067
[  152.356912] PUD 23709e067
[  152.356930] PMD 22e8e0067
[  152.356947] PTE 0

[  152.356989] Oops: 0000 [#1] SMP
[  152.357010] Modules linked in: rc_hauppauge em28xx_rc rc_core si2157 si2168 i2c_mux em28xx_dvb dvb_core vmw_vsock_vmci_transport vsock coretemp crct10dif_pclmul crc32_pclmul ghash_clmulni_intel pcbc em28xx tveeprom v4l2_common aesni_intel videodev aes_x86_64 crypto_simd glue_helper ppdev media cryptd vmw_balloon joydev input_leds serio_raw nfit mac_hid vmw_vmci parport_pc shpchp i2c_piix4 lp parport ip_tables x_tables autofs4 hid_generic usbhid hid psmouse vmwgfx ttm mptspi mptscsih drm_kms_helper mptbase syscopyarea scsi_transport_spi sysfillrect sysimgblt fb_sys_fops e1000 drm pata_acpi fjes
[  152.357324] CPU: 1 PID: 1496 Comm: kdvb-ad-1-fe-0 Not tainted 4.10.0-32201708161632-generic #0+mediatree+hauppauge-Ubuntu
[  152.357386] Hardware name: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 04/05/2016
[  152.357447] task: ffff94aef2ce6a40 task.stack: ffffb1bf41ed8000
[  152.357484] RIP: 0010:si2168_set_frontend+0x35d/0x5f0 [si2168]
[  152.357519] RSP: 0018:ffffb1bf41edb9b0 EFLAGS: 00010202
[  152.357550] RAX: 00000000000000f0 RBX: ffff94aeee141828 RCX: 0000000000000000
[  152.357591] RDX: ffff94aef2ce6a40 RSI: 0000000000000002 RDI: ffff94aeee141800
[  152.357632] RBP: ffffb1bf41edba30 R08: 000000000001c620 R09: ffffffff87ea3a4c
[  152.357673] R10: ffff94aeefe85130 R11: ffff94aeefe85251 R12: ffff94aeee363c00
[  152.357714] R13: ffffb1bf420cfac0 R14: 00000000000000f7 R15: ffff94aeee141800
[  152.357756] FS:  0000000000000000(0000) GS:ffff94aeffc80000(0000) knlGS:0000000000000000
[  152.357802] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  152.357836] CR2: ffffb1bf420cfad3 CR3: 0000000232966000 CR4: 00000000000006e0
[  152.357939] Call Trace:
[  152.357964]  ? usb_control_msg+0xec/0x120
[  152.357991]  ? cpumask_next_and+0x2d/0x50
[  152.358032]  dvb_frontend_swzigzag_autotune+0x1e2/0x2e0 [dvb_core]
[  152.359141]  dvb_frontend_swzigzag+0x335/0x3a0 [dvb_core]
[  152.360226]  ? dequeue_entity+0xed/0x420
[  152.361297]  ? dequeue_task_fair+0x5ab/0xaa0
[  152.362347]  ? pick_next_task_fair+0x10d/0x4b0
[  152.363370]  ? __switch_to+0x23c/0x520
[  152.364375]  ? lock_timer_base+0x81/0xa0
[  152.365456]  ? try_to_del_timer_sync+0x5a/0x80
[  152.366484]  ? del_timer_sync+0x48/0x50
[  152.367518]  ? schedule_timeout+0x1e7/0x3f0
[  152.368677]  ? si2157_init+0x3e3/0x620 [si2157]
[  152.369732]  ? si2168_init+0x1c7/0x520 [si2168]
[  152.370754]  ? del_timer_sync+0x50/0x50
[  152.371758]  dvb_frontend_thread+0x400/0x700 [dvb_core]
[  152.372756]  ? wake_atomic_t_function+0x60/0x60
[  152.373706]  kthread+0x109/0x140
[  152.374626]  ? dtv_set_frontend+0x420/0x420 [dvb_core]
[  152.375536]  ? kthread_create_on_node+0x60/0x60
[  152.376407]  ret_from_fork+0x2c/0x40
[  152.377247] Code: aa fd ff ff 8b 05 a9 1a 00 00 89 45 a8 0f b7 05 a3 1a 00 00 66 89 45 ac 0f b6 45 a4 41 09 c6 3c f0 44 88 75 ac 75 04 80 4d ad 02 <41> 80 7d 13 00 74 04 80 4d ad 01 48 8d 75 a8 4c 89 e7 c7 45 c8
[  152.379812] RIP: si2168_set_frontend+0x35d/0x5f0 [si2168] RSP: ffffb1bf41edb9b0
[  152.380665] CR2: ffffb1bf420cfad3
[  152.381512] ---[ end trace 1c04621115702ff0 ]---

This on: 
Linux ubuntu-tv01 4.10.0-32201708161632-generic #0+mediatree+hauppauge-Ubuntu SMP Thu Aug 17 00:27:10 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

Let me know if you require more info

Cheers
Evan

b-rad-NDi, 2017-10-22:
I will look into this issue. It would appear you're running in a virtualized environment? We technically don't support virtualization, because there are issues that crop up in VM's that we cannot reproduce natively. I will still review the driver though.

mrevjd, 2017-10-22:
No problems.
Correct on the virtual environment. However this only just started happening when I updated to 4.10.0-32201708161632-generic so thought you should know.

b-rad-NDi, 2017-10-27:
I have updated the 4.10 kernel, you can apt-get update and manually install it from the ppa to test out.

mrevjd, 2017-10-27:
Thanks for the heads up.

mrevjd, 2017-10-29:
So far so good. 
Thanks

b-rad-NDi, 2017-10-30:
Excellent, please file another issue if you find anything else.

closed 2017-10-30 by b-rad-NDi