b-rad-NDi/Ubuntu-media-tree-kernel-builder #27: Upstream drivers ?

CvH, 2017-10-29:
Hi, I am curious if there are any plans to "upstream" your drivers ? 
As OS maintainer I got several requests from users for those devices but unfortunately no proper solution yet. 
To add 20+ patches to the buildsystem doesn't sounds like a good idea long and midterm if there is no "timeframe" for upstreaming.

best regards

b-rad-NDi, 2017-10-30:
Apologies, the v4l2-core-Reverse-758d90e16-kvmalloc patch derailed upstreaming efforts for some months. I had too many other projects to manage and couldn't dedicate the time to find root of the issue. I didn't realize it was a core issue, and the effects were anything from GPF to invalid opcode to random oopses depending on the machine. Now that I've determined it is not any of my patches they will be heading to the mailing list for review soon.

CvH, 2017-10-30:
@b-rad-NDi thanks for reply, good to hear :) I was asking because several "other" brands just do nothing to upstream and that creates a real mess downstream. I guess nobody is too much happy to do that game again and again ;) 
Will add support to LE as soon your patches hit media_tree, thanks for your work !

b-rad-NDi, 2017-10-30:
The point of this project is to get Hauppauge back in the thick of things. We do not have a dedicated upstreamer, and unfortunately it is too easy to neglect upstream when there's so many other responsibilities in a smaller company. Thankfully the interest generated in this has further convinced my bosses of the importance of me absorbing the part. Keep an eye out, I hope most of the patches will be submitted within the next couple weeks.

CHBMB, 2017-10-30:
I'd echo @CvH I also am involved in maintaining DVB drivers for an OS and I find it difficult to recommend Hauppauge at the moment as it's a Slackware based and the Hauppauge driver situation is not great.

I'd be far more likely to recommend your hardware for the growing Linux PVR enthusiast market with upstream support or at least an easier driver package.

b-rad-NDi, 2017-10-31:
Passive aggressive guilt tripping won't make the process magically go any quicker.

CHBMB, 2017-10-31:
I'm not trying to rush you, and there's no need for the attitude, The point of my first post was to perhaps give you some more traction with your bosses, but with Plex supporting Hauppauge tuners I've seen a significant rise in interest in your products on Unraid.  Their price point is good compared to alternatives like Digital Devices and TBS, if the drivers were upstream. it'd be a no brainer, as that would be a definite market advantage over both of those companies.

Personally I have a TBS tuner and I'll quite happily keep on recommending them as I've found their support department to be very polite and responsive when I've approached them for help and advice.

b-rad-NDi, 2017-10-31:
Apologies then. The media tree maintainers have been quite busy. I'm hoping to get reviewed soon.

b-rad-NDi, 2017-10-31:
I've been trying to get someone with the equipment to review them for months. It's nothing to do with my bosses any more.

CHBMB, 2017-10-31:
Good luck! :+1:

b-rad-NDi, 2017-11-01:
btw, this is my personal github, unrelated to the company completely. If you'd like to request support from Hauppauge for anything, such as slackware support, especially so my bosses see it, you need to file a support ticket with the company.

CHBMB, 2017-11-01:
Happy to do so if it helps you out.  Would mentioning anything in particular be helpful to your cause?

b-rad-NDi, 2017-11-01:
Anything you can think of to further impress on them that me continuing driver development and upstream work is important helps.

b-rad-NDi, 2018-01-05:
Long process of authorship research is complete. Authorization received from any involved parties.

Initial batch of 9 patches sent upstream. A ton more to follow tomorrow :)

CvH, 2018-01-05:
awesome news :)

gentoolinux, 2018-01-05:
Thank you very much, Brad! Great work! Let's hope your patches will soon be accepted by the kernel maintainers.

parkerlreed, 2018-01-05:
@b-rad-NDi Can't wait for the dual-HD support :D Thanks again for all the wonderful work you have done for Hauppauge and the community.

CHBMB, 2018-01-06:
Well done that man!

b-rad-NDi, 2018-01-09:
All new devices submitted. Only things left are:
- analog: will probably require work
- second frontend on 975 (patch set pushed today for review/rfc)
- bits and bobs of leftover opimizations

parkerlreed, 2018-01-10:
@b-rad-NDi :heart: https://www.spinics.net/lists/linux-media/msg127060.html

b-rad-NDi, 2018-01-12:
30 patches submitted, 10 to go.

b-rad-NDi, 2018-03-06:
Only took 60 days and my patches have finally started working into mainline :D

CvH, 2018-03-06:
@b-rad-NDi do you maintain an recent patch tree `patches/mainline-extra/tip/` here ? (easier to grab all still need patches)

b-rad-NDi, 2018-03-06:
Once I work through the merge today and clean out what is in patchwork I will do some house cleaning.

b-rad-NDi, 2018-03-06:
As it is right now everything in patches/mainline-extra/tip is exactly what is in patchwork

CvH, 2018-03-06:
yea tx :)

b-rad-NDi, 2018-03-06:
All patches have been sorted into series directories. The moment my 'missing bits' series gets merged into mainline I will be removing the following sets:
```
- 01.em28xx.dual.hd
- 02.spectral.inversion
- 03.935.and.975
- 04.975.second.frontend
- 05.new.pcie.devices
- 06.cx25840.extra.clocking.option
```

CvH, 2018-03-07:
> The moment my 'missing bits' series gets merged into mainline I will be removing the following sets:

Hmm I guess you are too quick :> another 12 patches got merged

b-rad-NDi, 2018-03-07:
:-D
Will pull fresh media tree and clean house today.

b-rad-NDi, 2018-03-09:
All upstreamed patch sets removed :)

CHBMB, 2018-03-10:
@b-rad-NDi Thanks for all your hard work on this.  Much appreciated.

CvH, 2018-05-07:
tx for your efforts, are the other patches just "forgotten" upstream or why are they so long at limbo ?

b-rad-NDi, 2018-05-07:
A set or two is submitted at a time, to ensure minimal collisions. The first set took over two months to be accepted, and the remaining sets required additional testing and fixes. Plus that whole "time" thing. Some of my recent patches were picked up very quickly, so maybe the pace will get better.

b-rad-NDi, 2018-05-08:
Ryzen patch set submitted to the list. It is a bit more than trivial of a series, so we'll see how the process goes with that set.

b-rad-NDi, 2018-05-11:
Not bad, only took three days to achieve queued status on the Ryzen set.

CvH, 2018-05-11:
likely easy to reproduce :D

b-rad-NDi, 2018-05-11:
The ryzen work is not done yet though, I realized that QuadHD analog audio inputs causes an almost identical issue when I started validating the analog tuning series to submit it...