b-rad-NDi/Ubuntu-media-tree-kernel-builder #28: Cannot install linux-image-mediatree

sauma, 2017-10-30:
I recently bought a hauppauge usb tuner for use on a Plex server running Ubuntu 17.10. When following the instructions here I am able to add the PPA successfully, but installing the packages fails. 

sudo apt-get install linux-image-mediatree
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-image-mediatree

b-rad-NDi, 2017-10-30:
You can't install the virtual packages, because something changed in artful and now my scripts to build them fail. The kernel packages are in the ppa though, so you can manually install them.
`sudo apt-get update && apt-cache search mediatree`

sauma, 2017-10-31:
That looks like it ran successfully.

`Reading package lists... Done
linux-image-mediatree - linux-image packages wih integrated LinuxTV.org mediatree drivers
linux-image-extra-4.10.0-32201708161632-generic - Linux kernel extra modules with slipstreamed mediatree drivers for version 4.10.0 on 64 bit x86 SMP
linux-headers-mediatree - Header files related to linux-image-mediatree
linux-headers-4.10.0-32201708161632-generic - Linux kernel headers with slipstreamed mediatree drivers for version 4.10.0 on 64 bit x86 SMP
`
Is there a final step I'm missing? I rebooted and the devices doesn't show as available for Plex. 
`lsusb` 
does show 
`Bus 001 Device 003: ID 2040:826d Hauppauge`
which is encouraging. 

If this isn't the appropriate place to troubleshoot this, I can reach out to Hauppauge support directly. Thanks!

b-rad-NDi, 2017-10-31:
I'm lead engineer at Hauppauge, so it would come back to me ;) It looks like this is a DVB device, so you will also need to install the firmware:
`sudo apt-get install linux-firmware-hauppauge`
Lemme know if it works after that.

sauma, 2017-10-31:
I've definitely stumbled upon the right corner of the internet!

`The following NEW packages will be installed:
  linux-firmware-hauppauge
0 upgraded, 1 newly installed, 0 to remove and 3 not upgraded.
Need to get 0 B/6,916 B of archives.
After this operation, 18.4 kB of additional disk space will be used.
Selecting previously unselected package linux-firmware-hauppauge.
(Reading database ... 208974 files and directories currently installed.)
Preparing to unpack .../linux-firmware-hauppauge_0.1.0+zesty_all.deb ...
Unpacking linux-firmware-hauppauge (0.1.0+zesty) ...
Setting up linux-firmware-hauppauge (0.1.0+zesty) ...
`
After a reboot it still just spins when trying to find the tuner. 

Looking at the Plex Tuner Service logs, I see this repeating:
`Oct 31, 2017 14:27:17.221 [0x7fc6aeffd700] INFO - [I] web_service_t::process_device_discovery
Oct 31, 2017 14:27:17.221 [0x7fc6aeffd700] INFO - [I] device_reactor_t::get_device_list ()
Oct 31, 2017 14:27:17.222 [0x7fc6aeffd700] INFO - [I] dvb_device_factory_t::get_device_list ()
Oct 31, 2017 14:27:19.227 [0x7fc6aeffd700] INFO - [I] SAT2IP - discover_satip_servers(): no servers found
Oct 31, 2017 14:27:19.633 [0x7fc6aeffd700] INFO - [I] v4l::DeviceGetList_impl. getting v4l device list
Oct 31, 2017 14:27:19.640 [0x7fc6aeffd700] INFO - [I] onetv_factory::DeviceGetListEx >>
Oct 31, 2017 14:27:19.643 [0x7fc6aeffd700] INFO - [I] onetv_factory::DeviceGetListEx. Found device 5964:20742
Oct 31, 2017 14:27:19.643 [0x7fc6aeffd700] INFO - [I] onetv_factory::DeviceGetListEx. Found device 8256:33389
Oct 31, 2017 14:27:19.643 [0x7fc6aeffd700] INFO - [I] onetv_factory::DeviceGetListEx. Found device 2965:6032
Oct 31, 2017 14:27:19.643 [0x7fc6aeffd700] INFO - [I] onetv_factory::DeviceGetListEx. Found device 3034:5924
Oct 31, 2017 14:27:19.643 [0x7fc6aeffd700] INFO - [I] onetv_factory::DeviceGetListEx. Found device 8263:2133
Oct 31, 2017 14:27:19.644 [0x7fc6aeffd700] INFO - [I] onetv_factory::DeviceGetListEx. Found device 8457:10257
Oct 31, 2017 14:27:19.644 [0x7fc6aeffd700] INFO - [I] onetv_factory::DeviceGetListEx. Found device 1266:45858
Oct 31, 2017 14:27:19.644 [0x7fc6aeffd700] INFO - [I] onetv_factory::DeviceGetListEx. Found device 5964:20742
Oct 31, 2017 14:27:19.644 [0x7fc6aeffd700] INFO - [I] onetv_factory::DeviceGetListEx. Found device 3034:297
Oct 31, 2017 14:27:19.644 [0x7fc6aeffd700] INFO - [I] onetv_factory::DeviceGetListEx. Found device 32903:36
Oct 31, 2017 14:27:19.644 [0x7fc6aeffd700] INFO - [I] onetv_factory::DeviceGetListEx. Found device 7531:2
Oct 31, 2017 14:27:19.644 [0x7fc6aeffd700] INFO - [I] onetv_factory::DeviceGetListEx. Found device 1267:10
Oct 31, 2017 14:27:19.644 [0x7fc6aeffd700] INFO - [I] onetv_factory::DeviceGetListEx. Found device 32903:36
Oct 31, 2017 14:27:19.644 [0x7fc6aeffd700] INFO - [I] onetv_factory::DeviceGetListEx. Found device 7531:2
Oct 31, 2017 14:27:19.644 [0x7fc6aeffd700] INFO - [I] onetv_factory::DeviceGetListEx. Found device 7531:2
Oct 31, 2017 14:27:19.644 [0x7fc6aeffd700] INFO - [I] onetv_factory::DeviceGetListEx. Found device 8457:33040
Oct 31, 2017 14:27:19.644 [0x7fc6aeffd700] INFO - [I] onetv_factory::DeviceGetListEx. Found device 7531:3`

b-rad-NDi, 2017-10-31:
ugh i hate how github code tags eats newlines when it should just be a pre tag. At least Plex notes that it found a device.

Can you reboot without the device connected, then note where dmesg ends. Then plug the device in and paste all the dmesg logs after the initial point?

sauma, 2017-10-31:
I'll leave off the code tags this time. Here's what dmesg shows after I plug in the device. 

[  145.004713] usb 1-1.2: new high-speed USB device number 6 using ehci-pci
[  145.114292] usb 1-1.2: New USB device found, idVendor=2040, idProduct=826d
[  145.114299] usb 1-1.2: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[  145.114303] usb 1-1.2: Product: 955D
[  145.114306] usb 1-1.2: Manufacturer: HCW
[  145.114309] usb 1-1.2: SerialNumber: 0013965722

b-rad-NDi, 2017-11-01:
And nothing else? You should see some additional messages of the drivers initializing. I will review the driver later today and see if I notice anything.

sauma, 2017-11-01:
Tried it again and that's all that comes up (except for about fifty lines that seem to be about Ubuntu not sure what to do with a screen rotation command).

b-rad-NDi, 2017-11-01:
Will review now and get back to you.

b-rad-NDi, 2017-11-01:
I will have to spin up an artful VM, I have an ATSC DualHD with me and it is working with the PPA. The vid:pid is in there, so it 'should' work for you.

b-rad-NDi, 2017-11-01:
Wait a second, did you apt-get install the packages that apt-cache listed?
Can you do:
`dpkg -l | grep mediatree`
Also in your first post you indicate you're on 17.10 aka Artful, but your logs indicate kernel 4.10, which is 17.04 aka Zesty. Can you clarify what version of Ubuntu you are running:
`lsb_release -r`

sauma, 2017-11-01:
`sam@ubuntu:~$ sudo dpkg -l | grep mediatree`
```
ii  linux-headers-4.10.0-32201708161632             4.10.0-32201708161632.0+mediatree+hauppauge  all         Header files related to Linux kernel version 4.10.0
ii  linux-headers-4.10.0-32201708161632-generic     4.10.0-32201708161632.0+mediatree+hauppauge  amd64        Linux kernel headers with slipstreamed mediatree drivers for version 4.10.0 on 64 bit x86 SMP
ii  linux-headers-mediatree                         0.16.5+zesty                                 all          Header files related to linux-image-mediatree
ii  linux-image-4.10.0-32201708161632-generic       4.10.0-32201708161632.0+mediatree+hauppauge  amd64        Linux kernel image for version 4.10.0 on 64 bit x86 SMP
ii  linux-image-extra-4.10.0-32201708161632-generic 4.10.0-32201708161632.0+mediatree+hauppauge  amd64        Linux kernel extra modules with slipstreamed mediatree drivers for version 4.10.0 on 64 bit x86 SMP
ii  linux-image-mediatree                           0.16.5+zesty                                 all          linux-image packages wih integrated LinuxTV.org mediatree drivers

```
You're right! I am on 17.10 but changed to an older kernel when I saw that that was the latest on listed us supported on. I've changed it back to the default, as shown below. 
```
sam@ubuntu:~$ lsb_release -r
Release:        17.10
```
```
sam@ubuntu:~$ uname -a
Linux ubuntu 4.13.0-16-generic #19-Ubuntu SMP Wed Oct 11 18:35:14 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```

b-rad-NDi, 2017-11-01:
I'm not sure how you managed to install 4.10 on Artful, did you manually configure /etc/apt/sources.list? You should have gotten 4.13 from the ppa if things were configured correctly, but you somehow have the Zesty kernel installed.

b-rad-NDi, 2017-11-01:
The kernels you want should look like:
linux-headers-4.13.0-16201710261111
linux-image-4.13.0-16201710261111
linux-image-extra-4.13.0-16201710261111

The end is the timestamp of build, 26 October, 2017 11:11

sauma, 2017-11-01:
[image]
I saw this and figured since it didn't work by default I could install an older kernel and tell grub to boot from there. That must be what I was in when I installed mediatree. I'll clean this up and try again.

b-rad-NDi, 2017-11-01:
You should be able to add the ppa, then 'apt-get update' and manually install the packages. If you desire to download them directly and install via 'dpkg -i' then you can go here:
https://launchpad.net/~b-rad/+archive/ubuntu/kernel+mediatree+hauppauge/+packages
Then expand linux-4.13.0-16201710261111.0+mediatree+hauppauge  and download the four packages I listed above for your arch (amd64 or --gasp-- i386)

sauma, 2017-11-01:
It's really odd that it just doesn't find those packages after I add the repository. 

I downloaded the .debs directly, installed them, and rebooted. 

[image]

Success! If you want me to try anything else to troubleshoot the PPA issue, I'm happy to help but downloading those 4 files directly and installing manually did the trick. 

Thanks for the help!

b-rad-NDi, 2017-11-01:
:D I will still spin up an Artful VM and see if I can determine what's up with the packages not showing up. I will close this issue, since it is 'solved'. If you have any further issues please file another issue. Cheers.

closed 2017-11-01 by b-rad-NDi

parkerlreed, 2017-11-03:
Another dualHD user! :D 
[image]