b-rad-NDi/Ubuntu-media-tree-kernel-builder #29: BUG: unable to handle kernel paging request at XXX in si2168_set_frontend call

comio, 2017-11-24:
Hi,

Before all, I'm using a gentoo kernel based on 4.14.1 (gentoo-sources-4.14.1, see [4.14-1 Patch List](https://dev.gentoo.org/~mpagano/genpatches/patches-4.14-1.htm)) and I known that you are supporting only Ubuntu based kernel.

I applied all patches provided by you for 4.12 vanilla, but I'm having the crash dump attached. Can you give me some hint?
will these patches pushed on kernel mainstream?

Thanks a lot,

luigi

`[34408.703615] BUG: unable to handle kernel paging request at ffffb58f4162f8d3
[34408.703649] IP: si2168_set_frontend+0x341/0x5b0 [si2168]
[34408.703657] PGD 46e125067 P4D 46e125067 PUD 46e130067 PMD 46bc74067 PTE 0
[34408.703678] Oops: 0000 [#1] SMP
[34408.703686] Modules linked in: veth cfg80211 xt_nat nf_conntrack_netlink xfrm_user xfrm_algo xt_CHECKSUM xt_addrtype br_netfilter overlay devlink arc4 md4 cifs ccm fscache nf_conntrack_netbios_ns nf_conntrack_broadcast ip6t_REJECT nf_reject_ipv6 ip6t_rpfilter xt_tcpudp ipt_REJECT nf_reject_ipv4 xt_conntrack ipt_MASQUERADE nf_nat_masquerade_ipv4 ip_set ebtable_nat ebtable_broute ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack em28xx_rc rc_core libcrc32c si2157 si2168 i2c_mux em28xx_dvb dvb_core bridge iptable_mangle iptable_raw stp llc iptable_security ebtable_filter ebtables ip6table_filter ip6_tables iptable_filter nf_tables nfnetlink em28xx tveeprom v4l2_common videodev
[34408.703867]  input_leds led_class media vfat fat intel_rapl intel_powerclamp coretemp kvm_intel kvm iTCO_wdt irqbypass punit_atom_debug crct10dif_pclmul iTCO_vendor_support crc32_pclmul ghash_clmulni_intel pcbc snd_intel_sst_acpi aesni_intel snd_hda_codec_hdmi snd_intel_sst_core aes_x86_64 snd_hda_codec_realtek crypto_simd snd_hda_codec_generic glue_helper cryptd snd_soc_rt5670 snd_soc_sst_atom_hifi2_platform snd_soc_rl6231 snd_hda_intel snd_soc_sst_match snd_hda_codec r8169 snd_soc_core lpc_ich i2c_i801 pcspkr snd_hwdep mii snd_hda_core snd_compress ac97_bus snd_pcm snd_timer snd acpi_pad intel_int0002_vgpio rtc_cmos mei_txe mei soundcore binfmt_misc sch_fq_codel ip_tables x_tables uas usb_storage xhci_pci xhci_hcd ahci libahci i2c_hid pwm_lpss_platform 8250_dw pwm_lpss pinctrl_cherryview nct6775 hwmon_vid
[34408.704067] CPU: 1 PID: 24331 Comm: kdvb-ad-0-fe-0 Tainted: G        W       4.14.1-gentoo #1
[34408.704074] Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./N3700-ITX, BIOS P1.80 01/11/2017
[34408.704082] task: ffff961fe4d00000 task.stack: ffffb58f432a0000
[34408.704094] RIP: 0010:si2168_set_frontend+0x341/0x5b0 [si2168]
[34408.704102] RSP: 0018:ffffb58f432a3a00 EFLAGS: 00010202
[34408.704111] RAX: 00000000000000f0 RBX: ffff961fd448d828 RCX: 0000000000000000
[34408.704118] RDX: ffff961fe4d00000 RSI: 0000000000000002 RDI: ffff961fd448d800
[34408.704126] RBP: ffffb58f432a3a68 R08: 000000000001ee20 R09: ffffffffb87b2eed
[34408.704133] R10: ffff961fec7d71e1 R11: 0000000000000000 R12: ffff961fd5d37800
[34408.704140] R13: ffffb58f4162f8c0 R14: 00000000000000f8 R15: ffff961fd448d800
[34408.704149] FS:  0000000000000000(0000) GS:ffff961fffc80000(0000) knlGS:0000000000000000
[34408.704156] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[34408.704164] CR2: ffffb58f4162f8d3 CR3: 0000000402e08000 CR4: 00000000001006e0
[34408.704170] Call Trace:
[34408.704204]  dvb_frontend_swzigzag_autotune+0x16e/0x2f0 [dvb_core]
[34408.704239]  dvb_frontend_swzigzag+0x2fa/0x3a0 [dvb_core]
[34408.704254]  ? i2c_adapter_unlock_bus+0x12/0x20
[34408.704265]  ? i2c_transfer+0x61/0xd0
[34408.704279]  ? put_prev_entity+0x2f/0x5e0
[34408.704290]  ? sched_clock+0x9/0x10
[34408.704301]  ? pick_next_task_fair+0x155/0x590
[34408.704312]  ? __switch_to+0x1f3/0x4e0
[34408.704324]  ? lock_timer_base+0x75/0x90
[34408.704335]  ? try_to_del_timer_sync+0x53/0x80
[34408.704345]  ? del_timer_sync+0x39/0x40
[34408.704358]  ? schedule_timeout+0x18a/0x350
[34408.704368]  ? call_timer_fn+0x130/0x130
[34408.704395]  dvb_frontend_thread+0x4dc/0x6d0 [dvb_core]
[34408.704407]  ? wait_woken+0x80/0x80
[34408.704420]  kthread+0x125/0x140
[34408.704454]  ? dtv_set_frontend+0x410/0x410 [dvb_core]
[34408.704467]  ? kthread_create_on_node+0x70/0x70
[34408.704479]  ret_from_fork+0x25/0x30
[34408.704488] Code: 61 fd ff ff 8b 05 c5 19 00 00 89 45 a8 0f b7 05 bf 19 00 00 66 89 45 ac 0f b6 45 a4 41 09 c6 3c f0 44 88 75 ac 0f 84 1e 02 00 00 <41> 80 7d 13 00 74 04 80 4d ad 01 48 8d 75 a8 49 bd 06 00 00 00 
[34408.704561] RIP: si2168_set_frontend+0x341/0x5b0 [si2168] RSP: ffffb58f432a3a00
[34408.704563] CR2: ffffb58f4162f8d3
[34408.704567] ---[ end trace ef5132db24bb0955 ]---
`

comio, 2017-11-24:
```
gentoobox dvb-frontends # gdb ./si2168.o
GNU gdb (Gentoo 8.0.1 vanilla) 8.0.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://bugs.gentoo.org/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./si2168.o...done.
(gdb) list si2168_set_frontend+0x341/0x5b0
Function "si2168_set_frontend+0x341/0x5b0" not defined.
(gdb) list *(si2168_set_frontend+0x341/0x5b0)
0x630 is in si2168_set_frontend (drivers/media/dvb-frontends/si2168.c:224).
219		dev_dbg(&client->dev, "failed=%d\n", ret);
220		return ret;
221	}
222	
223	static int si2168_set_frontend(struct dvb_frontend *fe)
224	{
225		struct i2c_client *client = fe->demodulator_priv;
226		struct si2168_dev *dev = i2c_get_clientdata(client);
227		struct dtv_frontend_properties *c = &fe->dtv_property_cache;
228		struct si2168_config *config = client->dev.platform_data;
(gdb) 

```

comio, 2017-11-24:
dmesg:

[    6.389331] em28xx 1-5.2:1.0: New device HCW dualHD @ 480 Mbps (2040:0265, interface 0, class 0)
[    6.389343] em28xx 1-5.2:1.0: DVB interface 0 found: isoc
[    6.389389] em28xx 1-5.2:1.0: chip ID is em28174
[    7.595105] em28xx 1-5.2:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0x2be4c2c8
[    7.595110] em28xx 1-5.2:1.0: EEPROM info:
[    7.595114] em28xx 1-5.2:1.0: 	microcode start address = 0x0004, boot configuration = 0x01
[    7.601916] em28xx 1-5.2:1.0: 	AC97 audio (5 sample rates)
[    7.601920] em28xx 1-5.2:1.0: 	500mA max power
[    7.601924] em28xx 1-5.2:1.0: 	Table at offset 0x27, strings=0x0e6a, 0x1888, 0x087e
[    7.602429] em28xx 1-5.2:1.0: Identified as Hauppauge WinTV-dualHD DVB (card=99)
[    7.604815] tveeprom: Hauppauge model 204109, rev B3I6, serial# 13962917
[    7.604819] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    7.604821] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[    7.604822] tveeprom: audio processor is None (idx 0)
[    7.604823] tveeprom: has no radio, has IR receiver, has no IR transmitter
[    7.604829] em28xx 1-5.2:1.0: dvb set to isoc mode.
[    7.604899] em28xx 1-5.2:1.0: chip ID is em28174
[    8.810920] em28xx 1-5.2:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0x2be4c2c8
[    8.810923] em28xx 1-5.2:1.0: EEPROM info:
[    8.810925] em28xx 1-5.2:1.0: 	microcode start address = 0x0004, boot configuration = 0x01
[    8.817318] em28xx 1-5.2:1.0: 	AC97 audio (5 sample rates)
[    8.817322] em28xx 1-5.2:1.0: 	500mA max power
[    8.817326] em28xx 1-5.2:1.0: 	Table at offset 0x27, strings=0x0e6a, 0x1888, 0x087e
[    8.817541] em28xx 1-5.2:1.0: Identified as Hauppauge WinTV-dualHD DVB (card=99)
[    8.819762] tveeprom: Hauppauge model 204109, rev B3I6, serial# 13962917
[    8.819765] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    8.819767] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[    8.819769] tveeprom: audio processor is None (idx 0)
[    8.819770] tveeprom: has no radio, has IR receiver, has no IR transmitter
[    8.819777] em28xx 1-5.2:1.0: dvb ts2 set to isoc mode.
[    9.018610] usbcore: registered new interface driver em28xx
[    9.064255] em28xx 1-5.2:1.0: Binding DVB extension
[    9.083848] i2c i2c-11: Added multiplexed i2c bus 14
[    9.083853] si2168 11-0064: Silicon Labs Si2168-B40 successfully identified
[    9.083855] si2168 11-0064: firmware version: B 4.0.2
[    9.094255] si2157_probe: probing si2157 tuner fe=ffff8d509a2e7828 cfg=ffffa3d380e838d8 addr=0X60
[    9.094258] si2157_setup_configuration: client=ffff8d508bddd000
[    9.094260] si2157_setup_configuration: dev=          (null)
[    9.094268] si2157_attach (14-0060)
[    9.094270] si2157 14-0060: creating new instance
[    9.094271] si2157_attach: instance=1
[    9.094272] si2157_attach(): new instance for tuner @0x60
[    9.094273] si2157_setup_configuration: client=ffff8d508bddd000
[    9.094274] si2157_setup_configuration: dev=ffff8d508bdddc00
[    9.094276] si2157_setup_configuration(0x60): dvb driver submitted configuration; port=1 invert=0
[    9.094753] cfg->mdev=ffff8d50a9aaf800
[    9.094762] si2157 14-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    9.094791] dvbdev: DVB: registering new adapter (1-5.2:1.0)
[    9.094796] em28xx 1-5.2:1.0: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[    9.094801] dvbdev: dvb_create_media_entity: media entity 'Silicon Labs Si2168' registered.
[    9.095325] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[    9.097041] em28xx 1-5.2:1.0: DVB extension successfully initialized
[    9.097047] em28xx 1-5.2:1.0: Binding DVB extension
[    9.104587] i2c i2c-13: Added multiplexed i2c bus 15
[    9.104592] si2168 13-0067: Silicon Labs Si2168-B40 successfully identified
[    9.104594] si2168 13-0067: firmware version: B 4.0.2
[    9.106869] si2157_probe: probing si2157 tuner fe=ffff8d5092136828 cfg=ffffa3d380e838d8 addr=0X63
[    9.106871] si2157_setup_configuration: client=ffff8d508a42c400
[    9.106873] si2157_setup_configuration: dev=          (null)
[    9.106883] si2157_attach (15-0063)
[    9.106885] si2157 15-0063: creating new instance
[    9.106886] si2157_attach: instance=1
[    9.106887] si2157_attach(): new instance for tuner @0x63
[    9.106888] si2157_setup_configuration: client=ffff8d508a42c400
[    9.106889] si2157_setup_configuration: dev=ffff8d508a42cc00
[    9.106891] si2157_setup_configuration(0x63): dvb driver submitted configuration; port=1 invert=0
[    9.107461] cfg->mdev=ffff8d5090c0a800
[    9.107474] si2157 15-0063: Silicon Labs Si2147/2148/2157/2158 successfully attached
[... 34 lines cut ...]
[  539.053459] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  539.053467] CR2: ffffa3d380e838d3 CR3: 00000001f2e08000 CR4: 00000000001006e0
[  539.053475] Call Trace:
[  539.053513]  dvb_frontend_swzigzag_autotune+0x16e/0x2f0 [dvb_core]
[  539.053552]  dvb_frontend_swzigzag+0x2fa/0x3a0 [dvb_core]
[  539.053570]  ? __enqueue_entity+0x5c/0x60
[  539.053583]  ? put_prev_entity+0x2f/0x5e0
[  539.053596]  ? pick_next_task_fair+0x155/0x590
[  539.053609]  ? __switch_to+0x1f3/0x4e0
[  539.053622]  ? lock_timer_base+0x75/0x90
[  539.053634]  ? try_to_del_timer_sync+0x53/0x80
[  539.053646]  ? del_timer_sync+0x39/0x40
[  539.053660]  ? schedule_timeout+0x18a/0x350
[  539.053676]  ? si2157_init+0x30e/0x640 [si2157]
[... cut ...]
[  539.053772]  ? dtv_set_frontend+0x410/0x410 [dvb_core]
[  539.053785]  ? kthread_create_on_node+0x70/0x70
[  539.053796]  ret_from_fork+0x25/0x30
[  539.053806] Code: 61 fd ff ff 8b 05 c5 19 00 00 89 45 a8 0f b7 05 bf 19 00 00 66 89 45 ac 0f b6 45 a4 41 09 c6 3c f0 44 88 75 ac 0f 84 1e 02 00 00 <41> 80 7d 13 00 74 04 80 4d ad 01 48 8d 75 a8 49 bd 06 00 00 00 
[  539.053970] RIP: si2168_set_frontend+0x341/0x5b0 [si2168] RSP: ffffa3d380d2fa00
[  539.053977] CR2: ffffa3d380e838d3
[  539.053988] ---[ end trace 505f4534db9a4a36 ]---

comio, 2017-11-29:
I removed the 0024 (rollback of vmalloc usage)... now it seems more stable.

ciao

luigi

b-rad-NDi, 2017-11-29:
Apologies, been super busy with other responsibilities. Interesting that this patch appears to break mainline, when Ubuntu is broken with out it. I should move this to only apply to Ubuntu builds.

I am still going to investigate this issue though, it seems a strange place to oops.

I've been massaging all the patches lately to pass checkpatch, pretty much done now so will begin sending them to the mailing list.

b-rad-NDi, 2017-12-16:
@comio : I have removed the vmalloc revert. Have you gotten this particular oops again since this report? For me on 4.13 with the current patch set DualHD has been quite stable.

comio, 2017-12-17:
Hi @b-rad-NDi ,

I rebased your 4.13 patches on the 4.15.6 (I like dangerous life). For now all seems to work fine. I everyday record a on-the-air channel and I still haven't kernel crash.
I hope to see your effort on mainline kernel.

ciao

luigi

b-rad-NDi, 2017-12-17:
Still awaiting lineage confirmation on a few patches for sign off, then they will start getting submitted in groups.

closed 2017-12-17 by b-rad-NDi