b-rad-NDi/Ubuntu-media-tree-kernel-builder #30: WinTV Starburst 2 Frontend not supported

netbus76, 2017-12-07:
Hi,

I recently purchased a WinTV Starburst 2 to run it on Ubuntu 16.04.

`
uname -a

4.4.0-92201708161632-generic #0+mediatree+hauppauge-Ubuntu SMP Thu Aug 17 00:28:00 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
`

The problem is, that after installing the packages to make the card available, it's recognized, but not the frontend:
`
lspci -vnn

03:00.0 Multimedia video controller [0400]: Conexant Systems, Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb [14f1:8880] (rev 04)
	Subsystem: Hauppauge computer works Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb [0070:f02a]
	Flags: bus master, fast devsel, latency 0, IRQ 16
	Memory at fda00000 (64-bit, non-prefetchable) [size=2M]
	Capabilities: <access denied>
	Kernel driver in use: cx23885
	Kernel modules: cx23885
`

`
dmesg

[    3.130335] cx23885: cx23885 driver version 0.0.4 loaded
[    3.130545] cx23885: CORE cx23885[0]: subsystem: 0070:f02a, board: Hauppauge WinTV-Starburst2 [card=59,autodetected]
[    3.501615] tveeprom: Hauppauge model 150300, rev C2I9, serial# 4035992633
[    3.501616] tveeprom: MAC address is 00:0d:fe:90:5c:39
[    3.501617] tveeprom: tuner model is unknown (idx 189, type 4)
[    3.501618] tveeprom: TV standards UNKNOWN (eeprom 0x01)
[    3.501619] tveeprom: audio processor is CX23888 (idx 40)
[    3.501620] tveeprom: decoder processor is CX23888 (idx 34)
[    3.501621] tveeprom: has no radio, has IR receiver, has no IR transmitter
[    3.501622] cx23885: cx23885[0]: warning: unknown hauppauge model #150300
[    3.501622] cx23885: cx23885[0]: hauppauge eeprom: model=150300
[    3.501624] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    3.501637] cx23885: cx23885[0]: cx23885 based dvb card
[    3.501639] cx23885: cx23885[0]: The frontend of your DVB/ATSC card  isn't supported yet
[    3.501639] cx23885: cx23885[0]: frontend initialization failed
[    3.501640] cx23885: cx23885_dvb_register() dvb_register failed err = -22
[    3.501641] cx23885: cx23885_dev_setup() Failed to register dvb adapters on VID_B
[    3.501644] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xd0
[    3.501647] cx23885: cx23885[0]/0: found at 0000:03:00.0, rev: 4, irq: 16, latency: 0, mmio: 0xfda00000
`

I saw that in it was not added to "drivers/media/pci/cx23885/cx23885-dvb.c", when support was added.
Is the frontend the same as for the older Starburst card?

b-rad-NDi, 2017-12-07:
I will investigate. I thought it used the same frontend as old starburst. I will look up the specs and fix it up.

b-rad-NDi, 2017-12-07:
I've fixed this in the 4.13.0 patches.
c57c41dd97071e18470110705f3599fefeb4a1e3
&
1f31336ebf3b5ede261f5eb816005102ee2432fb
I will try and regenerate the 4.4 patches soon, but you might be able to just replace the two patches above in 4.4 with the 4.13 versions and rebuild.

netbus76, 2017-12-09:
Thank you for the fast reaction, I will try to do this!

netbus76, 2017-12-10:
I tried to use the shell script from the repo, first with original patches with option -i then -p, but got errors when applying the patch. Should this work out-of-box?
Sorry for the questions, but i'm not so familiar with git and patches.

b-rad-NDi, 2017-12-11:
Ubuntu released a new rev, which messed up the media tree patch. Backports changed a bit so I have to fix things up. Working on that now, I'll upload some pre built packages for you to test when it's complete.

b-rad-NDi, 2017-12-11:
I have the media tree patch fixed up. I also had to do a small fix the starburst2 dvb patch so it will build. I'll be updating the repo briefly in case you'd like to compile, otherwise you can install these four packages to test out:
https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0

netbus76, 2017-12-12:
No luck with installing the deb files:
```
htpc@ro-htpc:~/Downloads$ sudo apt install ./linux-headers-4.4.0-103201712111959_4.4.0-103201712111959.0+mediatree+hauppauge_all.deb 
[sudo] password for htpc: 
Reading package lists... Error!
E: Sub-process Popen returned an error code (2)
E: Encountered a section with no Package: header
E: Problem with MergeList /home/htpc/Downloads/linux-headers-4.4.0-103201712111959_4.4.0-103201712111959.0+mediatree+hauppauge_all.deb
E: The package lists or status file could not be parsed or opened.
htpc@ro-htpc:~/Downloads$ sudo apt install ./linux-image-4.4.0-103201712111959-generic_4.4.0-103201712111959.0+mediatree+hauppauge_amd64.deb 
Reading package lists... Error!
E: Sub-process Popen returned an error code (2)
E: Encountered a section with no Package: header
E: Problem with MergeList /home/htpc/Downloads/linux-image-4.4.0-103201712111959-generic_4.4.0-103201712111959.0+mediatree+hauppauge_amd64.deb
E: The package lists or status file could not be parsed or opened.
htpc@ro-htpc:~/Downloads$ sudo apt install ./linux-image-extra-4.4.0-103201712111959-generic_4.4.0-103201712111959.0+mediatree+hauppauge_amd64.deb 
Reading package lists... Error!
E: Sub-process Popen returned an error code (2)
E: Encountered a section with no Package: header
E: Problem with MergeList /home/htpc/Downloads/linux-image-extra-4.4.0-103201712111959-generic_4.4.0-103201712111959.0+mediatree+hauppauge_amd64.deb
E: The package lists or status file could not be parsed or opened.
htpc@ro-htpc:~/Downloads$ sudo apt install ./linux-headers-4.4.0-103201712111959-generic_4.4.0-103201712111959.0+mediatree+hauppauge_amd64.deb 
Reading package lists... Error!
E: Sub-process Popen returned an error code (2)
E: Encountered a section with no Package: header
E: Problem with MergeList /home/htpc/Downloads/linux-headers-4.4.0-103201712111959-generic_4.4.0-103201712111959.0+mediatree+hauppauge_amd64.deb
E: The package lists or status file could not be parsed or opened.
```

b-rad-NDi, 2017-12-12:
Use:
`sudo dpkg -i`

b-rad-NDi, 2017-12-12:
You will have to install all four in one command, due to dependencies. But first, verify your files to make sure they are valid. Mine are installable locally:

9e337ca4c3a687230d1859d7ab142ca2  linux-headers-4.4.0-103201712111959_4.4.0-103201712111959.0+mediatree+hauppauge_all.deb

42f9704beefca679e0979ba5108ddc0c linux-headers-4.4.0-103201712111959-generic_4.4.0-103201712111959.0+mediatree+hauppauge_amd64.deb

fc5f5c08e9d73a333cedf61549622db7  linux-image-4.4.0-103201712111959-generic_4.4.0-103201712111959.0+mediatree+hauppauge_amd64.deb

fbc5f114eae16925a5abdff37cb50d6f  linux-image-extra-4.4.0-103201712111959-generic_4.4.0-103201712111959.0+mediatree+hauppauge_amd64.deb

netbus76, 2017-12-12:
Great, it worked, now I'm only missing the FW, it tried to load FW dvb-demod-m88rs6000.fw
Try to get it, and answer as soon as i know more.

Thank you!

netbus76, 2017-12-12:
Got it! Starburst is now supported :+1:

b-rad-NDi, 2017-12-12:
yay :)

closed 2017-12-12 by b-rad-NDi