b-rad-NDi/Ubuntu-media-tree-kernel-builder #31: No Channels found on Ubuntu 16.04.3

ocha, 2017-12-15:
Hi.
I installed kernel from PPA (4.4-apt/ppa and 4.13 - using deb files from ppa).
Device is detected in dmesg and using w_scan, however there are 0 channels found. If i connect it to win10 computer, channels are found, but on ubuntu xenial there are no channels found. 

Have you seen similar issue? below is output of dmesg and w_scan:

[   12.016556] em28xx 3-2:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0x46990a4e
[   12.016557] em28xx 3-2:1.0: EEPROM info:
[   12.016558] em28xx 3-2:1.0: 	microcode start address = 0x0004, boot configuration = 0x01
[   12.025585] em28xx 3-2:1.0: 	AC97 audio (5 sample rates)
[   12.025586] em28xx 3-2:1.0: 	500mA max power
[   12.025588] em28xx 3-2:1.0: 	Table at offset 0x27, strings=0x0a72, 0x187c, 0x086a
[   12.025643] em28xx 3-2:1.0: Identified as Hauppauge WinTV-dualHD 01595 ATSC/QAM (card=100)
[   12.026273] tveeprom: Hauppauge model 204201, rev B3I6, serial# 13965808
[   12.026274] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[   12.026275] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[   12.026276] tveeprom: audio processor is None (idx 0)
[   12.026276] tveeprom: has no radio, has IR receiver, has no IR transmitter
[   12.026278] em28xx 3-2:1.0: dvb set to bulk mode.
[   12.027250] em28xx 3-2:1.0: chip ID is em28174
[   13.228772] em28xx 3-2:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0x46990a4e
[   13.228773] em28xx 3-2:1.0: EEPROM info:
[   13.228774] em28xx 3-2:1.0: 	microcode start address = 0x0004, boot configuration = 0x01
[   13.237875] em28xx 3-2:1.0: 	AC97 audio (5 sample rates)
[   13.237876] em28xx 3-2:1.0: 	500mA max power
[   13.237877] em28xx 3-2:1.0: 	Table at offset 0x27, strings=0x0a72, 0x187c, 0x086a
[   13.237924] em28xx 3-2:1.0: Identified as Hauppauge WinTV-dualHD 01595 ATSC/QAM (card=100)
[   13.238598] tveeprom: Hauppauge model 204201, rev B3I6, serial# 13965808
[   13.238599] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[   13.238600] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[   13.238600] tveeprom: audio processor is None (idx 0)
[   13.238601] tveeprom: has no radio, has IR receiver, has no IR transmitter
[   13.238602] em28xx 3-2:1.0: dvb ts2 set to bulk mode.
[   13.437694] usbcore: registered new interface driver em28xx
[   13.442684] WARNING: You are using an experimental version of the media stack.
               	As the driver is backported to an older kernel, it doesn't offer
               	enough quality for its usage in production.
               	Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
               	330dada5957e3ca0c8811b14c45e3ac42c694651 media: dvb_frontend: fix return error code
               	5e0d3c3a2c67c060e65e152558fa3cbdabdafe6c media: MAINTAINERS: add si2165 driver
               	7c4f4d1d7e5e3a5efc0a75410c7dad1c6c0be838 media: cx23885: Use semicolon after assignment instead of comma
[   13.446786] em28xx 3-2:1.0: Binding DVB extension
[   13.466372] i2c i2c-1: Added multiplexed i2c bus 4
[   13.468056] si2157_probe: probing si2157 tuner fe=ffff9377e22aa810 cfg=ffffa01082a338e8 addr=0X60
[   13.468057] si2157_setup_configuration: client=ffff9377e2810c00
[   13.468057] si2157_setup_configuration: dev=          (null)
[   13.468059] si2157_attach (4-0060)
[   13.468059] si2157 4-0060: creating new instance
[   13.468060] si2157_attach: instance=1
[   13.468060] si2157_attach(): new instance for tuner @0x60
[   13.468060] si2157_setup_configuration: client=ffff9377e2810c00
[   13.468061] si2157_setup_configuration: dev=ffff9377e22cf600
[   13.468061] si2157_setup_configuration(0x60): dvb driver submitted configuration; port=1 invert=1
[   13.472874] si2157 4-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[   13.472879] dvbdev: DVB: registering new adapter (3-2:1.0)
[   13.472880] em28xx 3-2:1.0: DVB: registering adapter 0 frontend 0 (LG Electronics LGDT3306A VSB/QAM Frontend)...
[   13.473219] em28xx 3-2:1.0: DVB extension successfully initialized
[   13.473221] em28xx 3-2:1.0: Binding DVB extension
[   13.485036] i2c i2c-3: Added multiplexed i2c bus 5
[   13.485504] si2157_probe: probing si2157 tuner fe=ffff9377e22ab810 cfg=ffffa01082a338e8 addr=0X62
[   13.485504] si2157_setup_configuration: client=ffff9377e272a400
[   13.485505] si2157_setup_configuration: dev=          (null)
[   13.485507] si2157_attach (5-0062)
[   13.485507] si2157 5-0062: creating new instance
[   13.485508] si2157_attach: instance=1
[   13.485508] si2157_attach(): new instance for tuner @0x62
[   13.485508] si2157_setup_configuration: client=ffff9377e272a400
[   13.485509] si2157_setup_configuration: dev=ffff9377e22ce600
[   13.485509] si2157_setup_configuration(0x62): dvb driver submitted configuration; port=1 invert=1
[   13.489196] si2157 5-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[   13.489201] dvbdev: DVB: registering new adapter (3-2:1.0)
[   13.489202] em28xx 3-2:1.0: DVB: registering adapter 1 frontend 0 (LG Electronics LGDT3306A VSB/QAM Frontend)...
[   13.489877] em28xx 3-2:1.0: DVB extension successfully initialized
[   13.489878] em28xx: Registered (Em28xx dvb Extension) extension
[   13.492266] WARNING: You are using an experimental version of the media stack.
               	As the driver is backported to an older kernel, it doesn't offer
               	enough quality for its usage in production.
               	Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
               	330dada5957e3ca0c8811b14c45e3ac42c694651 media: dvb_frontend: fix return error code
               	5e0d3c3a2c67c060e65e152558fa3cbdabdafe6c media: MAINTAINERS: add si2165 driver
               	7c4f4d1d7e5e3a5efc0a75410c7dad1c6c0be838 media: cx23885: Use semicolon after assignment instead of comma
[   13.496844] em28xx 3-2:1.0: Registering input extension
[   13.524014] Registered IR keymap rc-hauppauge
[   13.524960] rc rc0: 3-2:1.0 IR as /devices/pci0000:00/0000:00:16.0/0000:0b:00.0/usb3/3-2/3-2:1.0/rc/rc0
[   13.525031] input: 3-2:1.0 IR as /devices/pci0000:00/0000:00:16.0/0000:0b:00.0/usb3/3-2/3-2:1.0/rc/rc0/input6
[   13.525443] em28xx 3-2:1.0: Input extension successfully initialized
[   13.525444] em28xx 3-2:1.0: Remote control support is not available for this card.
[   13.525445] em28xx: Registered (Em28xx Input Extension) extension

w_scan -fa -c US -Z
w_scan version 20141122 (compiled for DVB API 5.10)
using settings for UNITED STATES
ATSC
VSB US/CA, DVB-T TW
scan type TERRCABLE_ATSC, channellist 1
output format w_scan XML tuning data
output charset 'ISO-8859-1'
Info: using DVB adapter auto detection.
	/dev/dvb/adapter0/frontend0 -> TERRCABLE_ATSC "LG Electronics LGDT3306A VSB/QAM Frontend": good :-)
	/dev/dvb/adapter1/frontend0 -> TERRCABLE_ATSC "LG Electronics LGDT3306A VSB/QAM Frontend": good :-)
Using TERRCABLE_ATSC frontend (adapter /dev/dvb/adapter0/frontend0)
-_-_-_-_ Getting frontend capabilities-_-_-_-_
Using DVB API 5.10
frontend 'LG Electronics LGDT3306A VSB/QAM Frontend' supports
INVERSION_AUTO
8VSB
QAM_64
QAM_256
FREQ (54.00MHz ... 858.00MHz)
-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_
57000: 8VSB(time: 00:00.640)
63000: 8VSB(time: 00:03.876)
.... ... ....
ERROR: Sorry - i couldn't get any working frequency/transponder
 Nothing to scan!!

b-rad-NDi, 2017-12-15:
Everything looks good in dmesg. I will have this tested in the morning. I've tested the lgdt3306a/si2157 combo on the current packages on received QAM signal. I will have your model tested in the morning.

ocha, 2017-12-15:
One thing I noticed was that Green LED is always ON on the dongle and there is no Si2168 demux like some other people have in their dmesg.

I also see some errors in dmesg sometimes when I scan for a channel:
[ 5541.971350] si2157 4-0060: found a 'Silicon Labs Si2157-A30'
[ 5542.024966] si2157 4-0060: firmware version: 3.0.5
[ 5586.608540] em28xx 3-2:1.0: failed to clear USB bulk endpoint stall/halt condition (error=-75)
[ 5586.608578] em28xx 3-2:1.0: write to i2c device at 0xc0 failed with unknown error (status=133)

b-rad-NDi, 2017-12-15:
Your demod is lgdt3306a, si2168 is for dvb models. It is a bit strange though there is not a single message from that driver though. w_scan sees it, even if it can't use it.

ocha, 2017-12-15:
yea. I have been through so many kernels, even tried compiling mediatree locally but no fix. Closest I've gotten to is this - detected device but cant scan channels

ocha, 2017-12-15:
I can set up a vanilla ubuntu vm if you want to poke around it. just give me a link to your pub key and I will add it to the server and give you ip.

b-rad-NDi, 2017-12-15:
I have a tester in the office setting up to repro this. If it works for him I might take you up on a closer inspection.
As for the lack of lgdt3306a messages, it does appear that both probe and attach are silent functions, so that is not necessarily bad. I will let you know what my tester finds.

b-rad-NDi, 2017-12-15:
Initial tests came back fine for an ISOC model DualHD using my ppa 4.4 packages. ATSC scanning found channels and unplug/replug tests passed. They are digging up a bulk model to test out now. They will test ATSC/QAM as well as watching.

b-rad-NDi, 2017-12-15:
Your w_scan command is close to what we typically use, but you're missing the -A1 parameter.

Scanning ATSC VSB (United States, Over-the-Air)

`$ w_scan -fa -A1 -c US -X`

Scanning ATSC QAM annex B (United States, digital cable TV)

`$ w_scan -fa -A2 -c US -X`

b-rad-NDi, 2017-12-15:
However even without that parameter my w_scan is finding ATSC channels in the first iteration. I'm guessing it'll do QAM scan next.

ocha, 2017-12-15:
I have done with both -A1 and -A2 but it didnt change anything. I also used some more advanced params on w_scan, still no success

b-rad-NDi, 2017-12-15:
My tester has OK results with US DualHD, both ISOC and bulk models, scanning and viewing ATSC and QAM. So it would appear the packages are fine, unsure what could be your issue here.

It will be a little noisy, but can you enable debug on the three drivers and then post the dmesg produced on device connection?

Put this into a conf file in /etc/modprobe.d/
`options lgdt3306a debug=2
options si2157 debug=2
options em28xx debug=2 `
The logs produced by a scan might be too much to post here, I guess if you could just attach it as a file.

b-rad-NDi, 2017-12-15:
btw...you don't happen to be running Ubuntu under vmware, do you?

ocha, 2017-12-15:
Yes, I am running ubuntu under vmware. Would that be a problem? I saw people having stuttering when running under vmware, but in my case I cannot even get a channel scanned...

b-rad-NDi, 2017-12-15:
All bets are off if vmware is involved. We do not support VM usage for the exact reasons you are experiencing. I use kvm/qemu and have zero issues with stuttering or hardware performance when using any of our usb devices. What you experience is specific to vmware.

Please file a ticket with vmware about this. I spoke with a couple of their people at a conference recently and they told me to "hire our big guns and we'll figure out why it doesn't work". When I laughed and said why should we pay when our hardware works everywhere else, including in kvm/qemu, and they balked at it being their issue. All they could tell me was to tell any affected users to file tickets saying that passthru of lots of usb tuners does not work and maybe one of their open source people will get interested.

ocha, 2017-12-15:
hmm.  I will put the card on physical hardware and see if that will fix it. I thought vmware issue was limited to actual performance (stuttering) but not the functionality (not being able to get any channels). fingers crossed I get it working on physical hardware

b-rad-NDi, 2017-12-15:
Do let me know if everything works satisfactory native.

b-rad-NDi, 2017-12-15:
Hold the phone, I misunderstood my tester. I might have indeed botched something on the bulk models. He is seeing similar issue as you, even natively. I am inspecting related changes now.

b-rad-NDi, 2017-12-15:
806bbf444
If you look at
`patches/mainline-extra/4.13.0/0002-em28xx-bulk-transfer-implementation-fix.patch`
You'll see I replaced a hard coded value with the defined constant it claimed to be. The problem is 0xef is actually one less than the constant. Could this be the problem? This should be the sole difference between an ISOC DualHD and bulk DualHD.
I'm recompiling now, I'll upload packages to dropbox for you to test when they're done.

ocha, 2017-12-15:
Great. thank you. I should be able to test it in an hour or so.

b-rad-NDi, 2017-12-15:
https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0

b-rad-NDi, 2017-12-15:
Reported to be fixed with this build. Let me know if your findings concur.
(still zero guarantees about performance under vmware though :-p)

ocha, 2017-12-15:
Same problem. and save eeprom address 0xfc. 
below is kernel version loaded and dmesg output

Linux ubuntu 4.4.0-103201712151353-generic #0+mediatree+hauppauge SMP Fri Dec 15 13:55:43 CST 2017 x86_64 x86_64 x86_64 GNU/Linux

[   11.655210] em28xx 1-2:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0x46990a4e
[   11.655212] em28xx 1-2:1.0: EEPROM info:
[   11.655213] em28xx 1-2:1.0: 	microcode start address = 0x0004, boot configuration = 0x01
[   11.663883] em28xx 1-2:1.0: 	AC97 audio (5 sample rates)
[   11.663884] em28xx 1-2:1.0: 	500mA max power
[   11.663886] em28xx 1-2:1.0: 	Table at offset 0x27, strings=0x0a72, 0x187c, 0x086a
[   11.663958] em28xx 1-2:1.0: Identified as Hauppauge WinTV-dualHD 01595 ATSC/QAM (card=100)
[   11.665074] tveeprom: Hauppauge model 204201, rev B3I6, serial# 13965808
[   11.665075] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[   11.665076] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[   11.665077] tveeprom: audio processor is None (idx 0)
[   11.665078] tveeprom: has no radio, has IR receiver, has no IR transmitter
[   11.665079] em28xx 1-2:1.0: dvb set to bulk mode.
[   11.665830] em28xx 1-2:1.0: chip ID is em28174
[   12.823196] em28xx 1-2:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0x46990a4e
[   12.823198] em28xx 1-2:1.0: EEPROM info:
[   12.823200] em28xx 1-2:1.0: 	microcode start address = 0x0004, boot configuration = 0x01
[   12.832042] em28xx 1-2:1.0: 	AC97 audio (5 sample rates)
[   12.832044] em28xx 1-2:1.0: 	500mA max power
[   12.832045] em28xx 1-2:1.0: 	Table at offset 0x27, strings=0x0a72, 0x187c, 0x086a
[   12.832097] em28xx 1-2:1.0: Identified as Hauppauge WinTV-dualHD 01595 ATSC/QAM (card=100)
[   12.832835] tveeprom: Hauppauge model 204201, rev B3I6, serial# 13965808
[   12.832836] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[   12.832837] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[   12.832838] tveeprom: audio processor is None (idx 0)
[   12.832839] tveeprom: has no radio, has IR receiver, has no IR transmitter
[   12.832840] em28xx 1-2:1.0: dvb ts2 set to bulk mode.
[   13.031983] usbcore: registered new interface driver em28xx
[   13.036761] WARNING: You are using an experimental version of the media stack.
               	As the driver is backported to an older kernel, it doesn't offer
               	enough quality for its usage in production.
               	Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
               	330dada5957e3ca0c8811b14c45e3ac42c694651 media: dvb_frontend: fix return error code
               	5e0d3c3a2c67c060e65e152558fa3cbdabdafe6c media: MAINTAINERS: add si2165 driver
               	7c4f4d1d7e5e3a5efc0a75410c7dad1c6c0be838 media: cx23885: Use semicolon after assignment instead of comma
[   13.040922] em28xx 1-2:1.0: Binding DVB extension
[   13.055388] i2c i2c-1: Added multiplexed i2c bus 4
[   13.057096] si2157_probe: probing si2157 tuner fe=ffff880088285810 cfg=ffff88008915fa98 addr=0X60
[   13.057097] si2157_setup_configuration: client=ffff88008849f800
[   13.057098] si2157_setup_configuration: dev=          (null)
[   13.057100] si2157_attach (4-0060)
[   13.057101] si2157 4-0060: creating new instance
[   13.057101] si2157_attach: instance=1
[   13.057102] si2157_attach(): new instance for tuner @0x60
[   13.057102] si2157_setup_configuration: client=ffff88008849f800
[   13.057103] si2157_setup_configuration: dev=ffff8800875e7000
[   13.057104] si2157_setup_configuration(0x60): dvb driver submitted configuration; port=1 invert=1
[   13.060914] cfg->mdev=ffff880295796c00
[   13.060920] si2157 4-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[   13.060924] dvbdev: DVB: registering new adapter (1-2:1.0)
[   13.060926] em28xx 1-2:1.0: DVB: registering adapter 0 frontend 0 (LG Electronics LGDT3306A VSB/QAM Frontend)...
[   13.060928] dvbdev: dvb_create_media_entity: media entity 'LG Electronics LGDT3306A VSB/QAM Frontend' registered.
[   13.061356] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[   13.062367] em28xx 1-2:1.0: DVB extension successfully initialized
[   13.062369] em28xx 1-2:1.0: Binding DVB extension
[   13.074209] i2c i2c-3: Added multiplexed i2c bus 5
[   13.074717] si2157_probe: probing si2157 tuner fe=ffff880088285010 cfg=ffff88008915fa98 addr=0X62
[   13.074718] si2157_setup_configuration: client=ffff8802988a6000
[   13.074719] si2157_setup_configuration: dev=          (null)
[   13.074721] si2157_attach (5-0062)
[   13.074722] si2157 5-0062: creating new instance
[   13.074722] si2157_attach: instance=1
[   13.074723] si2157_attach(): new instance for tuner @0x62
[   13.074723] si2157_setup_configuration: client=ffff8802988a6000
[... 9 lines cut ...]
[   13.079941] em28xx: Registered (Em28xx dvb Extension) extension
[   13.082120] WARNING: You are using an experimental version of the media stack.
               	As the driver is backported to an older kernel, it doesn't offer
               	enough quality for its usage in production.
               	Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
               	330dada5957e3ca0c8811b14c45e3ac42c694651 media: dvb_frontend: fix return error code
               	5e0d3c3a2c67c060e65e152558fa3cbdabdafe6c media: MAINTAINERS: add si2165 driver
[... cut ...]
[   85.550441] si2157 4-0060: firmware version: 3.0.5
[   85.714582] si2157 5-0062: found a 'Silicon Labs Si2157-A30'
[   85.752657] si2157 5-0062: firmware version: 3.0.5
[   85.955170] si2157 4-0060: found a 'Silicon Labs Si2157-A30'
[   86.007258] si2157 4-0060: firmware version: 3.0.5
[  262.583928] si2157 4-0060: found a 'Silicon Labs Si2157-A30'
[  262.626153] si2157 4-0060: firmware version: 3.0.5
[  295.581107] em28xx 1-2:1.0: failed to clear USB bulk endpoint stall/halt condition (error=-75)

b-rad-NDi, 2017-12-16:
Is this native or still virtualized? My tester reported scanning and viewing worked in kaffeine with this particular build.

ocha, 2017-12-16:
It is in virtual. I am building physical machine right now to try. that way we can figure out if its vmware or something else

ocha, 2017-12-16:
On real hardware it works. I compared dmesg and it looks like only meaningful difference is errors after scanning. I guess it wont work with vmware. I was really hopping to have stuttering issues instead of this. :(
[14895.259239] lgdt3306a: error (addr 59 reg 0080 error (ret == -110)
[14900.450059] lgdt3306a: error (addr 59 reg 217a error (ret == -110)
[14931.247521] lgdt3306a: error (addr 59 reg 0080 error (ret == -110)
[14936.377248] lgdt3306a: error (addr 59 reg 0080 error (ret == -110)
[15023.613104] lgdt3306a: error (addr 59 reg 0080 error (ret == -110)
[15090.320466] lgdt3306a: error (addr 59 reg 2179 error (ret == -110)

ocha, 2017-12-16:
nvm, it works in vmware too. Thank you for the help.

closed 2017-12-16 by ocha

b-rad-NDi, 2017-12-16:
What was the fix for vmware?

ocha, 2017-12-16:
I reconnected USB and then restarted. That helped. Its still flaky but thats vmware issue. Im going to add pcie usb3 card and do a passthrough. That way there should be no problem.

b-rad-NDi, 2017-12-16:
Ok, so you experience the stuttering still. Good luck on the pcie pass through, we've had reports of that working fine.

b-rad-NDi, 2017-12-17:
@ocha : In the builds I'm pushing to the ppa tonight I've tripled the bulk transfer size. It's still less than original, so within the current value and mainlines. There's a small chance this helps out streaming in vmware. Testing builds are in dropbox linked to above, ppa will be updated in a few hours.

ocha, 2017-12-17:
@b-rad-NDi this latest kernel is working amazing. No buffering and plex channel scan got much more channels (60 vs 4). 
I am running two parallel streams on an esxi hypervisor with USB2 vmware passthrough (not PCIe passthrough) and other vms are taking around 60% cpu usage.

b-rad-NDi, 2017-12-17:
:-O No more stuttering, continuation errors, and/or assorted kernel messages with two simultaneous vmware vm's both doing tv stuff? That would be pretty phenomenal if optimizing transfer alignment had such an impact.

ocha, 2017-12-17:
pretty much. so it is one vmware vm running plex and doing two streams. first 3~5 seconds there is some buffering on plex but after that it is flawless. I had two different channels up in parallel for about 10 minutes and worked well. lgdt3306a: errors in dmesg are much less. esxi host is running other vms doing other things and total cpu load on host is around 80%, even under those conditions it streamed perfectly. 

one thing to note, plex does MPEG2 to H264 transcoding and might be doing couple seconds buffering for transcoding which could hide the issue of tearing and stuttering...

b-rad-NDi, 2017-12-22:
@ocha : I have new packages for you to test out. It is possible you noticed a disconnect GPF/oops on unplugging DualHD. That is fixed. Also I just realized that my so called 'vmware fix' had an overflow issue causing a compiler warning. I reduced this value to the maximum value it could take, but left the bulk multiplier the same. If you have a chance to test these packages out, please let me know if vmware still works.
https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0

ocha, 2017-12-22:
@b-rad-NDi This build is not working well. It is impossible to watch any channel. Looks like as if reception is bad or not enough bandwidth: jittery (1~2 frames a second) and each frame is not complete. 

I reverted back to previous kernel and everything is working again.

b-rad-NDi, 2017-12-22:
Quite interesting that changes affect vmware so dramatically. Are you running the ppa kernel?

ocha, 2017-12-22:
No. I am running previous kernel you had on dropbox (20171216 if i'm not
mistaken)

b-rad-NDi, 2017-12-22:
I believe those two versions are identical.

ocha, 2017-12-22:
by two version I assume you mean PPA and 20171216.

b-rad-NDi, 2017-12-22:
Correct, I believe after your confirmation of vmware working I pushed new builds to the ppa.

b-rad-NDi, 2017-12-28:
@ocha : I'm still experimenting with suitable multipliers. If you're interested in testing out the latest builds they are in the same dropbox previously linked.

ocha, 2017-12-28:
@b-rad-NDi Just tried latest kernel 4.4 from the dropbox. around 1 frame every 3~4 seconds (both video and audio). Reverted back to previous kernel and works good (3~5 second buffering when starting up but no buffering/stuttering/pixelation afterwards).

b-rad-NDi, 2017-12-28:
Jeez, vmware is a fickle mistress.

ocha, 2017-12-28:
Yes. Im not sure exactly what is going on. I am using USB2 physical port and USB2 virtual port. I thought this way it would be least amount of "virtualization" happening. I would guess, bigger the buffer/batch size of data, better it will work.

b-rad-NDi, 2017-12-28:
Yeh, the problem is finding the exact level which makes everyone happy. The strange part is I believe the version you're running has an overflow, causing wraparound on the multiplier...so it's strange that it works.
Right now I have the em28xx packet multiplier set to 128, which means it will transmit 188*128. 
The usb bulk packets though were set to a multiple of that. I am going to align the values now and rebuild. Perhaps if both multiples are the exact same vmware will become happier.
Unfortunately, according to the documentation from Empia the packet multiplier can only be set to a single byte value, which means 255 maximum. 255*188 however is not a multiple of 512 to make usb spec happy, whereas 256*188 is.
I will be rebuilding with the same timestamp, the prior packages have been deleted. New ones should be up in 10min.

ocha, 2017-12-28:
sounds good. I can test them in a bit and see if it helps. BTW, I think my device is operating in bulk mode.

b-rad-NDi, 2017-12-28:
There are new builds there with both values set to 24064, which is the lowest common multiple of 188 and 512.
I am also going to make you a special build now if you'd like to experiment. I am going to make both multipliers module options. Then you can explicitly set them to find the 'sweet spot'.

I am a little curious about what is getting written to the em28xx packet multiplier register for you now. You can enable register debug in the em28xx driver to find out. It will be exceptionally noisy dumping all the regs though.

ocha, 2017-12-28:
Im not sure how to enable debug for the registers. if you have a link to a guide I can try and get you the output. Trying new kernel right now.

b-rad-NDi, 2017-12-28:
Add this to /etc/modprobe.d/em28xx.conf:
``options em28xx-i2c i2c_debug=2``
You should now see a TON of spew in dmesg on next boot.
Boot like that once, connect a dualHD, then just try to view a channel briefly. After that you can upload the log here and I should be able to pull the multiplier out.

b-rad-NDi, 2017-12-28:
edited, looks like there is not em28xx-i2c.

ocha, 2017-12-28:
Im not sure where log is located - i couldnt find it. either way it broke something and none of the channels would work. after removing debug flag, it started working again. this latest kernel updates screen more often ` fram every 1-2 seconds but it is very pixelated - as if only some part of the frame had data.

b-rad-NDi, 2017-12-28:
The log would be in dmesg. It's not very surprising things didn't work, since that option will cause so much output combined with vmware already having issue.
If you try again just do this:
``dmesg > /tmp/dmesg.em28xx.i2c.log``
Very unfortunate that aligning the buffers was not successful this time.

ocha, 2017-12-28:
I tried again. it was not that noisy. i was able to tune one channel. here is the link to dmesg after 5 seconds of tuning to a channel. http://termbin.com/4ze4

ocha, 2017-12-28:
And this is the old kernel that works fine with debug flag on: http://termbin.com/mz83

b-rad-NDi, 2017-12-28:
Unfortunately that flag doesn't appear helpful, it only logs the eeprom dump. Oh well.
I went back and re-re-reviewed the slight documentation I have and turns out I again botched the multiplier. For the multiplier 0 is 1x188, so the multiple needs to be one less than is desired. Because of this I am able to max out the multiplier. So now, by default I have the bulk multiplier set to transmit 188x256 packets at a time. Both usb urb alloc and em28xx are configured the same. This could have been why the recent builds were fubar.

I have also added two kernel parameters for you to experiment with the multipliers. 
In /etc/modprobe.d/em28xx.conf put this:
``options em28xx-dvb bulk_multiplier=94``
``options em28xx pkt_mutiplier=256``

where:
``bulk_multiplier * 512 should equal pkt_multiplier * 188``

94 and 256 are the default values in the driver, this should be as optimal as possible since the em28xx bulk multiplier is maxed out. You will see both multipliers printed out in dmesg when they are configured.

Packages with these new, proper, defaults and the module options are in the dropbox. Fingers crossed these defaults are good for you!

ocha, 2017-12-28:
This one looks like it is working... Here is dmesg for it http://termbin.com/tria
I used values that you provided... Do you have values that was used with the previous kernel that worked for me?

b-rad-NDi, 2017-12-28:
Nice. Believe it or not, before the values should have been completely in err. In your 'old good build' I believe I had:
``options em28xx-dvb bulk_multiplier=282``
``options em28xx pkt_mutiplier=768``
Now 768 is bigger than 1B, so it gets wrapped. This combo should not have really worked due to the wrapping.

ocha, 2017-12-28:
Interesting that it worked. I wonder if compiler or hardware detected and
"corrected" those values to max possible

b-rad-NDi, 2017-12-28:
One curious combo to try though, would be a bulk multiplier that was a multiple of the packet multiplier:
``options em28xx-dvb bulk_multiplier=188``
``options em28xx pkt_mutiplier=256``
Original is (47 x 2), above is (47 x 4), if that works, then you could try (47 x 8) for bulk multiplier.