b-rad-NDi/Ubuntu-media-tree-kernel-builder #32: Wrong parameter size.

comio, 2017-12-27:
Hi @b-rad-NDi ,

I'm rebuild again my sources (4.14.8 this time) and I noticed a strange warning:

```
In file included from drivers/media/usb/em28xx/em28xx-core.c:25:0:
drivers/media/usb/em28xx/em28xx-core.c: In function ‘em28xx_capture_start’:
drivers/media/usb/em28xx/em28xx.h:194:43: warning: large integer implicitly truncated to unsigned type [-Woverflow]
 #define EM28XX_DVB_BULK_PACKET_MULTIPLIER (384 * 2)
                                           ^
drivers/media/usb/em28xx/em28xx-core.c:646:6: note: in expansion of macro ‘EM28XX_DVB_BULK_PACKET_MULTIPLIER’
      EM28XX_DVB_BULK_PACKET_MULTIPLIER - 1);
      ^
```

This the call that generates the warning:

```
			em28xx_write_reg(dev, (dev->ts == PRIMARY_TS) ?
					EM2874_R5D_TS1_PKT_SIZE :
					EM2874_R5E_TS2_PKT_SIZE,
					EM28XX_DVB_BULK_PACKET_MULTIPLIER - 1);

```

and the last parameter should be a u8 value. Instead the EM28XX_DVB_BULK_PACKET_MULTIPLIER is a 16bit value.

Best regards,

luigi

b-rad-NDi, 2017-12-27:
This was pointed out in your previous issue. The rebuild with that u8 maxed out, and not wrapped, had horrible performance in vmware according to you. Strangely, with the wrap around the multiplier is set to 19...and you said that had fine performance.

This is the multiplier to experiment with. There are two spots the same multiplier is used, but differently. There should probably be two different multipliers.

This constant **EM28XX_DVB_BULK_PACKET_MULTIPLIER** should be set to 47. This configures bulk transfers to be (512 * 47) == 24064B.

Then the register write, which is throwing this warning should be set to 0x80. This configures the em28xx packet multiplier to (188 * 128) == 24064B.

This gives us even packet alignment, for 188B packets at least. Unfortunately the next lcm is 256, so cannot be used.

b-rad-NDi, 2017-12-27:
Ooops, apologies. It was another user I was talking with about this issue.

b-rad-NDi, 2017-12-27:
I've fixed the patch now, and set the multiplier to 128. This is the lowest common multipler of 188 and 512. Please let me know if you find this value non-optimal.

b-rad-NDi, 2017-12-28:
Closing this since the parameter is no longer wrapping.

closed 2017-12-28 by b-rad-NDi