b-rad-NDi/Ubuntu-media-tree-kernel-builder #34: WinTV-Nova HD

CvH, 2018-01-28:
Hi we have a user with and WinTV-Nova HD that is not supported. After searching at the kernel code and your repo I can't find the id either. 

```
New USB device found, idVendor=2040, idProduct=d901
New USB device strings: Mfr=1, Product=2, SerialNumber                                                                                                                                                             =3
Product: Hauppauge
Manufacturer: Max
SerialNumber: 101142
```
Any idea what device (maybe new rev) this is actually ?

b-rad-NDi, 2018-01-28:
I will figure it out and get it supported (if possible) first thing Monday.

CvH, 2018-01-28:
tx for support :)

CvH, 2018-01-29:
got an pic of the box, maybe helpful
[image]

b-rad-NDi, 2018-01-29:
The device has pids 0x9300 and 0x9301 supported via dvb-usb device. I am having someone look up what pid 0xd901 is.

b-rad-NDi, 2018-01-30:
This contains Montage ds3103 demod and Montage ts2020 tuner. Both have upstream drivers. I will add this to the queue of devices to get supported. Thanks for reporting.

CvH, 2018-01-30:
tx, if you have an patch you can ping me if you like

b-rad-NDi, 2018-01-30:
Will do.

CvH, 2018-02-15:
@b-rad-NDi any news on that ?

b-rad-NDi, 2018-02-15:
Turns out this is an ancient device, not made by us, but resold by us. It is possible I got the innards correct. I'll look again at the driver to see if I can easily add a pid and get you a package to test.

CvH, 2018-02-15:
@b-rad-NDi do you have access to such an device?

b-rad-NDi, 2018-02-15:
Negative, one was looked around for in our warehouse and was not found either.

CvH, 2018-02-15:
okay so the usual problem :D, I try it with the user to fix it :)

CvH, 2018-03-08:
tried it with that patch (thought it is a TT-4600 clone)
https://github.com/CvH/LibreELEC.tv/commit/15c488b95800de3dcca1e7b787096e52e58fdbf0

sadly driver crashs https://pastebin.com/cFYdgMbH

as I have not too much knowledge about it I could be plain wrong :)

b-rad-NDi, 2018-03-08:
As soon as I finish the ryzen patch I will look into this.

CvH, 2018-03-08:
btw after realising I had read the number wrong (thought TS2022) I wondering if you are right with 
> Montage ds3103 demod and Montage ts2020 tuner

common is `ds3103 + ts2022` or `M88DS3000/2 + ts2020` so did you made an typo or is that box "uncommon" ?

CvH, 2018-03-08:
got an image
[image]

so it looks like `ds3103 + ts2022` :)

b-rad-NDi, 2018-03-08:
Noice, I was going to tell you to just bust it open and let's find out. Should be trivial then if it's already a supported combo.

b-rad-NDi, 2018-03-08:
Ok. So your patch seems like it could work, as long as the components all match. The one detail that is different though, is the lack of FX2 usb bridge support anywhere. The CY7C63013 is an FX2 compatible bridge. In other dvb-usb drivers you'll see .usb_ctrl = CYPRESS_FX2, but in the dw2102.c it is .usb_ctrl = DEVICE_SPECIFIC

When I load up dw2102.o in gdb and list the line of your NPE I get the second line as the source.
```
        request_module("m88ds3103");
        client = i2c_new_device(&d->i2c_adap, &board_info);
        if (client == NULL || client->dev.driver == NULL)
                return -ENODEV;
```
Can you recompile your kernel with **_#define DEBUG 1_** inside of
- drivers/media/dvb-frontends/m88ds3103.c
- drivers/media/dvb-frontends/ts2020.c

and then the following options inside of a modprobe.d conf file
```
options tuner debug=1
options dvb-core frontend_debug=1
options dw2102 debug=1
```

CvH, 2018-03-08:
Compiled and shipped to test user :) I guess we have an answer in ~12-16h

b-rad-NDi, 2018-03-08:
One problem that I'm not sure if is causing any issue is I don't see support for ds3103**_b_** in the m88ds3103 driver.

CvH, 2018-03-12:
dmesg http://ix.io/W0R
tbh not too much debug for my understanding
used that patch for doing it https://github.com/CvH/LibreELEC.tv/commit/6626aca3305a655fe49dcd622d150dac7f138961

CvH, 2018-03-12:
just got an response at an thread about an TT-4600, the new revision (still same vendor + product id) of that box uses `TS2022 + DS3103B + CY7C68013A` and crash similar to the WinTV

b-rad-NDi, 2018-03-12:
Interesting that another device also causes a crash, a bug must have crept in. I'll review the drivers in the backtrace and see if I can find an obvious culprit.

CvH, 2018-03-14:
urgs,  I think https://git.linuxtv.org/media_tree.git/commit/drivers/media/dvb-frontends/m88ds3103.c?id=b9c97c67fd19262c002d94ced2bfb513083e161e should fix the crash

build an media_tree master build with the patch, lets see what happens there :)