b-rad-NDi/Ubuntu-media-tree-kernel-builder #35: Critical security vulnerabilities (Spectre 1 & 2 and Meltdown) in kernel

WyseNynja, 2018-02-06:
```
$ git clone https://github.com/speed47/spectre-meltdown-checker.git
$ cd ./spectre-meltdown-checker
$ ./spectre-meltdown-checker.sh 
Spectre and Meltdown mitigation detection tool v0.34+

Checking for vulnerabilities on current system
Kernel is Linux 4.4.0-103201712311231-generic #0+mediatree+hauppauge-Ubuntu SMP Sun Dec 31 21:15:04 UTC 2017 x86_64
CPU is Intel(R) Xeon(R) CPU E3-1275 v5 @ 3.60GHz

Hardware check
...

CVE-2017-5753 [bounds check bypass] aka 'Spectre Variant 1'
* Kernel has array_index_mask_nospec:  NO 
* Checking count of LFENCE instructions following a jump in kernel:  NO  (only 8 jump-then-lfence instructions found, should be >= 30 (heuristic))
> STATUS:  VULNERABLE  (Kernel source needs to be patched to mitigate the vulnerability)

CVE-2017-5715 [branch target injection] aka 'Spectre Variant 2'
* Mitigation 1
  * Kernel is compiled with IBRS/IBPB support:  NO 
  * Currently enabled features
    * IBRS enabled for Kernel space:  NO 
    * IBRS enabled for User space:  NO 
    * IBPB enabled:  NO 
* Mitigation 2
  * Kernel compiled with retpoline option:  NO 
  * Kernel compiled with a retpoline-aware compiler:  NO 
  * Retpoline enabled:  NO 
> STATUS:  VULNERABLE  (IBRS hardware + kernel support OR kernel with retpoline are needed to mitigate the vulnerability)

CVE-2017-5754 [rogue data cache load] aka 'Meltdown' aka 'Variant 3'
* Kernel supports Page Table Isolation (PTI):  NO 
* PTI enabled and active:  NO 
* Running as a Xen PV DomU:  NO 
> STATUS:  VULNERABLE  (PTI is needed to mitigate the vulnerability)

A false sense of security is worse than no security at all, see --disclaimer
```

Please rebuild the kernel with an updated version of GCC.

b-rad-NDi, 2018-02-06:
The release is built by Launchpad, but duly noted. I will do my best to push out new builds tomorrow.

b-rad-NDi, 2018-02-08:
Even fully up to date, Xenial fails to pass completely. We shall see how a Launchpad build differs.
```
Spectre and Meltdown mitigation detection tool v0.34+

Checking for vulnerabilities on current system
Kernel is Linux 4.4.0-112201802080420-generic #0+mediatree+hauppauge SMP Thu Feb 8 10:09:03 CST 2018 x86_64
CPU is Intel(R) Core(TM)2 Duo CPU     T7700  @ 2.40GHz

Hardware check
* Hardware support (CPU microcode) for mitigation techniques
  * Indirect Branch Restricted Speculation (IBRS)
    * SPEC_CTRL MSR is available:  NO 
    * CPU indicates IBRS capability:  NO 
  * Indirect Branch Prediction Barrier (IBPB)
    * PRED_CMD MSR is available:  NO 
    * CPU indicates IBPB capability:  NO 
  * Single Thread Indirect Branch Predictors (STIBP)
    * SPEC_CTRL MSR is available:  NO 
    * CPU indicates STIBP capability:  NO 
  * Enhanced IBRS (IBRS_ALL)
    * CPU indicates ARCH_CAPABILITIES MSR availability:  NO 
    * ARCH_CAPABILITIES MSR advertises IBRS_ALL capability:  NO 
  * CPU explicitly indicates not being vulnerable to Meltdown (RDCL_NO):  NO 
  * CPU microcode is known to cause stability problems:  NO  (model 15 stepping 11 ucode 0x1)
* CPU vulnerability to the three speculative execution attacks variants
  * Vulnerable to Variant 1:  YES 
  * Vulnerable to Variant 2:  YES 
  * Vulnerable to Variant 3:  YES 

CVE-2017-5753 [bounds check bypass] aka 'Spectre Variant 1'
* Kernel has array_index_mask_nospec:  NO 
* Checking count of LFENCE instructions following a jump in kernel...  YES  (71 jump-then-lfence instructions found, which is >= 30 (heuristic))
> STATUS:  NOT VULNERABLE  (Kernel source has PROBABLY been patched to mitigate the vulnerability (jump-then-lfence instructions heuristic))

CVE-2017-5715 [branch target injection] aka 'Spectre Variant 2'
* Mitigation 1
  * Kernel is compiled with IBRS/IBPB support:  YES 
  * Currently enabled features
    * IBRS enabled for Kernel space:  NO 
    * IBRS enabled for User space:  NO 
    * IBPB enabled:  NO 
* Mitigation 2
  * Kernel compiled with retpoline option:  NO 
  * Kernel compiled with a retpoline-aware compiler:  NO 
  * Retpoline enabled:  NO 
> STATUS:  VULNERABLE  (IBRS hardware + kernel support OR kernel with retpoline are needed to mitigate the vulnerability)

CVE-2017-5754 [rogue data cache load] aka 'Meltdown' aka 'Variant 3'
* Kernel supports Page Table Isolation (PTI):  YES 
* PTI enabled and active:  YES 
* Running as a Xen PV DomU:  NO 
> STATUS:  NOT VULNERABLE  (PTI mitigates the vulnerability)

A false sense of security is worse than no security at all, see --disclaimer
```

b-rad-NDi, 2018-02-08:
Artful reports similar. Both vulnerable to CVE-2017-5715 mitigation 2.
```
Spectre and Meltdown mitigation detection tool v0.34+

Checking for vulnerabilities on current system
Kernel is Linux 4.13.0-32201802080420-generic #0+mediatree+hauppauge SMP Thu Feb 8 12:13:50 CST 2018 x86_64
CPU is Intel Xeon E312xx (Sandy Bridge)

Hardware check
* Hardware support (CPU microcode) for mitigation techniques
  * Indirect Branch Restricted Speculation (IBRS)
    * SPEC_CTRL MSR is available:  NO 
    * CPU indicates IBRS capability:  NO 
  * Indirect Branch Prediction Barrier (IBPB)
    * PRED_CMD MSR is available:  NO 
    * CPU indicates IBPB capability:  NO 
  * Single Thread Indirect Branch Predictors (STIBP)
    * SPEC_CTRL MSR is available:  NO 
    * CPU indicates STIBP capability:  NO 
  * Enhanced IBRS (IBRS_ALL)
    * CPU indicates ARCH_CAPABILITIES MSR availability:  NO 
    * ARCH_CAPABILITIES MSR advertises IBRS_ALL capability:  NO 
  * CPU explicitly indicates not being vulnerable to Meltdown (RDCL_NO):  NO 
  * CPU microcode is known to cause stability problems:  NO  (model 42 stepping 1 ucode 0x1)
* CPU vulnerability to the three speculative execution attacks variants
  * Vulnerable to Variant 1:  YES 
  * Vulnerable to Variant 2:  YES 
  * Vulnerable to Variant 3:  YES 

CVE-2017-5753 [bounds check bypass] aka 'Spectre Variant 1'
* Kernel has array_index_mask_nospec:  NO 
* Checking count of LFENCE instructions following a jump in kernel...  YES  (65 jump-then-lfence instructions found, which is >= 30 (heuristic))
> STATUS:  NOT VULNERABLE  (Kernel source has PROBABLY been patched to mitigate the vulnerability (jump-then-lfence instructions heuristic))

CVE-2017-5715 [branch target injection] aka 'Spectre Variant 2'
* Mitigation 1
  * Kernel is compiled with IBRS/IBPB support:  YES 
  * Currently enabled features
    * IBRS enabled for Kernel space:  NO 
    * IBRS enabled for User space:  NO 
    * IBPB enabled:  NO 
* Mitigation 2
  * Kernel compiled with retpoline option:  NO 
  * Kernel compiled with a retpoline-aware compiler:  NO 
  * Retpoline enabled:  NO 
> STATUS:  VULNERABLE  (IBRS hardware + kernel support OR kernel with retpoline are needed to mitigate the vulnerability)

CVE-2017-5754 [rogue data cache load] aka 'Meltdown' aka 'Variant 3'
* Kernel supports Page Table Isolation (PTI):  YES 
* PTI enabled and active:  YES 
* Running as a Xen PV DomU:  NO 
> STATUS:  NOT VULNERABLE  (PTI mitigates the vulnerability)

A false sense of security is worse than no security at all, see --disclaimer
```

cweemin, 2018-02-08:
launchpad is not patched?

b-rad-NDi, 2018-02-08:
These were self built packages from Ubuntu kernel git repo's using fully updated Xenial and Artful environments.

I will be submitting new builds to launchpad in an hour or two. We will see how the results vary then.

b-rad-NDi, 2018-02-08:
My mainline builds return the same as Ubuntu, 'Spectre variant 2 - Mitigation 2' must still be a WiP.

b-rad-NDi, 2018-02-09:
PPA contains new builds, this is as fixed as possible for now. Closing.

Please re-open if new CVE or fixes are produced.

closed 2018-02-09 by b-rad-NDi