b-rad-NDi/Ubuntu-media-tree-kernel-builder #37: Kernel 4.13.0-32201802081620-generic #0+mediatree+hauppauge-Ubuntu

MikeB2013, 2018-02-09:
Just updated my Xubuntu 17.10 test system (not a Ryzen system) which pulled in the latest Hauppauge kernel, rebooted and my QuadHD 885 no longer works. Note it was working on the previous kernel.
I am attaching dmesg outputs from both kernels:

[dmesg_4.13.0-19201712311231-generic.txt](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1711853/dmesg_4.13.0-19201712311231-generic.txt)

[dmesg_4.13.0-32201802081620-generic.txt](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1711857/dmesg_4.13.0-32201802081620-generic.txt)

running dvb-fe-tool -g -a 0 on latest kernel I get 
ERROR    Remote I/O error while opening /dev/dvb/adapter0/frontend0

whilst on the previous kernel I get the expected response
FREQUENCY = 522000000
MODULATION = QAM/AUTO
BANDWIDTH_HZ = 8000000
INVERSION = OFF
CODE_RATE_HP = AUTO
CODE_RATE_LP = AUTO
GUARD_INTERVAL = 1/32
TRANSMISSION_MODE = AUTO
HIERARCHY = NONE
DELIVERY_SYSTEM = DVBT

b-rad-NDi, 2018-02-09:
Hmmm...and not a single error in your dmesg. I have a dvb quadhd here, sans any signal, but I'll try w-scan and see if if I get the same error. I have a suspicion this is related to a new clock fix patch, but what I found was the cx25840 firmware would fail to load if the clock was incorrect.

MikeB2013, 2018-02-09:
w_scan output on my system (all the QuadHD adapter frontends fail with same error code 121)

`mike@1710-haup-test:~$ w_scan -c GB -a 0
w_scan -c GB -a 0 
w_scan version 20170107 (compiled for DVB API 5.10)
using settings for UNITED KINGDOM
DVB aerial
DVB-T GB
scan type TERRESTRIAL, channellist 6
output format vdr-2.0
WARNING: could not guess your codepage. Falling back to 'UTF-8'
output charset 'UTF-8', use -C <charset> to override
main:4012: FATAL: failed to open '/dev/dvb/adapter0/frontend0': 121 Remote I/O error
`

I can run with some debug settings, if it will help, just let me know which settings

Mike

b-rad-NDi, 2018-02-09:
This is what I have in /etc/modprobe.d/linuxtv.conf while testing:
```
#blacklist cx23885

options tuner debug=1
options dvb-core frontend_debug=1

options cx231xx-core core_debug=1
options cx231xx-video video_debug=1
options cx231xx-417 v4l_debug=1
options cx2311xx-dvb debug=2

options lgdt3306a debug=1

options cx25840 debug=1
```

b-rad-NDi, 2018-02-09:
Ok, I get the exact same result as you. And oddly not a single bad looking message in the logs. I'm investigating.

b-rad-NDi, 2018-02-09:
I'm having someone in the main office check 885 and 888 atsc quadhd's now. This issue is strange, because your two logs are identical. My atsc 1265 (single tuner 888 quadhd) has no issue. So this either only affects 885 boards or only dvb boards. I'll know soon.

b-rad-NDi, 2018-02-09:
Okay, the plot thickens. Both 885 and 888 ATSC QuadHD's work completely.

I am bisecting my si2168 patches.

b-rad-NDi, 2018-02-09:
Got it fixed. Apparently cx23885 is incompatible with a frontend who has ts_bus_ctrl.

Will submit new builds briefly.

b-rad-NDi, 2018-02-09:
Doing xenial & artful sanity builds, then pushing immediately to launchpad. You should have an update available sometime tonight.

MikeB2013, 2018-02-10:
Hi Brad,
Tried latest build from launchpad. It solves the dvb-fe-tool -g -a 0 issue in that I now get the expected response.

However when I try and tune to any channel it fails, w_scan does not find any channels (I aborted after a few seconds).

I am attaching some files which may help, in particular the dvbv5-scan output which I ran in verbose mode.

[dmesg-4.13.0-32201802091620-generic.txt](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1713269/dmesg-4.13.0-32201802091620-generic.txt)

[w_scan.txt](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1713270/w_scan.txt)

[dvbv5-scan.txt](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1713273/dvbv5-scan.txt)

Mike

b-rad-NDi, 2018-02-10:
Hmmm...I will back the ts bus control patch off completely and rebuild now.

MikeB2013, 2018-02-10:
If it helps while testing I can pull any builds from your dropbox, saves launchpad.

Mike

b-rad-NDi, 2018-02-10:
Saves me the humiliation of continuing to put up builds where our main dvb demod is broken too lol.

New builds are up, ts_bus_ctrl is no longer tri-stated when going to sleep. If that fixes you up, then I will need to figure out how to better wake back up the ts bus on resume.

MikeB2013, 2018-02-10:
Hi Brad,
Tested 4.13.0-32201802100743-generic #0+mediatree+hauppauge  and it is working, apart from spamming logs, cx23885_wakeup:, cx23885_stop_dma: , cx23885_reg_dump2: etc, debug in /etc/modprobe.d/linuxtv.conf was not active.

I have had all 4 tuners on QuadHD DVB running simultaneously using mythtv. 

I also ran a very quick test on 4.4 and 4.15 builds, they seem to be ok as well.

Mike

b-rad-NDi, 2018-02-10:
Nice, so I just need to figure out how to truely wake the ts bus back up. Too bad I don't have a DVB generator so I can actually test what I do.

All that spam is debug helping to fix the Ryzen issue, it will not be in a ppa build.

b-rad-NDi, 2018-02-10:
@MikeB2013 : I put new 4.15 packages up for testing if you want to try them out. This should have ts bus control fixed now. Enabling the ts bus was missing. Ooops. Thanks for reporting this and helping to test.

MikeB2013, 2018-02-11:
@b-rad-NDi 

Tested new 4.15 package my QuadHD works.
Attached is dmesg output for reference.

[dmesg-4.15.0-rc6+.txt](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1714523/dmesg-4.15.0-rc6.txt)

Mike

b-rad-NDi, 2018-02-11:
Thank you for the assistance.

closed 2018-02-11 by b-rad-NDi