b-rad-NDi/Ubuntu-media-tree-kernel-builder #38: Frontend missing Type information

Saentist, 2018-02-13:
quadHD (855)
https://cdn.pbrd.co/images/H7tPOHd.png
other brand's
https://cdn.pbrd.co/images/H7tQW2v.png
https://cdn.pbrd.co/images/H7tRWFk.png
something missing

b-rad-NDi, 2018-02-13:
You find where that 'master for' string is set and I'll get the proper frontend type filled in there. My cursory search just found nothing though.

b-rad-NDi, 2018-02-13:
You need to tell me exactly what boards these are if I'm expected to figure out differences.

b-rad-NDi, 2018-02-13:
What utility are you using to view this heirarchy?

Saentist, 2018-02-13:
This is TVHeadend
----
how to make and install
https://tvheadend.org/boards/4/topics/24116
----
manual for beginners, but good documentation make product attractive  ...
https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/wiki 
oops someting missing ;)

by the way I not see any other device with will not include frontend type DVB-S/C/T
except quadHD
good is that dvb-c and dvb-t have different options
dvb-t have LNA (low noise amplifier) dvb-c hasn't

b-rad-NDi, 2018-02-13:
Not going to install and debug tvheadend to figure this out.

b-rad-NDi, 2018-02-13:
Any binary trying to pick a frontend would request the delivery systems supported. They would not be going by a string match.

I have not found a single place that appears to hold this value, so again, what cards are you showing? I can compare the board profile of QuadHD to any other card in cx23885-cards.c and there is zero difference.

Saentist, 2018-02-13:
https://cdn.pbrd.co/images/H7uqz5G.png
this is Hauppauge WinTV-HVR400

b-rad-NDi, 2018-02-13:
The formatting is completely different for that entry. I want to know what this is:
https://cdn.pbrd.co/images/H7tRWFk.png

Saentist, 2018-02-13:
TurboSight TBS 6981 is only cars with i have with cx23885

b-rad-NDi, 2018-02-13:
Specifically adapter2, since it uses si2168.

Saentist, 2018-02-13:
https://www.linuxtv.org/wiki/index.php/Silicon_Labs_Si2168

b-rad-NDi, 2018-02-13:
I am extensively familiar with the si2168. The point is we use that frontend a lot, and in our QuadHD it shows 'master for' and for one of your devices it shows 'DVB-C'.

The TBS6981 appears to only use cx24117, so that is not the device I'm looking for. It's cx23885 board profile is braindead simple too, along with the rest of the support code. I wonder if this is part of the eeprom.

b-rad-NDi, 2018-02-13:
Ok, fyi this erroneous data is coming directly from tvheadend. tvheadend does some checks and prints 'master for #%d' based on some checks I'm reviewing.

b-rad-NDi, 2018-02-13:
You should file a ticket with tvheadend about this, I'm not digging further into why they are displaying output the way they do. This is not my bug, unless someone at tvheadend tells me what I'm doing wrong.

b-rad-NDi, 2018-02-13:
It's related to their lfe_master checks. If lfe_master is set, you get 'master for #%d', if it is not set you get 'DELSYS #%d'

b-rad-NDi, 2018-02-13:
You may re-open this issue if you get word from tvheaded that this needs anything fixed on my end.

closed 2018-02-13 by b-rad-NDi

Saentist, 2018-02-13:
Will add issue

Saentist, 2018-02-15:
Fixed by TVHeadend team 
changed way how frontend was read
https://github.com/tvheadend/tvheadend/commit/74368d0447b0167e33022d8bf8c46644ddd270ba

b-rad-NDi, 2018-02-15:
Nice one!