b-rad-NDi/Ubuntu-media-tree-kernel-builder #39: Unneeded flood in syslog and kern.log log

Saentist, 2018-02-15:
Kernel and CPU  
Linux 4.13.0-32201802101823-generic on x86_64
AMD A6-5400B APU with Radeon(tm) HD Graphics, 2 cores
QuadHD 855

**No debug options set.**
Just one second of **syslog** or **kern.log**, include ~860 lines of messages from cx23885.
this take 10+Gb each 24h

**syslog**

Feb 15 09:11:41 tvserv2 kernel: [172197.386219] cx23885 0000:06:00.0: cx23885_wakeup: [ffff98820f1f6000/24] wakeup reg=47897 buf=10861337
Feb 15 09:11:41 tvserv2 kernel: [172197.387742] cx23885 0000:05:00.0: cx23885_wakeup: [ffff988218490000/12] wakeup reg=8333 buf=8200333
Feb 15 09:11:41 tvserv2 kernel: [172197.388638] cx23885 0000:06:00.0: cx23885_wakeup: [ffff9881902d8000/27] wakeup reg=10748 buf=8202748
Feb 15 09:11:41 tvserv2 kernel: [172197.389963] cx23885 0000:06:00.0: cx23885_wakeup: [ffff98820f1f4c00/25] wakeup reg=47898 buf=10861338
Feb 15 09:11:41 tvserv2 kernel: [172197.390751] cx23885 0000:05:00.0: cx23885_wakeup: [ffff9882193a4400/21] wakeup reg=20214 buf=8998646
Feb 15 09:11:41 tvserv2 kernel: [172197.392811] cx23885 0000:05:00.0: cx23885_wakeup: [ffff9881a6438000/13] wakeup reg=8334 buf=8200334
Feb 15 09:11:41 tvserv2 kernel: [172197.393710] cx23885 0000:06:00.0: cx23885_wakeup: [ffff9881902da000/28] wakeup reg=10749 buf=8202749
Feb 15 09:11:41 tvserv2 kernel: [172197.393758] cx23885 0000:06:00.0: cx23885_wakeup: [ffff98821594f400/26] wakeup reg=47899 buf=10861339
Feb 15 09:11:41 tvserv2 kernel: [172197.395811] cx23885 0000:05:00.0: cx23885_wakeup: [ffff988214673000/22] wakeup reg=20215 buf=8998647
Feb 15 09:11:41 tvserv2 kernel: [172197.397557] cx23885 0000:06:00.0: cx23885_wakeup: [ffff98821934b800/27] wakeup reg=47900 buf=10861340

**kern.log**

Feb 15 09:11:42 tvserv2 kernel: [172198.344834] cx23885 0000:05:00.0: cx23885_wakeup: [ffff98820f12ec00/9] wakeup reg=8522 buf=8200522
Feb 15 09:11:42 tvserv2 kernel: [172198.345740] cx23885 0000:06:00.0: cx23885_wakeup: [ffff9881902da800/24] wakeup reg=10937 buf=8202937
Feb 15 09:11:42 tvserv2 kernel: [172198.347054] cx23885 0000:06:00.0: cx23885_wakeup: [ffff988217c06400/21] wakeup reg=48150 buf=10861590
Feb 15 09:11:42 tvserv2 kernel: [172198.347835] cx23885 0000:05:00.0: cx23885_wakeup: [ffff9882193a6000/18] wakeup reg=20403 buf=8998835
Feb 15 09:11:42 tvserv2 kernel: [172198.349898] cx23885 0000:05:00.0: cx23885_wakeup: [ffff98820f12e400/10] wakeup reg=8523 buf=8200523
Feb 15 09:11:42 tvserv2 kernel: [172198.350795] cx23885 0000:06:00.0: cx23885_wakeup: [ffff9881902db400/25] wakeup reg=10938 buf=8202938
Feb 15 09:11:42 tvserv2 kernel: [172198.350844] cx23885 0000:06:00.0: cx23885_wakeup: [ffff988217c07400/22] wakeup reg=48151 buf=10861591
Feb 15 09:11:42 tvserv2 kernel: [172198.352908] cx23885 0000:05:00.0: cx23885_wakeup: [ffff9882193a7000/19] wakeup reg=20404 buf=8998836
Feb 15 09:11:42 tvserv2 kernel: [172198.354650] cx23885 0000:06:00.0: cx23885_wakeup: [ffff98820f1f4800/23] wakeup reg=48152 buf=10861592
Feb 15 09:11:42 tvserv2 kernel: [172198.354958] cx23885 0000:05:00.0: cx23885_wakeup: [ffff988218490c00/11] wakeup reg=8524 buf=8200524
Feb 15 09:11:42 tvserv2 kernel: [172198.355858] cx23885 0000:06:00.0: cx23885_wakeup: [ffff9881902dac00/26] wakeup reg=10939 buf=8202939
Feb 15 09:11:42 tvserv2 kernel: [172198.357972] cx23885 0000:05:00.0: cx23885_wakeup: [ffff9882193a4c00/20] wakeup reg=20405 buf=8998837

**it is not mistake in both places messages are same**

b-rad-NDi, 2018-02-15:
There is no reason to file an issue about a _**debug build with debug options enabled which is only offered in the cx23885 issues thread**_

b-rad-NDi, 2018-02-15:
This build does not come from the ppa and none of the debug patch is part of the repo.

closed 2018-02-15 by b-rad-NDi