b-rad-NDi/Ubuntu-media-tree-kernel-builder #40: i want to use four QuadHD but only two shows up

aabrekk, 2018-03-07:


b-rad-NDi, 2018-03-07:
Edit /etc/modprobe.d/linuxtv.conf and enter:
```
option cx23885 debug=1
option cx25840 debug=1
```
Then reboot and paste the following:
```
dmesg | egrep 'cx2|lgdt|si21|dvb'
```
You can remove /etc/modprobe.d/linuxtv.conf after that.

aabrekk, 2018-03-08:
root@mediabox:~# dmesg | egrep 'cx2|lgdt|si21|dvb'
[    5.930879] cx23885: cx23885 driver version 0.0.4 loaded
[    5.931009] cx23885: CORE cx23885[0]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB(885) [card=60,autodetected]
[    6.271811] cx23885: cx23885[0]: hauppauge eeprom: model=166200
[    6.271812] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.271814] cx23885: cx23885[0]: cx23885 based dvb card
[    6.271815] cx23885: dvb_register(): board=60 port=1
[    6.278120] si2168 3-0064: Silicon Labs Si2168-B40 successfully identified
[    6.278121] si2168 3-0064: firmware version: B 4.0.2
[    6.279791] si2157 4-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    6.279798] dvbdev: DVB: registering new adapter (cx23885[0])
[    6.279800] cx23885 0000:03:00.0: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[    6.279895] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.279896] cx23885: cx23885[0]: cx23885 based dvb card
[    6.279897] cx23885: dvb_register(): board=60 port=2
[    6.285233] si2168 3-0066: Silicon Labs Si2168-B40 successfully identified
[    6.285234] si2168 3-0066: firmware version: B 4.0.2
[    6.286079] si2157 4-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    6.286086] dvbdev: DVB: registering new adapter (cx23885[0])
[    6.286088] cx23885 0000:03:00.0: DVB: registering adapter 1 frontend 0 (Silicon Labs Si2168)...
[    6.286335] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xa5
[    6.286338] cx23885: cx23885[0]/0: found at 0000:03:00.0, rev: 4, irq: 17, latency: 0, mmio: 0xdea00000
[    6.286420] cx23885: CORE cx23885[1]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB(885) [card=60,autodetected]
[    6.627552] cx23885: cx23885[1]: hauppauge eeprom: model=166201
[    6.627553] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.627553] cx23885: cx23885[1]: cx23885 based dvb card
[    6.627554] cx23885: dvb_register(): board=60 port=1
[    6.633012] si2168 8-0064: Silicon Labs Si2168-B40 successfully identified
[    6.633013] si2168 8-0064: firmware version: B 4.0.2
[    6.633722] si2157 9-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    6.633729] dvbdev: DVB: registering new adapter (cx23885[1])
[    6.633730] cx23885 0000:04:00.0: DVB: registering adapter 2 frontend 0 (Silicon Labs Si2168)...
[    6.634086] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.634087] cx23885: cx23885[1]: cx23885 based dvb card
[    6.634087] cx23885: dvb_register(): board=60 port=2
[    6.639519] si2168 8-0066: Silicon Labs Si2168-B40 successfully identified
[    6.639520] si2168 8-0066: firmware version: B 4.0.2
[    6.640230] si2157 9-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    6.640236] dvbdev: DVB: registering new adapter (cx23885[1])
[    6.640237] cx23885 0000:04:00.0: DVB: registering adapter 3 frontend 0 (Silicon Labs Si2168)...
[    6.640672] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xa5
[    6.640675] cx23885: cx23885[1]/0: found at 0000:04:00.0, rev: 4, irq: 18, latency: 0, mmio: 0xde800000
[    6.640785] cx23885: CORE cx23885[2]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB(885) [card=60,autodetected]
[    6.984106] cx23885: cx23885[2]: hauppauge eeprom: model=166200
[    6.984107] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.984107] cx23885: cx23885[2]: cx23885 based dvb card
[    6.984107] cx23885: dvb_register(): board=60 port=1
[    6.989641] si2168 13-0064: Silicon Labs Si2168-B40 successfully identified
[    6.989642] si2168 13-0064: firmware version: B 4.0.2
[    6.990387] si2157 14-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    6.990393] dvbdev: DVB: registering new adapter (cx23885[2])
[    6.990394] cx23885 0000:10:00.0: DVB: registering adapter 4 frontend 0 (Silicon Labs Si2168)...
[    6.990764] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.990765] cx23885: cx23885[2]: cx23885 based dvb card
[    6.990765] cx23885: dvb_register(): board=60 port=2
[    6.996298] si2168 13-0066: Silicon Labs Si2168-B40 successfully identified
[    6.996313] si2168 13-0066: firmware version: B 4.0.2
[    6.997031] si2157 14-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    6.997036] dvbdev: DVB: registering new adapter (cx23885[2])
[    6.997037] cx23885 0000:10:00.0: DVB: registering adapter 5 frontend 0 (Silicon Labs Si2168)...
[    6.997376] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xa5
[    6.997379] cx23885: cx23885[2]/0: found at 0000:10:00.0, rev: 4, irq: 18, latency: 0, mmio: 0xde600000
[    6.997438] cx23885: CORE cx23885[3]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB(885) [card=60,autodetected]
[    7.340109] cx23885: cx23885[3]: hauppauge eeprom: model=166201
[    7.340109] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    7.340110] cx23885: cx23885[3]: cx23885 based dvb card
[    7.340110] cx23885: dvb_register(): board=60 port=1
[    7.345641] si2168 18-0064: Silicon Labs Si2168-B40 successfully identified
[    7.345641] si2168 18-0064: firmware version: B 4.0.2
[    7.346364] si2157 19-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[... 104 lines cut ...]
[   13.390453] si2157 9-0062: firmware version: 3.0.5
[   13.394364] cx23885 0000:04:00.0: DVB: adapter 2 frontend 0 frequency 0 out of range (42000000..870000000)
[... cut ...]
[   15.295113] si2157 4-0060: firmware version: 3.0.5
[   15.299012] si2157 4-0062: firmware version: 3.0.5
[   15.301380] cx23885 0000:03:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (42000000..870000000)
[   15.310763] si2168 3-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   15.912151] si2168 3-0064: firmware version: B 4.0.11
[   15.919100] si2157 4-0060: found a 'Silicon Labs Si2157-A30'
[   15.941746] si2157 4-0060: firmware version: 3.0.5
root@mediabox:~#

aabrekk, 2018-03-08:
The last quad card is in use on another pc until i get this thingy fixed.  So there is 3 Quad cards at the moment in this pc. just to let you know.

MikeB2013, 2018-03-08:
I think some of this is to do with kernel config DVB_MAX_ADAPTERS=8

Mike

b-rad-NDi, 2018-03-08:
Ubuntu is overriding the kernel default of 16 and forcing 8. I will override this in my builds. Something in mainline is forcing the config option to 8 as well, so I will also submit a patch upstream.

As of now the only way to override this is by compiling the kernel yourself.

aabrekk, 2018-03-08:
Ive tried to compile the kernel myself. 

cat /boot/config-4.14.24-custom1  |grep MAX_ADAPTERS
CONFIG_DVB_MAX_ADAPTERS=16

But still it views only 8 adapters in /dev/dvb/ So. I wonder. Is there anything we can do to fix it?

b-rad-NDi, 2018-03-08:
Your logs indicate that DVB_MAX_ADAPTERS is not getting set correctly. You are failing at a spot which explicitly checks this limit and fails if it is too high. I will make a build for you where I've made sure to increase the value.

aabrekk, 2018-03-08:
You got the dmessage yesterday. i did rebuild the kernel today. Thats why. Thanks ;)

b-rad-NDi, 2018-03-08:
Add these to the contents of a modprobe.d conf file:
```
option cx23885 debug=1
option cx25840 debug=1
options tuner debug=1
options dvb-core frontend_debug=1
```
Then install these packages:
https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0

and then paste here the following:
```
uname -a
dmesg | egrep 'si2|lgdt|cx2|dvb|tune|frontend|adapter'
```

aabrekk, 2018-03-08:
erikandre@mediabox:~$ uname -a
Linux mediabox 4.16.0-rc4+ #12 SMP Tue Mar 6 11:18:26 CST 2018 x86_64 x86_64 x86_64 GNU/Linux
erikandre@mediabox:~$ dmesg | egrep 'si2|lgdt|cx2|dvb|tune|frontend|adapter'
[    5.959688] cx23885: cx23885 driver version 0.0.4 loaded
[    5.959786] cx23885: CORE cx23885[0]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB(885) [card=60,autodetected]
[    6.299426] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    6.299428] cx23885: cx23885[0]: hauppauge eeprom: model=166200
[    6.299429] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.299435] cx23885: cx23885[0]: cx23885 based dvb card
[    6.299436] cx23885: dvb_register(): board=60 port=1
[    6.305648] si2168 3-0064: Silicon Labs Si2168-B40 successfully identified
[    6.305649] si2168 3-0064: firmware version: B 4.0.2
[    6.307026] si2157 4-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    6.307033] dvbdev: DVB: registering new adapter (cx23885[0])
[    6.307034] cx23885 0000:03:00.0: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[    6.307121] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.307121] cx23885: cx23885[0]: cx23885 based dvb card
[    6.307122] cx23885: dvb_register(): board=60 port=2
[    6.312369] si2168 3-0066: Silicon Labs Si2168-B40 successfully identified
[    6.312370] si2168 3-0066: firmware version: B 4.0.2
[    6.313010] si2157 4-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    6.313014] dvbdev: DVB: registering new adapter (cx23885[0])
[    6.313015] cx23885 0000:03:00.0: DVB: registering adapter 1 frontend 0 (Silicon Labs Si2168)...
[    6.313224] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xa5
[    6.313227] cx23885: cx23885[0]/0: found at 0000:03:00.0, rev: 4, irq: 17, latency: 0, mmio: 0xdea00000
[    6.313305] cx23885: CORE cx23885[1]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB(885) [card=60,autodetected]
[    6.655248] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    6.655250] cx23885: cx23885[1]: hauppauge eeprom: model=166201
[    6.655250] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.655251] cx23885: cx23885[1]: cx23885 based dvb card
[    6.655251] cx23885: dvb_register(): board=60 port=1
[    6.660580] si2168 8-0064: Silicon Labs Si2168-B40 successfully identified
[    6.660581] si2168 8-0064: firmware version: B 4.0.2
[    6.661187] si2157 9-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    6.661192] dvbdev: DVB: registering new adapter (cx23885[1])
[    6.661193] cx23885 0000:04:00.0: DVB: registering adapter 2 frontend 0 (Silicon Labs Si2168)...
[    6.661328] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.661329] cx23885: cx23885[1]: cx23885 based dvb card
[    6.661329] cx23885: dvb_register(): board=60 port=2
[    6.666634] si2168 8-0066: Silicon Labs Si2168-B40 successfully identified
[    6.666634] si2168 8-0066: firmware version: B 4.0.2
[    6.667338] si2157 9-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    6.667343] dvbdev: DVB: registering new adapter (cx23885[1])
[    6.667343] cx23885 0000:04:00.0: DVB: registering adapter 3 frontend 0 (Silicon Labs Si2168)...
[    6.667530] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xa5
[    6.667533] cx23885: cx23885[1]/0: found at 0000:04:00.0, rev: 4, irq: 18, latency: 0, mmio: 0xde800000
[    6.667624] cx23885: CORE cx23885[2]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB(885) [card=60,autodetected]
[    7.008103] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    7.008105] cx23885: cx23885[2]: hauppauge eeprom: model=166200
[    7.008105] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    7.008106] cx23885: cx23885[2]: cx23885 based dvb card
[    7.008106] cx23885: dvb_register(): board=60 port=1
[    7.013608] si2168 13-0064: Silicon Labs Si2168-B40 successfully identified
[    7.013608] si2168 13-0064: firmware version: B 4.0.2
[    7.014325] si2157 14-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    7.014331] dvbdev: DVB: registering new adapter (cx23885[2])
[    7.014332] cx23885 0000:10:00.0: DVB: registering adapter 4 frontend 0 (Silicon Labs Si2168)...
[    7.014593] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    7.014594] cx23885: cx23885[2]: cx23885 based dvb card
[    7.014594] cx23885: dvb_register(): board=60 port=2
[    7.020027] si2168 13-0066: Silicon Labs Si2168-B40 successfully identified
[    7.020027] si2168 13-0066: firmware version: B 4.0.2
[    7.020715] si2157 14-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    7.020719] dvbdev: DVB: registering new adapter (cx23885[2])
[    7.020720] cx23885 0000:10:00.0: DVB: registering adapter 5 frontend 0 (Silicon Labs Si2168)...
[    7.020883] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xa5
[    7.020886] cx23885: cx23885[2]/0: found at 0000:10:00.0, rev: 4, irq: 18, latency: 0, mmio: 0xde600000
[    7.020945] cx23885: CORE cx23885[3]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB(885) [card=60,autodetected]
[    7.365418] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    7.365420] cx23885: cx23885[3]: hauppauge eeprom: model=166201
[... 80 lines cut ...]
[   10.995201] si2168 18-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   11.609723] si2168 18-0066: firmware version: B 4.0.11
[... cut ...]
[   14.707254] si2168 3-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   15.313603] si2168 3-0066: firmware version: B 4.0.11
[   15.316673] cx23885 0000:03:00.0: DVB: adapter 1 frontend 0 frequency 0 out of range (42000000..870000000)
[   15.324880] si2168 3-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   15.927139] si2168 3-0064: firmware version: B 4.0.11
[   15.930210] cx23885 0000:03:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (42000000..870000000)
erikandre@mediabox:~$

b-rad-NDi, 2018-03-08:
Looks good?

aabrekk, 2018-03-08:
it shows 12 adapters. I might try to add the last card. to see if it shows all.

b-rad-NDi, 2018-03-08:
the cx23885 driver has a limit of 8 boards, each quad hd is two boards to the driver, so the max possible QuadHD should be four.

b-rad-NDi, 2018-03-08:
Which coincidentally maxes out the number of possible dvb adapters.

aabrekk, 2018-03-08:
Have just inserted the last card. 

But only 3 cards are showing in lspci. Cant understand why? 

Any ideah?

b-rad-NDi, 2018-03-09:
Unfortunately no. If you see no errors in dmesg, then you should start looking around at pcie debug options to enable. You should also try asking on irc and seeing if anyone has an idea.

aabrekk, 2018-03-09:
Could it be that the driver doesnt want to show more than 3 cards? And then gives a big apple about the last card?

b-rad-NDi, 2018-03-09:
If your card does not show up in lspci then there is a much lower level issue involved. This is outside of the media tree. You will need to find someone who knows more about pcie and various limitations that might be encountered.

tluciomiranda, 2018-06-14:
Hi @b-rad-NDi 

I am having also the same problem with 3 cards, as only two of them work correctly. The following messages appear at boot time in dmesg for the third card (the other 2 work ok):

[    4.774931] tveeprom: Hauppauge model 166200, rev B5I6, serial# 4035902800
[    4.774933] tveeprom: MAC address is 00:0d:fe:8e:fd:50
[    4.774935] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    4.774936] tveeprom: TV standards ATSC/DVB Digital (eeprom 0x80)
[    4.774938] tveeprom: audio processor is CX23885 (idx 39)
[    4.774939] tveeprom: decoder processor is CX23885 (idx 33)
[    4.774940] tveeprom: has no radio, has IR receiver, has no IR transmitter
[    4.774941] cx23885: cx23885[4]: hauppauge eeprom: model=166200
[    4.774943] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    4.774944] cx23885: cx23885[4]: cx23885 based dvb card
[    4.774946] cx23885: dvb_register(): board=60 port=1
[    4.781039] i2c i2c-24: Added multiplexed i2c bus 27
[    4.781042] si2168 24-0064: Silicon Labs Si2168-B40 successfully identified
[    4.781044] si2168 24-0064: firmware version: B 4.0.2
[    4.782061] si2157 25-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    4.782068] cx23885[4]: dvb_register_adapter failed (errno = -23)
[    4.782069] vb2_dvb_register_adapter failed (errno = -23)
[    4.782145] cx23885: cx23885_dvb_register() dvb_register failed err = -22
[    4.782579] cx23885: cx23885_dev_setup() Failed to register dvb adapters on VID_B
[    4.782996] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    4.782998] cx23885: cx23885[4]: cx23885 based dvb card
[    4.783000] cx23885: dvb_register(): board=60 port=2
[    4.788985] i2c i2c-24: Added multiplexed i2c bus 27
[    4.788988] si2168 24-0066: Silicon Labs Si2168-B40 successfully identified
[    4.788990] si2168 24-0066: firmware version: B 4.0.2
[    4.790003] si2157 25-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    4.790009] cx23885[4]: dvb_register_adapter failed (errno = -23)
[    4.790010] vb2_dvb_register_adapter failed (errno = -23)
[    4.790098] cx23885: cx23885_dvb_register() dvb_register failed err = -22
[    4.790540] cx23885: cx23885_dev_setup() Failed to register dvb on VID_C
[    4.790962] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xa4
[    4.790967] cx23885: cx23885[4]/0: found at 0000:0e:00.0, rev: 3, irq: 17, latency: 0, mmio: 0xef200000
[    4.791080] cx23885: CORE cx23885[5]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB(885) [card=60,autodetected]
[    5.126947] tveeprom: Hauppauge model 166201, rev B5I6, serial# 4035902800
[    5.126949] tveeprom: MAC address is 00:0d:fe:8e:fd:50
[    5.126951] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    5.126953] tveeprom: TV standards ATSC/DVB Digital (eeprom 0x80)
[    5.126954] tveeprom: audio processor is CX23885 (idx 39)
[    5.126955] tveeprom: decoder processor is CX23885 (idx 33)
[    5.126956] tveeprom: has no radio
[    5.126957] cx23885: cx23885[5]: hauppauge eeprom: model=166201
[    5.126959] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    5.126960] cx23885: cx23885[5]: cx23885 based dvb card
[    5.126962] cx23885: dvb_register(): board=60 port=1
[    5.133055] i2c i2c-27: Added multiplexed i2c bus 30
[    5.133058] si2168 27-0064: Silicon Labs Si2168-B40 successfully identified
[    5.133060] si2168 27-0064: firmware version: B 4.0.2
[    5.134075] si2157 28-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    5.134081] cx23885[5]: dvb_register_adapter failed (errno = -23)
[    5.134082] vb2_dvb_register_adapter failed (errno = -23)
[    5.134160] cx23885: cx23885_dvb_register() dvb_register failed err = -22
[    5.134601] cx23885: cx23885_dev_setup() Failed to register dvb adapters on VID_B
[    5.135030] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    5.135031] cx23885: cx23885[5]: cx23885 based dvb card
[    5.135033] cx23885: dvb_register(): board=60 port=2
[    5.141111] i2c i2c-27: Added multiplexed i2c bus 30
[    5.141114] si2168 27-0066: Silicon Labs Si2168-B40 successfully identified
[    5.141115] si2168 27-0066: firmware version: B 4.0.2
[    5.142161] si2157 28-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    5.142168] cx23885[5]: dvb_register_adapter failed (errno = -23)
[    5.142169] vb2_dvb_register_adapter failed (errno = -23)
[    5.142245] cx23885: cx23885_dvb_register() dvb_register failed err = -22
[    5.142690] cx23885: cx23885_dev_setup() Failed to register dvb on VID_C
[    5.143126] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xa4
[    5.143131] cx23885: cx23885[5]/0: found at 0000:0f:00.0, rev: 3, irq: 18, latency: 0, mmio: 0xef000000

Build:
administrator@iptv-server:~$ uname -r
4.4.0-122201805060708-generic

Will you patch something in your next builds to allow for more cards to be used (ideally four) ?
Also, appreciate your efforts on this project.

Regards

b-rad-NDi, 2018-07-03:
Patched the build now to hopefully support 8 quadhd's at once.

tluciomiranda, 2018-07-03:
Great! 
Thank you again.

aabrekk, 2018-07-04:
Hi! ive tried your 4.17 kernel and header.

Got

Begin: Running script/local-block ... done and then

Gave up waiting for root device

Missing modules cat /proc/modules dropping into a shell

Running ubuntu 16.04

What might be wrong?

best wishes, Aabrekk

b-rad-NDi, 2018-07-04:
What's wrong is the builds with this modified have not been pushed to production yet :-p

When this last batch of sanity builds completes I'll push them to launchpad.

b-rad-NDi, 2018-07-04:
I dunno what 4.17 of mine you've tested though, as those builds have been mostly private'ish.

b-rad-NDi, 2018-07-04:
Wait...your issues are completely unrelated to this if you're misssing root device...

b-rad-NDi, 2018-07-04:
ubuntu 16.04 users should be downloading the 4.4 kernel releases and nothing else, at this time.

aabrekk, 2018-07-05:
Now installed ubuntu 18.04 LTS and wonder where the header and kernel stuff
you had in dropbox is?

aabrekk, 2018-07-05:
8 QuadHD`s at once. Hmm. I see 8 adapters (2 cards) not 16  (4 cards)

So i wonder what kind of files you have patched to hopefully support 8
quadhd`s at once.

At the moment running Ubuntu server 18.04 LTS

aabrekk, 2018-07-05:
Found it. But, You said you added support for 8 Cards

tried linux-headers-mediatree_0.17.4+bionic_all.deb
linux-image-mediatree_0.17.4+bionic_all.deb

8 QuadHD`s at once. Hmm. I see 8 adapters (2 cards) not 16  (4 cards)

So i wonder what kind of files you have patched to hopefully support 8
quadhd`s at once.

At the moment running Ubuntu server 18.04 LTS

2018-07-05 11:37 GMT+02:00 Erik Andre Aabrekk <erikandre@oneway.no>:

aabrekk, 2018-07-05:
ive tried is  linux-headers-mediatree_0.17.4+bionic_all.deb
linux-image-mediatree_0.17.4+bionic_all.deb

8 QuadHD`s at once. Hmm. I see 8 adapters (2 cards) not 16  (4 cards)

So i wonder what kind of files you have patched to hopefully support 8
quadhd`s at once.

At the moment running Ubuntu server 18.04 LTS

Aabrekk

2018-07-05 12:14 GMT+02:00 Erik Andre Aabrekk <erikandre@oneway.no>:

b-rad-NDi, 2018-07-05:
0.17.4 is old and unpatched. The 0.17.5 release will be linked to the new packages.
You may find the packages directly here:
https://launchpad.net/~b-rad/+archive/ubuntu/kernel+mediatree+hauppauge/+packages
You want timestamp 201807041620

b-rad-NDi, 2018-07-05:
You should have the possibility of maximum 8 cx23885 cards and 32 maximum dvb adapters.

b-rad-NDi, 2018-07-05:
I cleaned out the dropbox because the contents were out of date.

aabrekk, 2018-07-05:
Whitch of the files do i need to download and install. There was a ocean of
files. And just want to grab the files needed. All, Or can you tell me
whitch i need?

b-rad-NDi, 2018-07-05:
New virtual packages have been pushed, install the ppa using the normal instructions and you'll get what you need.

b-rad-NDi, 2018-07-26:
@aabrekk : Have you tried the 20180704 packages to confirm the limit should now be 8 quad hd's?

b-rad-NDi, 2018-08-30:
beep beep @aabrekk @tluciomiranda

tluciomiranda, 2018-08-30:
Hi Brad,
Sorry for not saying anything. Yes I've upgraded the system, and at least with the last two packages 0.17.5 and 0.17.6 all is working fine. Using 12 tuners now.
Thank you!

b-rad-NDi, 2018-08-30:
Excellent! Thanks for the confirmation. I will submit patches to Ubuntu to rectify this issue in general. Closing this issue.

closed 2018-08-30 by b-rad-NDi

aabrekk, 2018-08-31:
Beep Beep. it Did work and i got 12 Tuners on my 3 Twin cards.

But now after some time i have been having some issues (Kernelmodule)
issues.
The computer just start freeze and NOTHING is responding. So i had to
remove my tvtuner cards to get the computer working as i wished.

http://paste.ubuntu.com/p/6fFhpsbmNN/

http://paste.ubuntu.com/p/SBQHnSddt6/

Not been that much fun. More a annoying problem. I thought it was my
mainboard or my cpu or the CPU cooler. But neeh. It was the kernelmodule
that dropped my system so i had a running computer that didnt respond to my
operations.

b-rad-NDi, 2018-08-31:
@aabrekk Those appear to be network related oopses dealing with memory/cpuidle, they should be unrelated to cx23885. Could indicate memory issues or a driver bug when changing performance level. You could potentially pass some kernel options to aid your issues (which ones I don't know). On my ryzen platform I was seeing something similar with the onboard ethernet and had to disable it.

estimadarocha, 2018-09-04:
i am having issues with this... what should i do to run more than 2 quad's? ppa isn't enought?

estimadarocha, 2018-09-04:
after reboot they appear... however:

[ 2904.125871] si2168 39-0066: Direct firmware load for dvb-demod-si2168-02.fw failed with error -2
[ 2904.125875] si2168 39-0066: Direct firmware load for dvb-demod-si2168-02.fw failed with error -2
[ 2904.125875] si2168 39-0066: firmware file 'dvb-demod-si2168-02.fw' not found

lots of those!

aabrekk, 2018-09-05:
Issue was used i7 8 gen CPU on 7 gen mainboard. It didnt like it. Found one
i5 CPU 7 gen hopefully things is running more smoooth now. Thanks for the
info in last reply. Pointed me in the correct direction.

b-rad-NDi, 2018-09-05:
@estimadarocha You must install linux-firmware-hauppauge package from my ppa to obtain that firmware. I am leaving this bug closed, as it only relates to getting 4 quadhd's recognized. Please open new tickets for other issues encountered, along with kernel logs / system data / etc so I can assess them.