b-rad-NDi/Ubuntu-media-tree-kernel-builder #41: WinTV-QuadHD-ATSC cx23885_wakeup flood

tomsoutherland, 2018-03-24:
I've just built an Ubuntu 16.04.4 server on a Supermicro X10SRi motherboard and am using the Hauppauge WinTV-QuadHD-ATSC model 165100 (card=57). I'm using Plex to manage the streaming.

I can get everything working but shortly after several clients begin streaming live TV, the cx23885_wakeup message begins flooding the logs and slows performance.

My version:

Linux snoopy 4.4.0-116201803011620-generic #0+mediatree+hauppauge-Ubuntu SMP Thu Mar 1 22:40:04 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

Messages relevant to the card at boot:

Mar 24 06:59:57 snoopy kernel: [   18.211239] Linux video capture interface: v2.00
Mar 24 06:59:57 snoopy kernel: [   18.278773] Adding 33554428k swap on /dev/zd0.  Priority:-1 extents:1 across:33554428k SSFS
Mar 24 06:59:57 snoopy kernel: [   18.329447] rc_core: IR Remote Control driver registered, major 242
Mar 24 06:59:57 snoopy kernel: [   18.569988] cx23885: cx23885 driver version 0.0.4 loaded
Mar 24 06:59:57 snoopy kernel: [   18.570107] cx23885: CORE cx23885[0]: subsystem: 0070:6a18, board: Hauppauge WinTV-QuadHD-ATSC [card=57,autodetected]
Mar 24 06:59:57 snoopy kernel: [   18.902901] tveeprom: Hauppauge model 165100, rev B4I6, serial# 4036037378
Mar 24 06:59:57 snoopy kernel: [   18.902904] tveeprom: MAC address is 00:0d:fe:91:0b:02
Mar 24 06:59:57 snoopy kernel: [   18.902905] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
Mar 24 06:59:57 snoopy kernel: [   18.902907] tveeprom: TV standards NTSC(M) ATSC/DVB Digital (eeprom 0x88)
Mar 24 06:59:57 snoopy kernel: [   18.902908] tveeprom: audio processor is CX23888 (idx 40)
Mar 24 06:59:57 snoopy kernel: [   18.902909] tveeprom: decoder processor is CX23888 (idx 34)
Mar 24 06:59:57 snoopy kernel: [   18.902910] tveeprom: has no radio, has IR receiver, has no IR transmitter
Mar 24 06:59:57 snoopy kernel: [   18.902911] cx23885: cx23885[0]: hauppauge eeprom: model=165100
Mar 24 06:59:57 snoopy kernel: [   18.908494] cx25840 5-0044: cx23888 A/V decoder found @ 0x88 (cx23885[0])
Mar 24 06:59:57 snoopy kernel: [   19.673868] cx25840 5-0044: loaded v4l-cx23885-avcore-01.fw firmware (16382 bytes)
Mar 24 06:59:57 snoopy kernel: [   19.751286] cx23885: cx23885[0]: registered device video0 [v4l2]
Mar 24 06:59:57 snoopy kernel: [   19.751322] cx23885: cx23885[0]: registered device vbi0
Mar 24 06:59:57 snoopy kernel: [   19.751421] cx23885: cx23885[0]: alsa: registered ALSA audio device
Mar 24 06:59:57 snoopy kernel: [   19.751423] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
Mar 24 06:59:57 snoopy kernel: [   19.751425] cx23885: cx23885[0]: cx23885 based dvb card
Mar 24 06:59:57 snoopy kernel: [   19.751427] cx23885: dvb_register(): board=57 port=1
Mar 24 06:59:57 snoopy kernel: [   19.784633] si2157 4-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
Mar 24 06:59:57 snoopy kernel: [   19.784637] cx23885: dvb_register(): QUADHD_ATSC analog setup
Mar 24 06:59:57 snoopy kernel: [   19.784640] dvbdev: DVB: registering new adapter (cx23885[0])
Mar 24 06:59:57 snoopy kernel: [   19.784643] cx23885 0000:04:00.0: DVB: registering adapter 0 frontend 0 (LG Electronics LGDT3306A VSB/QAM Frontend)...
Mar 24 06:59:57 snoopy kernel: [   19.784820] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
Mar 24 06:59:57 snoopy kernel: [   19.784821] cx23885: cx23885[0]: cx23885 based dvb card
Mar 24 06:59:57 snoopy kernel: [   19.784823] cx23885: dvb_register(): board=57 port=2
Mar 24 06:59:57 snoopy kernel: [   19.791594] si2157 4-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
Mar 24 06:59:57 snoopy kernel: [   19.791598] dvbdev: DVB: registering new adapter (cx23885[0])
Mar 24 06:59:57 snoopy kernel: [   19.791601] cx23885 0000:04:00.0: DVB: registering adapter 1 frontend 0 (LG Electronics LGDT3306A VSB/QAM Frontend)...
Mar 24 06:59:57 snoopy kernel: [   19.791785] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xd0
Mar 24 06:59:57 snoopy kernel: [   19.791789] cx23885: cx23885[0]/0: found at 0000:04:00.0, rev: 4, irq: 74, latency: 0, mmio: 0xfb400000
Mar 24 06:59:57 snoopy kernel: [   19.791902] cx23885: CORE cx23885[1]: subsystem: 0070:6b18, board: Hauppauge WinTV-QuadHD-ATSC [card=57,autodetected]
Mar 24 06:59:57 snoopy kernel: [   20.122742] tveeprom: Hauppauge model 165101, rev B4I6, serial# 4036037378
Mar 24 06:59:57 snoopy kernel: [   20.122743] tveeprom: MAC address is 00:0d:fe:91:0b:02
Mar 24 06:59:57 snoopy kernel: [   20.122745] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
Mar 24 06:59:57 snoopy kernel: [   20.122746] tveeprom: TV standards NTSC(M) ATSC/DVB Digital (eeprom 0x88)
Mar 24 06:59:57 snoopy kernel: [   20.122747] tveeprom: audio processor is CX23888 (idx 40)
Mar 24 06:59:57 snoopy kernel: [   20.122748] tveeprom: decoder processor is CX23888 (idx 34)
Mar 24 06:59:57 snoopy kernel: [   20.122749] tveeprom: has no radio
Mar 24 06:59:57 snoopy kernel: [   20.122750] cx23885: cx23885[1]: hauppauge eeprom: model=165101
Mar 24 06:59:57 snoopy kernel: [   20.124434] cx25840 8-0044: cx23888 A/V decoder found @ 0x88 (cx23885[1])
Mar 24 06:59:57 snoopy kernel: [   20.738058] cx25840 8-0044: loaded v4l-cx23885-avcore-01.fw firmware (16382 bytes)
Mar 24 06:59:57 snoopy kernel: [   20.815493] cx23885: cx23885[1]: registered device video1 [v4l2]
Mar 24 06:59:57 snoopy kernel: [   20.815521] cx23885: cx23885[1]: registered device vbi1
Mar 24 06:59:57 snoopy kernel: [   20.815593] cx23885: cx23885[1]: alsa: registered ALSA audio device
Mar 24 06:59:57 snoopy kernel: [   20.815595] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
Mar 24 06:59:57 snoopy kernel: [   20.815596] cx23885: cx23885[1]: cx23885 based dvb card
Mar 24 06:59:57 snoopy kernel: [   20.815598] cx23885: dvb_register(): board=57 port=1
Mar 24 06:59:57 snoopy kernel: [   20.822456] si2157 7-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
Mar 24 06:59:57 snoopy kernel: [   20.822461] cx23885: dvb_register(): QUADHD_ATSC analog setup
Mar 24 06:59:57 snoopy kernel: [   20.822463] dvbdev: DVB: registering new adapter (cx23885[1])
Mar 24 06:59:57 snoopy kernel: [   20.822477] cx23885 0000:05:00.0: DVB: registering adapter 2 frontend 0 (LG Electronics LGDT3306A VSB/QAM Frontend)...
Mar 24 06:59:57 snoopy kernel: [   20.822717] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
Mar 24 06:59:57 snoopy kernel: [   20.822719] cx23885: cx23885[1]: cx23885 based dvb card
Mar 24 06:59:57 snoopy kernel: [   20.822720] cx23885: dvb_register(): board=57 port=2
Mar 24 06:59:57 snoopy kernel: [   20.829554] si2157 7-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
Mar 24 06:59:57 snoopy kernel: [   20.829557] dvbdev: DVB: registering new adapter (cx23885[1])
Mar 24 06:59:57 snoopy kernel: [   20.829559] cx23885 0000:05:00.0: DVB: registering adapter 3 frontend 0 (LG Electronics LGDT3306A VSB/QAM Frontend)...
Mar 24 06:59:57 snoopy kernel: [   20.829782] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xd0
Mar 24 06:59:57 snoopy kernel: [   20.829787] cx23885: cx23885[1]/0: found at 0000:05:00.0, rev: 4, irq: 75, latency: 0, mmio: 0xfb200000

When it goes spam mode:

Mar 24 06:55:04 snoopy kernel: [  738.607358] cx23885 0000:04:00.0: cx23885_wakeup: [ffff88082e796400/31] wakeup reg=0 buf=65536
Mar 24 06:55:04 snoopy kernel: [  738.617282] cx23885 0000:04:00.0: cx23885_wakeup: [ffff880857335400/0] wakeup reg=1 buf=65537
Mar 24 06:55:04 snoopy kernel: [  738.627210] cx23885 0000:04:00.0: cx23885_wakeup: [ffff880857335000/1] wakeup reg=2 buf=65538
Mar 24 06:55:04 snoopy kernel: [  738.637150] cx23885 0000:04:00.0: cx23885_wakeup: [ffff880857334c00/2] wakeup reg=3 buf=65539
Mar 24 06:55:04 snoopy kernel: [  738.647073] cx23885 0000:04:00.0: cx23885_wakeup: [ffff880857334800/3] wakeup reg=4 buf=65540
Mar 24 06:55:04 snoopy kernel: [  738.657007] cx23885 0000:04:00.0: cx23885_wakeup: [ffff880857334400/4] wakeup reg=5 buf=65541
Mar 24 06:55:04 snoopy kernel: [  738.666952] cx23885 0000:04:00.0: cx23885_wakeup: [ffff880857334000/5] wakeup reg=6 buf=65542
Mar 24 06:55:04 snoopy kernel: [  738.676882] cx23885 0000:04:00.0: cx23885_wakeup: [ffff880857333c00/6] wakeup reg=7 buf=65543
Mar 24 06:55:04 snoopy kernel: [  738.686817] cx23885 0000:04:00.0: cx23885_wakeup: [ffff880857333800/7] wakeup reg=8 buf=65544
Mar 24 06:55:04 snoopy kernel: [  738.696749] cx23885 0000:04:00.0: cx23885_wakeup: [ffff880857333400/8] wakeup reg=9 buf=65545
Mar 24 06:55:04 snoopy kernel: [  738.706683] cx23885 0000:04:00.0: cx23885_wakeup: [ffff880857333000/9] wakeup reg=10 buf=65546
Mar 24 06:55:04 snoopy kernel: [  738.716615] cx23885 0000:04:00.0: cx23885_wakeup: [ffff880857332c00/10] wakeup reg=11 buf=65547

This will continue until I reboot the system.

tomsoutherland, 2018-03-27:
I wonder if this is the same issue observed in "QuadHD missed interrupts #19"?

As of now, I've disabled all kernel logging to prevent this logging from filling up my disk. I have plenty of CPU cycles so the system seems to be tolerating the additional load. Is this the appropriate place to seek a resolution? Should I be communicating with Hauppauge Computer Works instead?

b-rad-NDi, 2018-03-27:
Pretty much the same. I will be investigating this today.

b-rad-NDi, 2018-03-27:
Did you happen to catch the first spam? Very suspicious it is at a 16bit boundary if so.

MikeB2013, 2018-03-27:
@b-rad-NDi 

Not sure if this is the same but my two logs in quadHD 885 mpeg risc op code error #17 show the start of the spam streams

kern-ryzen-quadhd-885.log.tar.gz
kern-cx23885-tbs-6981.log.tar.gz

Mike

b-rad-NDi, 2018-03-27:
Same thing, you hit the 16 bit limit and then SHTF. Very good hint. Something is getting incorrectly wrapped.

b-rad-NDi, 2018-03-27:
Workstation boot disk death :/ Should be able to work on this tomorrow.

b-rad-NDi, 2018-03-28:
I have packages I just started testing out here:
https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0

This just changes the counter to u32 from u16, so there will be no wrapping, which is the one obvious issue.

I'm letting the test cook on my R3 system, spinning up new pipelines every 15s on all four tuners.

b-rad-NDi, 2018-03-28:
Note after staring at this patch I do not believe simply changing a u16 to u32 is a fix. Digging into this further.

MikeB2013, 2018-03-29:
@b-rad-NDi 
Another datapoint in case it helps.

I ran the new packages on my test amd (not Ryzen) system with a single TBS DVB-S/S2 tuner card, which works correctly with current Ubuntu stock kernel 4.13.0-37-generic.  

The spam flood started around 7 minutes in from a reboot, followed soon after with
Mar 29 08:38:09 xubuntu1604 kernel: [  527.386088] cx23885: cx23885[0]: mpeg risc op code error

Attached is the syslog
[syslog-20180329-tbs-4.4-hau.log.tar.gz](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1859221/syslog-20180329-tbs-4.4-hau.log.tar.gz)

Note my test system is not heavily loaded, I would go as far as to say it has a minimal load. The only application of consequence is mythtv backend and all it was doing was gathering data over EIT from the TBS card, not even recording anything. So there is something causing a regression.

Mike

tomsoutherland, 2018-03-29:
It took longer for the spam flood to kick in but sadly, it did again after around 10 minutes.
[kern-log-4.4.0-116201803281617-generic.gz](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1859745/kern-log-4.4.0-116201803281617-generic.gz)

b-rad-NDi, 2018-03-29:
Both of you exhibit the exact same condition.
```
Mar 29 08:37:55 xubuntu1604 kernel: [  513.379379] cx23885 0000:02:00.0: [ffff8800cbb47000/31] wakeup reg=65535 buf=65536
Mar 29 08:37:55 xubuntu1604 kernel: [  513.379395] cx23885 0000:02:00.0: [ffff8800cf4b6400/0] wakeup reg=65535 buf=65537
Mar 29 08:37:55 xubuntu1604 kernel: [  513.379404] cx23885 0000:02:00.0: [ffff8800cbb5d000/1] wakeup reg=65535 buf=65538
Mar 29 08:37:55 xubuntu1604 kernel: [  513.379412] cx23885 0000:02:00.0: [ffff8800cbb58800/2] wakeup reg=65535 buf=65539
Mar 29 08:37:55 xubuntu1604 kernel: [  513.383919] cx23885 0000:02:00.0: [ffff880204e6b800/3] wakeup reg=0 buf=65540
Mar 29 08:37:55 xubuntu1604 kernel: [  513.383942] cx23885 0000:02:00.0: [ffff880204e6fc00/4] wakeup reg=0 buf=65541
Mar 29 08:37:55 xubuntu1604 kernel: [  513.383950] cx23885 0000:02:00.0: [ffff880204e69800/5] wakeup reg=0 buf=65542
Mar 29 08:37:55 xubuntu1604 kernel: [  513.383957] cx23885 0000:02:00.0: [ffff880204e68000/6] wakeup reg=0 buf=65543
Mar 29 08:37:55 xubuntu1604 kernel: [  513.383964] cx23885 0000:02:00.0: [ffff880204e68800/7] wakeup reg=0 buf=65544
Mar 29 08:37:55 xubuntu1604 kernel: [  513.388477] cx23885 0000:02:00.0: [ffff8800c8bec400/8] wakeup reg=1 buf=65545
Mar 29 08:37:55 xubuntu1604 kernel: [  513.388500] cx23885 0000:02:00.0: [ffff8800c8be9c00/9] wakeup reg=1 buf=65546
Mar 29 08:37:55 xubuntu1604 kernel: [  513.388508] cx23885 0000:02:00.0: [ffff8800cbba9800/10] wakeup reg=1 buf=65547
Mar 29 08:37:55 xubuntu1604 kernel: [  513.388516] cx23885 0000:02:00.0: [ffff8800cbba9000/11] wakeup reg=1 buf=65548
Mar 29 08:37:55 xubuntu1604 kernel: [  513.388523] cx23885 0000:02:00.0: [ffff8800cbbaec00/12] wakeup reg=1 buf=65549
Mar 29 08:37:55 xubuntu1604 kernel: [  513.393033] cx23885 0000:02:00.0: [ffff8800cbbadc00/13] wakeup reg=2 buf=65550
Mar 29 08:37:55 xubuntu1604 kernel: [  513.393058] cx23885 0000:02:00.0: [ffff8800cbbaf000/14] wakeup reg=2 buf=65551
Mar 29 08:37:55 xubuntu1604 kernel: [  513.393065] cx23885 0000:02:00.0: [ffff8800cbbaa000/15] wakeup reg=2 buf=65552
```
Reviewing data sheets and appropriate parts of driver. I believe my constant cycle test does not trigger this because the counter is reset when starting dma, which is only done at the start of a 'session'. When I spin for only 15s it was not possible for me to hit it.

b-rad-NDi, 2018-03-29:
Problem (potentially) identified. When the u16 (reg above) rolls over, but the u32 (buf above) does not, this made the delta calc negative...and since it was unsigned, it became positive. Meaning that once the u16 register rolled over every single interrupt spun through the loop de-queueing buffers (incorrectly) and spewing debug (very expensively inside spinlock) five times.

I think I have the math addressing this issue now. The calculations are now done s32, and the u32 'buf' is modded(65536) before subtracting it from the u16 to check if there were missed interrupts.

This should prevent the loop from spinning multiple times after 0xffff interrupts.

New builds will be in the dropbox briefly.

MikeB2013, 2018-03-30:
Both my systems (Ryzen 3 with QuadHD 885 and AMD Phenom II X4 965 Processor with TBS 6981)  are running 4.4.0-116201803291004-generic #0+mediatree+hauppauge with no issues so far.

Mike

MikeB2013, 2018-03-31:
@b-rad-NDi 

Further to my previous post.

On my  system AMD Phenom II X4 965 Processor with only a TBS 6981 tuner after some hours I got a few 

`Mar 30 23:02:54 xubuntu1604 kernel: [23397.762182] cx23885 0000:02:00.0: cx23885_clear_bridge_error() dma in progress detected 0x00000001 0x00000001, clearing
Mar 30 23:23:43 xubuntu1604 kernel: [24647.179424] cx23885 0000:02:00.0: cx23885_clear_bridge_error() dma in progress detected 0x00000001 0x00000001, clearing
Mar 31 01:44:00 xubuntu1604 kernel: [33064.641206] cx23885 0000:02:00.0: cx23885_clear_bridge_error() dma in progress detected 0x00000001 0x00000001, clearing
Mar 31 02:49:47 xubuntu1604 kernel: [37011.363028] cx23885 0000:02:00.0: cx23885_clear_bridge_error() dma in progress detected 0x00000001 0x00000001, clearing
`

The TBS 6981 was non responsive to tuning requests and required a system reboot (soft, not power cycle) to resolve. 
Attached are syslog.1 dmesg and lspci output (in case it helps)

[syslog.1.zip](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1865264/syslog.1.zip)
[dmesg-20180331.txt.zip](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1865265/dmesg-20180331.txt.zip)
[lspci.txt.zip](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1865266/lspci.txt.zip)

It seems to me that cx23885_clear_bridge_error is being invoked for non Hauppauge tuners.

Mike

b-rad-NDi, 2018-03-31:
cx23885_clear_bridge_error is invoked in theory for any device which encounters a stall in the RiSC engine. I will go back and re-analyze this patch set. I will add back in any parts trimmed from Marks so called 'fully working' original patch.

MikeB2013, 2018-04-01:
My Ryzen system is showing risc errors  after a number of hours running see attached dmesg

[dmesg-20180401.txt.zip](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1866549/dmesg-20180401.txt.zip)

Mike

b-rad-NDi, 2018-04-02:
@MikeB2013 : I'm building two versions. One will have Marks original patch in its entirety. The other will contain only the 'fixed' cx23885_wakeup function handling skipped interrupts. Hopefully the 'original patch' build will exhibit no risc errors on either device and either system. Curious though, if you encounter risc errors on your either device/system with only the interrupt fix applied.

Saentist, 2018-04-02:
This wake up messages stay more large if activate power save feature,
then and firmware reload ;)

b-rad-NDi, 2018-04-02:
I have that message turned to debug level 1 now @Saentist in recent builds, so you should not see anything except the bridge error message unless you explicitly enable debug.

Saentist, 2018-04-02:
echo 'auto' > '/sys/bus/i2c/devices/i2c-7/device/power/control'

and result 
[3021626.892589] cx23885 0000:05:00.0: cx23885_stop_dma: delay=0 reg1=0x00000000 reg2=0x00000000
[3021636.875663] si2168 5-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[3021637.497192] si2168 5-0064: firmware version: B 4.0.11
[3021637.504185] si2157 6-0060: found a 'Silicon Labs Si2157-A30'
[3021637.526814] si2157 6-0060: firmware version: 3.0.5

b-rad-NDi, 2018-04-02:
The stop_dma message has been demoted and will not be seen. The other messages have always been in those drivers...

b-rad-NDi, 2018-04-02:
https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0

Two builds:
- 201804021100 - Marks full original patch. Might have some extra noise in it to be dialed out.
For the 'full original effect' you must put this in /etc/modprobe.d/cx23885.conf
```
options cx23885 startreadmask 0x3fffffff
```
- 201804021200 - Contains only the patch to fix skipped interrupts.

b-rad-NDi, 2018-04-02:
Fudge, actually I messed up something. Deleting 1200 build until I can fix.

MikeB2013, 2018-04-03:
Tried build 201804021100 on both my systems, with options cx23885 startreadmask 0x3fffffff in cx23885.conf both systems failed to initialize dvb frontends  dmesg showed 
Apr  3 10:14:00 xubuntu1604 kernel: [    4.214155] cx23885: `' invalid for parameter `startreadmask'
Apr  3 10:14:00 xubuntu1604 kernel: [    4.218169] cx23885: unknown parameter '0x3fffffff' ignored

 Changing to "options cx23885 startreadmask=0x3fffffff" allowed dvb frontends to be created.

b-rad-NDi, 2018-04-03:
Fixed my comment. Do let me know how these builds perform. If the 1100 build works, with startreadmask of 0x3fffffff, then you can try and go cold turkey with it set to 0. If 0 still worked then I could remove a big block of register dump.

MikeB2013, 2018-04-03:
Attached are dmesg outputs for each of my systems. Both are using mythtv application, which can work in one of two ways for /dev/dvb/adapters, in one mode the adapter is opened on boot and stays open, in the other mode the adapter is opened and closed on demand, both systems are currently set to open and close on demand. 
[dmesg-ryzen-20180403.txt.zip](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1871623/dmesg-ryzen-20180403.txt.zip)
[dmesg-amd-965.txt.zip](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1871624/dmesg-amd-965.txt.zip)

For the ryzen system( has quadhd DVB-T/T2 tuner) a firmware download seems to be triggered on every open,  the amd-965 system (has TBS DVB-S/S2 tuner).

I will switch startreadmask to 0 and let you know, if it works I will leave running for 24 hours.
Mike

b-rad-NDi, 2018-04-03:
startreadmask=0 should at least get rid of all that regdump2 spam.

The si2168 unfortunately needs to reload firmware under certain conditions depending on the version. I will look into this a bit, as every open is a bit excessive. It should only happen after resuming from sleep.

MikeB2013, 2018-04-03:
I am still seeing cx23885_reg_dump2 messages on both systems with options cx23885 startreadmask=0 on the 1100 build

Mike

b-rad-NDi, 2018-04-03:
Post snippets of any suspect spam and I'll turn them off.

MikeB2013, 2018-04-03:
From the ryzen system

[Apr 3 14:58] cx23885 0000:07:00.0: cx23885_reg_dump2: 00040000: 00000020 00000000 00000000 00000000
[  +0.000017] cx23885 0000:07:00.0: cx23885_reg_dump2: 00040010: 00001f06 001f8000 00000000 00000000
[  +0.000015] cx23885 0000:07:00.0: cx23885_reg_dump2: 00040020: 00000000 00000000 00000000 00000000
[  +0.000017] cx23885 0000:07:00.0: cx23885_reg_dump2: 00040030: 00001111 00000001 00000000 00000000
[  +0.000016] cx23885 0000:07:00.0: cx23885_reg_dump2: 00040040: 00000000 00000000 00000000 00000000
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 00040050: 00000000 00000000 00000000 00000000
[  +0.000014] cx23885 0000:07:00.0: cx23885_reg_dump2: 00040060: 00000000 00000000 00000000 00000000
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 00040070: 00000000 00000000 00000000 00000000
[  +0.000014] cx23885 0000:07:00.0: cx23885_reg_dump2: 00040080: 00000000 00000000 00000000 00000000
[  +0.000014] cx23885 0000:07:00.0: cx23885_reg_dump2: 00040090: 00000000 00000000 00000000 00000000
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 000400a0: ffffffff 00000000 00000000 00000000
[  +0.000014] cx23885 0000:07:00.0: cx23885_reg_dump2: 000400b0: 00000000 00000000 00000000 00000000
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 000400c0: 00000000 00000000 00000000 00000000
[  +0.000014] cx23885 0000:07:00.0: cx23885_reg_dump2: 000400d0: 00000000 00000000 00000000 00000000
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 000400e0: 00000000 00000000 00000000 00000000
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 000400f0: 00000000 00000000 00000000 00000000
[  +0.000014] cx23885 0000:07:00.0: cx23885_reg_dump2: 00130120: 00000be6 00000000 00000000 00000000
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 00130130: 00000000 00000000 00000000 00000000
[  +0.000014] cx23885 0000:07:00.0: cx23885_reg_dump2: 00130140: 00000011 00000001 00000000 00000000
[  +0.000017] cx23885 0000:07:00.0: cx23885_reg_dump2: 00130150: 000002f0 00470bc0 0000000d 00000000
[  +0.000014] cx23885 0000:07:00.0: cx23885_reg_dump2: 00130160: 00000000 00000000 00000000 00000001
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 00130170: 00000000 00000000 00000000 00000000
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 00130180: 00000000 0000000c 00000000 00000000
[  +0.000014] cx23885 0000:07:00.0: cx23885_reg_dump2: 00130190: 00000000 00000000 00000000 00000000
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 001301a0: 00000000 00000000 00000000 00000000
[  +0.000014] cx23885 0000:07:00.0: cx23885_reg_dump2: 001301b0: 00000000 00000000 00000000 00000000
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 001301c0: 00000000 00000000 00000000 00000000
[  +0.000014] cx23885 0000:07:00.0: cx23885_reg_dump2: 001301d0: 00000000 00000000 00000000 00000000
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 001301e0: 00000000 00000000 00000000 00000000
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 001301f0: 00000000 00000000 00000000 00000000
[  +0.000014] cx23885 0000:07:00.0: cx23885_reg_dump2: 00130220: 00009da8 00000000 00000000 00000000
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 00130230: 00000000 00000000 00000000 00000000
[  +0.000014] cx23885 0000:07:00.0: cx23885_reg_dump2: 00130240: 00000000 00000000 00000000 00000000
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 00130250: 000002f0 00470bc0 0000000c 00000000
[  +0.000014] cx23885 0000:07:00.0: cx23885_reg_dump2: 00130260: 00000000 00000000 00000000 00000001
[  +0.000016] cx23885 0000:07:00.0: cx23885_reg_dump2: 00130270: 00000000 00000000 00000000 00000000
[  +0.000014] cx23885 0000:07:00.0: cx23885_reg_dump2: 00130280: 00000000 00000000 00000000 00000000
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 00130290: 00000000 00000000 00000000 00000000
[  +0.000014] cx23885 0000:07:00.0: cx23885_reg_dump2: 001302a0: 00000000 00000000 00000000 00000000
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 001302b0: 00000000 00000000 00000000 00000000
[  +0.000015] cx23885 0000:07:00.0: cx23885_reg_dump2: 001302c0: 00000000 00000000 00000000 00000000
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 001302d0: 00000000 00000000 00000000 00000000
[  +0.000014] cx23885 0000:07:00.0: cx23885_reg_dump2: 001302e0: 00000000 00000000 00000000 00000000
[  +0.000013] cx23885 0000:07:00.0: cx23885_reg_dump2: 001302f0: 00000000 00000000 00000000 00000000

From amd-965 system
[Apr 3 14:51] cx23885 0000:02:00.0: cx23885_reg_dump2: 00040000: 00000020 00000000 00000000 00000000
[  +0.000014] cx23885 0000:02:00.0: cx23885_reg_dump2: 00040010: 00001f06 01bf8000 00000000 00000000
[  +0.000010] cx23885 0000:02:00.0: cx23885_reg_dump2: 00040020: 00000000 00000000 00000000 00000000
[  +0.000010] cx23885 0000:02:00.0: cx23885_reg_dump2: 00040030: 00000000 00000000 00000000 00000000
[  +0.000010] cx23885 0000:02:00.0: cx23885_reg_dump2: 00040040: 00001111 00000000 00000000 00000000
[  +0.000010] cx23885 0000:02:00.0: cx23885_reg_dump2: 00040050: 00000000 00000000 00000000 00000000
[  +0.000010] cx23885 0000:02:00.0: cx23885_reg_dump2: 00040060: 00000000 00000000 00000000 00000000
[  +0.000011] cx23885 0000:02:00.0: cx23885_reg_dump2: 00040070: 00000000 00000000 00000000 00000000
[  +0.000010] cx23885 0000:02:00.0: cx23885_reg_dump2: 00040080: 00000000 00000000 00000000 00000000
[  +0.000010] cx23885 0000:02:00.0: cx23885_reg_dump2: 00040090: 00000000 00000000 00000000 00000000
[... cut ...]
[  +0.000009] cx23885 0000:02:00.0: cx23885_reg_dump2: 001302b0: 00000000 00000000 00000000 00000000
[  +0.000010] cx23885 0000:02:00.0: cx23885_reg_dump2: 001302c0: 00000000 00000000 00000000 00000000
[  +0.000010] cx23885 0000:02:00.0: cx23885_reg_dump2: 001302d0: 00000000 00000000 00000000 00000000
[  +0.000010] cx23885 0000:02:00.0: cx23885_reg_dump2: 001302e0: 00000000 00000000 00000000 00000000
[  +0.000010] cx23885 0000:02:00.0: cx23885_reg_dump2: 001302f0: 00000000 00000000 00000000 00000000

b-rad-NDi, 2018-04-03:
rebuilding, didn't realize that reg_dump2 function as at level 0 entirely.

b-rad-NDi, 2018-04-03:
Build 201804030931 has reg_dump2 at debug level 1, it should not display unless enabled.

MikeB2013, 2018-04-03:
Build 201804030931 installed on both systems, reg_dump2 messages gone! I deleted cx23885.conf
Will leave running and report any issues.

Mike

b-rad-NDi, 2018-04-03:
@tomsoutherland : Have you tried any of these builds to see if they fix your issue?

MikeB2013, 2018-04-04:
@b-rad-NDi 

Build 201804030931 has been running successfully for 24 hours on both my systems with application mythtv.

My amd-965 system showed  2 instances of clear_bridge_error,  tuner (TBS-6981 DVB-S/S2) continues to work. 

[Apr 4 00:36] cx23885 0000:02:00.0: cx23885_clear_bridge_error() dma in progress detected 0x00000001 0x00000001, clearing
[Apr 4 07:39] cx23885 0000:02:00.0: cx23885_clear_bridge_error() dma in progress detected 0x00000001 0x00000001, clearing

On my ryzen system nothing reported except the firmware downloading, which would be nice to reduce.
 I can get rid of many of these messages  by adjusting my mythtv application, for the testing I had it set to run EIT scans on each of the 4 tuners on the QuadHD card, normal running would be to do this for 1 tuner.

Attached are dmesg outputs for reference.
[dmesg-amd-965-0931.txt.zip](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1876559/dmesg-amd-965-0931.txt.zip)
[dmesg-ryzen-0931.txt.zip](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1876560/dmesg-ryzen-0931.txt.zip)

Mike

b-rad-NDi, 2018-04-04:
Great to hear that with debug off completley the patch still is working. Hopefully the systems continue to perform the same.

MikeB2013, 2018-04-05:
@b-rad-NDi 
Here is an update.
On my amd-965 system (TBS-6981 DVB-S/S2 tuner) I had a couple more clear_bridge_errors, the last one was followed some minutes later by a risc error, TBS tuners still working.
[Apr 5 09:51] cx23885 0000:02:00.0: cx23885_clear_bridge_error() dma in progress detected 0x00000001 0x00000001, clearing
[Apr 5 10:16] cx23885: cx23885[0]: mpeg risc op code error
[  +0.000009] cx23885: cx23885[0]: TS2 C - dma channel status dump
[  +0.000004] cx23885: cx23885[0]:   cmds: init risc lo   : 0xcbb35000
[  +0.000004] cx23885: cx23885[0]:   cmds: init risc hi   : 0x00000000
[  +0.000004] cx23885: cx23885[0]:   cmds: cdt base       : 0x000105e0
[  +0.000003] cx23885: cx23885[0]:   cmds: cdt size       : 0x0000000a
[  +0.000004] cx23885: cx23885[0]:   cmds: iq base        : 0x00010440
[  +0.000003] cx23885: cx23885[0]:   cmds: iq size        : 0x00000010
[  +0.000004] cx23885: cx23885[0]:   cmds: risc pc lo     : 0xcbb35004
[  +0.000003] cx23885: cx23885[0]:   cmds: risc pc hi     : 0x00000000
[  +0.000003] cx23885: cx23885[0]:   cmds: iq wr ptr      : 0x00004111
[  +0.000004] cx23885: cx23885[0]:   cmds: iq rd ptr      : 0x00004110
[  +0.000003] cx23885: cx23885[0]:   cmds: cdt current    : 0x00010618
[  +0.000004] cx23885: cx23885[0]:   cmds: pci target lo  : 0x0c6491a0
[  +0.000003] cx23885: cx23885[0]:   cmds: pci target hi  : 0x00000000
[  +0.000004] cx23885: cx23885[0]:   cmds: line / byte    : 0x01860000
[  +0.000004] cx23885: cx23885[0]:   risc0: 
[  +0.000002] 0x1c0002f0 [ write
[  +0.000001]  sol eol count=752 ]
[  +0.000004] cx23885: cx23885[0]:   risc1: 
[  +0.000001] 0x0c6491a0 [ INVALID
[  +0.000001]  sol eol 22 21 18 resync 12 count=416 ]
[  +0.000006] cx23885: cx23885[0]:   risc2: 
[  +0.000001] 0x00000000 [ INVALID
[  +0.000002]  count=0 ]
[  +0.000002] cx23885: cx23885[0]:   risc3: 
[  +0.000001] 0x1c0002f0 [ write
[  +0.000002]  sol eol count=752 ]
[  +0.000003] cx23885: cx23885[0]:   (0x00010440) iq 0: 
[  +0.000001] 0x3b3a2e8c [ INVALID
[  +0.000001]  sol irq2 irq1 21 20 19 cnt1 13 count=3724 ]
[  +0.000007] cx23885: cx23885[0]:   (0x00010444) iq 1: 
[  +0.000001] 0x00000000 [ INVALID
[  +0.000002]  count=0 ]
[  +0.000003] cx23885: cx23885[0]:   (0x00010448) iq 2: 
[  +0.000001] 0x1c0002f0 [ write
[  +0.000001]  sol eol count=752 ]
[  +0.000003] cx23885: cx23885[0]:   iq 3: 0x0c649490 [ arg #1 ]
[  +0.000004] cx23885: cx23885[0]:   iq 4: 0x00000000 [ arg #2 ]
[  +0.000003] cx23885: cx23885[0]:   (0x00010454) iq 5: 
[  +0.000001] 0x1c0002f0 [ write
[  +0.000001]  sol eol count=752 ]
[  +0.000004] cx23885: cx23885[0]:   iq 6: 0x0c649780 [ arg #1 ]
[  +0.000004] cx23885: cx23885[0]:   iq 7: 0x00000000 [ arg #2 ]
[  +0.000003] cx23885: cx23885[0]:   (0x00010460) iq 8: 
[  +0.000001] 0x1c0002f0 [ write
[  +0.000001]  sol eol count=752 ]
[  +0.000004] cx23885: cx23885[0]:   iq 9: 0x0c649a70 [ arg #1 ]
[  +0.000003] cx23885: cx23885[0]:   iq a: 0x00000000 [ arg #2 ]
[  +0.000003] cx23885: cx23885[0]:   (0x0001046c) iq b: 
[  +0.000001] 0x00000000 [ INVALID
[  +0.000001]  count=0 ]
[  +0.000003] cx23885: cx23885[0]:   (0x00010470) iq c: 
[  +0.000001] 0x1c0002f0 [ write
[  +0.000001]  sol eol count=752 ]
[  +0.000004] cx23885: cx23885[0]:   iq d: 0x0c648eb0 [ arg #1 ]
[  +0.000003] cx23885: cx23885[0]:   iq e: 0x00000000 [ arg #2 ]
[  +0.000004] cx23885: cx23885[0]:   (0x0001047c) iq f: 
[  +0.000001] 0x1c0002f0 [ write
[  +0.000001]  sol eol count=752 ]
[  +0.000003] cx23885: cx23885[0]:   iq 10: 0xce7c88e7 [ arg #1 ]
[  +0.000004] cx23885: cx23885[0]:   iq 11: 0x9cf32f4f [ arg #2 ]
[  +0.000002] cx23885: cx23885[0]: fifo: 0x00006000 -> 0x7000
[  +0.000001] cx23885: cx23885[0]: ctrl: 0x00010440 -> 0x104a0
[  +0.000003] cx23885: cx23885[0]:   ptr1_reg: 0x000068f0
[  +0.000003] cx23885: cx23885[0]:   ptr2_reg: 0x00010618
[  +0.000004] cx23885: cx23885[0]:   cnt1_reg: 0x00000002
[  +0.000003] cx23885: cx23885[0]:   cnt2_reg: 0x00000003

Full dmesg output attached
[dmesg-amd-965-0931-20180405.txt.zip](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1880167/dmesg-amd-965-0931-20180405.txt.zip)

My ryzen system has not shown any errors, just the firmware downloading messages.

Mike

b-rad-NDi, 2018-04-29:
Thanks for the update @MikeB2013 : I think I have had enough confirmations from various sources to push a v1 set for this upstream on Monday.

b-rad-NDi, 2018-05-04:
I believe this particular issue is solved. I have left my R3 1300 system streaming four programs on four different frequencies non-stop for over 48hr and have not experience the long term wakeup flood.

@MikeB2013 : The firmware loading messages for si2168 should be gone now as well. I am looking at how to elegantly address the superfluous si2157 output as well.

Latest packages can be found here:
https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0

@MikeB2013 : If you try these packages and find them do not fix your issue please re-open and I'll get back to this.

closed 2018-05-04 by b-rad-NDi