b-rad-NDi/Ubuntu-media-tree-kernel-builder #42: Use dev_info() instead dev_err()

comio, 2018-04-03:
Hi @b-rad-NDi ,

Why do you not use  dev_info() call to show information messages instead dev_err() call?

```
em28xx-cards.c:3798:            dev_err(&intf->dev, "DVB interface %i found:%s%s\n",
```
to
```
em28xx-cards.c:3798:            dev_info(&intf->dev, "DVB interface %i found:%s%s\n",
```

b-rad-NDi, 2018-04-03:
Not my doing ;) I tend to only use dev_err and dev_warn when it is an actual error or warning. I'll submit a patch about this issue though, because it does seem strange.

comio, 2018-04-03:
Perfect. Thanks in advance.

luigi

b-rad-NDi, 2018-05-09:
Looks like someone beat me to this one.

b-rad-NDi, 2018-05-09:
Actually this was already upstream when you filed the ticket ;)

closed 2018-05-09 by b-rad-NDi