b-rad-NDi/Ubuntu-media-tree-kernel-builder #43: Ubuntu 18.04 support

ocha, 2018-04-24:
Hi. I was wondering if there are plans to support latest LTS release of Ubuntu in your PPA? 

Thanks.

b-rad-NDi, 2018-04-26:
This is queued. Trying to work through 'day job duties' and then I'll produce new builds.

ocha, 2018-04-26:
Thank you @b-rad-NDi ! I appreciate work and support you are putting in this

b-rad-NDi, 2018-05-03:
Someone upstream broke the second tuner on ATSC DualHD models, a patch for which has been submitted upstream.

The Bionic packages have been mostly qualified now and besides the DualHD issue, appear to be working completely.

ocha, 2018-05-03:
@b-rad-NDi I have DualHD model. Which version has the path for second tuner fix. I can install it and test it out

b-rad-NDi, 2018-05-03:
I will start all new builds soon. If you tell me which flavour you prefer I can prioritize it.

ocha, 2018-05-03:
im on Ubuntu Bionic (18.04). I added PPA but havent installed new kernel since PPA didnt have bionic.. I can install kernel manually and then switch to PPA once available. It's not a pressing issue so I can wait if needed.

b-rad-NDi, 2018-05-04:
Fully patched packages are up on the dropbox. These have ATSC DualHD fixed.

If you're patient, then I'll be pushing builds to the ppa tomorrow morning.

ocha, 2018-05-04:
Sounds good. Thank you @b-rad-NDi

MikeB2013, 2018-05-06:
@b-rad-NDi 
Just for info.

The 4.15.0-20201805041217 build from your dropbox is working well on Xubuntu 18.04 (AMD Ryzen 3 1200 Quad-Core Processor on Gigabyte GA-AB350 Gaming 3 Motherboard, BIOS F22 03/15/2018) with my QuadHD (885) and MythTV Version : v30-Pre-599-gba4c52b

Mike

b-rad-NDi, 2018-05-06:
Thanks for the update @MikeB2013. I pushed new builds to Launchpad for Xenail, Artful, and Bionic. If they are all successful I will push new virtual packages up to trigger updates.

smattord, 2018-05-06:
I have funky results on Xubuntu 18.04. In plex the device initially showed up after driver installation but scanning channels became hung and then the device was no longer visible.

b-rad-NDi, 2018-05-06:
@smattord : if you find problem, then please reproduce, file a new issue, and include an appropriate amount of dmesg. The Bionic packages have been fairly well qualified by us now and no issues were found.

b-rad-NDi, 2018-05-06:
Bionic virtual packages pushed to Launchpad. 18.04 support is complete. If you find any issues at all, then please file new ticket. Closing this now.

closed 2018-05-06 by b-rad-NDi