b-rad-NDi/Ubuntu-media-tree-kernel-builder #44: Different ATSC frequencies on adjacent device #s locks up quickly

ejlion14, 2018-04-26:
Forgive me if I come off as a bit of a noob here.  I'm experienced in IT, intermediate in Linux at best, and trying something new.

I'm running Ubuntu 16.04 and tvheadend with a WinTV quadHD (4 x CX23885) PCI-E card, using it as a back-end for OTA channels.  I have encountered a very interesting issue that's taken some time to troubleshoot.

What I've concluded is as follows:

Of the four adapters that appear on the back-end, I can watch as many channels as I want on the same frequency... e.g. 3-1, 3-2, 3-3, etc., consuming one adapter, indefinitely, without incident.

Similarly, I can watch two channels on different frequencies on Tuner instances #0 and #3 indefinitely, or even multiple channels on each of two frequencies, without incident.  For example, one PC watching 6-1 and 6-2, another PC watching 10-1 and 10-2.  I believe this is the same if I use adapter instances #1 and #2.

However, if I attempt to watch two different frequencies on #0 and #1 adapters, or #2 and #3 adapters, it works for a short time, then freezes. This happens whether I'm using Kodi, downloading the m3u into VNC, or using the embedded player from the EPG screen.  It happens regardless of HD or not.  

I'm wondering if there's some relationship between each pair of adapters.  It's the type of problem that's very difficult to find someone else experiencing.  Further, if there IS a relationship, I'm not sure how to prevent this conflict from occurring.

Any help/advice would be greatly appreciated.  I can acquire any information from the server if pointed in the right direction.

b-rad-NDi, 2018-04-26:
Sounds strange. I can't say I have ever heard of this happening. Do you see any errors/warnings/etc in dmesg logs? I've run tests with each of the four tuners on a different frequency without issue. What sort of system is this?

ejlion14, 2018-04-26:
Hello!  Thank you for replying, I truly appreciate the access to your expertise.

It's not always clear to me what's worrisome in dmesg.  How would I best construct an experiment to produce a useful output?  I do see "wakeups," which I've seen referenced elsewhere.

The system is a VMware VM with the PCI-E card passed through.  It's on a Dell PowerEdge T320.

Also important to note - I can temporarily resolve the issue without restarting the server or service.  At worst, I have to reset a connection from tvheadend.  The server does not appear to be degraded in any way.  When I watch channels in a configuration that I know to be stable, it's gone 96+ hours without incident.  When I attempt to use different frequencies on 0+1 or 2+3 simultaneously, it lasts 40 minutes maximum, but the time is variable.

b-rad-NDi, 2018-04-26:
The fact you're using VMWare is very informative. You might be encountering issues in their pass thru system. Sometimes 'pass thru' is not completely identical to normal usage.

To get dmesg logs do something like this:
```dmesg -T | tail```
This shows you timestamp of your 'start point'.

Then after encountering an error, increase the amount of lines shown and paste all that happened after the starting point here.
```dmesg -T | tail -n 150```

ejlion14, 2018-04-26:
dmesg currently looks like this (below), during smooth operation of two frequencies, one on adapter #1 and the other on adapter #2.  #0 and #3 unused.  Literally the entire queue is this, maybe ten pages of it.  Since this is happening during normal operation, I'm inclined to believe it's harmless, but perhaps not?

I saw someone with a similar problem elsewhere, not exactly the same, who said that if they set debug to 7, their issue was resolved, granted, with enormous logs.  I'm not sure if/how that's helpful here, I'm sharing it in the interest of giving you all the information available to me.  I'm a bit discouraged that you haven't seen this.  Is there a "special" or arbitrary relationship between each pair of adapters?

Thank you again.

[ 5727.591557] cx23885 0000:13:00.0: cx23885_wakeup: [ffff88003ba1d800/0] wakeup
 reg=57953 buf=385633
[ 5727.599660] cx23885 0000:03:00.0: cx23885_wakeup: [ffff88003b1d9800/5] wakeup
 reg=64006 buf=391686
[ 5727.601490] cx23885 0000:13:00.0: cx23885_wakeup: [ffff88003ba1c400/1] wakeup
 reg=57954 buf=385634
[ 5727.609572] cx23885 0000:03:00.0: cx23885_wakeup: [ffff880037292800/6] wakeup
 reg=64007 buf=391687
[ 5727.611423] cx23885 0000:13:00.0: cx23885_wakeup: [ffff88003ba1c000/2] wakeup
 reg=57955 buf=385635
[ 5727.619485] cx23885 0000:03:00.0: cx23885_wakeup: [ffff88003da0ec00/7] wakeup
 reg=64008 buf=391688
[ 5727.621355] cx23885 0000:13:00.0: cx23885_wakeup: [ffff88003ba1c800/3] wakeup
 reg=57956 buf=385636
[ 5727.629396] cx23885 0000:03:00.0: cx23885_wakeup: [ffff88003da0d400/8] wakeup
 reg=64009 buf=391689
[ 5727.631288] cx23885 0000:13:00.0: cx23885_wakeup: [ffff88003ba1e000/4] wakeup
 reg=57957 buf=385637
[ 5727.639308] cx23885 0000:03:00.0: cx23885_wakeup: [ffff88003da0f400/9] wakeup
 reg=64010 buf=391690
[ 5727.641220] cx23885 0000:13:00.0: cx23885_wakeup: [ffff88003ba1e400/5] wakeup
 reg=57958 buf=385638
[ 5727.649220] cx23885 0000:03:00.0: cx23885_wakeup: [ffff88003da0cc00/10] wakeu
p reg=64011 buf=391691
[ 5727.651153] cx23885 0000:13:00.0: cx23885_wakeup: [ffff88003ba1cc00/6] wakeup
 reg=57959 buf=385639
[ 5727.659133] cx23885 0000:03:00.0: cx23885_wakeup: [ffff880037125000/11] wakeu
p reg=64012 buf=391692
[ 5727.661085] cx23885 0000:13:00.0: cx23885_wakeup: [ffff88003cb6f400/7] wakeup
 reg=57960 buf=385640
[ 5727.669044] cx23885 0000:03:00.0: cx23885_wakeup: [ffff88003a847800/12] wakeu
p reg=64013 buf=391693
[ 5727.671020] cx23885 0000:13:00.0: cx23885_wakeup: [ffff88003b3c9000/8] wakeup
 reg=57961 buf=385641
[ 5727.678955] cx23885 0000:03:00.0: cx23885_wakeup: [ffff88003a844c00/13] wakeu
p reg=64014 buf=391694
[ 5727.680950] cx23885 0000:13:00.0: cx23885_wakeup: [ffff88003b3cb000/9] wakeup
 reg=57962 buf=385642
[ 5727.688869] cx23885 0000:03:00.0: cx23885_wakeup: [ffff88003a845400/14] wakeu
p reg=64015 buf=391695
[ 5727.690882] cx23885 0000:13:00.0: cx23885_wakeup: [ffff88003b3c8000/10] wakeu
p reg=57963 buf=385643
[ 5727.698779] cx23885 0000:03:00.0: cx23885_wakeup: [ffff88003a845000/15] wakeu
p reg=64016 buf=391696
[ 5727.700815] cx23885 0000:13:00.0: cx23885_wakeup: [ffff88003b3c8400/11] wakeu
p reg=57964 buf=385644

ejlion14, 2018-04-26:
I just saw your dmesg instructions, will do it now and will produce the error.

ejlion14, 2018-04-26:
Hello.  My output right now is:

Linux dvr 4.4.0-116201803011620-generic #0+mediatree+hauppauge-Ubuntu SMP Thu Mar 1 22:40:04 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

I am currently waiting for my configuration to fail so I can capture dmesg.  Should I abandon that in favor of installing those files?  

Thank you again.

b-rad-NDi, 2018-04-26:
The issue being displayed in your previous logs is sorted out by these packages. It might be worth letting your current test go for a little while, but I believe the messages you get will be the same as above but with values like 65535, 65536 / etc. It is very likely your problems will go away after installing the set on my dropbox.

ejlion14, 2018-04-26:
I've been good for 31 minutes thus far - too soon to be sure, but promising.

I don't want to take up to much more of your time, but could you please teach me what exactly was going wrong, and what about those installations you expect will resolve it?

b-rad-NDi, 2018-04-26:
A simple explanation is mismatched field sizes. The register that counts handled buffers is only 16bit, but the subsystem counter is 32bit. When the 16bit value rolls over the two values are no longer equal, since the 32bit value obviously does not rollover at the same value. When this happens the handler thinks there are missing interrupts to handle and thus spins through the loop five times trying to handle interrupts. The big problem with that is the wakeup function then does five printks, which are very expensive since they happen inside of a spin lock in this case. One thing leads to another and the system goes fubar until the device is closed and re-opened. Comprende? If you're further interested you can see the original investigation here: #41

b-rad-NDi, 2018-04-26:
I sort out the issue, with some mod magic, to convert the 32bit value into it's equivalent 16bit value.

ejlion14, 2018-04-26:
It appears I spoke too soon.  The failure re-occurred.  I'm attaching a Kodi screenshot and my dmesg.

[image]

[dmesg.txt](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1952773/dmesg.txt)

All that said, thank you for the explanation.  It's still worthwhile reading.

b-rad-NDi, 2018-04-26:
Odd...not a single blurb of any sort in dmesg. That doesn't make this very easy to debug. I will set up my test pc to do a long time test on all four tuners with four different frequencies.

ejlion14, 2018-04-26:
In case this is meaningful at all, bandwidth and in/out all drop to 0 on all streams at the same time, and all exhibit that behavior.  If I were running VNC using a downloaded m4u, it would stop.

[image]

[image]

[image]

Thank you for offering to test.  Do you have this card?  Again, it only happens if I use 0 and 1 together, or 2 and 3 together.  

I did see one thread elsewhere that said the card behaved differently in a PCI x8 slot, even though the card itself is a x1.  Is there any plausibility in that?  In your opinion, is it worth trying?  It's not trivial to do, but I'm willing.

ejlion14, 2018-04-26:
Are there perhaps application logs that would be useful?  Or perhaps a setting(s) I can change to generate more comprehensive logs?

b-rad-NDi, 2018-04-26:
I have the card, I'm lead software engineer ;)

If I were you I'd investigate possible vmware pci passthru options as well. Also various kernel command line options. If my native usage doesn't exhibit your issue, then it would point to some vmware peculiarity. If that is the case it is very hard to debug on my end.

b-rad-NDi, 2018-04-26:
In theory, any errors or warnings encountered should output to dmesg. The fact you see none means you probably wouldn't see anything more informative if you dialed up debug. That pcie bridge driver produces a LOT of spew if you enable debug.

ejlion14, 2018-04-27:
Ha!  It took a few hours, but I finally 
[dmesg_apr27.txt](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/1956996/dmesg_apr27.txt)
said something comically stupid - of course you have access to the card!

I moved the card to a different slot, disabled hardware acceleration on the VM, and raised the debug to 6.  The behavior didn't change, but it produced the attached.  It is useful?  If I crack the host open again, I'll likely install a couple other Hauppauge Tuners for some more troubleshooting variables.  I've considered perhaps installing Windows on the VM (e.g. NextPVR), but the efficiency of Linux is very attractive here.  I considered MythTV, but I'm afraid whatever underlying issue is causing this will affect that, too.  Of course, if you have any thoughts, I'd welcome them.

ejlion14, 2018-04-28:
Something new happened.  I was watching channels 6-1 and 17-1 (WPVI and WPHL, respectively).  For the first time ever, one froze, yet the other keeps running.  

In dmesg, the first "count" went to 0x0 and stayed there.  Until the one feed froze, both had incremented:
[ 2879.378948] cx23885 0000:13:00.0: cx23885_irq: ts1_status: 0x00000000  ts1_mask: 0x00001111 count: 0x748
[ 2879.378950] cx23885 0000:13:00.0: cx23885_irq: ts2_status: 0x00000001  ts2_mask: 0x00001111 count: 0xabc

After the freeze of one feed:
[ 4533.989902] cx23885 0000:03:00.0: cx23885_irq: vida_status: 0x00000000 vida_mask: 0x00000000 count: 0x0
[ 4533.989905] cx23885 0000:03:00.0: cx23885_irq: audint_status: 0x00000000 audint_mask: 0x00000000 count: 0x0
[ 4533.989907] cx23885 0000:03:00.0: cx23885_irq: ts1_status: 0x00000000  ts1_mask: 0x00000000 count: 0x0
[ 4533.989909] cx23885 0000:03:00.0: cx23885_irq: ts2_status: 0x00000001  ts2_mask: 0x00001111 count: 0x21fc
[ 4533.989911] cx23885 0000:03:00.0: cx23885_irq:  (PCI_MSK_VID_C     0x00000004)
[ 4533.989913] cx23885 0000:03:00.0: cx23885_irq:  (PCI_MSK_IR        0x10000000)
[ 4533.989915] cx23885 0000:03:00.0: cx23885_irq_ts:  (RISCI1            0x00000001)
[ 4533.989921] cx23885 0000:03:00.0: cx23885_wakeup: [ffff880037827c00/27] wakeup reg=8700 buf=74236
[ 4533.989975] cx23885 0000:03:00.0: cx23885_buf_prepare: cx23885_buf_prepare: ffff880037827c00
[ 4533.989995] cx23885 0000:03:00.0: cx23885_buf_queue: [ffff880037827c00/27] cx23885_buf_queue - append to active

I understand we may be straying outside the area of responsibility of the driver, but if you can infer anything from this that might help me chase things down, I would be grateful.

Thanks again for your time.

b-rad-NDi, 2018-04-28:
Also, have you looked into BIOS update?

ejlion14, 2018-04-28:
I've considered KVM specifically, but I'm not sure I want to make such a wholesale change to solve one relatively small issue.  I would likely sooner just use a separate workstation, and may soon use one for testing as a control group.  Although a real control group would be to just install Ubuntu and tvheadend directly on the PowerEdge T320, without ESXi.  As I boot from USB, I could do that if I committed to the test.

I did increase the debug to 6 after reading another thread here, where someone was able to stop errors by using debug level 6 or 7.  It did not help in my case.

Here's a question: it appears to see the quadhd card as two separate devices: one a "WinTV-quadhd" and another a ""CX23887/8 PCIe Broadcast Audio and Video Decoder..."  If I attach either one, I get two tuners.  Does this sound familiar/meaningful at all?

ejlion14, 2018-04-28:
I actually looked at the BIOS earlier today when reading another thread - the one where the user saw relief in a PCIe x16 slot and eventually unearthed a BIOS issue.  Apparently I'm one version behind.  I'm not TOO optimistic about the BIOS, as it appears to really just mitigate Meltdown and Spectre.

b-rad-NDi, 2018-05-04:
FYI I have had my QuadHD streaming four different programs off four different frequencies for 48hr now on my Ryzen3 test system. Not a single slip or error or warning in dmesg.

b-rad-NDi, 2018-05-04:
The builds located here are contain all the latest patches:
https://www.dropbox.com/sh/2waefdgue62fcha/AABixnkFjZObrYcJ7dR0-cnBa?dl=0
Just install the four 4.4.0 packages and give it a shot. Not really optimistic, since I believe this is a vmware issue, but you never know.

closed 2018-12-13 by b-rad-NDi