b-rad-NDi/Ubuntu-media-tree-kernel-builder #45: hwe packages

b-rad-NDi, 2018-05-06:
TODO:

I intend on copying the 'newest' kernel packages from future Ubuntus back to the previous LTS, up to the next LTS.

So, Xenial HWE would have gotten Yakkety 4.8, then Zesty 4.10, then Artful 4.13, then freeze at and tail Bionic release kernel 4.15.

MikeB2013, 2018-07-21:
Brad,

Just in case you are not yet aware, Ubuntu 16.04 LTS (from 16.04.4 install) is now using 4.15 series kernel, was 4.13 series. 

Also Ubuntu Artful has now gone EOL.

Mike

b-rad-NDi, 2018-07-30:
For you consideration:
https://launchpad.net/~b-rad/+archive/ubuntu/test-test-hw-staging

I'm doing HWE tests now. The above ppa is the initial testing area to ensure no collisions.

The ppa is strictly temporary, but is open for testing. If you add the ppa then this should install all kernel packages:
```
sudo apt-get install linux-hwe-image-mediatree linux-hwe-headers-mediatree
```

Once everything has been qualified the linux-hwe packages will be distributed via the main ppa.

Please comment here if you find any issues.

b-rad-NDi, 2018-07-30:
The work just entailed simple branch support. Now the hwe branch is selected and compiled, whatever version it may be.

MikeB2013, 2018-07-31:
Brad,
Just for information. 
I have tried the ppa on a fully updated test xubuntu 16.04 system running mythtv 29 with a network tuner (vbox 3442 dual DVB-T/T2)  with no problems, no collisions.

b-rad-NDi, 2018-07-31:
Thanks for the update, good to hear. I just fiddled a bit and extended the abi tag to include branch name, so I think after a bit more testing these hwe packages will be available in the main ppa.

b-rad-NDi, 2018-09-05:
HWE packages are live in the main ppa. Closing this for now, if issues are found with the HWE packages please file a new ticket.

closed 2018-09-05 by b-rad-NDi