b-rad-NDi/Ubuntu-media-tree-kernel-builder #46: dkms-mediatree

b-rad-NDi, 2018-05-06:
TODO:

Implement this as a configurable dkms package.

jpunzel, 2018-06-05:
+1 for this. This is a great project, but I'm a little wary of relying on a third party to provide an entire kernel image for my machine. DKMS seems like a much better way to do this.

b-rad-NDi, 2018-07-03:
"Technically" you're putting your faith in Launchpad to produce secure/auditable builds, but I get you. Have had way too much on my plate recently, but this is queued up for August.

PartialVolume, 2018-08-10:
Looking forward to the DKMS release, like @jpunzel I have my reservations about an entire kernel image from a third party when DKMS was designed to do just this sort of thing. I was wondering whether there was any timeline on the DKMS release ? If it's fairly soon I might stick with my old Hauppaugue MyTV.t stick and delay installing the WinTV-dualHD for the time being. My kernel is now at 4.15.0-30

b-rad-NDi, 2018-08-30:
I will prioritize this. Much on my plate, but I'll try and work on this next week.

PartialVolume, 2018-08-30:
Thanks for putting your time into this, much appreciated.

walterav1984, 2018-10-10:
A example for dkms build whole mediatree for both ubuntu and debian, the forum link is german language, maybe this can help/hint building a hauppage dkms (debian ;-) package?

https://github.com/jasmin-j/media_build/tree/dkms/0006-14_Jul_2018
https://www.vdr-portal.de/forum/index.php?thread/131033-media-build-dkms/

Thanks for current dvb-c support of the WinTV dualhd and upstreaming most of your patches into mainline kernel.

techiegirl2day, 2019-02-03:
I want to put my hand up for a dkms mediatree as well.  I am not a fan of Ubuntu preferring plain Debian. Any update on when a dkms will become available?  I looked at the above links but was discouraged by the fact the build was over 1 yr old & seems to have been abandoned.  I would like to be able to keep a current running build like the ppa offered. Please give us debian users an alternative. Full disclosure I use a Systinit based distro.

techiegirl2day, 2020-06-28:
hi,

Just wondering what is the status of this Wishlist item.  Is this still a possibility or has it been shelved?

b-rad-NDi, 2020-07-01:
I have loads and loads of priorities that come first, but I'm starting to lean towards prioritizing this method. No timeline though.

PartialVolume, 2020-07-01:
Must admit I'm also looking forward to this. I'm so tempted to get involved but like @b-rad-NDi I'm up to my eyeballs in work and my other project that I spend a lot of time on [nwipe](https://github.com/martijnvanbrummelen/nwipe). So tempted though. I'm still using my old hauppauge USB stick as it always just works with KDE Neon (Ubuntu 18.04LTS), rather than my new Hauppauge HD stick. There's just not enough programmers for all the work that needs doing.

[image]