b-rad-NDi/Ubuntu-media-tree-kernel-builder #47: Errors

Saentist, 2018-05-21:
> [   61.523003] si2157 6-0060: found a 'Silicon Labs Si2157-A30'
[   61.545583] si2157 6-0060: firmware version: 3.0.5
[11596.597103] cx23885 0000:06:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x0009 address=0x00000000ffd7b000 flags=0x0000]
[13639.651330] cx23885 0000:05:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[14657.389306] cx23885 0000:05:00.0: dma in progress detected 0x00000001 0x00000001, clearing

> uname -a
Linux tvserv2 4.15.0-20201805041217-generic #0+mediatree+hauppauge SMP Fri May 4 12:19:11 CDT 2018 x86_64 x86_64 x86_64 GNU/Linux

>lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu Bionic Beaver (development branch)
Release:        18.04
Codename:       bionic

Is this error are fatal or ignorable?

b-rad-NDi, 2018-05-21:
Well, did it kill your signal or does everything still work? If all four tuners still work, then the error condition was recovered from and the fix worked as desired.

The message below is a long standing problem with AMD chipsets, dating back to 2013 or earlier:
```AMD-Vi: Event logged [IO_PAGE_FAULT ...```
It's an AMD issue, and apparently no one since the original reporting of the bug has been able to make it go away for affected drivers. I don't see it while using cx23885 digital television, but cx23885 analog would generate it immediately. The onboard ethernet I have on my Ryzen b350m motherboard generates this within minutes and essentially does not work at all, I have to blacklist the driver completely otherwise it messes up the PCIe bus royally. The same ethernet chip has been causing the same issue on AMD motherboards for over four years with supposed fixes dating all the way back to kernel early 3.x kernels.

I made it go away while working to get cx23885 analog fully working on Ryzen by adding this kernel command line parameter in /etc/default/grub :
```iommu=pt```

Saentist, 2018-05-21:
Cpu is 
>AMD A6-5400B APU with Radeon(tm) HD Graphics

on 
>HP Compaq Pro 6305 SFF/1850

don't see any problems, but who knows.

b-rad-NDi, 2018-05-21:
The fatal symptom of this is the affected tuner stops functioning. If you are all still functional, then you just hit an AMD burp and should be good. Bursts in PCIe traffic are what typically generate the AMD-Vi IO_PAGE_FAULT error, based reading loads of reports on this.

closed 2018-07-03 by b-rad-NDi