b-rad-NDi/Ubuntu-media-tree-kernel-builder #49: Unable to apply against Synology 4.4 kernel

th0ma7, 2018-06-24:
I've been trying to apply the patches from:
https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/tree/master/patches/mainline-extra/4.4.0

Onto the Synology 4.4 kernel as provided for my device (DS-918+) from the sources here:
https://sourceforge.net/projects/dsgpl/files/Synology%20NAS%20GPL%20Source/22259branch/apollolake-source/

Some go fine but other fails such as:
```
$ patch -p1 < ../patches/0001-si2168-Change-from-DVB-T-to-DVB-T-T2-autodetect.patch 
patching file drivers/media/dvb-frontends/si2168.c
Hunk #1 succeeded at 16 (offset -2 lines).
Hunk #2 FAILED at 155.
Hunk #3 succeeded at 139 (offset -27 lines).
Hunk #4 succeeded at 177 (offset -27 lines).
Hunk #5 succeeded at 252 (offset -82 lines).
Hunk #6 succeeded at 325 (offset -82 lines).
Hunk #7 FAILED at 456.
2 out of 7 hunks FAILED -- saving rejects to file drivers/media/dvb-frontends/si2168.c.rej
$ patch -p1 < ../patches/0002-cx23885-Set-Hauppauge-HVR1255-tuner-type-to-TDA18271.patch 
patching file drivers/media/pci/cx23885/cx23885-cards.c
Hunk #1 succeeded at 324 (offset -1 lines).
Hunk #2 succeeded at 352 (offset -1 lines).
```

Having a way to apply the media tree on top of the synology kernel would then allow creating updated modules and use tvheadend directly within the synology linux os.

Thnx in advance.

b-rad-NDi, 2018-07-03:
The patches in that directory are designed to be applied against a backported "tip" media tree, aka the latest kernel. The entire media tree is backported, then the series in mainline-extra are applied. Hence the path "mainline-extra", the patches only apply against mainline. So what you're trying to do will not work. You also cannot use the strategy that I'm using here, due to the fact Synology likes to customize their kernels and include assorted backports. This system is designed to apply and build cleanly against a fairly stock kernel.

With that said I have a project underway to support DVB devices on all sorts of NAS's. I have WD covered, QNAP and Synology are next. I will be on vacation until the end of the month though, so Synology support would not happen until then.

b-rad-NDi, 2018-07-03:
Closing this for now, but be on the lookout for a new repo addressing this topic around the end of the month.

closed 2018-07-03 by b-rad-NDi

th0ma7, 2018-07-03:
Thnx for the heads-up!  Will certainly look into it!  Cheers and have a great vacation!

th0ma7, 2018-09-06:
Out of curiosity, have you had a chance to look onto this new repo? Cheers!

b-rad-NDi, 2018-09-07:
Close. The initial stage of the project has been accepted and a stub website is created. I will be pushing up the first repo probably some time early next week.

th0ma7, 2018-10-21:
Sorry for asking again... Any news on this front?  Is there any way I can help test things out?  Cheers!

b-rad-NDi, 2018-11-01:
My TV driver package has been accepted by WD, I'll queue up Synology next since I have two of them.

I'll try to upload the code here soon, then it'll be testable by anyone.

th0ma7, 2018-12-18:
Any updates since last time?  I've been trying to use synology VM in the meantime but I guess I'm running into other qemu related issue with passthrough (#62, #60).
Having the package directly available through SynoCommunity would be really awesome.
Cheers and thnx in advance!