b-rad-NDi/Ubuntu-media-tree-kernel-builder #52: multible dvb sticks crash at startup

CvH, 2018-09-05:
4 x Hauppauge HVR950Q connected, driver seems to crash after 1 device is initialised (1 tuner works at Tvh)
dmesg http://ix.io/1lYc
problem appears at vanilla 4.18 and 4.18+your drivers

b-rad-NDi, 2018-09-05:
I believe I have at least a couple 950Q, unsure if I have four of them. I will try and repro. Does the issue require 4 tuners?

CvH, 2018-09-05:
I already requested a log where he attaches 1 stick after another, LIKELY already broken with 2 sticks

b-rad-NDi, 2018-09-05:
I can only find one :/ and at least on 4.15 there is no oopsing with attach/disconnect cycles. I'm building 4.19 mainline now to test it.

b-rad-NDi, 2018-09-06:
No failure on the 4.19-rc1 i had handy. You'll need to sort out if this requires two tuners or if one will trigger it.

CvH, 2018-09-06:
user didn't reply yet, I post as soon I have the info

b-rad-NDi, 2018-09-06:
Ok, so I've located the exact cause of the oops. I'll submit a patch to fix it. But that is just fixing the oops. The symptom that triggers the oops is the analog video setup failing.

```
[    7.454271] usb 1-2.3: Not enough bandwidth for new device state.
[    7.454302] usb 1-2.3: Not enough bandwidth for altsetting 5
[    7.454304] au0828: Failure setting usb interface0 to as5
[    7.454307] au0828: au0828_usb_probe() au0282_dev_register failed to register on V4L2
```
That is not good. Does this particular dongle fail on its own I wonder? If this dongle was set aside, does another verified 950Q work when plugged in second?

b-rad-NDi, 2018-09-06:
gist:cddc90a138f10de3ba2aec575c648fa8

CvH, 2018-09-08:
http://ix.io/1mbh
booted, attach one stick after the other -> looks working
after reboot same crash as before

b-rad-NDi, 2018-09-19:
Indeed appears working in that log. If a reboot, without detaching the dongles, triggers the oops it sounds like a reset state issue. On fresh insertion it is fully reset, but when it thinks it's warm, it really needs a reset as well.

Did you try the patch? That will prevent the oops, but obviously nothing about any possible reset issue.

CvH, 2018-09-19:
contacted user that he tries the patch, no response yet

CvH, 2018-09-20:
with patch and still just one device shows up
http://ix.io/1n4T

b-rad-NDi, 2018-09-20:
Ooops is fixed, but the second dongle still fails initialization for some reason.
```
[    5.799868] usb 1-2.3: Not enough bandwidth for new device state.
[    5.799893] usb 1-2.3: Not enough bandwidth for altsetting 5
[    5.799894] au0828: Failure setting usb interface0 to as5
[    5.799896] au0828: au0828_usb_probe() au0282_dev_register failed to register on V4L2
```

b-rad-NDi, 2018-09-20:
Here is a small patch to output the USB error code.
https://gist.github.com/b-rad-NDi/aa5a877c760149c2fb24cd0e850fd7d7

Also, while searching around for similar issues I've come across something that indicates there is a long standing issue with some classes of devices, when using powered hubs with 'single transaction translator'. I'm not sure if it is related to this user, but a symptom is the inability to set enough bandwidth for an interface. The stated solution is to use a multi TT hub.
Is this user using a hub? Are they putting both dongles in the same hub? Have they tried not using a hub and/or splitting devices across host interfaces?

b-rad-NDi, 2018-09-20:
I asked a tester in the office to do some tests to try to repro this. I've instructed them to not use a hub at first, then try one if everything is fine.

CvH, 2018-09-22:
just quoting the reply
```
http://ix.io/1ncw

I am using a super special powered USB hub. I'd like to continue using a hub as I won't have access to all the tuners without it. Also, since the tuners work when plugged into the hub after LE boots, I'd assume (perhaps incorrectly) that the problem isn't one of the hub.

Anyway, here's the dmesg | paste with tuners plugged directly into the Chromebox:

http://ix.io/1ncf
So the tuners are showing. Humbug. I'm not sure there's a way to get the driver working with the hub, as that's my preference.

```

b-rad-NDi, 2018-09-28:
Not all hubs are made the same. I see in the logs a USB3 hub is being used. Perhaps try a different powered hub. If the two dongles always work when connected directly to the pc, even after reboot, but do not through the hub, then it is a problem with the hub and/or hub drivers in the kernel.

b-rad-NDi, 2018-12-13:
This oops is fixed upstream now. The core cause being usb hub issues, still not handled. Closing this issue, feel free to start a new one soley for the usb mode setting issue if desired.

closed 2018-12-13 by b-rad-NDi