b-rad-NDi/Ubuntu-media-tree-kernel-builder #53: Hauppauge DVB-S2 461e not supported

CvH, 2018-09-15:
http://ix.io/1mGG
This is an recently bought stick that doesn't appear at the dmesg at all, with 4.18 the same happens.

As the stick has Montage M88DS3103 and every vendor out there has updated it to Montage M88DS3103**b** for 2018+ production it is likely not supported anymore.

So likely every Hauppauge product with M88DS3103 that is produced 2018+ is not supported anymore at the kernel.

b-rad-NDi, 2018-09-15:
Unfortuately this stick does pop up...
```
[   11.656937] usbcore: registered new interface driver cx231xx
```
There was a production issue and batches of this stick have the same vid:pid as a 521e. This is on my very full radar, but seeing as it's duplicate vid:pid does not use the same bridge, it's a bit of a clusterf*ck. Trying to handle this is on my todo.

CvH, 2018-09-15:
Tx for replying, thought it is due the 3103b problem that is sadly making all usb DVB-S2 sticks from 2018+ unusable at linux. Wouldn't be surprised if this was another stick for the list.

CvH, 2018-09-15:
Btw is it possible to manually change it at the meanwhile (force usage of some driver)?
Afaik I did it long time ago, but can't remember what was needed.

b-rad-NDi, 2018-09-19:
You would most likely have to recompile the kernel to remove the vid:pid from the cx231xx driver first. Then you can add the new vid:pid to the proper driver via sysfs interface.

I'm checking out what the 3103b requires for support.

CvH, 2018-09-19:
there is some ugly driver from a "vendor" (likely from geniatech) that should make it work but nobody could confirm it working https://gist.github.com/CvH/afc38e6d9edf23aa4f1d9837f9860dcd yet

inline firmware and other great stuff but maybe helpful at some point - I extracted that patch from an prepatched media_tree with media_build they shipped (reverted media_build patches and compared against the used media_tree).

b-rad-NDi, 2018-09-19:
Ugly'ish and huge indeed. I'll see what the main office says about just the 3103b part. This looks like it contains more than that.

b-rad-NDi, 2018-09-19:
Do you have a contact email for whoever supplied the driver package? I'd like to ask permission to tear it apart and see about getting it upstreamed.

CvH, 2018-09-19:
I have no real idea (I can only guess geniatech), but I got an email address to someone (due another guy) who should know whats up zxl@geniatech.com and wrote him but only got some boilerplate back. So I am still out of info :D

b-rad-NDi, 2018-09-19:
We have a relationship with Geniatech, I'll see if I can get a couple more emails to cc.

CvH, 2018-09-25:
https://bitbucket.org/CrazyCat/media_build/downloads/linux-media-LATEST.tar.bz2
supports M88DS3103b now :) (likely he has no intention to upstream it)

`ds3k driver updated with ds3103b support + dvb-usb-dw2102 use ds3k for su3000 cards (HDStar, Tevii S662 and etc.)`

b-rad-NDi, 2018-09-28:
hmmm...i wish git_log was filled in. I'll queue this up to analyze. I also have what we've done to the windows driver, which sounds like it's the same as the geniatech driver.

CvH, 2018-09-28:
I could ask him if he has some kind of patch, I did not even tried to compare it and create a patchfile because he has a lot other stuff at his media_tree too (tbs + some other ot of tree drivers)

CvH, 2018-09-28:
here is a dmesg from an working `Terratec Cinergy S2 USB Box`
https://pastebin.com/Vbwrn7pL

b-rad-NDi, 2018-10-09:
ok...so been told to take care of this now and have been looking/comparing. The 'working' tree essentially just uses the geniatech driver, from the patch you linked first, doing a global string swap of ds3000 to ds3k and changing author to "Geniatech". I'm just going to massage the geniatech modifications out and integrate only the bits necessary. If it works for crazycat, then the original patch should be fine, nothing is different.

b-rad-NDi, 2018-10-11:
Such a weird patch...I'll probably provide two options. One with ds3000 modified and the other with m88ds3103 modified. The two drivers are equivalent as far as I can tell.

b-rad-NDi, 2018-10-11:
As in m88ds3103 already had most of the reference code geniatech released.

NukeMania, 2018-10-11:
Hello, 
Can i ask a question about ds3103b 
I think geniatech's windows and linux drivers not working well with ds3103b when i tested on both systems they can't lock to some frequencies
I found a working windows driver for ds3103b it can lock all frequencies and it was Wintv-nova-hd-usb2 driver
I think There is something wrong on geniatech implementation

b-rad-NDi, 2018-10-12:
Well that's good if so, because then I can get that windows source and pull the secret sauce out.

NukeMania, 2018-10-13:
one more thing i didn't mention I leave you to your decision
m88ds3103b is very similar to m88rs6000 and they share same firmware 
there is support in m88ds3103.c for rs6000 and for tuner m88rs6000t.c
Even the chip id jumps from 0x74(rs6000) to 0x75(ds3103b)
I didn't try change chip id to match for ds3103b and modifying dw2102.c  for m88ds3103/m88rs6000t configuration 
I don't have much skills for that but my opinion adding ds3103b in ms88ds3103 will be more accurate than ds3000

b-rad-NDi, 2018-11-01:
@NukeMania : I have noticed this and is what partially stalled this work. I'm not sure why the ds3000 driver is being used when the m88rs6000 driver appears so similar.

I cannot test the signal part, but I'm trying to have one of these mailed to me so I can at least verify connect/probe/disconnect properly.

NukeMania, 2018-11-01:
Hi Brad,
I did too It partially worked with m88rs6000 driver I can confirm as your results alike
I can connect/probe/disconnect but tuning is falling
I can be tester for tuning if you want

b-rad-NDi, 2018-12-13:
Hello, I have worked on this now.
I have two drivers:
-- the Geniatech "ds3k" driver with ALL tuner stuff removed
-- the upstream m88ds3103 driver extended to support 3103b

The em28xx bridge driver has been customized a little to support testing between both demod driver options. To test, create a file **/etc/modprobe.d/linuxtv.conf** and put this in it:
```
#options em28xx-dvb HPG461E_rev=1
```
Without the option set, you will use the m88ds3103 driver.
With the "#" removed you will use the ds3k driver.

Both driver versions detect and communicate with demod, frontend, tuner, and SEC.
I have no idea if either version works. Our German office will be testing these packages out when they have time. Feel free to give them a spin.

https://drive.google.com/drive/folders/1ZJCOSd5cxfbVqCFtZ4A06HzrWN9h6cmJ?usp=sharing

Hopefully the m88ds3103 driver works, because it will be quite easy to upstream what I've done to it.

acidowl, 2018-12-13:
Hi Brad, I would love to test it on my Raspberry Pi 3B. Is there any chance you could compile this to an ARM target or are there any instructions to install the firmware myself by extracting it from the .deb file?

b-rad-NDi, 2018-12-13:
The "firmware" was forgotten about, I just uploaded it. It is called dvb-fe-ds3103.fw.
I don't have an environment set up right now to produce rpi packages. A little busy right now, I'll consider setting up a kernel build when I have time. Are you running mainline kernel at least?

acidowl, 2018-12-13:
Ah, awesome, thanks! I'll setup my Raspberry to the last mainline kernel and test it.

CvH, 2018-12-13:
@b-rad-NDi I can build some LE images if you like - could you commit/post some patch files ?

b-rad-NDi, 2018-12-14:
I remembered I had a rpi3 environment setup. I cloned the official 4.14 rpi kernel, quickly backported the drivers, then compiled some debs. There are available in the same gdrive location.
@CvH I'll make a branch with patches tomorrow.

b-rad-NDi, 2018-12-14:
you still need the firmware also in /lib/firmware

b-rad-NDi, 2018-12-17:
Our testers in Germany have confirmed that the m88ds3103 upstream driver is still missing secret sauce, it does not find any signal. However after setting the em28xx-dvb driver option to switch to the "ds3k" driver signal is detected and all programs are found.
Now begins the secret sauce extraction...

zygotic99, 2018-12-20:
+1 on this issue.  Thanks for looking at it @b-rad-NDi

Before finding this issue I compiled modified em28xx to register 2013:0259 and cx231xx to not do so.  My 461e is now much closer to working - lots of what the kernel logs looks as it should and it has a lit LED - but it still doesn't fully work.  As far as I can tell no v4l device is being registered.  So I guess I need the demodulator fix, don't think I can get any further.

It looks like there is or was some relationship between Hauppauge and PCTV, and now each company has a different device using 2013:0259, from what I can tell.  Seems odd.

I'm not sure whether I need ts2020.ko for the tuner, or something related to Allegro for the LNB.  I'd be interested to know how it all hangs together, but I'll wait and watch for progress here.

Mostly posting to register interest in a fix for this.  I bought Hauppauge for Linux support, as I have a few times over the years.

b-rad-NDi, 2018-12-20:
@zygotic99 : thanks for joining the mix. You have a misprint device. I'm still not sure how I'll handle that in the kernel, but the builds I've posted here have that modification in place.
Hauppauge bought and owns PCTV ;)
If you have a 461e, you'll eventually need m88ds3103 (demod), ts2020 (tuner), and af2093 (lnb controller).
I was working on something else the last couple days, but I'll post the patches and "reverse engineering" aids I'm using tomorrow.

b-rad-NDi, 2018-12-27:
@CvH 'work in progress' 3103b series is uploaded. The ds3k driver works, the m88ds3103 driver does not. The 461e v2 uses the ds3k driver by default in this build. You can switch to the broken m88ds3103 driver if desired, it's there for capturing i2c traffic to determine what is missing.

zimchaa, 2019-01-19:
Hi, I'm not sure if this is the right place to add my bit of extra info - I have a PCTV 461e, [LF Rev B8H9 printed on the back] - the thread on the forum and the above mention **2013:0259** as the USB ID, but I have a different one: **2013:0461** - which I've only found mentioned here: [raspberrypi.de Forum](https://forum-raspberrypi.de/forum/thread/40312-dvb-s2-461e/) the same problem I'm having - will my issue be resolved alongside the resolution of the above or is it different?

lsusb and relevant dmesg (also attached)
```
Bus 001 Device 006: ID 1c4f:0002 SiGma Micro Keyboard TRACER Gamma Ivory
Bus 001 Device 005: ID 2013:0461 PCTV Systems 
Bus 001 Device 004: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

[    2.326302] usb 1-1.3: new high-speed USB device number 5 using dwc_otg
[    2.417366] usb 1-1.3: New USB device found, idVendor=2013, idProduct=0461
[    2.417388] usb 1-1.3: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[    2.417400] usb 1-1.3: Product: PCTV 461
[    2.417409] usb 1-1.3: Manufacturer: PCTV
[    2.417418] usb 1-1.3: SerialNumber: 0014050895
```
[terminal libreelec dvb-s2 tuner.txt](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/2775916/terminal.libreelec.dvb-s2.tuner.txt)

mikebaldry, 2019-02-06:
I also have the same issue as @zimchaa with the same dmesg output. Any help would very much be appreciated @b-rad-NDi!

b-rad-NDi, 2019-02-06:
There's two different issues at play. One, is some devices have an incorrect pid in the eeprom. I have a work in progress utility that will be able to fix that.
The other issue is the ds3k driver for the 3103b demod and the fact that it's not anywhere close to upstreamable. I have fixed it to work though. Figuring out what little bits from it are necessary to get the upstream driver working is one of my main priorities right now.

b-rad-NDi, 2019-02-06:
btw @zimchaa & @mikebaldry 
2013:0461 is supported in my ppa and *should* work. It is just the upstream driver that I am working on.
2013:0259 is also supported with a module option to "disable" it in the cx231xx driver.

zimchaa, 2019-02-08:
Hey @b-rad-NDi @CvH - a minor update but I don't know if this should be on a different thread as it's not the incorrect PID problem, incase you're interested to see my progress however: I'm trying this on a Raspi 3B+ using the files from your Google Drive (link below) and I can't quite see where I'm going wrong, any ideas?

https://drive.google.com/drive/folders/1ZJCOSd5cxfbVqCFtZ4A06HzrWN9h6cmJ?usp=sharing

I notice there's a slight difference in the kernel versions, but I'm on the very latest according to apt on Raspbian.

dmesg output:
[dmesg.txt](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/2845459/dmesg.txt)

Device info:
```
pi@raspi3b:~ $ uname -a
Linux raspi3b 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
```

USB Device ID:
```
pi@raspi3b:~ $ lsusb
Bus 001 Device 004: ID 2013:0461 PCTV Systems 
Bus 001 Device 005: ID 1c4f:0002 SiGma Micro Keyboard TRACER Gamma Ivory
Bus 001 Device 006: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Files I'm using:
```
pi@raspi3b:~ $ ls -ltr
total 27440
-rw-r--r-- 1 pi pi 10986896 Feb  8 12:05 linux-headers-4.14.87-v7+_4.14.87-v7+-1_armhf.deb
-rw-r--r-- 1 pi pi 17087230 Feb  8 12:05 linux-image-4.14.87-v7+_4.14.87-v7+-1_armhf.deb
-rw-r--r-- 1 pi pi    11645 Feb  8 13:11 dvb-tuner-si2157-a30-05.fw
-rw-r--r-- 1 pi pi     8192 Feb  8 13:11 dvb-fe-ds3103.fw
```

Installation status:
```
pi@raspi3b:~ $ dpkg -l | grep 4.14
ii  linux-headers-4.14.87-v7+       4.14.87-v7+-1                armhf        Linux kernel headers for 4.14.87-v7+ on armhf
ii  linux-image-4.14.87-v7+         4.14.87-v7+-1                armhf        Linux kernel, version 4.14.87-v7+
```

Firmware folder:
```
pi@raspi3b:~ $ ls -ltr /lib/firmware | grep dvb
-rw-r--r--  1 root root  50222 Aug 20 10:53 dvb-usb-terratec-h5-drxk.fw
-rw-r--r--  1 root root   5834 Aug 20 10:53 dvb-usb-it9135-02.fw
-rw-r--r--  1 root root   8128 Aug 20 10:53 dvb-usb-it9135-01.fw
-rw-r--r--  1 root root  33768 Aug 20 10:53 dvb-usb-dib0700-1.20.fw
-rw-r--r--  1 root root  16497 Aug 20 10:53 dvb-fe-xc5000c-4.1.30.7.fw
-rw-r--r--  1 root root  12401 Aug 20 10:53 dvb-fe-xc5000-1.6.114.fw
-rw-r--r--  1 root root  18643 Aug 20 10:53 dvb-fe-xc4000-1.4.1.fw
-rw-r--r--  1 root root  11645 Feb  8 13:12 dvb-tuner-si2157-a30-05.fw
-rw-r--r--  1 root root   8192 Feb  8 13:12 dvb-fe-ds3103.fw
```

linuxtv.conf (not sure if this is required):
```
pi@raspi3b:~ $ cat /etc/modprobe.d/linuxtv.conf 
#options em28xx-dvb HPG461E_rev=1
```

martinund, 2019-04-01:
b-rad-NDi, how is work on this progressing?

I've just bought a 461e based on favourable reviews on Amazon and in the TVHeadend and Raspberry Pi forums, and now it looks as if the product has been updated to a new revision, invalidating the "it works" info :-(

I'm debating whether to send back the 461e and try to find a DVB-S2 adaptor which *is* supported on Raspbian Stretch (RasPi 3B+) for TV headend, or whether to keep it in the hope that you'll get the drivers working eventually.

b-rad-NDi, 2019-04-01:
@martinund  The ds3k driver included in this repo works fine with the new 461e. In addition I've backported it to stretch for rpi3b+ and it works there perfectly. If you're willing to build your own kernel enabling the new device is pretty easy.
When I have time, I am still continuing to work on the upstream driver. I'm hoping to clear out enough of my priorities to work on it again this week.

martinund, 2019-04-01:
What is the upstream driver used for? I'm wondering whether or not I need it for recording as opposed to playing live. What is involved in building a new kernel that includes your ds3k driver?

Presumably everything will eventually make it to a general release kernel that can be updated by "sudo apt upgrade", maybe with a firmware file in /lib/firmware. Any very rough estimate of when that might be?

b-rad-NDi, 2019-04-01:
The upstream driver only supports the ds3103 demod. It lacks support for the ds3103b demod. The ds3k driver supports both. Timeline for me to get the upstream driver working is indeterminate. 
Rpi debian stretch kernel upgrades don't happen via "apt-get dist-upgrade" for whatever reason they went another route. Building a kernel is quite easy, but if you've never done it before I can provide a tarball to unpack at / which provides the kernel and drivers integrated. Just gimme what "uname -a" shows you, if it's 4.9.x then I can provide a package immediately.

martinund, 2019-04-02:
Many thanks Brad. The output of uname -a on my Pi 3B+ is

Linux martin-pi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

so a slightly later version of the kernel.

I presume I should keep copies of all the files which will be overwritten when I unpack the tarball, in case of problems. I'll take an up-to-date image of the system drive (a 32 GB SD card) beforehand, just in case... 15 mins spent imaging the drive (and another 15 to restore it if necessary) is quicker than starting afresh and then customising everything all over again...

martinund, 2019-04-02:
I ordered the device - I can always use it on a Windows PC running NextPVR until the RasPi support is working. The one I received is labelled "200000 Rev B6H9" - I presume that's the new version. Until we move house, I don't even have a satellite dish to connect it to (!), but I want to be prepared in case we get there and find that the terrestrial TV aerial is little more than an "ornament". TV reception sites show no usable signal from any transmitter and yet our house and many nearby all have aerials pointing at a transmitter which is over a nearby hill, and the existing owners have various TVs which are plugged into RF rather than satellite outputs - most intriguing. Given that terrestrial reception surely will be very borderline and liable to change when there is an atmospheric "lift", I'm hedging my bets and going for satellite in case...

martinund, 2019-04-02:
Can I check what dmesg output I should see, because I have a sneaking suspicion that the device is actually working OK. ;-)

The dmesg trace is for rebooting with the 461e on USB 1.1.3 already plugged in.

TVHeadend Config | DVB Inputs sees the device https://i.postimg.cc/sD5kPvSy/TVH-sat.png . I presume the red dots are because I haven't fully configured that device in TVH - I can't scan for channels until I have a satellite dish to plug into, which is a *bit* of a showstopper to further testing ;-)

I have two adaptor0 and adaptor1 (#1 is my Hauppauge WinTV Nova DVB-T)  in /dev/dvb and both have devices demux0, dvr0, frontend0 and net0 - the fact that a working device and the 461 have the same devices looks promising.

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.98-v7+ (dom@dom-XPS-13-9370) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1200 SMP Tue Feb 12 20:27:48 GMT 2019
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] On node 0 totalpages: 242688
[    0.000000] free_area_init_node: node 0, pgdat 80c85780, node_mem_map ba39f000
[    0.000000]   Normal zone: 2133 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:31
[    0.000000] percpu: Embedded 17 pages/cpu @ba348000 s38720 r8192 d22720 u69632
[    0.000000] pcpu-alloc: s38720 r8192 d22720 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait splash plymouth.ignore-serial-consoles quiet
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 940232K/970752K available (7168K kernel code, 577K rwdata, 2080K rodata, 1024K init, 698K bss, 22328K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
                   lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x80800000   (8160 kB)
                     .init : 0x80b00000 - 0x80c00000   (1024 kB)
                     .data : 0x80c00000 - 0x80c906d4   ( 578 kB)
                      .bss : 0x80c97ef8 - 0x80d468f0   ( 699 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 25298 entries in 75 pages
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000018] Switching to timer-based delay loop, resolution 52ns
[    0.000269] Console: colour dummy device 80x30
[    0.000287] console [tty1] enabled
[    0.000311] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000327] pid_max: default: 32768 minimum: 301
[    0.000649] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000664] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001605] Disabling memory control group subsystem
[    0.001681] CPU: Testing write buffer coherency: ok
[    0.002103] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002506] Setting up static identity map for 0x100000 - 0x10003c
[    0.002631] Hierarchical SRCU implementation.
[    0.003309] smp: Bringing up secondary CPUs ...
[    0.004089] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.004934] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.005759] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.005864] smp: Brought up 1 node, 4 CPUs
[    0.005875] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.005881] CPU: All CPU(s) started in HYP mode.
[    0.005885] CPU: Virtualization extensions available.
[    0.006800] devtmpfs: initialized
[    0.017309] random: get_random_u32 called from bucket_table_alloc+0xfc/0x24c with crng_init=0
[    0.018081] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[... 378 lines cut ...]
[   11.452902] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   11.471328] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[... cut ...]
[   16.317709] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   16.317834] Bluetooth: HCI UART protocol Broadcom registered
[   16.722437] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   16.722452] Bluetooth: BNEP filters: protocol multicast
[   16.722478] Bluetooth: BNEP socket layer initialized
[   16.773581] Bluetooth: RFCOMM TTY layer initialized
[   16.773612] Bluetooth: RFCOMM socket layer initialized
[   16.773629] Bluetooth: RFCOMM ver 1.11
[   22.451091] EXT4-fs (mmcblk0p5): mounted filesystem with ordered data mode. Opts: (null)

b-rad-NDi, 2019-04-02:
@martinund B6H9 is the original 461e, B7H9 is the new revisions with 3103b demod. You're good to go :)

withakw, 2019-05-01:
Hi @b-rad-NDi 
I have just bought one of the new 461e tuners and I was wondering if you have finally got it working and whether you could send me the driver file please?
Thank you very much for your time

Keith

b-rad-NDi, 2019-05-01:
I do have the m88ds3103b demod working with the m88ds3103 driver now, but I need to distill it down to only what is required. Of course I didn't get the upstream driver working until I had included every single last bit of difference...so there's a lot I'll need to trim.
The "ds3k" driver works and is deployed by many users. If you want to use it, then you can download my patch series and apply it to your own builds: patches/mainline-extra/tip/99.montage.3103b

withakw, 2019-05-02:
@b-rad-NDi I am using OSMC and I am new to this so I am not sure how to incorporate it into the build. Do you have any instructions please

georgemailo, 2019-06-04:
Instructions would be very much appreciated. I tried to build the kernel for Raspberry Pi but no success yet.

b-rad-NDi, 2019-06-04:
I have trimmed down the driver to be close to upstreamable. I have just had a crazy amount of other priorities. I will be seeing about submitting the patch to the media list soon.
Here is a guide on compiling and running mainline kernel:
https://www.collabora.com/news-and-blog/blog/2019/03/12/quick-hack-raspberry-pi-meets-linux-kernel-mainline/
It requires you running something linux, like a vm/container/native. If you can figure out how to build and run that I will produce a patch to apply against mainline and then you should be up & running.

b-rad-NDi, 2019-06-04:
@withakw 
Here is a guide to building OSMC kernel:
https://discourse.osmc.tv/t/compile-kernel-on-rpi2-from-rbp2-sources-osmc/13366/2
If you can get it built and installed, and tell me what kernel version it is I can provide you a patch.

CvH, 2019-06-05:
or just use media_build and manually patch the source :)

> I have trimmed down the driver to be close to upstreamable.
thats great news, if you have something to show off pls commit it to your git repo so we can try

btw an rebase against 5.2 would be nice :) (I did it manually and it was done rather quickly)

b-rad-NDi, 2019-06-05:
@CvH : the dma-buf and fence abi changed. It has thrown a wrench into my latest sync and I just haven't had the chance to make the fixes required for backwards compatibility. However I do have trees at linuxtv.org now, so I'll just make a couple branches with my features series applied.

b-rad-NDi, 2019-06-13:
@cvh: I have a branch in my media_tree repo, Montage-3103b. This contains the distilled down support for the 3103b demod integrated into the m88ds3103b driver. It doesn't pass checkpatch, but it's in my cleanup queue for upstreaming.

CvH, 2019-06-13:
@b-rad-NDi tx, I hope I can add it to our next release so you should get feedback if it not work :D

b-rad-NDi, 2019-06-13:
note: branch subject to change :whistle:
now at v2, mostly passes check patch strict, close to what i'll try to upstream.
If you find any issues you can file them over there.

NukeMania, 2019-06-15:
@b-rad-NDi  : I tried v2 patch on geniatech device but I get no frontend was attached error 

[   71.084701] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[   71.403053] [2294] m88ds3103 7-006a: chip_id=74
[   71.406498] i2c i2c-7: Added multiplexed i2c bus 8
[   71.406512] [2294] m88ds3103 7-006a: 
[   71.406514] [2294] m88ds3103 7-006a: 
[   71.494240] dvb-usb: no frontend was attached by 'SU3000HD DVB-S USB2.0'
[   71.527597] dw2102: su3000_power_ctrl: 0, initialized 1
[   71.527599] dvb-usb: SU3000HD DVB-S USB2.0 successfully initialized and connected.
[   71.527668] usbcore: registered new interface driver dw2102

b-rad-NDi, 2019-08-09:
@NukeMania : Well there's no support for that device.

b-rad-NDi, 2019-08-09:
The demod support is included here:
b757b633b311e324085727d98bde3f1a11e5af21

I will look into the board profile required to get SU3000 working.

b-rad-NDi, 2019-08-09:
@NukeMania : wait, so you stuffed in something to support it. Did you look at how the board that I added creates the demod driver? You have to tell it the demod is a 3103b.

[43 more replies]

closed 2020-01-31 by b-rad-NDi