b-rad-NDi/Ubuntu-media-tree-kernel-builder #54: Hauppauge WinTV-HVR-2205 analog problem

muzena, 2018-09-18:
Hello @b-rad-NDi 
Since  I bought [Hauppauge WinTV-HVR-2205 card,](http://www.hauppauge.co.uk/site/products/data_hvr2205board.html?zoom_highlight=WinTV-HVR-2205) [Linux TV](https://www.linuxtv.org/wiki/index.php/Hauppauge_WinTV-HVR-2205) I searching patch or way to patch analog tuner.
I found for older [Hauppauge WinTV-HVR-2200](https://linuxtv.org/wiki/index.php/Hauppauge_WinTV-HVR-2200) card [these patch on github](https://github.com/reinhrst/nl.claude.tools/blob/2594dd91cb81a3c905a53b9168f52a42e317cd1f/misc/patch-HVR2200-PAL-3.2.1-gentoo).
Here I found frimwares for Hauppauge TV cards
http://www.steventoth.net/linux/hvr22xx/firmwares/
http://www.steventoth.net/linux/hvr22xx/firmwares/4038864/
http://www.steventoth.net/linux/hvr22xx/firmwares/4019072/
https://github.com/OpenELEC/dvb-firmware/tree/master/firmware

I try to modify and apply this patch, but without success.
Please, can you look at this patch, maybe you can adapt this patch, I can help you with debugging.

Also, can you include those frimware files in linux-firmware-hauppauge package in PPA.
Every time when I re/installing Ubuntu/Linux Mint I must manually download frimwares in /lib/frimware
NXP7164-2010-03-10.1.fw and dvb-demod-si2168-b40-01.fw

b-rad-NDi, 2018-11-15:
The firmware package has been updated to include the missing NXP7164 blobs.

muzena, 2018-11-16:
Thanks!
Can you also add these two frimwares
http://www.steventoth.net/linux/hvr22xx/firmwares/4038864/
I think they are needed for analog reciver.

Can you try something wit this patch, I try to add patch replace tuner and demodulators with hvr2205 cards: Si2168-B40, Si2157-A30 tuner, but it dosen't work.
I think the problem is in saa7164 driver, saa7164 encoder.

I try to add module debbug option: 'options saa7164 debug=2', but after boot system in dmesg i get this error:
 si2157: unknown parameter 'debug' ignored

b-rad-NDi, 2018-11-16:
I don't have any 22xx devices, so I can't try anything. I have requested the main office test out 2205 and get back to me.
I'll check out those firmwares and potentially add them as well.

Saentist, 2019-08-06:
Is there any movement on this task?

b-rad-NDi, 2019-08-07:
1) all the firmwares are already included
2) HVR_2200_4 support is already in the kernel
3) I see nothing in the patch that does anything except add some PAL video standards...and then overriding some existing values, without any reason stated, most likely affecting all other devices using the driver. aka not gonna happen.

So, what is wrong? Is the *ONLY* thing wrong the missing PAL video standards? This should not affect tuning, at all, this would manifest as "bad video" frames.

Saentist, 2019-08-07:
Is not possible to scan can be a problem?
no signal found or ioctrl failures is most of problems in analog reception.
No digital signal to test but frontend are loaded without problem in DVB part of board.

b-rad-NDi, 2019-08-07:
If a signal is found the tuner and frontend will signal lock. That is not the bridge and before any decoders are involved. The above patch does nothing besides add definitions for a bunch of PAL formats. An incorrect video format will just affect the video decoding, giving you a jumbled screen. Incorrect video format has nothing to do with tuner signal lock.
So, what is the problem? Is there a /dev/videoX analog tuner? I added the missing firmwares. Is anything else required? I see no logs or anything describing, whatsoever, what is still "broke". What works with the card now? I cannot act without info.

muzena, 2019-08-07:
I think that problem is in si2168 driver, I forgot what I tried in the posts above,  I will check again next week.

b-rad-NDi, 2019-08-08:
Please check and post full dmesg. I'm very unfamiliar with this bridge, it's before my time. The demods and tuners though, I'm very familiar with them and they work very well.

Saentist, 2019-08-08:
Analog reception cannot found signal on all band 0.25Mhz step.
Signal is from areal antenna.
On same PC on windows all work.

b-rad-NDi, 2019-08-08:
That means nothing to me without logs. Preferably with full debug enabled for bridge, demod, and tuner.

Saentist, 2019-08-11:
https://paste.ubuntu.com/p/thK5SkbtjJ/
https://paste.ubuntu.com/p/mY6nZ6V3xK/

b-rad-NDi, 2019-08-11:
Thanks for this, but I need the dmesg with full debug enabled via debugfs. Bridge, demod, and tuner drivers should have full logging enabled.

Saentist, 2019-08-11:
any examples 
p.s.
it's analog tuner important no digital signal to test digital part.

b-rad-NDi, 2019-08-11:
```
echo "file drivers/media/pci/* +pfl" > /sys/kernel/debug/dynamic_debug/control 
echo "file drivers/media/tuners/* +pfl" > /sys/kernel/debug/dynamic_debug/control 
```
If you see other drivers that might be relevant enable them similarly.

Saentist, 2019-08-11:
```
# lsmod  |grep vid
nvidia_uvm            761856  0
nvidia_drm             45056  2
nvidia_modeset       1048576  11 nvidia_drm
nvidia              14381056  448 nvidia_uvm,nvidia_modeset
drm_kms_helper        180224  1 nvidia_drm
drm                   479232  5 drm_kms_helper,nvidia_drm
ipmi_msghandler       102400  2 ipmi_devintf,nvidia
videodev              200704  1 saa7164
media                  53248  1 videodev

# lsmod  |grep tv
tveeprom               28672  1 saa7164

# lsmod  |grep dvb
dvb_core              131072  1 saa7164

# lsmod  |grep med
media                  53248  1 videodev

```

Saentist, 2019-08-11:
Where do I have to expect logs to be show?

b-rad-NDi, 2019-08-12:
they'll be in dmesg with everything else

Saentist, 2019-08-12:
Maybe command you give is wrong!