b-rad-NDi/Ubuntu-media-tree-kernel-builder #56: How to use on Raspbian (Hauppauge WinTV-HVR-935 HD)

jzobac, 2018-11-04:
Hi,
Hlw to use this script on latest Rasbian and rpi3?

I have Hauppauge WinTV-HVR-935 HD.

With best regards

Jz

walterav1984, 2018-11-04:
It may already work on raspbian with branch NEXT linux kernel 4.19 which can be upgraded with more ease than this dkms thing. Since most of these Hauppauge driver patches get into mainline linux kernel other projects who use that updated kernel (like rasbpian/Libreelec) directly benefit for out of the box drivers(not firmware...). 

According to this patch the hauppauge support 1b51 was done somewhere near kernel 4.17-4.19.
https://github.com/torvalds/linux/blob/master/drivers/media/usb/cx231xx/cx231xx-cards.c

Instructions for RPI3
Start without your USB tuner attached and use a recent updated raspbian image and update your kernel with the following command:
```
sudo apt-get update
sudo apt-get -y upgrade
sudo apt-get -y dist-upgrade
sudo BRANCH=next rpi-update #use capitals since command is case sensitive!

sudo apt-get -y install tvheadend #available in raspbian repo since DVB-t2 HAT ;-)

sudo reboot
#After finishing reboot type
dmesg
#insert your usb tuner
dmesg
#notice any thing about missing firmware.fw, google your firmware and download it
wget http://someadress/examplefirmware.fw
sudo cp examplefirmware.fw /lib/firmware/
sudo reboot

#enjoy your mainline hauppauge linux driver!
```

Good luck

jzobac, 2018-11-05:
Hi,
very thanks, seems that there is some progress.

I have installed linux image 4.19.0-v7+.

Problems:
1) RPI show only one tuner, it is dual DVB-T card
2) yes, I'm using tvheadend (4.2.4) and probably it catches only DVB-T1, I wish to use DVB-T2 or I doesn't know how to switch to DVB-T2
3) Kodi seems to stop working, menus are jumping, like someone is always clicking. Probably driver of IR on TV card is broken :(

I don't have time to investigate more, hope today I will try something. But I must probably report error to linux image.

JZobac

walterav1984, 2018-11-07:
Are you sure the WinTV-HVR-935 HD is a dual tuner, since it only has a analog part besides the hybrid DVB-C/DVB-T/T2. The WinTV dualHD (2040:0265) has actually 2 real tuners and works with kernel 4.19.

Tvheadend should only list one Tuner, but it should show you 2 frontends one for DVB-T(which can also be used for DVB-T2) and one for DVB-C which you mention is currently missing. Getting DVB-T2 working in Tvheadend may require some mux/QAM editing but it works the same as DVB-T and doesn't mention anything special for DVB-T2.

About the kodi thing, I wasn't aware you were also using this RPi besides a Tvheadend Server also as a Kodi client in which infrared/hdmi-cec/gpu/etc are still beta for 4.19(thats why you using BRANCH=next) you have to wait for a real Raspbian 4.19 release to get those things fixed but for server it may be enough.

b-rad-NDi, 2018-11-07:
The 935C is definitely not a dual tuner device. The 935C does have an ir receiver, but the patch to enable it has not been accepted upstream yet.

This project is mainly for users of old kernels who desire everything from the latest mainline media tree. The patches left to be upstreamed at this point is very few.

jzobac, 2018-11-07:
Hi guys,
thanks for your time. My big apologies, I will to buy dual tuner, instead buy single :( (they cost same price).
I will try to replace the ćard with Hauppauge WinTV-dualHD - since it should work. But, there are two revisions, I know, that I will get the second, will it work too (Model 01595 USB device ID 2040:026d)?
https://www.linuxtv.org/wiki/index.php/Hauppauge_WinTV-dualHD

I have discovered, that IR on card is not the problem, because Ï have to boot without the card and Kodi is still "unresponsible" (menus are jumping).

I need that it will work with Kodi, because I move server (x64 Ubuntu 16.04) to new location, where is no antenna, thus new TV server must be multimedia center - rpi3. Temporary I can use DVB-T card Evolveo Venus (my old card) or new dual tuner Hauppage with DVB-T2 with rpi1 (but it will not have probably sufficient CPU power).

Why is not rpi TV Hat dual tuner, that's solution for all problems on the planet :) (or option to use two single tuner Hats).

Thanks again JZ

b-rad-NDi, 2018-11-07:
You want the second revision. The only difference is it utilizes bulk transfer mode, which helps out a lot on embedded.

closed 2018-11-07 by b-rad-NDi