b-rad-NDi/Ubuntu-media-tree-kernel-builder #57: VIDIOC_QBUF issue

Winshipe, 2018-11-16:
When I try to capture video from the Hauppauge USB-Live 2 using V4L in any capacity through OpenCV, ffmpeg, or VLC I get the some variation on the error message VIDIOC_QBUF: message too long.  From ffmpeg I get this

[video4linux2,v4l2 @ 0x565076344820] ioctl(VIDIOC_QBUF): Message too long
/dev/video0: Message too long

The system is an Ubuntu 18.04 VM running on Gnome Boxes, although I've tried with VirtualBox with similar results.
Dmesg output from the Boxes VM:
[Fri Nov 16 09:36:17 2018] usb 2-1.3: new full-speed USB device number 5 using uhci_hcd
[Fri Nov 16 09:36:17 2018] usb 2-1.3: not running at top speed; connect to a high speed hub
[Fri Nov 16 09:36:17 2018] usb 2-1.3: config 1 interface 2 altsetting 0 endpoint 0x83 has invalid maxpacket 512, setting to 64
[Fri Nov 16 09:36:17 2018] usb 2-1.3: config 1 interface 3 altsetting 0 endpoint 0x84 has invalid maxpacket 512, setting to 64
[Fri Nov 16 09:36:17 2018] usb 2-1.3: config 1 interface 4 altsetting 0 endpoint 0x85 has invalid maxpacket 512, setting to 64
[Fri Nov 16 09:36:17 2018] usb 2-1.3: config 1 interface 4 altsetting 1 endpoint 0x5 has invalid maxpacket 512, setting to 64
[Fri Nov 16 09:36:17 2018] usb 2-1.3: config 1 interface 5 altsetting 0 endpoint 0x86 has invalid maxpacket 512, setting to 64
[Fri Nov 16 09:36:17 2018] usb 2-1.3: New USB device found, idVendor=2040, idProduct=c200
[Fri Nov 16 09:36:17 2018] usb 2-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Fri Nov 16 09:36:17 2018] usb 2-1.3: Product: Hauppauge Device
[Fri Nov 16 09:36:17 2018] usb 2-1.3: Manufacturer: Hauppauge
[Fri Nov 16 09:36:17 2018] usb 2-1.3: SerialNumber: 0014054134
[Fri Nov 16 09:36:17 2018] cx231xx 2-1.3:1.1: New device Hauppauge Hauppauge Device @ 12 Mbps (2040:c200) with 6 interfaces
[Fri Nov 16 09:36:17 2018] cx231xx 2-1.3:1.1: can't change interface 3 alt no. to 3: Max. Pkt size = 0
[Fri Nov 16 09:36:17 2018] cx231xx 2-1.3:1.1: Identified as Hauppauge USB Live 2 (card=9)
[Fri Nov 16 09:36:17 2018] i2c i2c-2: Added multiplexed i2c bus 4
[Fri Nov 16 09:36:17 2018] i2c i2c-2: Added multiplexed i2c bus 5
[Fri Nov 16 09:36:18 2018] cx25840 1-0044: cx23102 A/V decoder found @ 0x88 (cx231xx #0-0)
[Fri Nov 16 09:36:22 2018] cx25840 1-0044: loaded v4l-cx231xx-avcore-01.fw firmware (16382 bytes)
[Fri Nov 16 09:36:22 2018] cx231xx 2-1.3:1.1: v4l2 driver version 0.0.3
[Fri Nov 16 09:36:22 2018] cx231xx 2-1.3:1.1: Registered video device video0 [v4l2]
[Fri Nov 16 09:36:22 2018] cx231xx 2-1.3:1.1: Registered VBI device vbi0
[Fri Nov 16 09:36:22 2018] cx231xx 2-1.3:1.1: audio EndPoint Addr 0x83, Alternate settings: 3
[Fri Nov 16 09:36:22 2018] cx231xx 2-1.3:1.1: video EndPoint Addr 0x84, Alternate settings: 5
[Fri Nov 16 09:36:22 2018] cx231xx 2-1.3:1.1: VBI EndPoint Addr 0x85, Alternate settings: 2
[Fri Nov 16 09:36:22 2018] cx231xx 2-1.3:1.1: sliced CC EndPoint Addr 0x86, Alternate settings: 2

b-rad-NDi, 2018-11-16:
I have a live2, I'll give this a test. I only use kvm/qemu however, so the tests will only be pertinent to that hypervisor.
If it is determined that there is an issue with specific VM software, then that issue almost certainly lies in the USB passthru mechanism of the hypervisor. If that is so, then it is fairly well out of my control and should be reported to the upstream project.

0x3333, 2020-08-24:
I don't know if this is related, but, I was unable to read frames from this device, until I discovered that If the companion audio device is not initialized, you cannot read video frames from it. You need to init audio device and read 1 frame, others can be ignored.