b-rad-NDi/Ubuntu-media-tree-kernel-builder #58: Aborts when I try to install the headers

keithwwalker, 2018-11-21:
I am having trouble at the last command to install the PPA headers:
`sudo apt-get install linux-headers-mediatree`
The commands 'abort's without loading anything.
More information is at this thread here:
[https://forum.kodi.tv/showthread.php?tid=337672&pid=2793502#pid2793502](url)
What I am I doing wrong or not considering?
Thanks
Keith Walker

b-rad-NDi, 2018-11-21:
No idea what you've done "wrong", but something is awry. I just added the ppa to a Xenial install, then installed linux-image-mediatree and linux-headers-mediatree without issue. The issue must be on your end. Perhaps clean the apt cache or something?

keithwwalker, 2018-11-23:
I manually typed in the commands, and the headers did load.  Now the issue is that when I run Kaffeine to test WinTV-DualHD functionality, only one tuner is detected.

closed 2018-12-13 by b-rad-NDi