b-rad-NDi/Ubuntu-media-tree-kernel-builder #59: More than 1 Hauppauge quadHD PCIE causes a fatal crash

Neolo, 2018-11-26:
Hi, when more than 1 Hauppauge quadHD PCIE is installed on the same host, driver crashes fatally after 5-10 minutes of run, and the only way to reboot is a power reset. Confirmed with different tuners and different PCIE slots. 1 tuner works ok.
Syslog gets flooded out so fast, that I ran out of HDD space in a few minutes (200 Gb consumed).

**Ubuntu Bionic
PPA - host slows down with syslog being flooded out, tuners stops working. Reboot command hangs.
"Ubuntu-media-tree-kernel-builder" script - host hangs without crash reports. Reboot possible only via manual power reset.**

With driver from PPA it loops with following messages in syslog:

Nov 24 17:50:33 tvbuntu kernel: [  577.490392] cx23885 0000:03:00.0: dma in progress detected 0x00000001 0x00000001, clearing
Nov 24 17:50:33 tvbuntu kernel: [  577.490413] cx23885 0000:03:00.0: dma in progress detected 0x00000001 0x00000001, clearing
Nov 24 17:50:33 tvbuntu kernel: [  577.490440] cx23885: cx23885[0]: mpeg risc op code error
Nov 24 17:50:33 tvbuntu kernel: [  577.490450] cx23885: cx23885[0]: TS1 B - dma channel status dump
Nov 24 17:50:33 tvbuntu kernel: [  577.490454] cx23885: cx23885[0]:   cmds: init risc lo   : 0x345ea000
Nov 24 17:50:33 tvbuntu kernel: [  577.490458] cx23885: cx23885[0]:   cmds: init risc hi   : 0x00000000
Nov 24 17:50:33 tvbuntu kernel: [  577.490461] cx23885: cx23885[0]:   cmds: cdt base       : 0x00010870
Nov 24 17:50:33 tvbuntu kernel: [  577.490465] cx23885: cx23885[0]:   cmds: cdt size       : 0x0000000a
Nov 24 17:50:33 tvbuntu kernel: [  577.490468] cx23885: cx23885[0]:   cmds: iq base        : 0x00010630
Nov 24 17:50:33 tvbuntu kernel: [  577.490471] cx23885: cx23885[0]:   cmds: iq size        : 0x00000010
Nov 24 17:50:33 tvbuntu kernel: [  577.490474] cx23885: cx23885[0]:   cmds: risc pc lo     : 0x345ea00c
Nov 24 17:50:33 tvbuntu kernel: [  577.490477] cx23885: cx23885[0]:   cmds: risc pc hi     : 0x00000000
Nov 24 17:50:33 tvbuntu kernel: [  577.490480] cx23885: cx23885[0]:   cmds: iq wr ptr      : 0x0000418d
Nov 24 17:50:33 tvbuntu kernel: [  577.490483] cx23885: cx23885[0]:   cmds: iq rd ptr      : 0x0000418c
Nov 24 17:50:33 tvbuntu kernel: [  577.490486] cx23885: cx23885[0]:   cmds: cdt current    : 0x00010878
Nov 24 17:50:33 tvbuntu kernel: [  577.490489] cx23885: cx23885[0]:   cmds: pci target lo  : 0x3478af00
Nov 24 17:50:33 tvbuntu kernel: [  577.490493] cx23885: cx23885[0]:   cmds: pci target hi  : 0x00000000
Nov 24 17:50:33 tvbuntu kernel: [  577.490495] cx23885: cx23885[0]:   cmds: line / byte    : 0x00100000
Nov 24 17:50:33 tvbuntu kernel: [  577.490499] cx23885: cx23885[0]:   risc0:
Nov 24 17:50:33 tvbuntu kernel: [  577.490500] 0x1c0002f0 [ write sol eol count=752 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490506] cx23885: cx23885[0]:   risc1:
Nov 24 17:50:33 tvbuntu kernel: [  577.490507] 0x3478ac10 [ INVALID eol 22 21 20 19 resync 13 count=3088 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490516] cx23885: cx23885[0]:   risc2:
Nov 24 17:50:33 tvbuntu kernel: [  577.490517] 0x00000000 [ INVALID count=0 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490521] cx23885: cx23885[0]:   risc3:
Nov 24 17:50:33 tvbuntu kernel: [  577.490522] 0x1c0002f0 [ write sol eol count=752 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490528] cx23885: cx23885[0]:   (0x00010630) iq 0:
Nov 24 17:50:33 tvbuntu kernel: [  577.490529] 0x70000000 [ jump count=0 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490534] cx23885: cx23885[0]:   iq 1: 0x1c0002f0 [ arg #1 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490538] cx23885: cx23885[0]:   iq 2: 0x3478af00 [ arg #2 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490540] cx23885: cx23885[0]:   (0x0001063c) iq 3:
Nov 24 17:50:33 tvbuntu kernel: [  577.490541] 0x00000000 [ INVALID count=0 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490546] cx23885: cx23885[0]:   (0x00010640) iq 4:
Nov 24 17:50:33 tvbuntu kernel: [  577.490547] 0x00000000 [ INVALID count=0 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490551] cx23885: cx23885[0]:   (0x00010644) iq 5:
Nov 24 17:50:33 tvbuntu kernel: [  577.490552] 0x1c0002f0 [ write sol eol count=752 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490558] cx23885: cx23885[0]:   iq 6: 0x3478a340 [ arg #1 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490561] cx23885: cx23885[0]:   iq 7: 0x00000000 [ arg #2 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490564] cx23885: cx23885[0]:   (0x00010650) iq 8:
Nov 24 17:50:33 tvbuntu kernel: [  577.490564] 0x1c0002f0 [ write sol eol count=752 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490570] cx23885: cx23885[0]:   iq 9: 0x3478a630 [ arg #1 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490573] cx23885: cx23885[0]:   iq a: 0x00000000 [ arg #2 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490576] cx23885: cx23885[0]:   (0x0001065c) iq b:
Nov 24 17:50:33 tvbuntu kernel: [  577.490577] 0x1c0002f0 [ write sol eol count=752 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490582] cx23885: cx23885[0]:   iq c: 0x3478a920 [ arg #1 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490585] cx23885: cx23885[0]:   iq d: 0x00000000 [ arg #2 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490588] cx23885: cx23885[0]:   (0x00010668) iq e:
Nov 24 17:50:33 tvbuntu kernel: [  577.490589] 0x1c0002f0 [ write sol eol count=752 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490595] cx23885: cx23885[0]:   iq f: 0x3478ac10 [ arg #1 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490598] cx23885: cx23885[0]:   iq 10: 0x346d5530 [ arg #2 ]
Nov 24 17:50:33 tvbuntu kernel: [  577.490599] cx23885: cx23885[0]: fifo: 0x00005000 -> 0x6000
Nov 24 17:50:33 tvbuntu kernel: [  577.490600] cx23885: cx23885[0]: ctrl: 0x00010630 -> 0x10690
Nov 24 17:50:33 tvbuntu kernel: [  577.490603] cx23885: cx23885[0]:   ptr1_reg: 0x00005430
Nov 24 17:50:33 tvbuntu kernel: [  577.490606] cx23885: cx23885[0]:   ptr2_reg: 0x00010888

Nov 24 18:21:36 tvbuntu kernel: [ 2440.066839] cx23885: cx23885[0]: mpeg risc op code error
Nov 24 18:21:36 tvbuntu kernel: [ 2440.066853] cx23885: cx23885[0]: TS1 B - dma channel status dump
Nov 24 18:21:36 tvbuntu kernel: [ 2440.066859] cx23885: cx23885[0]:   cmds: init risc lo   : 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.066865] cx23885: cx23885[0]:   cmds: init risc hi   : 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.066876] cx23885: cx23885[0]:   cmds: cdt base       : 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.066877] cx23885: cx23885[0]:   cmds: cdt size       : 0xffffffff
[... 49 lines cut ...]
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067023] cx23885: cx23885[0]:   (0x0001066c) iq f:
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067023] 0xffffffff [ INVALID sol eol irq2 irq1 23 22 21 20 19 18 cnt1 cnt0 resync 14 13 12 count=4095 ]
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067028] cx23885: cx23885[0]: fifo: 0x00005000 -> 0x6000
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067029] cx23885: cx23885[0]: ctrl: 0x00010630 -> 0x10690
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067030] cx23885: cx23885[0]:   ptr1_reg: 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067031] cx23885: cx23885[0]:   ptr2_reg: 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067032] cx23885: cx23885[0]:   cnt1_reg: 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067033] cx23885: cx23885[0]:   cnt2_reg: 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067033] cx23885: cx23885[0]: mpeg risc op code error
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067037] cx23885: cx23885[0]: TS2 C - dma channel status dump
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067038] cx23885: cx23885[0]:   cmds: init risc lo   : 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067039] cx23885: cx23885[0]:   cmds: init risc hi   : 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067040] cx23885: cx23885[0]:   cmds: cdt base       : 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067041] cx23885: cx23885[0]:   cmds: cdt size       : 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067042] cx23885: cx23885[0]:   cmds: iq base        : 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067043] cx23885: cx23885[0]:   cmds: iq size        : 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067044] cx23885: cx23885[0]:   cmds: risc pc lo     : 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067045] cx23885: cx23885[0]:   cmds: risc pc hi     : 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067046] cx23885: cx23885[0]:   cmds: iq wr ptr      : 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067047] cx23885: cx23885[0]:   cmds: iq rd ptr      : 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067048] cx23885: cx23885[0]:   cmds: cdt current    : 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067050] cx23885: cx23885[0]:   cmds: pci target lo  : 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067051] cx23885: cx23885[0]:   cmds: pci target hi  : 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067052] cx23885: cx23885[0]:   cmds: line / byte    : 0xffffffff
Nov 24 18:21:36 tvbuntu kernel: [ 2440.067052] cx23885: cx23885[0]:   risc0:

Neolo, 2018-12-13:
Looks like nobody cares...

b-rad-NDi, 2018-12-13:
The one single person monitoring this personal github account has a finite amount of time. There are discussions about this happening on the kernel mailing list.

b-rad-NDi, 2018-12-18:
A patch has been submitted that disables a "Ryzen specific fix" on non-ryzen systems. The "fix" hs been shown to have adverse affects on some other platforms, while being perfectly fine on lots of others. 
Hopefully you are a non Ryzen system and this patch solves your problems. We have customers with six of these in a single pc without issue, so there is no inherent problem.
The patch *should* make it into 4.21

Neolo, 2018-12-19:
Thank you. Have confirmed this issue or it’s a “blind” fix?
I don’t understand why Ryzen is considered to be majority and other platforms are irrelevant... In fact, it’s vice versa. I never used anything AMD.

b-rad-NDi, 2018-12-19:
The problem is a fix that a required fix on on family of platforms, which seemed innocuous on a lot of other platforms, was in fact quite detrimental on some others.

b-rad-NDi, 2019-01-21:
This should be fixed in mainline kernel as well as this ppa. Closing. Please re-open if you are still affected.

closed 2019-01-21 by b-rad-NDi