b-rad-NDi/Ubuntu-media-tree-kernel-builder #61: Debian support or build instructions?

tophneal, 2018-12-17:
Is it possible to add support for Debian Stretch? Alternatively, since I know that Ubuntu and Debian share a lot, is it possible to build the necessary packages for Debian, instead of using apt to install?

b-rad-NDi, 2018-12-17:
It's possible you can just download packages off launchpad and install them directly with dpkg. Kernel packages *should* be compatible between the two.

b-rad-NDi, 2018-12-17:
Also, using this system to build your own kernels should work fairly simply if the source is debian compatible. It's just a matter of setting up the git repo, then providing a set of patches to put on top. The LinuxTV.org media tree overlay patch is generated by this script. Then you just have to sort out build fixes and additional kernel config options.

b-rad-NDi, 2018-12-17:
I've thought about better genericizing that support...I'll add it to my queue :)

ltworf01, 2021-12-16:
Hi,

I understand, that this is a 3yr old thread. However, I was wondering, whether there is any chance to have this build and tested for Debian as well. 

I've managed to get a Hauppauge WinTV-soloHD stick to been recognized (installed firmware) and I'm able to scan for and find my channels. The trouble starts, when I try to actually watch TV. Using Kaffeine under Debian 11, every channel starts with video and audio, but the video freezes after just a few frames, whilst the audio continues. Using a manually generated (with w_scan) m3u playlist as channel list for VLC, the same happens for the channels, except for one, which for some reason shows synchronous video/audio. Additional trouble for me is, that, while the stick is attached to the system, my webcam device also stops working.

Going through all the logs, there doesn't seem anything to indicate, that the stick is not working properly. So I'm thinking, that maybe simply installing the firmware and Kaffeine or w_scan/VLC is still missing some parts, which are not availabe on Debian. I do not know, what the actual Ubuntu repository has on top of a Debian system.

I will setup a test system with Ubuntu in a VM and transfer it to my actual hardware to give it a try and rule out any possible hardware incompatibilities, but this will take some time.

I also had a look at the actual packages at "http://ppa.launchpad.net/b-rad/kernel+mediatree+hauppauge/ubuntu/pool/main/l/linux-hwe/?C=M;O=D", but honetsly, my Linux is still not strong enough to make heads or tails of all the packages and wildly starting and installing quickly lead to some unresolvable dependcy issues and I had to stop and revert.

So I was hoping, whether you could provide maybe any guidance or help, what one would actually need to install in order to make things work.

Thank you for consideration and happy holidays!

johnappletree, 2022-02-02:
I would also be interested in detailed build descriptions for Debian!

b-rad-NDi, 2022-02-02:
It's possible that this would require very little work on my part to make happen. I'll try and prioritize this tomorrow.

StephanGR, 2022-02-05:
> It's possible that this would require very little work on my part to make happen. I'll try and prioritize this tomorrow.

It will be a great idea, I'm running plex on my Debian NAS (OpenMediaVault), i installed the firmware, but when i'm looking for channel I got this error : 

```
Feb  5 11:49:51  Plex Media Server[1470157]: Entity: line 2: parser error : Input is not proper UTF-8, indicate encoding !
Feb  5 11:49:52 Plex Media Server[1470157]: Bytes: 0xE9 0x72 0x69 0x65
Feb  5 11:49:52 Plex Media Server[1470157]: tainer size="39"><Channel channelIdentifier="triplet://8442:10:2561" name="TF1 S
Feb  5 11:49:52 Plex Media Server[1470157]:                                                                                ^
Feb  5 11:49:52 Plex Media Server[1470157]: error : xmlTextWriterWriteDocCallback : XML error 9 !
Feb  5 11:49:52 Plex Media Server[1470157]: I/O error : write error
Feb  5 11:49:52 Plex Media Server[1470157]: error : xmlTextWriterCloseDocCallback : XML error 9 !
```

Btw if you have any idea about this error it would be great :)

Thanks

65881687132, 2022-07-14:
Would love to see this happen. Trying to use a Hauppage WinTV-quadHD USB with Plex in a CT in Proxmox, and it seems I have to install this on the host (Proxmox is Debian based) to get a CT to be able to pick up the tuner.

johnappletree, 2022-07-17:
@65881687132 this is exactly my setup as well. I would be very happy to see this happen.

javers99, 2022-07-24:
@b-rad-NDi any joy with further detailed instructions for Debian? Like many here, I'm keen to get my WinTV-QuadHD up and running under TrueNas (Debian Kernel 5.10) to use Plex / Tvheadend / NextPVR in dockers.

Many Thanks,

Jon

b0ggyb33, 2022-12-04:
I have the same requirements. WinTV-QuadHD under Debian, kernel 5.10.
I had previously downloaded dvb-demod-si2168-d60-01.fw (not b40, as dmesg told me it wanted d60, this may be another issue) which allowed me to scan all channels except the HD ones. 
I thought I would try the PPA as apparently this fixes the HD channel issue.

I managed to force apt to install the packages by setting the /etc/apt/source for this PPA to a valid Ubuntu version (I chose bionic somewhat arbitrarily) and installed linux-mediatree and linux-firmware-hauppauge. I still cannot scan the HD channels.

I will freely admit I have little idea what I'm doing, but I thought I'd share how far I've got with this.