b-rad-NDi/Ubuntu-media-tree-kernel-builder #62: Dual-tuner 955D issue - only one tuner functional

th0ma7, 2018-12-18:
First of all, I'm using a winTV tuner on my synology from which I passthrough the usa adapter to a VM running under the synology nas which may have major implications on the expected end-result.  I can probably test things out without VM layer elsewhere as needed.

Hardware: 1x hauppauge 955D dual tuner. 
kernel: $ uname -a
Linux th0ma7-TV 4.15.0-38201811061723-generic #0+mediatree+hauppauge-Ubuntu SMP Tue Nov 6 19:03:14 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

Since using your kernel PPA a second is now being found e.g. now the OS sees 2x ATSC-C and 2x ATSC-T inputs in tvheadend.  Although only one of the two is actually functional.  Prior to upgrading to the ppa kernel only 1x ATSC-C + 1x ATSC-T tuners was being found.

Environment and dmesg is the same as #60 

Thnx in advance!

th0ma7, 2018-12-19:
Aaaahh.. !  You can't imagine how much I'd love testing this out!!
I noticed that currnet DSM does have a `si2157.ko` driver & the latest firmware... so I guess it's "mostly" a matter of having an updated `si2157.ko` driver compatible with syno's kernel... using your backport magic.

Perhaps Santa will be kind to me this year ;)

Again, thnx for the heads-up.

th0ma7, 2019-01-19:
Any luck at looking into this? :)
Cheers!

b-rad-NDi, 2019-01-19:
No, but point me at your specific model and I'll start next week.
I have:
- ds115j (ancient)
- ds1515+

ds115j I already have working, and if I recall it uses kernel 3.10, so all other models shouldn't be an issue. The only tricky part will be overriding synos modules, my US models do not come with any dvb drivers at all so I can't experiment with them.

th0ma7, 2019-01-19:
I have a ds918+ which is running a 4.4 kernel.  See the following output:
```
$ uname -a
Linux th0ma7-nas 4.4.59+ #23824 SMP PREEMPT Fri Oct 12 18:31:34 CST 2018 x86_64 GNU/Linux synology_apollolake_918+
```
Thnx for looking into this! And don't hesitate to let me know how I can help.  Have a good week-end!

th0ma7, 2019-02-09:
Out of curiosity, any updates on this front or anything I can do to help you out?
Cheers!

th0ma7, 2019-03-17:
pinging again, anything I can to to help out?  I'd really much like to get this working on my NAS, last missing piece so I can cut-the-cord efficiently.  Thnx in advance and Happy St-Patrick day!  Cheers!

th0ma7, 2019-04-22:
I was able to make some mileage on my own but ended-up hitting a wall.

Using synology 4.4x kernel sources with appropriate toolchain
https://sourceforge.net/projects/dsgpl/files/Synology%20NAS%20GPL%20Source/22259branch/apollolake-source/

To which I backport the following patches:
001 - https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/patch/?id=4f75189024f4186a7ff9d56f4a8cb690774412ec
002 - https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/patch/?id=11a2a949d05e9d2d9823f0c45fa476743d9e462b
003 - https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/patch/?id=1586342e428d80e53f9a926b2e238d2175b9f5b5

Adding the second tuner, using this patch ended-up creating build errors while trying to backport it:
004 - https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/patch/?id=be7fd3c3a8c5e9acbc69f887ca961df5e68cf6f0

I got to the point that I now have em28xx.ko and em28xx-dvb.ko kernel drivers and also able to load them in my system.  I then noticed I also needed a dvb-frontend, namely lgdt3306a.ko but ended-up getting build errors on that specific modules (due to patch 001).

I have an rc script that loads all the necessary drivers at boot-up (now located in /usr/local/etc/rc.d/ with DSM6.1+):
```
#!/bin/sh

# Load mandatory modules
sudo insmod /lib/modules/dvb-core.ko
sudo insmod /lib/modules/rc-core.ko
sudo insmod /lib/modules/dvb-usb.ko
sudo insmod /lib/modules/videodev.ko
sudo insmod /lib/modules/v4l2-common.ko
sudo insmod /lib/modules/tveeprom.ko

# Load Hauppauge updated drivers
sudo insmod /usr/local/lib/modules/$(uname -r)/em28xx.ko
sudo insmod /usr/local/lib/modules/$(uname -r)/em28xx-dvb.ko
```
Here is the synology output:
```
$ dmesg | grep -i -e em28xx -e dvb
[  110.428728] em28xx: module verification failed: signature and/or required key missing - tainting kernel
[  110.440082] em28xx: New device HCW 955D @ 480 Mbps (2040:026d, interface 0, class 0)
[  110.449305] em28xx: DVB interface 0 found: isoc
[  110.454903] em28xx: chip ID is em28174
[  111.694148] tveeprom 8-0050: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[  111.735490] em28174 #0: dvb set to isoc mode.
[  111.740473] usbcore: registered new interface driver em28xx
[  111.965448] em28174 #0: Binding DVB extension
[  111.982927] em28xx: Registered (Em28xx dvb Extension) extension
[ 1336.830376] em28xx: New device HCW 955D @ 480 Mbps (2040:026d, interface 0, class 0)
[ 1336.839049] em28xx: DVB interface 0 found: isoc
[ 1336.844190] em28xx: chip ID is em28174
[ 1338.034666] tveeprom 8-0050: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[ 1338.061566] em28174 #0: dvb set to isoc mode.
[ 1338.066450] em28174 #0: Binding DVB extension
```

Now, the theory is, if I can get a working backport patch allowing to get lgdt3306a.ko to build along with getting the second tuner to work, I should then be able to get it all running ok natively on the synology instead of issues hit using VM...   Help on getting there would be much appreciated.

th0ma7, 2019-05-04:
Gave a few more cycles on this (might be useful someday for someone else).

I was able to backport lgdt3306a patches from 4.4.x patches found in your repository.
I posted the results on my own github at:
https://github.com/th0ma7/synology/tree/master/hauppauge

- 1st tuner detected all right
- firmware seems to be loaded ok
- lgdt3306a crash (see info below)

_(disclaimer: not a C expert nor linux media driver coder)_
I believe the updated lgdt3306a makes calls to i2c mux in order to allow getting the second tuner accessible and as I haven't yet backported the second tuner patches to em28xx making use of that it ends-up crashing at lgdt3306a module loading time.

Kernel output can be found at:
https://github.com/th0ma7/synology/blob/master/hauppauge/dmesg.20190504-01

Next: figure how to backport second tuner functionality to em28xx and hopefully resolve the crashing issue.

th0ma7, 2019-05-07:
WOOOOOTTTT!  Got the second tuner to work!

Patches available here:
1st tuner:
https://github.com/th0ma7/synology/blob/master/hauppauge/002-Hauppauge955D-em28xx-Tuner1.patch
2nd tuner:
https://github.com/th0ma7/synology/blob/master/hauppauge/003-Hauppauge955D-em28xx-Tuner2-v6.patch

Kernel output (after loading em28xx-dvb:
```
[  189.248503] em28xx: module verification failed: signature and/or required key missing - tainting kernel
[  189.260107] em28xx: New device HCW 955D @ 480 Mbps (2040:026d, interface 0, class 0)
[  189.268829] em28xx: DVB interface 0 found: isoc
[  189.273967] em28xx: chip ID is em28174
[  190.394271] em28174 #0: EEPROM ID = 26 00 01 00, EEPROM hash = 0x3d790eca
[  190.401884] em28174 #0: EEPROM info:
[  190.408069] em28174 #0: 	microcode start address = 0x0004, boot configuration = 0x01
[  190.425253] em28174 #0: 	AC97 audio (5 sample rates)
[  190.430819] em28174 #0: 	500mA max power
[  190.435232] em28174 #0: 	Table at offset 0x27, strings=0x0a72, 0x187c, 0x086a
[  190.443398] em28174 #0: Identified as Hauppauge WinTV-dualHD 01595 ATSC/QAM (card=100)
[  190.453539] tveeprom 8-0050: Hauppauge model 204101, rev B2I6, serial# 11584195
[  190.461750] tveeprom 8-0050: tuner model is SiLabs Si2157 (idx 186, type 4)
[  190.469571] tveeprom 8-0050: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[  190.481769] tveeprom 8-0050: audio processor is None (idx 0)
[  190.488127] tveeprom 8-0050: has no radio, has IR receiver, has no IR transmitter
[  190.496516] em28174 #0: dvb set to isoc mode.
[  190.501480] em28xx: chip ID is em28174
[  191.621200] em28174 #1: EEPROM ID = 26 00 01 00, EEPROM hash = 0x3d790eca
[  191.628816] em28174 #1: EEPROM info:
[  191.632901] em28174 #1: 	microcode start address = 0x0004, boot configuration = 0x01
[  191.647736] em28174 #1: 	AC97 audio (5 sample rates)
[  191.653310] em28174 #1: 	500mA max power
[  191.657698] em28174 #1: 	Table at offset 0x27, strings=0x0a72, 0x187c, 0x086a
[  191.665857] em28174 #1: Identified as Hauppauge WinTV-dualHD 01595 ATSC/QAM (card=100)
[  191.675762] tveeprom 10-0050: Hauppauge model 204101, rev B2I6, serial# 11584195
[  191.684103] tveeprom 10-0050: tuner model is SiLabs Si2157 (idx 186, type 4)
[  191.692008] tveeprom 10-0050: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[  191.704284] tveeprom 10-0050: audio processor is None (idx 0)
[  191.710726] tveeprom 10-0050: has no radio, has IR receiver, has no IR transmitter
[  191.719287] em28xx: dvb ts2 set to isoc mode.
[  191.924358] usbcore: registered new interface driver em28xx
[  249.474012] em28174 #0: Binding DVB extension
[  249.480377] em28174 #1: Binding DVB extension
[  249.486629] em28xx: Registered (Em28xx dvb Extension) extension
```

Sadly the dvb-frontend lgdt3306a still crashes (help welcomed):
```
$ sudo insmod /usr/local/lib/modules/$(uname -r)/lgdt3306a.ko
Killed
```

Current patch available here:
https://github.com/th0ma7/synology/blob/master/hauppauge/001-Hauppauge955D-lgdt3306a-v2.patch

With the following kernel output:
```
[  290.305588] i2c i2c-8: Added multiplexed i2c bus 11
[  290.311094] BUG: unable to handle kernel NULL pointer dereference at           (null)
[  290.319852] IP: [<ffffffffa0e54d38>] lgdt3306a_probe+0xc8/0x140 [lgdt3306a]
[  290.327637] PGD 30360d067 PUD 3037ce067 PMD 0 
[  290.332609] Oops: 0002 [#1] PREEMPT SMP 
[  290.336994] Modules linked in: lgdt3306a(OE+) em28xx_dvb(OE) em28xx(OE) pci_stub vhost_net(O) tun kvm_intel kvm irqbypass xt_ipvs ip_vs_rr ip_vs xt_mark iptable_mangle br_netfilter bridge stp aufs macvlan veth xt_conntrack xt_addrtype nf_conntrack_ipv6 ip6table_filter ip6_tables ipt_MASQUERADE xt_REDIRECT nf_nat_masquerade_ipv4 xt_nat vhost_scsi(O) vhost(O) iptable_nat nf_nat_ipv4 nf_nat_redirect nf_nat xt_recent xt_iprange xt_limit xt_state xt_tcpudp xt_multiport xt_LOG nf_conntrack_ipv4 nf_defrag_ipv4 iptable_filter ip_tables x_tables tveeprom v4l2_common videodev dvb_usb rc_core dvb_core cifs udf isofs loop nfsd exportfs rpcsec_gss_krb5 tcm_loop(O) iscsi_target_mod(O) syno_hddmon(P) target_core_ep(O) target_core_multi_file(O) target_core_file(O) target_core_iblock(O) target_core_mod(O) syno_extent_pool(PO)
[  290.417230]  hid_generic rodsp_ep(O) usbhid hid usblp usb_storage uhci_hcd ehci_pci ehci_hcd flashcache_syno(O) flashcache(O) syno_flashcache_control(O) openvswitch gre nf_defrag_ipv6 nf_conntrack apollolake_synobios(PO) leds_lp3943 exfat(O) btrfs i915 drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops cfbfillrect cfbcopyarea cfbimgblt drm fb fbdev intel_agp intel_gtt agpgart video backlight button synoacl_vfs(PO) hfsplus md4 hmac r8168(O) igb(O) i2c_algo_bit e1000e(O) vxlan ip6_udp_tunnel udp_tunnel fuse vfat fat crc32c_intel aesni_intel glue_helper lrw gf128mul ablk_helper arc4 cryptd ecryptfs sha256_generic ecb aes_x86_64 authenc des_generic ansi_cprng cts md5 cbc cpufreq_powersave cpufreq_performance acpi_cpufreq processor cpufreq_stats dm_snapshot dm_bufio crc_itu_t crc_ccitt quota_v2
[  290.495048]  quota_tree psnap p8022 llc sit tunnel4 ip_tunnel ipv6 zram sg etxhci_hcd xhci_pci xhci_hcd usbcore usb_common [last unloaded: apollolake_synobios]
[  290.509597] CPU: 1 PID: 23933 Comm: insmod Tainted: P           OE   4.4.59+ #24922
[  290.518153] Hardware name: Synology DS918+/DS918+, BIOS M.033 12/06/2018
[  290.525639] task: ffff88006f5443c0 ti: ffff8802fbaf4000 task.ti: ffff8802fbaf4000
[... cut ...]
[  292.846523]  [<ffffffff810003b6>] do_one_initcall+0x86/0x1b0
[  292.852845]  [<ffffffff810e1b48>] do_init_module+0x56/0x1be
[  292.859069]  [<ffffffff810b7d8d>] load_module+0x1dfd/0x2080
[  292.865291]  [<ffffffff810b51f0>] ? __symbol_put+0x50/0x50
[  292.871419]  [<ffffffff810b8199>] SYSC_finit_module+0x79/0x80
[  292.877838]  [<ffffffff810b81b9>] SyS_finit_module+0x9/0x10
[  292.884061]  [<ffffffff8156a58a>] entry_SYSCALL_64_fastpath+0x1e/0x92
[  292.891256] ---[ end trace c46c327cea5021c5 ]---
```

b-rad-NDi, 2019-05-07:
hi @th0ma7, sorry to leave you fighting this issue on your own for so long. My schedule and workload has been very full and quite hectic. I just finished my upstream work for our "new" satellite demodulator though, so that is a big issue taken care of. Tomorrow I will (finally) upload my NAS-mediadrivers repo and add support for Synology. Help is on the way :)

th0ma7, 2019-05-07:
me = smile (you just can't imagine how much)
Can't wait to give that a shot (but geez, my own schedule is packed until next week)!

Still, I think I'm not missing much to make the magic happen... 
I believe I'm a few bits away from having lgdt3306a functional as well.
BTW, DSM 7.0 is coming out shortly; should (hopefully) simplify greatly the backporting process depending of the new kernel synology will provide.

th0ma7, 2019-05-11:
Until your source code / modules are made available I've been able to find a few cycles and documented where the current status of my work stands on my github:
https://github.com/th0ma7/synology

For my own learning experience I'll continue working on lgdt3306a crash (and might find some clues from your source code once available?).  When backporting I've done I'll I could to keep all other modules untouched.  Therefore I've tried re-using media-i2c from original 4.4 to limit impact over other provided modules from synology and I guess this is where it currently fails.

th0ma7, 2019-05-14:
Got things to work up to tvheadend detecting both tuners.
One main thing I found is that the lgdt3306a crashes in two different occasions:
1.  when the device is also share with my running VM (probably shouldn't crash but on the other trying to force the usage of it while there is a USB passthrough to QEMU using it... not a so good idea).
2. At boot time.

Added a few traces and found this:
```
[  114.701465] DEBUG: Passed lgdt3306a_probe 2353
[  114.715373] BUG: unable to handle kernel NULL pointer dereference at           (null)
[  114.724132] IP: [<ffffffffa0d92dbc>] lgdt3306a_probe+0x14c/0x1b0 [lgdt3306a]
[  114.732017] PGD 442e92067 PUD 44191b067 PMD 0
[  114.736989] Oops: 0002 [#1] PREEMPT SMP
[  114.741366] Modules linked in: lgdt3306a(OE+) em28xx_dvb(OE) xt_ipvs ip_vs_rr ip_vs xt_mark iptable_mangle br_netfilter bridge stp aufs macvlan veth xt_conntrack x
...
```
This correspond to:
```
2351         printk(KERN_ALERT "DEBUG: Passed %s %d \n",__FUNCTION__,__LINE__);
2352         fe->ops.i2c_gate_ctrl = NULL;
2353         printk(KERN_ALERT "DEBUG: Passed %s %d \n",__FUNCTION__,__LINE__);
2354         *config->i2c_adapter = state->i2c_adap;
2355         printk(KERN_ALERT "DEBUG: Passed %s %d \n",__FUNCTION__,__LINE__);
2356         *config->fe = fe;
2357         printk(KERN_ALERT "DEBUG: Passed %s %d \n",__FUNCTION__,__LINE__);
```

So at boot time it fails right after 2353:
`2354         *config->i2c_adapter = state->i2c_adap;`

And when shared it fails right after that after 2355:
`2356         *config->fe = fe;`

th0ma7, 2019-05-14:
Changing the module loading order solved the issue (still I guess a if = NULL exit should be added)

```
# Load mandatory modules
sudo insmod /lib/modules/dvb-core.ko
sudo insmod /lib/modules/rc-core.ko
sudo insmod /lib/modules/dvb-usb.ko
sudo insmod /lib/modules/videodev.ko
sudo insmod /lib/modules/v4l2-common.ko
sudo insmod /lib/modules/tveeprom.ko

# Load Hauppauge necessary tuners, dvb-frontend and updated drivers
sudo insmod /lib/modules/si2157.ko
sudo insmod /usr/local/lib/modules/$(uname -r)/media.ko
sudo insmod /usr/local/lib/modules/$(uname -r)/lgdt3306a.ko
sudo insmod /usr/local/lib/modules/$(uname -r)/em28xx.ko
sudo insmod /usr/local/lib/modules/$(uname -r)/em28xx-dvb.ko
```

Rsulting in:
```
[  183.149627] Linux video capture interface: v2.00
[  191.687921] media: module verification failed: signature and/or required key missing - tainting kernel
[  191.698955] media: Linux media interface: v0.10
[  198.215178] em28xx: New device HCW 955D @ 480 Mbps (2040:026d, interface 0, class 0)
[  198.223873] em28xx: DVB interface 0 found: isoc
[  198.229080] em28xx: chip ID is em28174
[  199.349152] em28174 #0: EEPROM ID = 26 00 01 00, EEPROM hash = 0x3d790eca
[  199.356747] em28174 #0: EEPROM info:
[  199.360743] em28174 #0: 	microcode start address = 0x0004, boot configuration = 0x01
[  199.375481] em28174 #0: 	AC97 audio (5 sample rates)
[  199.381031] em28174 #0: 	500mA max power
[  199.385412] em28174 #0: 	Table at offset 0x27, strings=0x0a72, 0x187c, 0x086a
[  199.393509] em28174 #0: Identified as Hauppauge WinTV-dualHD 01595 ATSC/QAM (card=100)
[  199.403420] tveeprom 8-0050: Hauppauge model 204101, rev B2I6, serial# 11584195
[  199.411596] tveeprom 8-0050: tuner model is SiLabs Si2157 (idx 186, type 4)
[  199.419378] tveeprom 8-0050: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[  199.431542] tveeprom 8-0050: audio processor is None (idx 0)
[  199.437868] tveeprom 8-0050: has no radio, has IR receiver, has no IR transmitter
[  199.446252] em28174 #0: dvb set to isoc mode.
[  199.451174] em28xx: chip ID is em28174
[  200.571226] em28174 #1: EEPROM ID = 26 00 01 00, EEPROM hash = 0x3d790eca
[  200.578835] em28174 #1: EEPROM info:
[  200.582844] em28174 #1: 	microcode start address = 0x0004, boot configuration = 0x01
[  200.597682] em28174 #1: 	AC97 audio (5 sample rates)
[  200.603234] em28174 #1: 	500mA max power
[  200.607617] em28174 #1: 	Table at offset 0x27, strings=0x0a72, 0x187c, 0x086a
[  200.615742] em28174 #1: Identified as Hauppauge WinTV-dualHD 01595 ATSC/QAM (card=100)
[  200.625468] tveeprom 10-0050: Hauppauge model 204101, rev B2I6, serial# 11584195
[  200.633753] tveeprom 10-0050: tuner model is SiLabs Si2157 (idx 186, type 4)
[  200.641636] tveeprom 10-0050: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[  200.653886] tveeprom 10-0050: audio processor is None (idx 0)
[  200.660313] tveeprom 10-0050: has no radio, has IR receiver, has no IR transmitter
[  200.668777] em28xx: dvb ts2 set to isoc mode.
[  200.873825] usbcore: registered new interface driver em28xx
[  204.639538] em28174 #0: Binding DVB extension
[  204.648646] i2c i2c-8: Added multiplexed i2c bus 11
[  204.654106] DEBUG: Passed lgdt3306a_probe 2351 
[  204.659176] DEBUG: Passed lgdt3306a_probe 2353 
[  204.664244] DEBUG: Passed lgdt3306a_probe 2355 
[  204.669313] DEBUG: Passed lgdt3306a_probe 2357 
[  204.674381] lgdt3306a 8-0059: LG Electronics LGDT3306A successfully identified
[  204.682624] DEBUG: Passed lgdt3306a_probe 2360 
[  204.689844] si2157 11-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[  204.698531] DVB: registering new adapter (em28174 #0)
[  204.704189] usb 1-3: DVB: registering adapter 0 frontend 0 (LG Electronics LGDT3306A VSB/QAM Frontend)...
[  204.716587] em28174 #0: DVB extension successfully initialized
[  204.723118] em28174 #1: Binding DVB extension
[  204.732352] i2c i2c-10: Added multiplexed i2c bus 12
[  204.738120] DEBUG: Passed lgdt3306a_probe 2351 
[  204.743806] DEBUG: Passed lgdt3306a_probe 2353 
[  204.749655] DEBUG: Passed lgdt3306a_probe 2355 
[  204.754831] DEBUG: Passed lgdt3306a_probe 2357 
[  204.759982] lgdt3306a 10-000e: LG Electronics LGDT3306A successfully identified
[  204.768201] DEBUG: Passed lgdt3306a_probe 2360 
[  204.775201] si2157 12-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[  204.783877] DVB: registering new adapter (em28174 #1)
[  204.789631] usb 1-3: DVB: registering adapter 1 frontend 0 (LG Electronics LGDT3306A VSB/QAM Frontend)...
[  204.800949] em28174 #1: DVB extension successfully initialized
[  204.807486] em28xx: Registered (Em28xx dvb Extension) extension
[  204.939523] si2157 11-0060: found a 'Silicon Labs Si2157-A30'
[  204.996063] si2157 11-0060: firmware version: 3.0.5
[  205.001647] usb 1-3: DVB: adapter 0 frontend 0 frequency 0 out of range (55000000..858000000)
[  205.204421] si2157 12-0062: found a 'Silicon Labs Si2157-A30'
[  205.261090] si2157 12-0062: firmware version: 3.0.5
[  205.266576] usb 1-3: DVB: adapter 1 frontend 0 frequency 0 out of range (55000000..858000000)
```

b-rad-NDi, 2019-05-14:
I finally push this repository up. I had to think of how to genericize the building procedure, and my initial work was rather hackish.
https://github.com/b-rad-NDi/Embedded-MediaDrivers
The first family supported is Western Digital EX family. I am now adding WD PR family. When that builds, I'll pull down the Synology source for your device and integrate it.

b-rad-NDi, 2019-05-14:
Note this is rough first draft and will require a little hand holding for a brief bit.

th0ma7, 2019-05-15:
If that can be of any help feel free to use the documentation I did on my git repository.  I think it goes into much of the required steps which should be "somewhat" similar using your modules.

b-rad-NDi, 2019-05-16:
I added support for Synology Avoton (3.10) and Apollolake (4.4) architectures. There wasn't really required anything special for Apollolake.
Drivers produced by my build system are backported mainline drivers.
Closing this issue, since the new project is better suited to producing media drivers for arbitrary kernels, without installing a new kernel.

closed 2019-05-16 by b-rad-NDi