b-rad-NDi/Ubuntu-media-tree-kernel-builder #64: Hauppauge WinTV-dualHD Ubuntu on Raspberry Pi

darrenswhite, 2019-01-25:
Any chance of support for Ubuntu on Raspberry Pi? I'm running Ubuntu 16.04 with 4.4.0 kernel. Installing via the PPA doesn't work as it requires a different kernel.

Output from dmesg:
```
[    5.255276] usb 1-1.4: new high-speed USB device number 4 using dwc_otg
[    5.351816] usb 1-1.4: New USB device found, idVendor=2040, idProduct=0265
[    5.360488] usb 1-1.4: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[    5.369567] usb 1-1.4: Product: dualHD
[    5.374967] usb 1-1.4: Manufacturer: HCW
[    5.380485] usb 1-1.4: SerialNumber: 0014020771
```

Kernel version:
```
4.4.0-1102-raspi2
```

Output from PPA install:
```
» sudo apt-get install linux-image-mediatree
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 linux-image-mediatree : Depends: linux-image-4.4.0-141201901161620-generic but it is not installable
                         Depends: linux-image-extra-4.4.0-141201901161620-generic but it is not installable
E: Unable to correct problems, you have held broken packages.
```

darrenswhite, 2019-01-26:
Installed Raspbian instead and it's working fine now.

closed 2019-01-26 by darrenswhite

b-rad-NDi, 2019-01-28:
They are pretty good at kernel updates, glad to hear it's working now.

parkerlreed, 2019-01-28:
How well does the DualHD perform on the Pi? I know it has MPEG2 royalty free now so that should help. Are you able to view/record on both tuners at the same time?

darrenswhite, 2019-01-28:
Currently only 1 tuner seems to be working. I'm using it with Plex.