b-rad-NDi/Ubuntu-media-tree-kernel-builder #65: Ubuntu 18.04 (Bionic, LTS) hwe packages

sindastra, 2019-02-08:
Hello, Ubuntu 18.04 (Bionic, LTS) has hwe available (package: linux-generic-hwe-18.04).

I require the hwe kernel/headers. Any plans to create a hwe version of linux-image-mediatree and linux-headers-mediatree?

Kind regards

b-rad-NDi, 2019-02-08:
Will try to get it supported next week.

b-rad-NDi, 2019-02-14:
Building on launchpad now, they'll land in the main ppa tonight

sindastra, 2019-02-14:
Will it be stable and safe to install on a production machine?

b-rad-NDi, 2019-02-14:
That is entirely up to you. There shouldn't be any inherent stability issues introduced. The entire media tree is integrated into the rest of the kernel, while keeping API's outside of the media tree as they are in the corresponding version. The base Ubuntu kernel gets all the security patches applied to it, and the media tree is pulled from mainline so gets any appropriate security patches whenever it is sync'd.

sindastra, 2019-02-15:
I will try to upgrade tomorrow. What are the names of the packages for the hwe version? And will it automatically remove the non-hwe version when installing, or is a manual removal required?

b-rad-NDi, 2019-02-15:
linux-hwe-mediatree-images
linux-hwe-mediatree-headers

You will have to remove whatever stock kernels and virtual packages you desire to yourself.

b-rad-NDi, 2019-02-15:
please re-open if there are any issues.

closed 2019-02-15 by b-rad-NDi

sindastra, 2019-02-15:
E: Unable to locate package linux-hwe-mediatree-images
E: Unable to locate package linux-hwe-mediatree-headers

Do I need a special ppa for hwe?

b-rad-NDi, 2019-02-16:
oops apologies i sydlexic'd it up.
```
linux-hwe-headers-mediatree
linux-hwe-image-mediatree
```
The same ppa holds all packages:
https://launchpad.net/~b-rad/+archive/ubuntu/kernel+mediatree+hauppauge

sindastra, 2019-02-16:
Upgraded to the new packages, seems to be working so far.

sindastra, 2019-03-13:
Will the HWE packages get updated too? They're version `4.18.0-16201902141402` since release if I'm correct. Just asking because that's important because of security updates and such.

b-rad-NDi, 2019-03-13:
Didn't notice a couple kernels were updated, i'll push new builds tonight.