b-rad-NDi/Ubuntu-media-tree-kernel-builder #67: State of Hauppauge USB Live 2 on Raspberry Pi

mk-fg, 2019-02-23:
Hi,

I've been trying to setup Hauppauge USB Live 2 on RPi3 board with fairly recent 4.19.23 kernel.
It works there with transfer_mode=0 parameter and hangs whole board with transfer_mode=1.

Trying to capture video from it though, only able to get any frames from this driver on very low resolutions, e.g. 320x200, while setting everything higher that output zero frames to ffmpeg or mpv.
(example: `mpv --vo=rpi tv:// --tv-width 640 --tv-height 480` or `ffmpeg -y -f v4l2 -video_size 720x480 -i /dev/video0 -preset ultrafast output.mp4` both get 0 frames)

Thinking that it might be an issue with RPi's USB performance (esp. given that transfer_mode= above makes big difference), tried to lower the FPS of the input (down from 25), but `v4l2-ctl --set-parm 12` and such get inevitable "VIDIOC_S_PARM: failed: Inappropriate ioctl for device" error.

So wanted to ask here if there are any open issues or applicable patches to try on top of 4.19 that might help with Live 2 model, or is it simply not supported (yet?) on RPi, or maybe not possible to support there due to firmware/hardware limitations (e.g. fixed FPS too much for USB bus there)?
Is WinTV-HVR-955Q model being the only well-supported one on RPi for somesuch technical reason?

Looking through media tree patches for cx231xx, as well as ones in "tip" here, there seem to be only recent init-related fix, which probably won't help, as that's not the problem here.

Would appreciate any information or suggestions on the issue.

b-rad-NDi, 2019-03-05:
I'll try and repro your issue tomorrow. Haven't heard of this issue myself. As for 955Q being "well supported", it is no more supported than any other device. The USB Live2 is pretty much a 955Q without the tuner parts. It uses the exact same drivers.

mk-fg, 2019-03-06:
Thanks a lot for the information and looking into it.

I was indeed able to reproduce same exact issue with Hauppauge WinTV-HVR-955Q (111401) (`tveeprom: Hauppauge model 111401, rev E3I6, serial# 4036075122`) before seeing that comment.

If you won't be able to get same result, would appreciate if you'd post which platform and kernel you've used, so that I can maybe try synchronizing kernel on RPi to that (i.e. cherry-pick or revert any patches that made it into raspberrypi 4.19.x branch).

mk-fg, 2019-03-13:
> I'll try and repro your issue tomorrow.

Wonder if you got around to trying it on RPi?

Would be very interested in knowing both if it works or fails, as at this point I've tried earlier 4.4 and 4.9 kernels (without any extra cx231xx-specific patches) to rule-out regression in 4.19, and they both fail with same exact symptoms.
Using default transfer_mode=1 doesn't always hang the board, and when it doesn't, grabber chip seem to hang instead, producing `cx231xx 1-1.2:1.1: cx231xx_send_usb_command: failed with status --71` errors from loaded module and failing to reinitialize card/module until it is power-cycled.
Searching for cx231xx + these USB errors digs up plenty of old threads, but guess it's just a common "device becomes unresponsive" outcome from different issues.

Might still try to rule-out power issues, but otherwise such results seem to suggest that card might've only worked on RPi boards with 4.1 or earlier kernels (not sure if these might predate RPi3) or never worked with higher resolutions there at all (while stuff like 320x200 seem to work fine, as mentioned).
Well, either that or I'm using it completely wrong somehow :)

b-rad-NDi, 2019-03-13:
Ok, just connected up an rpi3 + usb live2. I'll stretch an hdmi cable over now and try it out.

b-rad-NDi, 2019-03-14:
I've installed debian stretch with 4.14 kernel, but cannot run your command lines. The mpv provided does not support several of the options you use. Going to try other methods.

b-rad-NDi, 2019-03-14:
ooops lol, didn't notice that was two individual commands. My c&p had your "or" as a parameter.

mk-fg, 2019-03-14:
Oh, I was about to suggest ffmpeg, which should probably work.
And same thing with vlc, though v4l2 plugin there might optionally depend on libzvbi.

mk-fg, 2019-03-14:
Oh, also, not sure if mpv on stock armhf debian might be compiled with `--vo=rpi` for mpv, which is kinda niche, so that might not work, but it should be safe to drop that option and let it pick whatever default it wants.

b-rad-NDi, 2019-03-14:
Ok, I definitely see a ton of errors in dmesg when trying to capture. I have my usblive2 going through a usb3 powered hub to ensure there was no power issues either. I'll have to look into this further, there shouldn't be a reason for it to fail like this.

mk-fg, 2019-03-14:
Thanks for confirming that I'm not going insane :)
(as such issues seem to date back to 2011 and 2015, so definitely looked like maybe I'm doing something wrong there)

b-rad-NDi, 2019-03-14:
Unreported to me so far, I'll add this to my very long queue.

b-rad-NDi, 2019-03-14:
btw, I've used cx231xx analog capture on imx devices without issue, as well as pc, so this is not a universal issue. That might make it a little bit harder to diagnose though.

mk-fg, 2019-03-14:
Yeah, figured that much from the fact that it went unreported like that.
If anything, I'd randomly blame weird RPi-specific USB driver, which was a cause for bunch of issues on that board in the past, though seen same issue reported on other ARM boards as well (e.g. https://linux-media.vger.kernel.narkive.com/W9abY1r5/cx231xx-support-for-arm-omap-working-at-all or http://www.cubieforums.com/index.php?topic=3236.0 ), so maybe not.

Will report here if it'll work with 4.1 if I'll manage to run that on RPi3 as mentioned, but otherwise will probably end up running device with Intel NUC x86, where guess it won't have any issues.

b-rad-NDi, 2019-03-14:
On kernel 4.19 it freezes my rpi3 completely.

mk-fg, 2019-03-14:
Don't think it always froze for me with 4.19/4.14, but quite often, also only with higher resolutions.
Loading module with transfer_mode=0 fixes the freezing, just giving 0 frames with same parameters instead, but not sure if maybe driver isn't supposed to work right in that non-default mode anyway.

b-rad-NDi, 2019-03-14:
I've never had to do anything by default except set input and standard, the driver should "just work". Interestingly/sadly those omap errors from six years ago are the exact same as what i'm seeing today.

mk-fg, 2019-03-14:
Yeah, though endianness patch at the end of that thread is long-applied, as well as patches from https://git.linuxtv.org/hverkuil/media_tree.git/log/?h=cx231xx afaict.

And these usb errors seem to pop-up on x86 as well (e.g. https://bugzilla.kernel.org/show_bug.cgi?id=43332 ), hence my assumption that they're just common symptom of chip going into invalid state due to one of maybe dozen different communication issues with the fw.

But admittedly I didn't look much beyond that, don't know anything about these devices and none of these logs have debug modes enabled in the module to have more info on similarity.

mk-fg, 2019-03-14:
Just tried linux-4.1.21 branch of RPi-specific linux tree, and capturing at higher resolutions there indeed works !!! But only once :)
Right after closing ffmpeg stream, same usb -71 errors pop-up and from there device goes unresponsive.

So guess it worked better on RPi few years ago, and maybe going back further or switching to a more period-accurate RPi board (e.g. RPi2) might fix the problems entirely and give strating point for kernel/driver bisection. Oh well.

b-rad-NDi, 2019-03-14:
That is one hell of a bisection lol. That kinda sucks that something happened in their usb code to mess up cx231xx so much since then. I have lots of rpi's so maybe i'll spin up a container to slowly diagnose the issue.

mk-fg, 2019-03-16:
Few more notes from trying 955Q plugged through powered USB hub with kernel 4.19.27:

- ffmpeg sometimes works, recording video, but always only once, getting usb errors on the second try.
  Not sure why I couldn't get this result without USB hub, given other observations, guess hub affects something on USB protocol level.

- Other times, one of these happens (was able to get one of each in ~5 reboots):

  - Board hangs and reboots. It is definitely not kernel panic, since it's hw watchdog that does rebooting after 14s, and not longer panic=300 timeout.

  - ffmpeg gets no frames, but no other errors are logged anywhere. This also happens 100% reliably with transfer_mode=0.

  - Same as above, but onboard WiFi immediately disconnects (and reconnects) when I run ffmpeg.
    Seems strange as to why'd it happen with powered USB hub, and given that wifi is not connected to USB on RPi3 (unlike e.g. ethernet card).

- Reloading driver(s) after USB errors via rmmod/modprobe does not help - reports `cx231xx: probe of 1-1.2.1:1.1 failed with error -71` on init and doesn't work.

- When USB errors happen, rebooting RPi always gets rid of these, yet USB hub doesn't loose power when RPi reboots, so all these errors are purely on RPi side (hardware or software) and seem to have nothing to do with 955Q.

If nothing else, rules-out power issues (which are otherwise kinda common with RPis), and seem to once again point at something wrong with RPi 3B+ USB stack or kernel interfering with driver operation on that particular platform.

Have no other plans to test it with RPi again, so thought to post these last results and a kind of summary.
They seem to match what you had with similar setup, thanks again for confirming it independently.

Not sure if this issue needs to be open, feel free to close if not useful.
While RPi boards are quite popular, don't think vendors can reasonably be expected to track and work around all their bugs and quirks, like ones on display here.

mk-fg, 2019-03-22:
Just tested same exact card with 4.19.30 on amd64 machine without any tricks and have no problems with it whatsoever - everything indeed "just works".

b-rad-NDi, 2019-03-22:
Good to hear everything works as it should on PC. Thanks for the rpi test results. It's a shame the rpi usb stack is so problematic.

closed 2019-03-22 by b-rad-NDi

mk-fg, 2019-08-11:
For anyone who might stumble upon this thread in the future:

There seem to be a bunch of patches to RPi kernel attempting to address this issue - https://github.com/Dark-Sky/rpi-udl-linux - which seem to work with Hauppauge 950Q/955Q at least, and probably others.

b-rad-NDi, 2019-08-11:
I don't see any patch specifically dealing with this issue or optimizing the usb stack in their commit history. There's under 100 commits, so it didn't take very long to review them. That makes it seem like this was finally fixed upstream.

b-rad-NDi, 2019-08-11:
Unless it's just the bulk mode transfer module option, in which case there's no patching at all.
I wouldn't be terribly surprised if forcing to bulk mode helps the devices work much better.

mk-fg, 2019-08-11:
I unfortunately don't have access to setup where these tuners work now, so can't test myself, only heard about it on irc today and these devices specifically mentioned under "tested with" in the README there.
Also, I messed-up the repo link, which should've been pointing to gitlab instead of github, as repo was apparently moved there.

Wouldn't surprise me if it's indeed upstream, as there were USB-related changes there recently for sure, though with people mostly complaining about it not working right in some scenarios since 4.19.57 :)

mk-fg, 2019-08-11:
Also, despite RPi in the name and intention, looks like that kernel isn't based on raspberrypi/linux, but rather:

    Majortom's & DarkSky's repository based off of V4L media_tree,
    powARman's saa716x work, and Updatelee's work

And especially if bult for aarch64 and not armv7 that official RPi kernel/distros are using, guess there'd be plenty of other differences as well.

b-rad-NDi, 2019-08-11:
Yeh I went to gitlab and looked through all their modifications though and spy no secret sauce.
i'm guessing this is an upstream fix and nothing specific to their kernel at all. With any luck the usb fixes will be migrated back to stable.